evcc-io/evcc #16578: Für Kostal Plenticore (ohne Speicher) werden die DC-Werte statt der AC-Werte verwendet

Robbe64, 2024-10-09:
### Describe the bug

Ich hab folgende PV Komponenten:
- Kostal Plenticore (ohne angeschlossenem Speicher)
- Sungrow Hyprid (mit Speicher)

Da ich den Plenticor Hybrid nicht als Hybrid sondern nur als einfachen WR ohne Speicher nutze habe ich hier vermutlich das gleiche Problem wie in #8288. Könntet ihr für diesen Anwendungsfall auch noch ein angepasstes Template anlegen?

### Steps to reproduce

[image]

In dem Screenshot heute morgen wird AC seitig in beiden WR nichts erzeugt, dennoch zeigt EVCC eine Produktion an.
Das führt dann bei mir zu einem zu hohen Hausverbrauch.
Der Effekt wird deutlich stärker, wenn wenn die Sonne scheint.
An diesem Tag war niemand Zuhause und die Sonne schien durchgehend. Zu sehen ist der von EVCC berechnete Hausverbrauch:
[image]

### Configuration details

```yaml
plant: 5754cceb248af4f44db329f392e18f0b887a50b39ce1a16cf1a142def199f908
interval: 20s # control cycle interval

log: debug
levels:
  tesla: debug

sponsortoken: XXX

meters:
- type: template
  template: kostal-plenticore
  id: 71
  host: 192.168.178.101
  port: 1502
  usage: pv
  modbus: tcpip
  name: pv1
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.178.26
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.178.26
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.178.26
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3

chargers:

  - name: easee1
    type: easee
    user: xxx
    password: xxx
    charger: xxx
    timeout: 20s
    authorize: true
  - name: goe
    type: template
    template: go-e-v3
    host: 192.168.178.54

loadpoints:
  - title: Carport
    charger: easee1
    mode: pv
    mincurrent: 7
    maxcurrent: 32
    soc:
      estimate: true # Interpolation des SoC zwischen den Abfragen zum Auto
      poll:
        mode: always
        interval: 60m
    enable:
      delay: 3m
    disable:
      delay: 7m
      threshold: 100
    guardduration: 5m
    vehicle: egolf # Standardfahrzeug
  - title: Werkstatt
    charger: goe
    mode: pv
    mincurrent: 6
    maxcurrent: 16
    soc:
      estimate: true
      poll:
        mode: allways
        interval: 60m
    enable:
      delay: 4m
    disable:
      delay: 7m
      threshold: 100
    guardduration: 5m

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
      - pv1
      - pv2
    battery:
      - battery3
  residualPower: 100 # 100W soll immer eingespeist werden. Nötig für Batteriesystem
  maxGridSupplyWhileBatteryCharging: 50 # Verhindert den Netzbezug bei DC Leistung > AC Leistung und gleichzeitigem Batterieladen

vehicles:
  - name: egolf
    type: id
    title: eGolf
    capacity: 21
    user: xx
    password: xx
    phases: 1
    onIdentify:
      minCurrent: 7
      maxCurrent: 16
      priority: -1
  - name: tesla
    type: template
    template: tesla-command
    title: Magda
    priority: -2
    capacity: 60 # kWh
    accessToken: xx
    refreshToken: xx
    vin: xx

tariffs:
  currency: EUR
  grid:
    type: tibber
    token: xx

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.077 # [currency]/kWh
  co2:
    type: grünstromindex
    zip: 59609

influx:
  url: http://solaranzeige:8086
  database: evcc

telemetry: true

mqtt:
  broker: nodered:1884
  topic: evcc
  user: xx
  password: xx
```

### Log details

```text
[site  ] DEBUG 2024/10/09 08:31:33 ----
[lp-1  ] DEBUG 2024/10/09 08:31:33 charge power: 0W
[lp-1  ] DEBUG 2024/10/09 08:31:33 charge currents: [0.007 0.008 0.009]A
[lp-2  ] DEBUG 2024/10/09 08:31:33 charge power: 0W
[lp-2  ] DEBUG 2024/10/09 08:31:33 charge currents: [0 0 0]A
[site  ] DEBUG 2024/10/09 08:31:33 pv power: 33W
[site  ] DEBUG 2024/10/09 08:31:33 battery soc: 27%
[site  ] DEBUG 2024/10/09 08:31:33 battery power: 216W
[site  ] DEBUG 2024/10/09 08:31:33 grid meter: 16W
[site  ] DEBUG 2024/10/09 08:31:33 grid currents: [-0.233 -0.235 0.487]A
[site  ] DEBUG 2024/10/09 08:31:33 site power: 332W
[lp-1  ] DEBUG 2024/10/09 08:31:33 charge total import: 7442.032kWh
[lp-1  ] DEBUG 2024/10/09 08:31:33 charger status: B
[lp-1  ] DEBUG 2024/10/09 08:31:33 next soc poll remaining time: 10m40s
[lp-1  ] DEBUG 2024/10/09 08:31:33 pv charge current: 0A = 0A + -1.44A (332W @ 1p)
[site  ] DEBUG 2024/10/09 08:31:53 ----
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.13

VolkerK62, 2024-10-09:
> Könntet ihr für diesen Anwendungsfall auch noch ein angepasstes Template anlegen?

Hast du mal `sunspec-inverter` ausprobiert?

Robbe64, 2024-10-09:
Ah, sehr gute Idee! Sheint zu passen. Danke!

closed 2024-10-09 by Robbe64