evcc-io/evcc #16579: v0.130.13 - prioritySoc funktioniert nicht mehr

michiproep, 2024-10-09:
### Describe the bug

Nach dem update auf 0.130.13 scheint sich das Verhalten im prioritySoc-Bereich genau so wie bufferSoc-Bereich zu verhalten.
Zuvor wurden im "unteren Bereich" alle Loadpoints ausgeschaltet. Seit dieser Version - muss aber auch zugeben, dass ich ein paar Versionen übersprungen habe - wird weiter geladen bis bei zu wenig Sonne wieder Batterie-Entladung ensteht.

### Steps to reproduce

1. set prioSoc to 90% while current soc is 20%
2. confirm that loadpoint keeps charging while battery is still loading (enough sun)

### Configuration details

```yaml
---
```

### Log details

```text
---
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.13

VolkerK62, 2024-10-09:
Ohne Log schwierig.
Aber prinzipiell können Auto und priorisierte Batterie gleichzeitig laden, wenn genug Sonne da ist.
`residualpower` hast du auf einen kleinen positiven Wert (ca. 100) eingestellt?

michiproep, 2024-10-09:
ja, residualPower=100
ja, aber das sollte nur oberhalb von prioSoc so sein.
Unterhalb bedeutet (oder bedeutete???) ja im Prinzip "Ladesperre für alle Loadpoints"

VolkerK62, 2024-10-09:
>Unterhalb bedeutet (oder bedeutete???) ja im Prinzip "Ladesperre für alle Loadpoints"

Nein, noch nie. 
Wenn über die Batterieladung hinaus noch genug Überschuss vorhanden ist, dann werden auch die Loadpoints "versorgt".

michiproep, 2024-10-09:
Noch nie? Aber genau das Verhalten konnte ich bis dato beobachten. Seit ca. einem Jahr schau ich mir das täglich an. Hmmm
Worin genau liegt dann der Unterschied zwischen dem prioSoc und bufferSoc Bereich?

hier noch ein Log-Auszug

`
[site ] DEBUG 2024/10/09 14:35:36 ----
[lp-1 ] DEBUG 2024/10/09 14:35:36 charge power: 4940W
[lp-1 ] DEBUG 2024/10/09 14:35:36 charge currents: [7.2 7.1 6.9]A
[lp-2 ] DEBUG 2024/10/09 14:35:36 charge power: 82W
[lp-3 ] DEBUG 2024/10/09 14:35:36 charge power: 0W
[lp-4 ] DEBUG 2024/10/09 14:35:36 charge power: 0W
[lp-5 ] DEBUG 2024/10/09 14:35:36 charge power: 0W
[site ] DEBUG 2024/10/09 14:35:36 pv power: 4915W
[site ] DEBUG 2024/10/09 14:35:36 battery soc: 71%
[site ] DEBUG 2024/10/09 14:35:36 battery power: 740W
[site ] DEBUG 2024/10/09 14:35:36 grid meter: -2W
[site ] DEBUG 2024/10/09 14:35:36 grid powers: [16 -148 130]W
[site ] DEBUG 2024/10/09 14:35:36 grid currents: [1.2 -1.19 1.96]A
[site ] DEBUG 2024/10/09 14:35:36 site power: 938W
[lp-1 ] DEBUG 2024/10/09 14:35:36 charge voltages: [232 232 234]V
[lp-1 ] DEBUG 2024/10/09 14:35:36 detected connected phases: 3p
[lp-1 ] DEBUG 2024/10/09 14:35:36 detected active phases: 3p
[lp-1 ] DEBUG 2024/10/09 14:35:36 charge total import: 5608.950kWh
[lp-1 ] DEBUG 2024/10/09 14:35:36 charger status: C
[lp-1 ] DEBUG 2024/10/09 14:35:36 vehicle soc: 82%
[lp-1 ] DEBUG 2024/10/09 14:35:36 vehicle soc limit: 100%
[lp-1 ] DEBUG 2024/10/09 14:35:36 vehicle range: 283km
[lp-1 ] DEBUG 2024/10/09 14:35:36 pv charge current: 5.84A = 7.2A + -1.36A (938W @ 3p)
[lp-1 ] DEBUG 2024/10/09 14:35:36 projected site power 938W >= 200W disable threshold
[lp-1 ] DEBUG 2024/10/09 14:35:36 pv disable timer start: 3m0s
[lp-1 ] DEBUG 2024/10/09 14:35:36 pv disable in 3m0s
[lp-1 ] DEBUG 2024/10/09 14:35:37 max charge current: 6A
[site ] DEBUG 2024/10/09 14:35:42 ----
[lp-1 ] DEBUG 2024/10/09 14:35:42 charge power: 4940W
[lp-1 ] DEBUG 2024/10/09 14:35:42 charge currents: [7.2 7.1 6.9]A
[lp-2 ] DEBUG 2024/10/09 14:35:42 charge power: 81W
[lp-3 ] DEBUG 2024/10/09 14:35:42 charge power: 0W
[lp-4 ] DEBUG 2024/10/09 14:35:42 charge power: 0W
[lp-5 ] DEBUG 2024/10/09 14:35:42 charge power: 0W
[main ] DEBUG 2024/10/09 14:35:42 lp Heizung at prio 10 gets additional 4940W from Garage at prio 6, total 4940W
[site ] DEBUG 2024/10/09 14:35:42 pv power: 4891W
[site ] DEBUG 2024/10/09 14:35:42 battery soc: 71%
[site ] DEBUG 2024/10/09 14:35:42 battery power: -343W
[site ] DEBUG 2024/10/09 14:35:42 grid meter: -560W
[site ] DEBUG 2024/10/09 14:35:42 grid powers: [-167 -336 -58]W
[site ] DEBUG 2024/10/09 14:35:42 grid currents: [-1.44 -1.79 -1.95]A
[site ] DEBUG 2024/10/09 14:35:42 battery has priority at soc 71% (< 72%)
[site ] DEBUG 2024/10/09 14:35:42 giving loadpoint priority for additional: 4940W
[site ] DEBUG 2024/10/09 14:35:42 site power: -5300W
[lp-2 ] DEBUG 2024/10/09 14:35:42 charger status: C
[lp-2 ] DEBUG 2024/10/09 14:35:42 pv charge current: 14.7A = 7A + 7.68A (-5300W @ 3p)
[site ] DEBUG 2024/10/09 14:35:48 ---- 
`

VolkerK62, 2024-10-09:
formatier das mal bitte lesbar

michiproep, 2024-10-09:
soll ich mal ein großes LogFile anhängen? Der Auszug ist bestimmt zu wenig

michiproep, 2024-10-09:
Hab ne neue Info. Das Verhalten scheint nur beim Auto/Loadpoint 1 so zu sein.
Hab das Auto abgestöpselt und siehe da...
[image]

VolkerK62, 2024-10-09:
hier hatte die Batterie (noch) keine Priorität
`[site ] DEBUG 2024/10/09 14:35:36 battery soc: 71%`

Überschuss reicht nicht und Auto geht in die Abschaltung
```
[lp-1 ] DEBUG 2024/10/09 14:35:36 pv disable in 3m0s
[lp-1 ] DEBUG 2024/10/09 14:35:37 max charge current: 6A
```

dann hast du (vmtl. per API) prioritySoc auf 72% eingestellt
`[site ] DEBUG 2024/10/09 14:35:42 battery has priority at soc 71% (< 72%)`

Erschwernde kommt dann in dem Fall noch die Loadpoint-Priorität dazu.
`[main ] DEBUG 2024/10/09 14:35:42 lp Heizung at prio 10 gets additional 4940W from Garage at prio 6, total 4940W`
Deshalb geht die Heizung erst in die Abschaltung, wenn das Auto aus ist.

michiproep, 2024-10-09:
ok, danke.
Ja, wenn man auch noch mit Prios arbeitet und die SOC-settings per API anwachsen lässt...

Let's close this! Danke nochmal!

closed 2024-10-09 by michiproep

etecprojekt, 2024-10-10:
Was auch nicht mehr zu klappen scheint, ist generell die Priorisierung der Ladepunkte. 
Wird ggf. im gleichen Zusammenhang sein, werde ein Issue eröffnen.