evcc-io/evcc #16587: Entsoe prices assigned to wrong day

rwegnerGermany, 2024-10-09:
### Describe the bug

When using entsoe for BZN|DE-LU, todays prices are assigned to tomorrow.
Please see these examples:

[image]

[image]

The problem seems to occur only in the afternoon when the response comprises 48h of price data.
Pleas find attached a entsoe response from 15:00 local time.
[Energy_Prices_202410091400-202410101400.txt](https://github.com/user-attachments/files/17317269/Energy_Prices_202410091400-202410101400.txt)

Now in the evening after evcc restart no current prices are shown and data seems to be one day off because Thursday 20:00, for example should be 31.8 ct instead of > 33 ct:

[image]

Please also see discussions:
[https://github.com/evcc-io/evcc/issues/16519](url)
[https://github.com/evcc-io/evcc/discussions/16539](url)

(Sorry for not testing any nightly build, but I think v0.130.13 is up to date in this regard.)

### Steps to reproduce

1. Configure Entsoe template as grid tariff with BZN|DE-LU
2. start evcc
3. wait until afternoon when tomorrows prices are published
...

### Configuration details

```yaml
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: template
    template: entsoe
    domain: BZN|DE-LU
    securitytoken: xxx # api token
    charges: 0.16672 #0.1984 minus 19% # optional, additional charges per kWh
    tax: 0.19 # optional, additional tax (0.1 for 10%)
```

### Log details

```text
[entsoe] TRACE 2024/10/09 20:44:58 GET https://web-api.tp.entsoe.eu/api?DocumentType=A44&amp;In_Domain=10Y1001A1001A82H&amp;Out_Domain=10Y1001A1001A82H&amp;PeriodEnd=202410102000&amp;PeriodStart=202410092000&amp;securityToken=***

...the response is truncated in my log, but please see attached file as an example.
No further log items to find regarding this topic.
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.13

andig, 2024-10-09:
Screenshots are useless without the log dats.

rwegnerGermany, 2024-10-09:
here is the trace log, but the entsoe response is truncated
[evcc-20241009-222649-trace.log](https://github.com/user-attachments/files/17318289/evcc-20241009-222649-trace.log)

andig, 2024-10-10:
It looks like at least one bug with Entso-E and would probably help if someone wanted to open a ticket:

```
[entsoe] TRACE 2024/10/10 08:24:20 GET https://web-api.tp.entsoe.eu/api?DocumentType=A44&In_Domain=10YPT-REN------W&Out_Domain=10YPT-REN------W&PeriodEnd=202410110800&PeriodStart=202410100800&securityToken=***
[entsoe] TRACE 2024/10/10 08:24:21 <?xml version="1.0" encoding="utf-8"?>
  <Publication_MarketDocument xmlns="urn:iec62325.351:tc57wg16:451-3:publicationdocument:7:3">
    <mRID>8116828a4635487daf7c7d3dea2b624b</mRID>
    <revisionNumber>1</revisionNumber>
    <type>A44</type>
    <sender_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</sender_MarketParticipant.mRID>
    <sender_MarketParticipant.marketRole.type>A32</sender_MarketParticipant.marketRole.type>
    <receiver_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</receiver_MarketParticipant.mRID>
    <receiver_MarketParticipant.marketRole.type>A33</receiver_MarketParticipant.marketRole.type>
    <createdDateTime>2024-10-10T06:24:21Z</createdDateTime>
    <period.timeInterval>
      <start>2024-10-09T22:00Z</start>
      <end>2024-10-10T22:00Z</end>
    </period.timeInterval>
      <TimeSeries>
        <mRID>1</mRID>
        <auction.type>A01</auction.type>
        <businessType>A62</businessType>
        <in_Domain.mRID codingScheme="A01">10YPT-REN------W</in_Domain.mRID>
        <out_Domain.mRID codingScheme="A01">10YPT-REN------W</out_Domain.mRID>
        <contract_MarketAgreement.type>A01</contract_MarketAgreement.type>
        <currency_Unit.name>EUR</currency_Unit.name>
        <price_Measure_Unit.name>MWH</price_Measure_Unit.name>
        <curveType>A03</curveType>
          <Period>
            <timeInterval>
              <start>2024-10-09T22:00Z</start>
              <end>2024-10-10T22:00Z</end>
            </timeInterval>
            <resolution>PT60M</resolution>
              <Point>
                <position>1</position>
                 <price.amount>40.08</price.amount>
              </Point>
              <Point>
                <position>2</position>
                 <price.amount>40</price.amount>
              </Point>
              <Point>
                <position>5</position>
                 <price.amount>44</price.amount>
              </Point>
              <Point>
                <position>6</position>
                 <price.amount>55.01</price.amount>
              </Point>
              <Point>
                <position>7</position>
                 <price.amount>69</price.amount>
              </Point>
              <Point>
                <position>8</position>
                 <price.amount>91.48</price.amount>
              </Point>
              <Point>
                <position>9</position>
                 <price.amount>108.8</price.amount>
              </Point>
              <Point>
                <position>10</position>
                 <price.amount>92</price.amount>
              </Point>
              <Point>
                <position>11</position>
                 <price.amount>69</price.amount>
              </Point>
[... 58 lines cut ...]
2024-10-10 00:00:00    2024-10-10 01:00:00    0.040
2024-10-10 01:00:00    2024-10-10 02:00:00    0.040
2024-10-10 02:00:00    2024-10-10 03:00:00    0.044
2024-10-10 03:00:00    2024-10-10 04:00:00    0.055
2024-10-10 04:00:00    2024-10-10 05:00:00    0.069
2024-10-10 05:00:00    2024-10-10 06:00:00    0.091
2024-10-10 06:00:00    2024-10-10 07:00:00    0.109
2024-10-10 07:00:00    2024-10-10 08:00:00    0.092
2024-10-10 08:00:00    2024-10-10 09:00:00    0.069
2024-10-10 09:00:00    2024-10-10 10:00:00    0.040
2024-10-10 10:00:00    2024-10-10 11:00:00    0.040
2024-10-10 11:00:00    2024-10-10 12:00:00    0.040
2024-10-10 12:00:00    2024-10-10 13:00:00    0.040
2024-10-10 13:00:00    2024-10-10 14:00:00    0.041
2024-10-10 14:00:00    2024-10-10 15:00:00    0.055
2024-10-10 15:00:00    2024-10-10 16:00:00    0.072
2024-10-10 16:00:00    2024-10-10 17:00:00    0.100
2024-10-10 17:00:00    2024-10-10 18:00:00    0.129
2024-10-10 18:00:00    2024-10-10 19:00:00    0.180
2024-10-10 19:00:00    2024-10-10 20:00:00    0.173
2024-10-10 20:00:00    2024-10-10 21:00:00    0.120
2024-10-10 21:00:00    2024-10-10 22:00:00    0.109
```

Positions 3 and 4 are missing for both days, which the parser is not aware of.

rwegnerGermany, 2024-10-10:
To my understanding thats not a bug, it's feature of the now used curve type A03: If the data hasn't changed, the sample isn't transmitted. So just fill up with the previous sample.
See here: [https://github.com/evcc-io/evcc/issues/16519#issuecomment-2397378903](url)

andig, 2024-10-10:
Wow, what an utterly horrible idea. To use an expressive data format like XML and the try to save data by imposing arbitrary logic on any consumer.  

That said: it seems the encoding has changed to curve type A03 although it's not clear, when or why this was changed. But yes, for A03 that seems to be expected.

closed 2024-10-10 by andig