evcc-io/evcc #16616: Trail token is not working.

andkanstc, 2024-10-12:
### Describe the bug

The trail token is not working. I copied it in the evcc.yaml and restart evcc. I get the error "sponsortoken: rpc error: code = Unknown desc = token blocked due to public exposure, please generate new token or contact info@evcc.io for support". What am I doing wrong?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# trial token, valid until 2024-10-18
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidHJpYWwiLCJleHAiOjE3MjkyNzQ0MDAsImlhdCI6MTcyODQ5NjgwMCwic3BlIjp0cnVlLCJzcmMiOiJtYSJ9.gnfLjIPCnFt3TmCUhoXCmA3nHK5RkQcOrLIejCQfc4w
```

### Log details

```text
[main ] INFO 2024/10/12 10:44:32 evcc 0.130.13
[main ] INFO 2024/10/12 10:44:32 using config file: /etc/evcc.yaml
[db ] INFO 2024/10/12 10:44:32 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2024/10/12 10:44:33 listening at :7070
[main ] FATAL 2024/10/12 10:44:33 sponsortoken: rpc error: code = Unknown desc = token blocked due to public exposure, please generate new token or contact info@evcc.io for support
[main ] FATAL 2024/10/12 10:44:33 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.13

andig, 2024-10-12:
Should be fixed

closed 2024-10-12 by andig

andkanstc, 2024-10-12:
I tried again. I copied the token. I restarted my raspi. But it don't work. The same error again.