evcc-io/evcc #16627: Erweiterung der Template Definition, um mehrere modbus Bus-IDs zu unterstützen

juergen-weber, 2024-10-13:
**Is your feature request related to a problem? Please describe.**
Die modbus-Implementierung des SAX Home Batteriespeichers implementiert 2 modbus-Tabelle die zwar mit der gleichen IP/Port Adresse angesprochen werden, aber unterschiedliche Bus-IDs (64 und 40) haben. Während über die Bus-ID 40 alle Schreibvorgänge und das Lesen von SOC, Lade-, Entladeleistung, Smartmeter Leistung abgewickelt werden, werden über die Bus-ID 40 quasi alle ADL400-Smartmeter Parameter in einer Sunspec angelehnten Art hier verfügbar gemacht.
Ich möchte gern über eine Erweiterung des SAX Templates diese Tabelle für den use case "grid" nutzen, finde aber keine Möglichkeit, 2 Bus-IDs in einer Template Definition unterzubringen.

**Describe the solution you'd like**
Implementierung mehrerer modbus Definitionen in einem Template.

**Describe alternatives you've considered**
Definition eines zusätzlichen Template nur für die grid-Meter Funktion.

**Additional context**

andig, 2024-10-13:
Was ist ADL400?

juergen-weber, 2024-10-13:
Du hast Recht mit Deiner Frage. Das war nicht zu verstehen. Das ADL400 Smart Meter wird standardmäßig vom SAX Home Batteriespeicher ausgeliefert. Dessen Werte werden mehr oder weniger Sunspec-konform über modbus mit der Bus ID 40 zugänglich gemacht. Das existierende SAX Template nutzt Bus ID 64. Alternativ könnte man natürlich die Grid-Funktion in ein eigenes Meter-Template auslagern. 

Bei der Durchsicht des Templates ist mir übrigens aufgefallen, dass das Schreiben des Registers mit der Funktion "writesingle" durchgeführt wird. Das funktioniert nicht. Es muss hier die modbus-Funktion 10h (write multiple registers) genutzt werden.

andig, 2024-10-13:
> Bei der Durchsicht des Templates ist mir übrigens aufgefallen, dass das Schreiben des Registers mit der Funktion "writesingle" durchgeführt wird. Das funktioniert nicht. Es muss hier die modbus-Funktion 10h (write multiple registers) genutzt werden.

Mach gerne direkt PR mit den gewünschten Änderungen- einfach die Datei hier in Github editieren.

> Das ADL400 Smart Meter wird standardmäßig vom SAX Home Batteriespeicher ausgeliefert. Dessen Werte werden mehr oder weniger Sunspec-konform über modbus mit der Bus ID 40 zugänglich gemacht. Das [...]

Ich frag mich eher, woher das aktuelle Template denn die Werte bekommt die es da über die 64 zugänglich macht? Also anders gesagt: was ist der Nutzen der 40 und Direktansprache des Zählers?

andig, 2024-10-13:
>  Die Tabelle mit der 40er Bus-ID stellt quasi alle Werte des ADL400 Smartmeters (Spannungen, Ströme, Leistungen, etc. alles 3 Phasen) zur Verfügung

Das wird einfach der Zähler selbst sein.

> Zum Testen muss ich mir aber noch eine Testumgebung bauen

Es reicht den einfach als `custom` Zähler zu konfigurieren

closed 2024-10-13 by andig