evcc-io/evcc #16633: Ladestand Kia Niro

sosarasp, 2024-10-13:
### Describe the bug

Hallo,

Seit dem Update auf 0.130.0 bis 0.130.13 wird der aktuelle Ladestand vom Kia Niro nicht mehr angezeigt. Bis 0129.0 hat das immer funktioniert. Jetzt wird nur noch der geladene Anteil angezeigt. Unter Konfiguration wird bei Fahrzeug SoC der richtige Wert angezeigt. Wahrscheinlich selbe Ursache wie Issue #15561  
[image]
[image]

### Steps to reproduce

Update von 0.129.0 auf 0.130.0 bis 0.130.13

### Configuration details

```yaml
chargers:
- name: Wallbox_Hof
  type: template
  template: ocpp
  stationid: 10383320
  remotestart: true
  
vehicles:
- name: Kia_Sonja
  type: template
  template: kia
  title: Kia_Sonja
  user: xxx
  password: xxx
  vin: xxx
  capacity: 64
  language: de
  icon: car
  phases: 3
  cache: 30m
  mode: OFF
  minCurrent: 6 
  maxCurrent: 16
  # Kann meist erst später eingetragen werden, siehe: https://docs.evcc.io/docs/features/vehicle (optional)

    
loadpoints:
- title: Hof
  charger: Wallbox_Hof
  vehicle: Kia_Sonja
  mode: off
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  enable:
    threshold: 0
    delay: 1m
  disable:
    threshold: -100 # maximum import power (W)
    delay: 3m
  guardduration: 1m
  priority: 1
  soc:
    poll:
      mode: connected
      # mode: always
      interval: 30m
    estimate: true
```

### Log details

```text
Soc:           65%
Capacity:      64.0kWh
Charge status: C
Range:         293km
Odometer:      13816km
Finish time:   2024-10-13 21:56:00 +0200 CEST
Position:      50.081714,9.248683
Limit Soc:     100%
OnIdentified:  Mode:off, MinCurrent:6, MaxCurrent:16
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.130.13 (01c2d979)

sosarasp, 2024-10-13:
Hier noch das Trace Log
[evcc-20241013-143117-trace.log](https://github.com/user-attachments/files/17355735/evcc-20241013-143117-trace.log)

VolkerK62, 2024-10-13:
Die Info ist jedenfalls da
`[kia   ] TRACE 2024/10/13 14:08:41 ...`
```
 "evStatus": {
                    "batteryCharge": true,
                    "batteryStatus": 62,
                    "batteryPlugin": 2, 
```
Ich hab mit meinem Hyundai keine Probleme.
Ich habe aber auch kein `soc.poll` und kein `cache` konfiguriert.

sosarasp, 2024-10-13:
Ja, die Info habe ich auch gesehen. 
soc.poll und cache habe ich auch schon rausgenommen, jedoch ohne Erfolg.
Sonst noch eine Idee?

VolkerK62, 2024-10-13:
@andig Ich hab mal etwas ausprobiert. Hilft evtl. weiter.
Auszüge aus /api/state

Fahrzeug nicht verbunden und nicht am Loadpoint ausgewählt
[image]

Fahrzeug nicht verbunden, aber am Loadpoint ausgewählt
[image]

Fahrzeug verbunden und **automatisch** erkannt
[image]

@sosarasp 
nimm mal den `vehicle` Eintrag aus dem Loadpoint. (den hab ich auch nicht).
Dann sollte nach den anstecken der Soc angezeigt werden.

sosarasp, 2024-10-13:
@VolkerK62 
auch den vehicle Eintrag aus dem Loadpoint zu nehmen hat leider nichts gebracht.

VolkerK62, 2024-10-13:
und das Auto wird automatisch erkannt?

sosarasp, 2024-10-13:
Ja, Auto wird erkannt.

VolkerK62, 2024-10-13:
Hast du mal ein trace Log wo das Anstecken des Fahrzeug mit drin ist.

VolkerK62, 2024-10-14:
Keine Ahnung, ob das mit dem eigentümlichen Ansteck-Verhalten der Wallbox zu tun hat.
Da kommen einige timeouts vom Wallbox-Meter.
Und dann das:
`[lp-1  ] ERROR 2024/10/13 20:26:43 vehicle soc: timeout`

Die 100% sind hochgerechnet, da in kürzester Zeit eine riesige Menge geladen wurde.
`[lp-1  ] DEBUG 2024/10/13 20:27:02 soc estimated: 100.00% (vehicle: 0.00%)`

sosarasp, 2024-10-14:
Heute Morgen hatte ich das Auto nochmal getrennt und wieder neu verbunden. Danach wurde der Ladestand wieder nicht angezeigt. 
LOG
[lp-1  ] INFO 2024/10/14 06:38:10 vehicle updated: unknown -> Kia_Sonja
[lp-1  ] DEBUG 2024/10/14 06:38:10 set charge mode: off
[lp-1  ] DEBUG 2024/10/14 06:38:10 soc estimated: 0.00% (vehicle: 0.00%)
[lp-1  ] DEBUG 2024/10/14 06:38:10 vehicle soc: 0%
[kia   ] TRACE 2024/10/14 06:38:10 GET https://prd.eu-ccapi.kia.com:8080/api/v1/spa/vehicles/24623c6a-e547-4518-9493-95c2745a0c35/status/latest

Hier noch das vollständige Trace Log.
[evcc-20241014-064100-trace.log.txt](https://github.com/user-attachments/files/17367732/evcc-20241014-064100-trace.log.txt)

Wenn ich es aber richtig verstehe, wird der aktuelle Ladestand nicht erkannt , da wo 100% angezeigt wurde, war es ja der estimate.
Die Frage die ich mir Stelle, soll der Soc durch den Charger bereitgestellt werden, also ausgelesen werden oder wird dieser über die Kia Api bereitgestellt?
Wurde bei diesen OCPP Charger oder Kia Api etwas geändert in Bezug auf den Vihcle SoC zur Version 0.129.0.
Wallbox ist eine StarCharge Aurora 11KW ( wird auch von Enpal teilweise verbaut)
Wenn es am Charger liegt, wäre es möglich hier ein geändertes Template zur Verfügung zu stellen?

VolkerK62, 2024-10-14:
Das ab- und wieder anstecken war zu schnell.

Soc über Charger vermute ich mal eher nicht. Da gibt es nur ganz weniger Wallbox/Auto Kombinationen, die das können.
Es gab mit 0.130.0 jede Menge Änderungen bei OCPP.

Was mich wundert, das Auto meldet, dass es nicht verbunden ist
       `"batteryPlugin": 0,`
(und die andere Erkennungsart (offene Ladeklappe) ist in deiner Fahrzeugabfrage nicht enthalten)
trotzdem wird es offensichtlich "erkannt"
`[lp-1  ] INFO 2024/10/14 06:16:07 vehicle updated: unknown -> Kia_Sonja`
allerdings kommt kein Fahrzeugstatus.
Keine Ahnung, wie das funktioniert.

Das muss jemand ergründen, der tiefer drin steckt.

andig, 2024-10-14:
> Was mich wundert, das Auto meldet, dass es nicht verbunden ist

Fahrzeuge melden immer irgendwas- einfach ignorieren.

> wird der aktuelle Ladestand vom Kia Niro nicht mehr angezeigt. ... Jetzt wird nur noch der geladene Anteil angezeigt. 

Mir ist völlig unklar, was damit gemeint sein kann. Screenshot zeigt den SoC ja.

sosarasp, 2024-10-15:
> Mir ist völlig unklar, was damit gemeint sein kann. Screenshot zeigt den SoC ja.

Die % Angabe ist aber nur der aktuell hinzugeladene Teil siehe Issue Eröffnung hier werden 11% angezeigt, in der Konfiguration unter Fahrzeug wird der richtige Wert 63% angezeigt. 

> [evcc-20241013-202921-trace.log.txt](https://github.com/user-attachments/files/17356708/evcc-20241013-202921-trace.log.txt) Hier das Trace Log von Start bis Auto eingesteckt. Erst Gastfahrzeug, dann der Kia. Keine Ladung gestartet. Jetzt wird auf der Ui aber ein Ladestand von 100% gezeigt und eine Lademenge von 2,2 MWh. Tatsächlich ist der Ladestand aber bei 70%. ![Screenshot_20241013_203330_Samsung Internet](https://private-user-images.githubusercontent.com/143269770/376060440-144f8f14-e044-4513-ad2c-25ef1f1d4634.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg5NjQ2OTYsIm5iZiI6MTcyODk2NDM5NiwicGF0aCI6Ii8xNDMyNjk3NzAvMzc2MDYwNDQwLTE0NGY4ZjE0LWUwNDQtNDUxMy1hZDJjLTI1ZWYxZjFkNDYzNC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAxNVQwMzUzMTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lY2NhY2Q2ODc3ZDNmNmY5MDhiZDMyNzA3ZWI1ZjdiODEzMDI0OGUxMzEwNDMzYWU2M2Y1YTRiZjNiMWIxNTdmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.9VEsyeG_ao-DMb185VIBMxC1U-L5sJu5TcWDmcy7Re4)

Die hier angezeigten 100% stimmen ebenfalls nicht, tatsächlich lag der prozentuale SoC bei ca. 70%. Die angezeigten 100% kamen durch einen anderen Fehler. Ein Gastfahrzeug wurde mit 2,2 MWh angegeben. Gastfahrzeug wurde keins angesteckt oder geladen, sondern nur der Kia und das waren vielleicht ehr 2,2 KWh, also Kommafehler. Nach dem Ab und wieder Anstecken war dieser Fehler auch wieder weg, aber der Prozentuale SoC wird wieder nicht angezeigt.

VolkerK62, 2024-10-15:
@sosarasp  zur Eingrenzung: wenn du auf 0.129.0 zurückgehst, funktioniert es und ab 0.130.0 nicht mehr?

sosarasp, 2024-10-15:
Genau. Mit 0.129.0 funktioniert es.

andig, 2024-10-15:
Was sagt denn

    evcc vehicle

Stimmt das?

sosarasp, 2024-10-15:
Bei Issue Erstellung hatte ich evcc vehicle ausgeführt. Die Abweichung 63% (aus Konfiguration) zu 65% (evcc vehicle) ist dem zeitlichen Abstand zwischen der Erstellung der Screenshots geschuldet.

evcc vehicle :

Soc:           65%
Capacity:      64.0kWh
Charge status: C
Range:         293km
Odometer:      13816km
Finish time:   2024-10-13 21:56:00 +0200 CEST
Position:      50.081714,9.248683
Limit Soc:     100%
OnIdentified:  Mode:off, MinCurrent:6, MaxCurrent:16

Also  evcc vehicle gibt den richtigen Wert aus.

mfuchs1984, 2024-10-15:
Beim vehicle sich gibt es auch einen timeout Error, und zwar bevor die API überhaupt gefragt wurde. Das liegt meines Erachtens auch am Meter timeout: estimator.go Funktion Soc fragt erst den charger. Da Soc in den metervalues vom charger verfügbar ist (0, vermutlich weil kein smart vehicle), wird er erst dort abgefragt. Durch das Meter timeout kommt ErrTimeout raus https://github.com/evcc-io/evcc/blob/57aba0db314ff0917af64da450456b43dbe73748/charger/ocpp/connector.go#L308

Das führt meines Erachtens dazu, das die Funktion Soc im estimator direkt mit dem timeout rausspringt, ohne beim vehicle nachzufragen.
https://github.com/evcc-io/evcc/blob/57aba0db314ff0917af64da450456b43dbe73748/core/soc/estimator.go#L118

mfuchs1984, 2024-10-15:
Die veralteten MeterValues tauchen übrigens im Log nur in dem Moment auf, in dem das Fahrzeug angesteckt und eine Transaction gestartet wird. Sieht aus als würde die wallbox da das sampling pausieren bzw. anders ausgelastet sein (in dem Moment antwortet sie auch einmal nicht auf einen MeterValues requests). Davor und danach sind sie jeweils nur 2 Sekunden alt.

mfuchs1984, 2024-10-15:
https://github.com/evcc-io/evcc/blob/57aba0db314ff0917af64da450456b43dbe73748/core/soc/estimator.go#L113

Diese Annahme führt hier m.E. zu einem Problem, unabhängig vom timeout. Der charger kann laut seiner Angaben grundsätzlich SoC, das Fahrzeug jedoch nicht.  Deshalb kommt wohl immer 0 raus. Im der ocpp Spec finde ich auf den ersten Blick nichts, was das Verhalten in diesem Fall definiert.

andig, 2024-10-15:
Arrgh, das ist doof. Sehen wir im Log, ob/was er da liefert?

mfuchs1984, 2024-10-15:
Der charger liefert in diesem Fall 0

andig, 2024-10-15:
Das ist dann gelogen... deshalb bitte:

    metervalues: -SoC

mfuchs1984, 2024-10-15:
Und dann wäre das nächste Problem hier das 30 Sekunden timeout, weil der charger nach dem anstecken zunächst Werte liefert, die 33 Sekunden alt sind.

mfuchs1984, 2024-10-15:
> ```
> metervalues: -SoC
> ```
@sosarasp füge das mal deiner charger Konfiguration hinzu und mach ein neues trace Log vom Start + Ansteckvorgang.

sosarasp, 2024-10-15:
Hallo zusammen, anbei das Trace Log von Start + Anstecken und Ladebeginn mit der geänderten charger Konfiguration

chargers:
- name: Wallbox_Hof
  type: template
  template: ocpp
  stationid: 10383320
  remotestart: true
  metervalues: -SoC

[evcc-20241015-201504-trace.log](https://github.com/user-attachments/files/17382891/evcc-20241015-201504-trace.log)

Das hinzufügen von: 

> metervalues: -SoC

war erfolgreich, jetzt wird der Ladestand in % wieder angezeigt.

[image]

mfuchs1984, 2024-10-15:
Aus den logs heraus würde ich sagen:
1. Wallbox meldet Auto angesteckt 
2. Evcc startet eine Transaktion 
3. Das veranlasst die Wallbox dazu, zu starten (mit dem Default Profile)
4. Dann stoppt evcc diesen Vorgang wieder

sosarasp, 2024-10-16:
Ok. 
Den Remotestart Befehl habe ich mal rausgenommen und die Wallbox auf local PnC. Dann gibt es keinen autostart. Aber als Massage kommt nur die Meldung Fahrzeug angeschlossen,  da der Kia bei generieren der Massage noch nicht den Kia erkannt hat. Das dauert halt vielleicht ca. 20 sec.
Wäre hilfreich wenn es die Möglichkeit gibt, die Massage um diese Zeit (bis Fahrzeugerkennung erfolgt ist) zu verzögern.
Hierzu das Trace Log
[evcc-20241016-061438-trace.log](https://github.com/user-attachments/files/17389786/evcc-20241016-061438-trace.log)