evcc-io/evcc #16637: LG ESS Home: Batterie laden mit dynamischen Stromtarif

diehappy69, 2024-10-13:
Hallo Zusammen,

ich setze das LG ESS Home 10 mit 2x 11kwh Batterien ein und würde sie gerne mit Strom aus dem Netz laden, wenn dieser billig ist (dynamischer Stromtarif - Grenzwert muss einstellbar sein). Zusätzlich habe ich noch 2 Go e Charger im Einsatz, die das verbilligte Laden unterstützen. Es muss jedoch verhindert werden, dass sie den Akku leerziehen, wenn die Autos geladen werden.

Idealerweise kann man einstellen, dass der Akku mit PV Strom geladen wird, wenn dieser noch nicht voll ist und die Fahrzeuge per billigen Hausstrom. Ist der Akku nicht voll, aber es gibt noch billigen Strom, dann soll dieser ebenfalls gefüllt werden (besonders im Winter). 

Etwas änliches bietet woh E3DC an, aber ein Systemwechsel kommt nicht mehr in Frage.

Vielen Dank
DieHappy

andig, 2024-10-13:
Das wirds nur geben wenn jemand rausfindet wie.

lmfgpeace, 2024-10-14:
Naja bei Sungrow gibts ja die Möglichkeit der AC Zwangsladung. Da würde das teilweise sicherlich gehen.

If *Strompreis <10ct*
and *weather forecast next day no sun*
and *battery SOC <30%*
then *emergency charge battery to 90% SOC from grid*

if *Strompreis >10ct*
or *weather forecast next day sun*
or *battery SOC >30%*
then *stop emergency charge battery from grid*

premultiply, 2024-10-15:
Modbus-Register?
API-Calls?
...?

andig, 2024-10-15:
Glaskugel?

NI07077, 2024-10-19:
Ich hab im photovoltaikforum.com folgende Modbus Informationen gefunden. 
Dort hat es ein User geschafft den Systembetrieb nachts per Modbus zu deaktivieren um ein Entladen des Akkus beim Laden des E-Autos aus dem Netz zu verhindern.

[(Home 8,10) LG ESS sunspec ver 1.3-1 (1).pdf](https://github.com/user-attachments/files/17446996/Home.8.10.LG.ESS.sunspec.ver.1.3-1.1.pdf)

andig, 2024-10-20:
Das ist lustig. Wir haben eine "custom" Implementierung dafür, tatsächlich scheint das Geräte aber Modbus zu können? Dann kannst Du mal probieren den als `sunspec-battery-control` zu konfigurieren. Siehe auch https://github.com/evcc-io/evcc/pull/16750

NI07077, 2024-10-20:
Glaube an dem Punkt waren einige User schonmal: #16188 
Problem war: SOC und DOD wurden unsauber dargestellt?

andig, 2024-10-20:
Achja- das war das kaputte Teil 🙁

NI07077, 2024-10-20:
Ich hab [hier](https://www.loxforum.com/forum/german/software-konfiguration-programm-und-visualisierung/398568-einbindung-wechselrichter-lg-ess-home#post407961) noch Informationen aus dem Loxone Forum zur Problematik mit der unterschiedlichen Umrechnung im Sunspect Protokoll.

ReinholdEckert, 2024-10-26:
@diehappy69 : Ich kann deine Anforderung voll nachvollziehen, da ich es ähnlich haben wollte.
Da ich mein Auto NIE aus der Batterie laden wollte (die soll rein fürs Haus sein), habe ich mir die Wallbox einfach zwischen dem ganzen PV/Smartmeter/Akku-Kram und dem Zähler des Netzbetreibers anschließen lassen. Somit lade ich im Normalfall entweder mit Netzstrom oder (im Sommer) mit Überschuss, der sonst ins Netz eingespeist worden wäre. Da brauche ich keine Softwarelösung dafür. Ist natürlich nicht so flexibel, falls das jemand braucht.
Dank evcc kann ich nun aber meine Wallbox so steuern, dass sie sich den optimalen Zeitpunkt aussucht.

andig, 2024-10-26:
Das ist m.E. so die ungefähr schlechteste Lösung- denn damit kannst Du nicht mehr wählen. Das Problem hier ist doch ganz einfach: der ESS kann das über Modbus, nur ist da die Implementierung kaputt. Also beim Hersteller beschweren und reparieren lassen!

ReinholdEckert, 2024-10-26:
Ja, das kann man sicher so sehen. Aber mit Softwarelösungen würde das meist bedeuten, dass die Batterie gänzlich gesperrt wird und dann auch das Haus nicht mehr damit versorgt werden kann. Da muss glaub ich jeder rausfinden, was ihm wichtig ist.
Zum Zeitpunkt meiner Wallbox-Installation waren mir Lösungen wie evcc noch gar nicht bekannt, da war ich schon sehr froh über diese "Hardware"-Lösung. Aber selbst jetzt stört mich das nicht, da ich nie den Akku aussaugen will - und das würd schnell passieren, wenn ich das Auto am Abend in der Garage anstecke. Dank evcc habe ich jetzt die Möglichkeit, die Wallbox fürs Privatauto auf die günstigste Zeit und ggf. auch mit reduzierter Leistung zu begrenzen, während das Firmenauto mit voller Leistung und beliebiger Stromquelle (Netz oder Überschuss) laden darf. Ergänzt sich wunderbar.

andig, 2024-10-26:
Zurück zum Thema 👍🏻

andig, 2024-10-29:
Mangels Rückmeldung zu LG Ticket geschlossen.

NI07077, 2024-12-21:
Laden aus dem Netz ist durch die Firmware 10.05.7433 / 04.10.2024 R2159 nun gezielt möglich.

[image]

Durch folgende Anleitung klappt das Upgrade:

1. Die neue Firmware vom Server des LG Support runterladen:

https://ess-pr-web.s3.eu-central-1.amazonaws.com/firmware/P10.05.7433R2159_D5003.0198_USR_G4V_D10k.bin

2. Die Anleitung vom LG Support zum Firmware Update anwenden:

https://lg-electronics.zendesk.com/hc/de/articles/18228830313757-Home-8-10-Manuelles-Firmeware-Update

- Die "bin" Datei auf einen maximal 16GB großen FAT32 formatierten USB Stick kopieren.

- Den Wechselrichter in den PV Only Modus (https://lg-electronics.zendesk.com/hc/de/articles/16297653378205-HOME-5-8-10-PV-Only-Modus-aktivieren) versetzen und Betriebsmodus eingeschaltet lassen.

- WLAN Dongle herausziehen und den USB Stick mit der Firmware einsetzen. Die verschieden LEDS des Wechselrichter beginnen zu wie in de Anleitung zu leuchten. Sobald nur noch die WLAN LED blinkt ist das Update beendet.

- USB Stick herausziehen und den WLAN Dongle wieder einsetzen.

- PV-Only-Modus wieder deaktivieren.

In den Installateur Einstellungen muss unter PCS in einem der Unterpunkte der Backup Modus aktiviert werden.

Es erscheint in der Enervu APP im normalen Menüpunkt "Akku" nun eine aktiverbare Einstellung " Aufladen vom Netz":

[image]

Indem der Backup Modus unter dem Menüpunkt "Akku" aktiviert und der Backup Modus SOC (reservierter SOC) verändert wird, kann das Laden aus dem Netz gesteuert werden.

Ebenfalls kann durch das Erhöhen des Backup Modus SOC über den derzeitigen SOC des Akkus das Entlanden der Batterie bei jedem beliebigen SOC verhindert werden.

Da die Envervu APP nur im eigenen Netzwerk nutzbar ist, gibt es die weitere Möglichkeit den Wechselrichter und die Netzladeung über die LG ThingQ App weltweit zu steuern.

[image]

Einziger Nachteil: die Netzladung wird mit 6000Watt durchgeführt, dieser Wert ist nicht anpassbar. Gegenfalls aber modulierbar durch die Batterieladeoption "Schnellladen" oder "Batteriepflege". Dies konnte ich aber derzeit noch nicht testen.

Die auslesbaren und steuerbaren Modbus/Sunspec Register verändern sich durch das Update leider nicht. Klappt eine Integration des Netzladens/Entladesperre über das LG ESS Home 8/10 Template?

diehappy69, 2024-12-21:
Hi,
gilt die Firmware für alle Home 8/10 Versionen?
Ich habe folgenden Wechselrichter mit 2 neuen Akkus:
[image]
[image]

NI07077, 2024-12-21:
Ich habe einen LG Home 8 Wechselrichter mit den alten HB7 und HB10 Batterien. 
Ich kann dir natürlich keine Garantie geben, dass die Firmware bei deinem LG Home 10 ebenfalls funktioniert. 
Im anderen Forum haben sich schon Mitglieder getraut und ihre Wechselrichter geflasht: https://github.com/Morluktom/ioBroker.lg-ess-home/issues/49#issuecomment-2558111676
Frag doch sonst dort nochmal nach deren Hardware.

NI07077, 2024-12-29:
> Ich schließe erstmal- gerne Update hier wenn es entsprechende Informationen gibt, _wie_ das passieren kann.

Hier das Update:
Mit der Firmware 10.05.7433 / 04.10.2024 R2159 ist das Netzladen möglich. Die Steuerung erfolgt über den Backup SOC.
@andig Morluktom hat den iobroker Adapter für den LG ESS Home 8/10 bereits angepasst: https://github.com/Morluktom/ioBroker.lg-ess-home/commit/e867028f83e7d3b42e4053af28fc030cdcdd5ea5
Über den Adapter ist ein Laden aus den Netz möglich. Auch die Entladesperre kann über den Backup SOC gesetzt werden.
Können wir den Fall wieder aufmachen und das Netzladen im LG ESS 8/10 template wie im Adapter von Morluktom übernehmen? Sunspec Register wurde leider nicht geupdet. Eine Kontrolle über Sunspec 802 Battery Base Model ist daher immer noch nicht möglich.

andig, 2024-12-29:
Ich kann da leider nichts raus lesen. Es braucht die konkreten Informationen wie normal/hold/charge zu steuern sind.

uwen70, 2025-01-12:
OK, ich versuche mal einige Infos zusammenzutragen:
LG ESS bietet die Möglichkeit bei Stromausfall eine Notstromversorgung bereitzustellen. Dazu ist der Backupmodus mit `backup_setting` auf "on" zu setzen. Die reservierte Energiemenge des Akkus ist der `backup_soc`. Zur Steuerung, ob der Akku entladen wird, kann dieser verwendet werden.  

**Entladen**
Wenn `backup_soc < bat_user_soc` wird der Akku bis auf `backup_soc` entladen. 

**Hold**
Wenn `backup_soc >= bat_user_soc` geht der Akku in Standby, sofern ein aufladen mit autoload ausgeschaltet ("0") ist

**Laden**
Soll der Akku aus dem Netz geladen, werden muss der `backup_soc > bat_user_soc` sein und `autocharge `auf "1" gesetzt werden.

Die Werte für den Akku werden über die URI `/v1/user/setting/batt` gelesen und gesetzt. Lesen geht mit einem POST und dem authKey. Die JSON Antwort sieht dann z.B. so aus:
```
{
    "winter_setting": "on",
    "winter_status": "on",
    "backup_setting": "on",
    "backup_status": "off",
    "alg_setting": "fast_charge",
    "safety_soc": "10",
    "charging_from_grid_to_keep_soc": "1",
    "backup_soc": "50",
    "startdate": "1101",
    "stopdate": "0228",
    "auto_charge": "on",
    "internet_connection": "connected",
    "enervu_activated": "true",
    "enervu_upload": "on"
}
```
Sollen Werte geschrieben werden, erfolgt das mit einem PUT und folgenden JSON Daten:
```
{
    "auth_key":"12345678-1234-1234-1234-1234567890ab",
    "wintermode":"on",
    "alg_setting":1,
    "safety_soc":"10",
    "startdate":"1101",
    "stopdate":"0228",
    "backupmode":"on",
    "autocharge":"0",
    "backup_soc":"50 %"
}
```
Die Antwort erfolgt dann wie bei beim POST.

Ich habe die vorstehenden Infos aus einem LG ESS Home 8 mit Firmware `"pms_version": "10.05.7433", "pms_build_date": "2024-10-04 R2159"` gewonnen.
Welche Infos benötigt man noch?

andig, 2025-01-12:
Welche PUTs braucht es dann jeweils für die Modi? Gerne PR.

uwen70, 2025-01-12:
Der authKey kommt vom Login (siehe [lgpcs.go](https://github.com/evcc-io/evcc/blob/master/meter/lgpcs/lgpcs.go)).

Bezüglich der PUTs (immer BATT URI `/v1/user/setting/batt`) denke ich folgendes Vorgehen:

Backup Mode anschalten einmalig:  `backupmode: "on"`
Dies würde die Bateriesteuerung aktivieren. Man könnte dies auch in der LG App einschalten.

Netzladen des Akkus aus- bzw. einschalten mit `autocharge: "0"` bzw. `"1"`

Setzen des `backup_soc: "50 %"`
Beim Setzen wird auch das % übertragen, in der Antwort kommt nur `backup_soc: "50"` zurück.

In der LG App kann man die Werte immer nur einzeln verändern.
[image]

Habe ich die Frage richtig verstanden?

andig, 2025-01-12:
> denke ich folgendes Vorgehen:

kannst du die vollständigen, getesteten, minimalen Requests bitte posten?

uwen70, 2025-01-12:
Nein, kann ich leider nicht. Ich habe nur Remote auf eine LG ESS Zugriff. Um diese Infos zusammenzustellen habe ich mittels [LG-ESS-Home---Webviewer](https://github.com/savage8811/LG-ESS-Home---Webviewer/) und Firefox Netzwerkanalyse anschaut, was da abläuft. Für mehr bauche ich eine Anleitung.

uwen70, 2025-01-16:
Gibt es eine prototypische Implementierung der aktiven Bateriesteuerung, wo man sich das mal anschauen kann?

andig, 2025-01-16:
Überall wo es implementiert ist, natürlich!

NI07077, 2025-01-16:
Bitte beachten zum AutoCharge (Netzlade)-Verhalten mal folgende Zusammenfassung/Empfehlung:

Bevor AutoCharge aktiviert wird, den BackupSoc > 8% über Soc stellen => Entladen stoppt bevor über Netz geladen wird. Diese Reihenfolge ist nicht zwingend notwendig, hilft aber vielleicht der Elektronik und dem Akku, nicht direkt vom Entladen ins Laden umzuspringen. > 8% ist notwendig, dass der Ladevorgang überhaupt beginnt (kann evtl. auch 4 - 5 - 6% sein), ich habe es letzte Nacht bei 91% Soc und BackupSoc 100% probiert, da hat die Netzladung gestartet. Heißt auch, dass ein Aktivieren von AutoCharge bei einem Soc von z.B. 95% zu dem Verhalten führt, dass der Akku nicht mehr entlädt.

Beim Deaktivieren von AutoChage muss zwingend das Netzladen beendet werden indem der BackupSoc < Soc gesetzt wird (wahrscheinlich dann auch ein Delta von > 8%), sonst lädt der Akku aus dem Netz weiter. Wenn das Netzladen über Soc beendet ist, kann AutoCharge deaktiviert werden.

Sollte der Akku nicht mehr Entladen (Akku im Standby), obwohl Soc > BackupSoc hilft nur Operation stop -> start, um die Batterie wieder zu motivieren, mitzuspielen. Ich hatte bisher nicht die Geduld, die Situation zu provozieren und dann abzuwarten , wann der Akku von selbst den Betrieb wieder aufnimmt.

uwen70, 2025-01-16:
Anbei ein Skript, welches den Akku in normal/hold/charge versetzen kann. Passwort und IP müssen noch angepasst werden. Curl wird benötigt.

[lgessbattery.txt](https://github.com/user-attachments/files/18446290/lgessbattery.txt)

andig, 2025-01-17:
Perfekt- das muss jetzt nur noch ins Template eingebaut werden 👍🏻

andig, 2025-01-25:
Das scheint niemand zu tun wollen, closing. PR welcome.

andig, 2025-01-25:
Gerne testen. Danke.

uwen70, 2025-01-26:
@andig: Vielen Dank für die Implementierung. Ist die Funktion schon in der Nightly Entwicklerversion enthalten? Wenn nicht, wie kommt man an eine ausführbare Version? Ich habe in der [Dokumentation](https://docs.evcc.io/docs/Home) nichts gefunden und kam da nicht weiter.

uwen70, 2025-01-27:
Ich bin mir unsicher ob, ich das richtig gebaut habe. Jedenfalls geht es nicht.

Ich kann die Webseite nicht aufrufen. Ist das normal bei dem Branch?
```
[server] FATAL 2025/01/27 00:53:38 httpd: failed to load embedded template:open index.html: file does not exist
[server] FATAL 2025/01/27 00:53:38 Make sure templates are included using the `release` build tag or use `make build`

```
In /meter/lgpcs/lgpcs.go in Zeile 126 die URI korrigieren:
`uri := fmt.Sprintf("%s/v1/user/essinfo/home", m.uri)`

Wenn ich die alte lauffähige Konfiguration der ESS direkt in die evcc.yaml kopiere kommt: 
 ```
./evcc --database ./evcc.db -c evcc.yaml -l debug
...
[site  ] DEBUG 2025/01/27 02:01:25 ----
[lp-1  ] DEBUG 2025/01/27 02:01:25 charge power: 0W
[site  ] DEBUG 2025/01/27 02:01:25 grid power: 189W
[site  ] DEBUG 2025/01/27 02:01:25 battery 1 power: -0W
[site  ] DEBUG 2025/01/27 02:01:25 pv 1 power: 0W
[site  ] DEBUG 2025/01/27 02:01:25 battery 1 soc: 72%
[site  ] DEBUG 2025/01/27 02:01:25 site power: 289W
[lp-1  ] DEBUG 2025/01/27 02:01:25 charger status: A
[lp-1  ] INFO 2025/01/27 02:01:25 car disconnected
[lp-1  ] DEBUG 2025/01/27 02:01:25 set battery boost: false
[lp-1  ] DEBUG 2025/01/27 02:01:25 set session soc limit: 0
[lp-1  ] DEBUG 2025/01/27 02:01:25 set session energy limit: 0
[lp-1  ] DEBUG 2025/01/27 02:01:25 charger disable
[lp-1  ] DEBUG 2025/01/27 02:01:25 wake-up timer: stop
[site  ] DEBUG 2025/01/27 02:01:45 ----
```
Das sieht soweit erstmal gut aus. Weiter komme ich dann ohne Webseite nicht.

uwen70, 2025-01-27:
In /meter/lgpcs/lgpcs.go in Zeile 145 die URI korrigieren:
`uri := fmt.Sprintf("%s/v1/user/setting/batt", m.uri)`

andig, 2025-01-27:
Du kannst direkt im PR ändern 👍🏻

uwen70, 2025-01-27:
> Du kannst direkt im PR ändern 👍🏻

Hmm, fehlende Rechte? 
```
git push origin feat/batterymode-2
Username for 'https://github.com': uwen70
Password... (Token)
remote: Permission to evcc-io/evcc.git denied to uwen70.
Schwerwiegend: konnte nicht auf 'https://github.com/evcc-io/evcc.git/' zugreifen: The requested URL returned error: 403
```

andig, 2025-01-28:
Wie gesagt- bitte direkt im Code kommentieren!

zaj78, 2025-03-05:
Das wäre auch für mich eine nützliche Funktion. Muss noch was gemacht werden, damit die Änderungen aus #18416 ins Produkt finden?

uwen70, 2025-03-09:
Ich hatte berichtet, dass es bei mir gut läuft ([#18492](https://github.com/evcc-io/evcc/pull/18492#pullrequestreview-2583476131)). Mittlerweile kann man aber den PR [#18416](https://github.com/evcc-io/evcc/pull/18416) wegen Konflikten nicht mehr mit der neuen Version 0.200.x mergen.

zaj78, 2025-03-09:
Die Konflikte sind überschaubar. Leider habe ich keine Ahnung von go und muss erst rausfinden, wie ich das ganze für Raspi bauen und dort testen kann, weil meine Wallbox mit RS 485 dran hängt.

zaj78, 2025-03-09:
Mit dem Bauen und auf dem Raspi testen war einfacher als ich dachte, scheint zu funktionieren. Ich beobachte das weiter. Soll ich einen neuen PR machen, mit #18416 rebased auf 0.200.8?

zaj78, 2025-03-09:
Ist das ausreichend, dass die Batterie nach dem Ende des Ladevorgangs immer im Backupmode mit Autocharge verbleibt, sollte das einstellbar werden, oder sollte evcc den vorherigen Stand merken und dahin zurückkehren?

andig, 2025-03-09:
@uwen70 die Kommunikation ist leider bei https://github.com/evcc-io/evcc/pull/18416#issuecomment-2643133324 abgerissen.

andig, 2025-03-09:
> oder sollte evcc den vorherigen Stand merken und dahin zurückkehren?

@zaj78 tut es. Falls nein bitte separate Discussion dazu.

PR #19630 (2025-03-11): LG ESS: add battery control
Fix #16637, replace #18416, #18492
Rebased on 0.200.8, reworked installer login
Required firmware for active battery control on Home 8/10: at least R2155, tested with R2173.