evcc-io/evcc #16668: Q-Cells X3 Modbus Communication is not working anymore

bertos03, 2024-10-15:
### Describe the bug

Ich habe einen Q-Cells X3 Hybrid Inverter am Laufen. 
Ich greife auf dessen Daten über Modbus TCP (mittels RTU über einen RS485/Ethernet Converter) zu.
Das hat immer wunderbar funktioniert.
Der Converter und der Inverter funktionieren weiterhin - das habe ich mittels eines Python Skripts getestet (die Werte des Inverters lassen sich auslesen).

Ohne Änderung der Konfiguration o.Ä. funktioniert allerdings nun EVCC nicht mehr. In den EVCC-Logs sehe ich diesen Fehler: energy: read failed: modbus: response slave id '0' does not match request '1'
Ich habe wirklich nichts an der Konfiguration geändert. Insofern meine Hypothese: Kann es sein, dass mit dem letzten Update ein Bug eingeschleust wurde?

### Steps to reproduce

1. Check Config.
2. Start EVCC.
3. Check for functionality -> not working
4. Check logs -> Error message: energy: read failed: modbus: response slave id '0' does not match request '1'
5. Check connection and functionality of inverter/RS485-Converter via custom python script while EVCC is NOT running -> working
6. Check manual readout via custom python script while EVCC IS running -> not working (RS485 converter seems to get flooded by requests)

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

plant: XXXXXXXXXXXXXX

interval: 35s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: XXXXXXXXXXXXXX 

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: error
levels:
  site: error
  lp-1: error
  lp-2: error
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
[... 243 lines cut ...]
  #   tags: <tags>
  
  #influx:
  #  url: http://localhost:8086
  #  database: evcc
  #  user: evcc
  # password: dmhrsa69
```

### Log details

```text
[site  ] ERROR 2024/10/15 22:55:48 pv 1 power: add[0]: read failed: modbus: response transaction id '2' does not match request '1'
[site  ] ERROR 2024/10/15 22:56:23 battery 1 power: read failed: modbus: exception '3' (illegal data value), function '4'
[site  ] ERROR 2024/10/15 22:56:35 pv 1 power: add[0]: read failed: read tcp 192.168.178.27:36960->192.168.178.74:502: i/o timeout
[site  ] ERROR 2024/10/15 22:57:35 battery 1 power: read failed: modbus: response transaction id '5' does not match request '4'
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

bertos03, 2024-10-15:
Wie so oft ... der Fehler saß 50cm vor dem Monitor.
Für alle, die das gleiche Problem haben:
Mein RS485 Converter (Waveshare) war auf TCP Client anstelle TCP Server konfiguriert. Das habe ich gefixt - nun geht es wieder.

closed 2024-10-15 by bertos03

andig, 2024-10-16:
> Ich habe wirklich nichts an der Konfiguration geändert

Mein Lieblingszitat 😅