evcc-io/evcc #16671: Keine Änderung des dynamischen Limits über Grafik

longtom99, 2024-10-16:
**Is your feature request related to a problem? Please describe.*

In der Grafik vom dynamischen Strompreis unter Einstellungen Parkplatz, Günstiges Netzladen verstelle ich dauernd (bis jetzt unabsichtlich weil ich es nicht wusste) den Schwellwert wenn ich auf die Balken tippe.
Da ich in der mobilen Ansicht keine Eerte sehe tippe ich auf die Balken und dadurch verstellt sich der Grenzwert.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

In der Grafik vom dynamischen Strompreis unter Einstellungen Parkplatz, Günstiges Netzladen die Möglichkeit NUR über das Dropdown Menü den Tarif einzustellen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Die Werte direkt in der Grafik ablesbar, das wird in der mobilen Ansicht aber nicht möglich sein,

**Additional context**
Add any other context or screenshots about the feature request here.

naltatis, 2024-10-16:
Das Auswählen des Limits direkt über das Diagramm ist in der Tat ein bewusstes Feature. Du bekommst ja auch sofortiges Feedback, dass du etwas eingestellt hast, wenn du da draufklickst.

Die Preis-Range (Min/Max) wird ja auch angezeigt. Heißt du hast zumindest schon mal den Wertebereich da. Ist für dich wichtig, dass du die genauen Zwischenwerte aller Balken auch auf Mobile sehen kannst?

longtom99, 2024-10-16:
Ja, irgendwie möchte ich schon wissen was z.B. die 3 kleinsten Balken sind.
Wenn ich das Auto anstecke, schaue ich nach wie der Tarif ist.
Und es wäre cool wenn ich da nicht in eine andere App vorher schauen müsste.

Eure Bedienung der ganzen App ist insgesamt wirklich MEGA!
Einfach und klar strukturiert, genau wie es mir gefällt.
Ich weiß auch nicht genau wie man es noch "optimieren" könnte.
Eventuell länger draufbleiben das man nur den Preis sieht?
Und bei kurz wird es ausgewählt?

closed 2024-10-17 by andig