evcc-io/evcc #16673: Günstiges Netzladen Dynamische Stromtarife /  Reset Filter nach Ausstecken

geni0602, 2024-10-16:
Aktuell kann man bei den dynamischen Tarifen (Tibber etc.) einen Filter setzten, dass der Ladevorgang unter einem bestimmten Preis gestartet wird. Funktioniert perfekt. 
Mein Problem ist, dass ich den Filter setzte und oft vergesse den zu deaktivieren und somit lädt sich das Auto automatisch (ungewollt) beim nächsten Erreichen des Zielpreises (meistens nachts, da Preise günstiger) auf. Wenn der Fall eintritt wird das Auto nachts vollgeladen und bei einem sonnigen Tag kann man den PV Überschuss entsprechend nicht mehr nutzen.

Gibt es die Möglichkeit den Preisfilter nach ausstrecken des Fahrzeugs zu deaktivieren (Filterreset)? Für mich wäre so ein Feature sinnvoll um Vermeidung ungewollter Ladevorgänge beim vergessenen Preisfilter zu vermeiden.

sveba, 2024-10-16:
Finde ich sehr sinnvolles Feature. Man kann auch einfach nur Dauer der Rule vllt. setzen. Also +XX Stunden/Tage

TomF79, 2024-10-16:
> reisfilter zu vermei

Würde ich auch sinnvoll finden.

maatinh, 2024-10-16:
Ich auch. Auch, weil bei dynamischen Stromtarifen das, was heute günstig ist, morgen schon wieder teuer sein kann.

naltatis, 2024-10-17:
Ja, sehe ich. Spannende Frage ist nur ob wir das zeitbasiert (bspw. solange wie ich in die Zukunft schauen kann, Tarif, fix 24h, ...) machen sollten oder ob wir das "Ereignisbasiert" (bis zum Abstecken) machen. Am klassischen Ladepunkt finde ich das "bis zum Abstecken" gut und sinnvoll. Wir haben aber beim "Speicher aus dem Netz laden" genau das gleiche Thema. Da würde nur eine zeitbasierte Logik funktionieren.

geni0602, 2024-10-17:
> Ja, sehe ich. Spannende Frage ist nur ob wir das zeitbasiert (bspw. solange wie ich in die Zukunft schauen kann, Tarif, fix 24h, ...) machen sollten oder ob wir das "Ereignisbasiert" (bis zum [...]

Finde die zeitbasierte Logik nicht verkehrt. Beim Abstecken hat man das Nachteil, das man den Filter einstellt, vorm Eintreten des Ladevorgangs das Auto spontan absteckt (Fahrt notw. etc.) und dann wird der Filter beim erneuten Einstecken vorm günstigen Preisfenster nicht mehr aktiv.

batmanvane, 2024-10-19:
Eventbasierte Logik geht ja schon bereits mit Boardmitteln via events und services. Das laeuft bei mir zb sehr gut fuer "automatisierten kostenopimalen" Ladezeitplan zu erstellen. Ich greife da aktuell 0 mal ein. Konkret: wochentags muss das auto frueh immer 80% haben. evcc immer auf pv, bei anstecken erstelle kostenoptimalen plan fuer naechsten tag inkl. ladeziel+wann fertig. falls noch sonne bis dahin aussreichend, orgelt er das rein, rest dann ueber den plan.

siehe hier:

typxxi, 2024-10-23:
> Ja, sehe ich. Spannende Frage ist nur ob wir das zeitbasiert (bspw. solange wie ich in die Zukunft schauen kann, Tarif, fix 24h, ...) machen sollten oder ob wir das "Ereignisbasiert" (bis zum [...]

Dann stellt sich an der gleichen Stelle die Frage, warum es noch keine Entladepause für den Hausakku gibt, um den Tibberkunden die morgendliche "Akku ist leer, aber Strom kostet 43 Cent" Frage zu ersparen.

Wenn ich automatisch von einem 3. Register aus dem Heimakku sagen kann, er solle Entlade Pause bei Preisen unter X Cent machen, wenn SOC um Mitternach unter 50% ist, dann spar ich mir die 20% Verluste über das Akku laden. 

Von daher wäre die Prio ggf. sogar höher, auch weil ich 20% Verluste in der CO2 Bilanz sparen kann. Akku billig voll machen (dann kommt der Hausstrom in dem Moment ja auch vom Netz) ist ja eine andere Geschichte als Netzstrom 1:1 ohne Verluste dann zu nutzen, wenn der günstig ist, wobei dann noch die Akkuladung erhalten bleibt, bis es eine Verwendung mit mehr Ersparnis gibt.

batmanvane, 2024-10-23:
Das Planen sowie das hier besprochene "optimierte" Laden macht EVCC ziemlich komplex - unter der Haube und auch das UI. Ich verfolge grd [Ladeplaner](https://github.com/evcc-io/evcc/pull/16091/files/fac2ab5b9dbc8c5ec63332970fc74d33a282779a..b5ef2c6c03e28335708627fc197092ab5ffe138e) - Cooles Feature aber wooow, eine Menge Schalter und Abhängigkeiten.

Ich finde den ursprünglichen EVCC Ansatz _weniger ist mehr sehr gut_. Sonst ist das nachher wie bei den Imbisskarten von Döner-Pizza-Frittenbuden, jede Variante hat ne Nummer/Schalter und keiner sieht mehr bei der Logik durch. 

Evtl. i) geht man den Weg Simple/Expert Modus (viele Schalter) oder ii) man schafft es, basierend auf einer adaptiven Strategie automatisiert im mittel kostenoptimal bis in den nächsten Tag (Planungshorizont) zu kommen. Das wäre dann nur ein Schalter. Die adaptive Strategie würde Lade/Entladevorgänge basierend auf historischen Verbauchsdaten und dynamischen Preispfaden der Anlage kostenoptimal durchführen. Kalman-Filter wöre ein möglicher Ansatz dafür.

geni0602, 2024-10-23:
Glaube, dass wir etwas vom Ursprungsthema abdriften.  Es geht hier nicht um komplexe Filtermöglichkeiten, sondern darum, dass der Preisfilter (dynamische Tarife) abschaltbar ist. Entweder nach ausstrecken, oder zeitlich begrenzt. Sowas würde das user experience nicht verkomplizieren, sondern meiner Meinung nach eher erleichtern/verbessern

floppybl, 2024-10-30:
Die Idee finde ich auch super, denn mir passiert dies auch des Öfteren.

Mir ist jedoch noch ein anderes "Problem" aufgefallen. Ich hatte jetzt das günstige Netzladen mehrfach aktiviert, aber evcc stoppt den Ladevorgang nicht, wenn der Preis über der Grenze ist. Bspw. heute Nacht war auf unter 33ct eingestellt. Laut Anzeige sollten nur 2 Stunden geladen werden. Gestartet hat er pünktlich nur heute morgen war das Auto voll und hat nach den 2 Stunden nicht gestoppt.

Ich weiß, ist ein bisschen offtopic aber passt vielleicht hier kurz rein :)

tvand, 2024-10-30:
> Wir haben aber beim "Speicher aus dem Netz laden" genau das gleiche Thema.

Die Hausbatterie wird ja mit dem Feature-Set immer mehr zum Loadpoint. Da würde ja die Ladeplanung auch passen.

maatinh, 2024-10-30:
> > Wir haben aber beim "Speicher aus dem Netz laden" genau das gleiche Thema.
> 
> Die Hausbatterie wird ja mit dem Feature-Set immer mehr zum Loadpoint. Da würde ja die Ladeplanung auch passen.

Oder wie ein Fahrzeug mit bidirektionaler Lademöglichkeit.

naltatis, 2024-11-04:
> Ich weiß, ist ein bisschen offtopic aber passt vielleicht hier kurz rein :)

@floppybl mach dazu gerne ein issue mit Log und Details auf, wenn das reproduzierbar ist.