evcc-io/evcc #16679: echarge / salia Phase switching

hanborns, 2024-10-16:
Hi,

dies ist ein Follow-Up zu Issue https://github.com/evcc-io/evcc/issues/11085

Alle seit 01.07.24 produzierten echarge / Hardy Barth cpµ2 Pro können nun zwischen 3 und 1 Phase umschalten. Das ist eine versteckte Konfiguration in der Wallbox-Web-UI (man muss erst auf das Zahnrad klicken). Man kann vorher produzierte cpµ2 Wallboxen auch nachrüsten lassen. Es muss auf jeden Fall der MID-Zähler verbaut, angeschlossen und konfiguriert sein, sonst geht es nicht. Ein Befehl zum Umschalten der Phasenzahl ist

curl -X PUT -d '{"salia/phase_switching/setphase":"toggle"}' http://192.168.223.225/api/secc
 {"result":"ok"}.

Ich habe bei echarge angefragt, ob man es auch explizit angeben kann, habe aber noch keine Antwort.

Man kann den Ist-Zustand in der Status-JSON auslesen unter secc.port0.salia.phase_switching.actual, da steht dann 3 bei 3 Phasen und 1 bei einer Phase. Ich kann gerne die JSON in den jeweiligen Zuständen zur Verfüng stellen bzw. Euch auch einen Zugang zu meiner Wallbox aufsetzen oder in einer Teams-Session debuggen ...

Mir ist nicht bekannt, ob es irgendeinen systematischen Weg gibt, das herauszufinden. echarge weiss vermutlich, ab welcher Seriennummer es auf jeden Fall geht (also die erste, die ab dem 1.7. produziert wurde), ansonsten müsste man annehmen, dass es nicht geht und man es explizit mitkonfigurieren muss per Configurationflag einschliesslich obiger Hinweise ...

Ich frage mal nach, ob man das systematisch abfragen kann.

Vielen Dank und viele Grüße, Hanno

[image]

[image]

andig, 2024-10-16:
Könnte jemand prüfen, ob es `secc.port0.salia.phase_switching.actual` nur gibt falls die Umschaltung verbaut ist?

hanborns, 2024-10-16:
Wenn ich es in der Konfiguration ausschalte, dann verschwindet der phase_switching Tree in der JSON ...
[image]

closed 2024-10-16 by andig

hanborns, 2024-10-17:
Hi, es geht noch nicht: evcc hat versucht zu schalten (2024/10/17 12:04:09), aber es ging nicht (log ist angehängt) Jetzt hat sich meine Wallbox aufgehängt und die Sonne ist weg ... ich probiere es wieder wenn es mehr Sonne gibt. Wie läuft denn der Wechsel ab, sendet evcc nur den Umschaltbefehl an die Wallbox oder wird davor der Strom reduiziert und nach dem Wechsel wieder hochgefahren, und wenn ja, sind da irgendwelche (konfigurierbaren) Verzögerungen? Da könnte es dann vielleicht eine Race Condition geben, weil die Wallbox selbst das Laden pausiert, die Umschaltung durchführt und dann das Laden fortsetzt. Vielleicht gerät sie aus dem Tritt, wenn man den Strom zu früh wieder hochdreht während sie noch in ihrer eigenen Prozedur ist.

[evcc-20241017-122043-debug.log](https://github.com/user-attachments/files/17412421/evcc-20241017-122043-debug.log)

andig, 2024-10-17:
Kannst du im trace log bitte schauen was an den Befehlen falsch ist?

hanborns, 2024-10-17:
Ich hatte die VM runtergefahren und die Logs scheinen weg zu sein. Ich versuche es nochmal nachzustellen und ziehe dann das Log.

hanborns, 2024-10-18:
Jetzt scheint es zu gehen, ich hatte die Wallbox stromlos gemacht und dann neu gestartet. Als ich heute mittag nach Hause kam habe ich dann schnell geladen und dabei manuell die Anzahl der Phasen gewechselt, das ging, allerdings kam einmal ein WARN, dass die Phasen nicht wie erwartet seien (siehe unten nochmal). Ich habe dann auf PV+Min gestellt, und auch da wurde sauber geschaltet (auf eine Phase). Als ich jetzt gerade heimkam war das Auto unter 30%, also sollte er mit 3 Phasen schnell laden bis 30%, dabei hat er auch einen Wechsel gemacht:

[salia ] TRACE 2024/10/18 15:57:57 PUT http://192.168.223.225/api/secc
[salia ] TRACE 2024/10/18 15:57:57 {"salia/phase_switching/setphase":"toggle"}
--
{"result":"ok"}

Allerdings hat er dann noch diesen Eintrag ins Log gemacht:

[lp-1  ] WARN 2024/10/18 15:59:26 charger logic error: phases mismatch (got 1, expected 3)
[lp-1  ] DEBUG 2024/10/18 15:59:26 forced charging at vehicle soc 28% (< 30% min soc)
[salia ] TRACE 2024/10/18 15:59:26 PUT http://192.168.223.225/api/secc
[salia ] TRACE 2024/10/18 15:59:26 {"salia/phase_switching/setphase":"toggle"}
--
{"result":"ok"}
[lp-1  ] DEBUG 2024/10/18 15:59:26 switched phases: 3p

Danach dann nicht mehr. In der Konfiguration gibt es auch eine Konfiguration für ein "switch delay" von 20 Sekunden. Laut Dokumentation verhindert die Wallbox auch zu häufiges Umschalten (https://www.echarge.de/0_uploads/echarge/download/konfigurationsanleitung-salia-deutsch_13.pdf, Seite 15), eventuell gibt es da eine Race Condition weil im Log oben ging es ja um ca. 1,5 Minuten ...)

Ich würde vorschlagen ich beobachte es weiter, aber im Moment sieht es gut aus.

andig, 2024-10-19:
Sinnig wäre auch mal mit 

    evcc charger --log trace,db:error --phases 1|3

zu probieren was da im API kommt und ob die Umschaltung funktioniert.

hanborns, 2024-10-24:
Ich hatte Bilder vor und nach dem Umbau gemacht. Die Seriennummer ist gleich geblieben, die der Salia-Platine entspricht aber auch nicht der des Gehäuses. Aber, es ist ein Aufkleber auf die Wallbox gekommen ...
[image]
[image]
[image]
[image]
Mir scheint es so, als wenn sich rechts die Relais oder was das ist verändert hätten.

hanborns, 2024-10-24:
> Sinnig wäre auch mal mit
> 
> ```
> evcc charger --log trace,db:error --phases 1|3
> ```
> 
> zu probieren was da im API kommt und ob die Umschaltung funktioniert.

Bei meinen Tests wegen der Umschaltung hat es noch einmal nicht umgeschaltet, als ich es dann über das CLI probiert habe kurze Zeit später mit evcc charger --log trace --phases 1 ging es dann aber. Das gilt generell beim Schalten per CLI, da ging es seit dem Cold Reboot der Wallbox immer. Ich beobachte das mal noch weiter, aber es funktioniert meistens gerade.

premultiply, 2024-10-24:
Eigentlich wurde nur das 4-polige Schütz gegen zwei 2-polige getauscht.
Die beiden Relais und die Kontakte dazu waren schon vorher auf der Platine.

premultiply, 2024-10-25:
Die beiden Relais sind die weissen Teile unter dem grünen Kabel neben dem Schütz.
Die Anschlüsse für die Schütze sind oben an der Platine. D, R1/S1, R2/S2.

premultiply, 2024-10-25:
Die beiden Relais für die Schütze sitzen neben den Schützen auf der Platine unter dem grünen Kabel.

premultiply, 2024-10-25:
Ich seh da zumindest keinen Zähler.

hanborns, 2024-10-28:
Der ist im Sicherungskasten ...
[image]

1977michael, 2024-10-30:
Hallo @hanborns 

bei mir geht es problemlos - habe die gleiche Hard- und Softwareversion. 

Test doch mal, evcc auszuschalten, die WB neu zu starten und über die App oder Weboberfläche der WB, ob die Umschaltung (automatisch oder manuell) geht. Zur Sicherheit Browser-Cash löschen.

In Deinem Heimnetz kannst Du lokal die API-Werte der WB (Salia-Platine) über http://169.254.12.53/api/ auslesen...

hanborns, 2024-10-30:
Danke @1977michael,

ich habe das mal genau so gemacht, und auch nach dem Reboot der Wallbox ging es weder per manuellem Aufruf (curl -X PUT -d '{"salia/phase_switching/setphase":"toggle"}' http://192.168.223.225/api/secc), noch mittels der WB-App, noch mit dem Icon im Web-UI (hatte den Cache gelöscht). Dann habe ich evcc wieder gestartet und dort das Laden gestartet ("Schnell"), und habe dann von "automatischem Wechsel" auf "1-phasig" gestellt und dann ging es, allerdings hat das Auto zu dem Zeitpunkt dann geladen. Ich versuche eine Systematik zu finden, tue mich aber echt schwer. Ich könnte mir vorstellen, dass es entweder nach einiger Zeit nicht mehr geht, oder aber das es so eine Art Race-Condition ist. Wenn man in der Salia-Oberfläche auf Wechseln klickt, dann führt er ja nur diesen Toggle aus, ohne ein Pausieren der Ladung. EVCC schickt aber vorher noch das Pausieren der Ladung, und dann sehr schnell danach den Toggle. Ich war schon am überlegen, ob sie nicht umschaltet, wenn der Strom auf 0A begrenzt oder das Laden pausiert ist, aber den Call dennoch mit ok quittiert. Ich habe jetzt das Laden des Autos beendet (Strom steht auf 0A) und wenn ich jetzt versuche umzuschalten passiert wieder nichts, weder per Web-UI noch per curl ... Wenn es das nächste mal klemmt stelle ich mal auf PV+Min so dass er lädt und versuche dann umzuschalten ...

1977michael, 2024-10-30:
Hallo, beim manuellen Aufruf muss doch die WB auf manuell stehen, oder? Ob die Box auf manuell durch den Befehl stellt, weiß ich nicht.

Nach dem Reboot klappt es - ohne andere Befehle zuvor durchzuführen - in der Ui auch nicht?

Beim Umschalten über die WB-Oberfläche pausiert auch die Ladung kurz und dann beginnt die Box unter den neuen Bedingungen die Ladung wieder (bei EVCC auch). Die Pausenzeit kannst Du sogar einstellen (Standart = 10 sec.)

Übrigens funktioniert die Umschaltung nur während des Ladevorgangs.

Sollte das nicht funktionieren, solltest Du evtl. mal Kontakt zum Kundenservice von Hardy Barth wenden.

hanborns, 2024-10-31:
Hi,
ja, sie stand auf manuell und als zulässiger Strom war 0A gesetzt. Wenn ich versucht habe einen höheren Wert einzutragen wurde das ignoriert, und das manuelle Starten ging auch nicht. Da sie nicht lud, hat sie auch die Anforderung zum Phasenumschalten ignoriert. Das mit der Pause war auch das was ich mit der Race Condition meinte. Wenn man den Toggle sendet, dann führt die Wallbox die Umschaltung ja vollkommen automatisch aus (Laden pausieren, umschalten, warten, wieder starten). EVCC sendet aber Pause und direkt danach umschalten. In der Dokumentation ist beschrieben, dass die Wallbox zu häufige Anfragen zum Schutz der Ladeelektronik ignoriert, bzw ich könnte mir auch vorstellen, dass wenn man Pause sendet, und dann Umschalten (was ja eine Pause beinhaltet) sie aus dem Tritt kommt. Eventuell ist das Pausieren überflüssig und EVCC sollte direkt den Umschaltbefehl senden. Ich versuche das mal manuell nachzustellen, wenn es mal wieder klemmt. Was ich verwunderlich finde, ist, dass die Wallbox immer mit ok antwortet, selbst wenn sie beschliesst, keine Folge zu leisten. Ich habe bei Hardy Barth angefragt, und auch gefragt, ob man da irgendwelche Logs ziehen kann.

1977michael, 2024-11-01:
Hallo @hanborns 

ich kann nach Tests nur sagen, dass bei mir die Umschaltung sowohl über die Web-Ui als auch über die Hardy-Barth-App oder EVCC gut funktioniert. 

@andig Danke nochmal für die Umsetzung!

waldayer1, 2025-04-01:
Hallo 
Ich werde bald eine neue echarge / Hardy Barth cpµ2 Pro bekommen, betrieben mit einem RCT Power Wechselrichter, also, so weit ich das verstehe, mit der Salina Steuerung.
Bei einigen von euch scheint das mit der Phasenumstellung zu funktionieren, obwohl auf der evcc Seite die Phasenumstellung weiterhin nicht erwähnt wird. Ich habe aber auch nicht ganz mitbekommen ob dann etwas an der evcc Einstellng etwas geändert wurde. 
Kann ich die Wallbox mit Phasenumstellung mitlerweile "plug and play" mit evcc verwenden, oder sind grössere Anpassungen nötig.
Sonst warte ich mal lieber ab,
Schönen Tag und Dnake für Infos

hanborns, 2025-04-03:
> Hallo Ich werde bald eine neue echarge / Hardy Barth cpµ2 Pro bekommen, betrieben mit einem RCT Power Wechselrichter, also, so weit ich das verstehe, mit der Salina Steuerung. Bei einigen von euch [...]

Hi, laut Hardy Barth können es alle, die ab dem 1.7.2024 produziert wurden. Meine habe ich von Hardy Barth umbauen lassen, weil sie vorher produziert worden war. Man muss es in der Web-UI der Wallbox freischalten (ist aber versteckt, man muss erst auf das Zahnrad klicken). Das ist aber alles dokumentiert, danach hat es evcc mit den Änderungen automatisch erkannt nachdem es implementiert war. Anfangs gab es ein wenig Probleme mit der Umschaltung bei mir, da die Wallbox manchmal nicht sofort reagiert, sondern es durchaus ein paar Minuten dauern kann, dann wirft evcc Fehler weil es scheinbar nicht geklappt hat. Ich habe das bei Hardy Barth gemeldet, die wollen es sich irgendwann mal anschauen, vor allem weil die Wallbox immer sofort mit ok antwortet, auch wenn sie erst noch warten will. Bei mir hat es dann die Hausbatterie um ein paar % geleert, deshalb habe ich nun die Wartezeit vor und nach dem Phasenwechsel in evcc auf 1 Minute reduziert, damit läuft es jetzt echt gut. Wichtig zu beachten ist, dass manche EVs die Phasenumschaltung nicht mögen, ich hatte das bei Hardy Barth angefragt (ist aber auch in der Doku) und bei unserem i3 waren keine Probleme bekannt.
Viele Grüße, Hanno