evcc-io/evcc #16692: "Netzladen" disappeared with latest NB

maatinh, 2024-10-17:
### Describe the bug

With the latest NB I do not have the menu to charge battery from grid anymore.

### Steps to reproduce

1. update to latest NB
2. enable experimental UI functions
3. open Battery from the menu
...

### Configuration details

```yaml
no changes to config
```

### Log details

```text
-
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.13 / https://github.com/evcc-io/evcc/commit/96e6c972

VolkerK62, 2024-10-17:
you should at least add your configuration, to know what hardware it is.
And a log file including server start would also be good.

maatinh, 2024-10-17:
Restarting evcc or re-saving the battery settings didnt help but after Reloading the browser windows with overriding cache (Ctrl-F5) it appeared back.

closed 2024-10-17 by maatinh