evcc-io/evcc #16696: Sungrow Hybrid SH10RT - Modbus Register für PV werden doppelt ausgelesen

heifisch, 2024-10-17:
### Describe the bug

Linux raspberrypi 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux

evcc liest die Modbus-Register vom Sungrow Wechelrichter für die Berechnung der Erzeugungs-Leistung doppelt und addiert sie im Webinterface. Es werden damit zu hohe Werte angezeigt.
In der Konsole mit dem Befehl **evcc meter** werden die Werte korrekt angezeigt.

Hier ein Ausschnitt aus dem Trace:
**evcc meter -l trace**

[main  ] INFO 2024/10/17 13:55:11 evcc 0.130.13
[main  ] INFO 2024/10/17 13:55:11 using config file: /etc/evcc.yaml
...
pv2
---
[modbus] TRACE 2024/10/17 13:55:11 modbus: send 00 01 00 00 00 06 01 04 13 98 00 02
[modbus] TRACE 2024/10/17 13:55:11 modbus: recv 00 01 00 00 00 07 01 04 04 18 20 00 00
[modbus] TRACE 2024/10/17 13:55:11 modbus: send 00 02 00 00 00 06 01 04 32 ca 00 02
[modbus] TRACE 2024/10/17 13:55:11 modbus: recv 00 02 00 00 00 07 01 04 04 fe 1a 00 00
Power:  6176W
Energy: 6505.0kWh

grid1
-----
[modbus] TRACE 2024/10/17 13:55:11 modbus: send 00 03 00 00 00 06 01 04 32 d1 00 02
[modbus] TRACE 2024/10/17 13:55:11 modbus: recv 00 03 00 00 00 07 01 04 04 00 81 00 00
[modbus] TRACE 2024/10/17 13:55:11 modbus: send 00 04 00 00 00 06 01 04 32 ec 00 02
[modbus] TRACE 2024/10/17 13:55:11 modbus: recv 00 04 00 00 00 07 01 04 04 18 fd 00 00
[modbus] TRACE 2024/10/17 13:55:11 modbus: send 00 05 00 00 00 06 01 04 15 e2 00 02
[modbus] TRACE 2024/10/17 13:55:11 modbus: recv 00 05 00 00 00 07 01 04 04 ff 32 ff ff
[modbus] TRACE 2024/10/17 13:55:11 modbus: send 00 06 00 00 00 06 01 04 13 9a 00 01
[modbus] TRACE 2024/10/17 13:55:11 modbus: recv 00 06 00 00 00 05 01 04 02 09 35
[modbus] TRACE 2024/10/17 13:55:11 modbus: send 00 07 00 00 00 06 01 04 15 e4 00 02
[modbus] TRACE 2024/10/17 13:55:11 modbus: recv 00 07 00 00 00 07 01 04 04 ff 0d ff ff
[modbus] TRACE 2024/10/17 13:55:11 modbus: send 00 08 00 00 00 06 01 04 13 9b 00 01
[modbus] TRACE 2024/10/17 13:55:11 modbus: recv 00 08 00 00 00 05 01 04 02 09 49
[modbus] TRACE 2024/10/17 13:55:11 modbus: send 00 09 00 00 00 06 01 04 15 e6 00 02
[modbus] TRACE 2024/10/17 13:55:11 modbus: recv 00 09 00 00 00 07 01 04 04 ff f7 ff ff
[modbus] TRACE 2024/10/17 13:55:11 modbus: send 00 0a 00 00 00 06 01 04 13 9c 00 01
[modbus] TRACE 2024/10/17 13:55:11 modbus: recv 00 0a 00 00 00 05 01 04 02 09 3a
Power:          -129W
Energy:         639.7kWh
Current L1..L3: -0.874A -1.02A -0.0381A

battery3
--------
[modbus] TRACE 2024/10/17 13:55:11 modbus: send 00 0b 00 00 00 06 01 04 32 dd 00 01
[modbus] TRACE 2024/10/17 13:55:11 modbus: recv 00 0b 00 00 00 05 01 04 02 01 9f
[modbus] TRACE 2024/10/17 13:55:11 modbus: send 00 0c 00 00 00 06 01 04 32 c8 00 01
[modbus] TRACE 2024/10/17 13:55:11 modbus: recv 00 0c 00 00 00 05 01 04 02 00 1d
[modbus] TRACE 2024/10/17 13:55:11 modbus: send 00 0d 00 00 00 06 01 04 32 c8 00 01
[modbus] TRACE 2024/10/17 13:55:11 modbus: recv 00 0d 00 00 00 05 01 04 02 00 1d
[modbus] TRACE 2024/10/17 13:55:11 modbus: send 00 0e 00 00 00 06 01 04 32 e2 00 02
[modbus] TRACE 2024/10/17 13:55:11 modbus: recv 00 0e 00 00 00 07 01 04 04 2d fa 00 00
[modbus] TRACE 2024/10/17 13:55:11 modbus: send 00 0f 00 00 00 06 01 04 32 de 00 01
[modbus] TRACE 2024/10/17 13:55:11 modbus: recv 00 0f 00 00 00 05 01 04 02 03 dc
Power:        415W
Energy:       1177.0kWh
Soc:          99%
Controllable: true
 
Im Logfile sieht es im regulären Betrieb hingegen so aus:

Oct 17 13:52:05 raspberrypi evcc[5763]: [modbus] TRACE 2024/10/17 13:52:05 modbus: send 00 25 00 00 00 06 01 04 13 98 00 02
Oct 17 13:52:05 raspberrypi evcc[5763]: [modbus] TRACE 2024/10/17 13:52:05 modbus: recv 00 25 00 00 00 07 01 04 04 19 61 00 00
Oct 17 13:52:05 raspberrypi evcc[5763]: [modbus] TRACE 2024/10/17 13:52:05 modbus: send 00 26 00 00 00 06 01 04 32 ca 00 02
Oct 17 13:52:05 raspberrypi evcc[5763]: [modbus] TRACE 2024/10/17 13:52:05 modbus: recv 00 26 00 00 00 07 01 04 04 fe 18 00 00
Oct 17 13:52:05 raspberrypi evcc[5763]: [modbus] TRACE 2024/10/17 13:52:05 modbus: send 00 27 00 00 00 06 01 04 13 98 00 02
Oct 17 13:52:05 raspberrypi evcc[5763]: [modbus] TRACE 2024/10/17 13:52:05 modbus: recv 00 27 00 00 00 07 01 04 04 19 61 00 00
Oct 17 13:52:05 raspberrypi evcc[5763]: [modbus] TRACE 2024/10/17 13:52:05 modbus: send 00 28 00 00 00 06 01 04 32 ca 00 02
[... 189 lines cut ...]
Oct 17 13:52:05 raspberrypi evcc[5763]: [cache ] TRACE 2024/10/17 13:52:05 lp-1/sessionPricePerKWh: <nil>
Oct 17 13:52:05 raspberrypi evcc[5763]: [cache ] TRACE 2024/10/17 13:52:05 lp-1/sessionPrice: <nil>
Oct 17 13:52:05 raspberrypi evcc[5763]: [cache ] TRACE 2024/10/17 13:52:05 lp-1/sessionCo2PerKWh: <nil>
Oct 17 13:52:05 raspberrypi evcc[5763]: [cache ] TRACE 2024/10/17 13:52:05 lp-1/chargedEnergy: 0
Oct 17 13:52:05 raspberrypi evcc[5763]: [cache ] TRACE 2024/10/17 13:52:05 lp-1/chargeDuration: 0s
Oct 17 13:52:05 raspberrypi evcc[5763]: [cache ] TRACE 2024/10/17 13:52:05 lp-1/effectivePriority: 0
Oct 17 13:52:05 raspberrypi evcc[5763]: [cache ] TRACE 2024/10/17 13:52:05 lp-1/effectivePlanTime: 0001-01-01 00:00:00 +0000 UTC
Oct 17 13:52:05 raspberrypi evcc[5763]: [cache ] TRACE 2024/10/17 13:52:05 lp-1/effectivePlanSoc: 0
Oct 17 13:52:05 raspberrypi evcc[5763]: [cache ] TRACE 2024/10/17 13:52:05 lp-1/effectiveMinCurrent: 6
Oct 17 13:52:05 raspberrypi evcc[5763]: [cache ] TRACE 2024/10/17 13:52:05 lp-1/effectiveMaxCurrent: 16
Oct 17 13:52:05 raspberrypi evcc[5763]: [cache ] TRACE 2024/10/17 13:52:05 lp-1/effectiveLimitSoc: 100
O
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.13

andig, 2024-10-17:
Evtl. 2x über Config UI konfiguriert?

heifisch, 2024-10-18:
Die Konfiguration haben wir auf der Konsole gemacht.
Sie war am Anfang etwas hakelig, aber wo sieht man, dass da etwas schief gegangen sein könnte?
Bzw. an welcher Stelle kann man das korrigieren?
Im Konfig-File sehe ich nur einen PV-Meter.

Im Log ist mir noch aufgefallen, dass es bei site 2x pv gibt.

Oct 18 06:40:10 raspberrypi evcc[8966]: [site  ] INFO 2024/10/18 06:40:10 site config:
Oct 18 06:40:10 raspberrypi evcc[8966]: [site  ] INFO 2024/10/18 06:40:10   meters:      grid ✓ pv ✓ battery ✓
Oct 18 06:40:10 raspberrypi evcc[8966]: [site  ] INFO 2024/10/18 06:40:10     grid:      power ✓ energy ✓ currents ✓
Oct 18 06:40:10 raspberrypi evcc[8966]: [site  ] INFO 2024/10/18 06:40:10     pv 1:      power ✓ energy ✓ currents ✗
Oct 18 06:40:10 raspberrypi evcc[8966]: [site  ] INFO 2024/10/18 06:40:10     pv 2:      power ✓ energy ✓ currents ✗
Oct 18 06:40:10 raspberrypi evcc[8966]: [site  ] INFO 2024/10/18 06:40:10     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
Oct 18 06:40:10 raspberrypi evcc[8966]: [site  ] INFO 2024/10/18 06:40:10   vehicles:
Oct 18 06:40:10 raspberrypi evcc[8966]: [site  ] INFO 2024/10/18 06:40:10     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓

heifisch, 2024-10-18:
Ich sehe gerade das die Logs der UI sich bezüglich Modbus-Abfragen unterscheiden.
Da werden die Leistungen korrekt angezeigt:

[site ] DEBUG 2024/10/18 08:00:02 ----
[lp-1 ] DEBUG 2024/10/18 08:00:02 charge power: 0W
[lp-1 ] DEBUG 2024/10/18 08:00:02 charge currents: [0 0 0]A
[db:1 ] TRACE 2024/10/18 08:00:02 modbus: send 08 7c 00 00 00 06 01 04 13 98 00 02
[db:1 ] TRACE 2024/10/18 08:00:02 modbus: recv 08 7c 00 00 00 07 01 04 04 00 e8 00 00
[db:1 ] TRACE 2024/10/18 08:00:02 modbus: send 08 7d 00 00 00 06 01 04 32 ca 00 02
[db:1 ] TRACE 2024/10/18 08:00:02 modbus: recv 08 7d 00 00 00 07 01 04 04 fe 4a 00 00
[site ] DEBUG 2024/10/18 08:00:02 pv power: 464W
[site ] DEBUG 2024/10/18 08:00:02 battery soc: 28%
[site ] DEBUG 2024/10/18 08:00:02 battery power: 169W
[site ] DEBUG 2024/10/18 08:00:02 grid power: 2W
[site ] DEBUG 2024/10/18 08:00:02 grid currents: [-0.206 -0.373 0.56]A
[site ] DEBUG 2024/10/18 08:00:02 site power: 271W
[lp-1 ] DEBUG 2024/10/18 08:00:02 charger status: A
[site ] DEBUG 2024/10/18 08:00:32 ----
[lp-1 ] DEBUG 2024/10/18 08:00:32 charge power: 0W
[lp-1 ] DEBUG 2024/10/18 08:00:32 charge currents: [0 0 0]A
[db:1 ] TRACE 2024/10/18 08:00:32 modbus: send 08 8e 00 00 00 06 01 04 13 98 00 02
[db:1 ] TRACE 2024/10/18 08:00:32 modbus: recv 08 8e 00 00 00 07 01 04 04 00 eb 00 00
[db:1 ] TRACE 2024/10/18 08:00:32 modbus: send 08 8f 00 00 00 06 01 04 32 ca 00 02
[db:1 ] TRACE 2024/10/18 08:00:32 modbus: recv 08 8f 00 00 00 07 01 04 04 fe 4a 00 00
[site ] DEBUG 2024/10/18 08:00:32 pv power: 470W
[site ] DEBUG 2024/10/18 08:00:32 battery soc: 28%
[site ] DEBUG 2024/10/18 08:00:32 battery power: 177W
[site ] DEBUG 2024/10/18 08:00:32 grid power: 4W
[site ] DEBUG 2024/10/18 08:00:32 grid currents: [-0.219 -0.387 0.578]A
[site ] DEBUG 2024/10/18 08:00:32 site power: 281W
[lp-1 ] DEBUG 2024/10/18 08:00:32 charger status: A

andig, 2024-10-18:
Du hast PV 2x konfiguriert. Einfach i UI löschen, dann gehts.

closed 2024-10-18 by andig

heifisch, 2024-10-18:
Vielen Dank, dass war es.
Ein PV-System im config-File definiert und eins in der UI.

Das passiert wenn 2 Leute dran rumschrauben ;-)

VolkerK62, 2024-10-18:
@andig  aber hätten dann nicht auch beide bei `evcc meter` angezeigt werden müssen

andig, 2024-10-18:
...wenn man die gleiche DB verwenden würde... ja.