evcc-io/evcc #16700: There ist no option to charge the home battery at minimum if I use time zones

jekruege, 2024-10-17:
### Describe the bug

I´m switched my energy contract from tibber to octopus energy.de. I configured my contract with time zones: 
tariffs:
  grid:
    type: fixed
    price: 0.294 # EUR/kWh (default)
    zones:
      - days: Mo-So
        hours: 0-5
        price: 0.15 # EUR/kWh
  currency: EUR

Now I haven't anymore the option to charge the battery at low prices.

### Steps to reproduce

1. Tap at battery settings
2. No charging the battery at a price level is possibles
3.
...

### Configuration details

```yaml
tariffs:

  grid:
    type: fixed
    price: 0.294 # EUR/kWh (default)
    zones:
      - days: Mo-So
        hours: 0-5
```

### Log details

```text
no log
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.13

naltatis, 2024-10-17:
Hi @jekruege, this is not enough information for us to understand the problem. You config seems incomplete. Pleaes update your description an we'll reopen once we have enough information to debug.

closed 2024-10-17 by naltatis

jekruege, 2024-10-18:
I have not that configuration window ("Netzladen") any more with my new configuration with price zones (nach dem Wechsel meines Lieferanten [von Tibber mit dynamischen Preisen zu Oktopus mit Preiszonen]):
 
[image]

naltatis, 2024-10-18:
Please share a more complete and formatted configuration. In the above example you've only one zone with no price. Not sure if this is really your configuration (which would be incomplete). In the text above you've pasted another snipped which has text but is not formatted. Please update your issue description, otherwise it's hard to help. 

[image]

jekruege, 2024-10-18:
Sorry, I'm not really fit with linux and I didn't find a way to get the yaml file out of the system ....

Do you have a clue for me?