evcc-io/evcc #16704: Sungrow SH15T issues with reading home battery status

leonidnf, 2024-10-17:
### Describe the bug

Evcc cannot read home battery data while using "sungrow-hybrid" template.
The inverter SH15T, communication device Winet-S2, and the Battery SBR128 are updated to the latest firmware.

### Steps to reproduce

Restart the Evcc

### Configuration details

```yaml
interval: 30s # control cycle interval

log: debug
levels:
  cache: error

modbusproxy:
  port: 5021
  uri: 192.168.200.133:502

meters:
  - name: grid1
    type: template
    template: sungrow-hybrid
    usage: grid
    modbus: tcpip
    id: 1
    host: 192.168.200.133 # Hostname
    port: 502 # Port
    timeout: 30s # optional for modbus: tcpip?

  - name: pv2
    type: template
    template: sungrow-hybrid
    usage: pv
    modbus: tcpip
    id: 1
    host: 192.168.200.133 # Hostname
    port: 502 # Port
    timeout: 30s # optional for modbus: tcpip

  - name: battery3
    type: template
    template: sungrow-hybrid
    usage: battery
    modbus: tcpip
    id: 1
    host: 192.168.200.133 # Hostname
    port: 502 # Port
    capacity: 12.8 # Akkukapazität in kWh (optional)
    timeout: 30s # optional for modbus: tcpip

chargers:
  - type: template
    template: twc3
    host: 192.168.200.9
    name: teslaCharger

vehicles:
  - name: teslay
    type: template
    template: tesla
    title: Tesla Y
    icon: car
    vin: xxxxx
    capacity: 59
    phases: 3
    accessToken: xxxxx
[... 200 lines cut ...]
[proxy-5021] TRACE 2024/10/17 20:18:04 read input: 00000000
[proxy-5021] TRACE 2024/10/17 20:18:04 read input: id 1 addr 5018 qty 1
[proxy-5021] TRACE 2024/10/17 20:18:04 read input: 0965
[proxy-5021] TRACE 2024/10/17 20:18:04 read input: id 1 addr 5019 qty 1
[proxy-5021] TRACE 2024/10/17 20:18:04 read input: 0949
[proxy-5021] TRACE 2024/10/17 20:18:04 read input: id 1 addr 5020 qty 1
[proxy-5021] TRACE 2024/10/17 20:18:04 read input: 0940
[proxy-5021] TRACE 2024/10/17 20:18:04 read input: id 1 addr 5032 qty 2
[proxy-5021] TRACE 2024/10/17 20:18:04 read input: 00000000
[proxy-5021] TRACE 2024/10/17 20:18:04 read input: id 1 addr 5034 qty 1
[proxy-5021] TRACE 2024/10/17 20:18:04 read input: 03e7
[proxy-5021] TRACE 2024/10/17 20:18:04 read input: id 1 addr 5241 qty 1
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.130.13

leonidnf, 2024-10-18:
Hi @andig ,
thank you for your fast response!

The first screenshot shows that the Battery SOC is 73% (which is correct), but the battery discharging is 0.0 w (which is not correct). In the second screenshot, it can be seen that the battery discharging 1.05 kW. In the EVCC logs can be seen that 

`[site  ] DEBUG 2024/10/17 20:17:53 battery power: 0W`

I assume that the EVCC sungrow-hybrid template can not properly calculate the charging/discharging power of the battery.
Please let me know if additional info is needed.

andig, 2024-10-18:
Was sagt

    evcc meter --log trace,db:error

Welche Werte stimmen nicht?

andig, 2024-10-18:
Lets see `evcc meter` please.

leonidnf, 2024-10-18:
@andig 
I have installed the Evcc on standalone linux server just for debugging and was able to pull the `evcc meter logs`

`[main  ] INFO 2024/10/18 15:15:24 evcc 0.130.13 (b313e3c5)
[main  ] INFO 2024/10/18 15:15:24 using config file: /etc/evcc.yaml
pv2
---
[pv2   ] TRACE 2024/10/18 15:15:24 modbus: send 00 01 00 00 00 06 01 04 13 98 00 02
[pv2   ] TRACE 2024/10/18 15:15:24 modbus: recv 00 01 00 00 00 07 01 04 04 01 ed 00 00
[pv2   ] TRACE 2024/10/18 15:15:24 modbus: send 00 02 00 00 00 06 01 04 32 ca 00 02
[pv2   ] TRACE 2024/10/18 15:15:24 modbus: recv 00 02 00 00 00 07 01 04 04 11 0f 00 00
Power:  493W
Energy: 436.7kWh

grid1
-----
[grid1 ] TRACE 2024/10/18 15:15:24 modbus: send 00 03 00 00 00 06 01 04 32 d1 00 02
[grid1 ] TRACE 2024/10/18 15:15:24 modbus: recv 00 03 00 00 00 07 01 04 04 00 00 00 00
[grid1 ] TRACE 2024/10/18 15:15:24 modbus: send 00 04 00 00 00 06 01 04 32 ec 00 02
[grid1 ] TRACE 2024/10/18 15:15:24 modbus: recv 00 04 00 00 00 07 01 04 04 14 4a 00 00
[grid1 ] TRACE 2024/10/18 15:15:24 modbus: send 00 05 00 00 00 06 01 04 15 e2 00 02
[grid1 ] TRACE 2024/10/18 15:15:24 modbus: recv 00 05 00 00 00 07 01 04 04 ff e3 ff ff
[grid1 ] TRACE 2024/10/18 15:15:24 modbus: send 00 06 00 00 00 06 01 04 13 9a 00 01
[grid1 ] TRACE 2024/10/18 15:15:24 modbus: recv 00 06 00 00 00 05 01 04 02 09 56
[grid1 ] TRACE 2024/10/18 15:15:24 modbus: send 00 07 00 00 00 06 01 04 15 e4 00 02
[grid1 ] TRACE 2024/10/18 15:15:24 modbus: recv 00 07 00 00 00 07 01 04 04 00 03 00 00
[grid1 ] TRACE 2024/10/18 15:15:24 modbus: send 00 08 00 00 00 06 01 04 13 9b 00 01
[grid1 ] TRACE 2024/10/18 15:15:24 modbus: recv 00 08 00 00 00 05 01 04 02 09 43
[grid1 ] TRACE 2024/10/18 15:15:24 modbus: send 00 09 00 00 00 06 01 04 15 e6 00 02
[grid1 ] TRACE 2024/10/18 15:15:24 modbus: recv 00 09 00 00 00 07 01 04 04 00 71 00 00
[grid1 ] TRACE 2024/10/18 15:15:24 modbus: send 00 0a 00 00 00 06 01 04 13 9c 00 01
[grid1 ] TRACE 2024/10/18 15:15:24 modbus: recv 00 0a 00 00 00 05 01 04 02 09 4a
Power:          -0W
Energy:         519.4kWh
Current L1..L3: -0.121A 0.0127A 0.475A

battery3
--------
[battery3] TRACE 2024/10/18 15:15:24 modbus: send 00 0b 00 00 00 06 01 04 32 dd 00 01
[battery3] TRACE 2024/10/18 15:15:24 modbus: recv 00 0b 00 00 00 05 01 04 02 00 91
[battery3] TRACE 2024/10/18 15:15:24 modbus: send 00 0c 00 00 00 06 01 04 32 c8 00 01
[battery3] TRACE 2024/10/18 15:15:24 modbus: recv 00 0c 00 00 00 05 01 04 02 00 00
[battery3] TRACE 2024/10/18 15:15:24 modbus: send 00 0d 00 00 00 06 01 04 32 c8 00 01
[battery3] TRACE 2024/10/18 15:15:24 modbus: recv 00 0d 00 00 00 05 01 04 02 00 00
[battery3] TRACE 2024/10/18 15:15:24 modbus: send 00 0e 00 00 00 06 01 04 32 e2 00 02
[battery3] TRACE 2024/10/18 15:15:24 modbus: recv 00 0e 00 00 00 07 01 04 04 01 6c 00 00
[battery3] TRACE 2024/10/18 15:15:24 modbus: send 00 0f 00 00 00 06 01 04 32 de 00 01
[battery3] TRACE 2024/10/18 15:15:24 modbus: recv 00 0f 00 00 00 05 01 04 02 00 f6
Power:        0W
Energy:       36.4kWh
Soc:          25%
Controllable: true
`

Battery 3 power is shown as `0`, but actually is 129W

andig, 2024-10-18:
Battery power is the odd one:

```yaml
  power:
    source: calc
    mul:
    - source: modbus
      {{- include "modbus" . | indent 4 }}
      register:
        address: 13021 # Battery power 32DD
        type: input
        decode: int16
    - source: calc
      add:
      - source: modbus
        {{- include "modbus" . | indent 6 }}
        register:
          type: input
          address: 13000 # Battery running state 32C8
          decode: bool16
          bitmask: 2 # Charging
        scale: -1
      - source: modbus
        {{- include "modbus" . | indent 6 }}
        register:
          type: input
          address: 13000 # Battery running state 32C8
          decode: bool16
          bitmask: 4 # Discharging
```

leonidnf, 2024-10-18:
From another Sungrow resource I see small difference in decode type for address 13021:
`      - name: Battery power raw
        unique_id: sg_battery_power_raw
        device_address: !secret sungrow_modbus_slave
        address: 13021 # reg 13022
        input_type: input
        data_type: uint16
        precision: 0
        unit_of_measurement: W
        device_class: power
        state_class: measurement
        scale: 1
        scan_interval: 10`
        
`      - name: Running state
        unique_id: sg_running_state
        device_address: !secret sungrow_modbus_slave
        address: 13000 # reg 13001
        input_type: input
        data_type: uint16
        precision: 0
        state_class: measurement
        scale: 1
        scan_interval: 10`

Could this be the reason?

andig, 2024-10-19:
That doesn't tell me anything.

leonidnf, 2024-10-20:
@andig Thank you! Will check the docs and maybe some other open source materials.
I see that evcc users have a similar issue:
https://github.com/evcc-io/evcc/issues/15915

leonidrm, 2024-10-21:
@andig solved my issue by changing the battery power calculation logic. I'm multiplying battery voltage with battery current (which comes with +/-) and having the power value needed by EVCC.
For this purpose, I created a separate template for my EVCC configuration.
  power:
    source: calc
    mul:
    - source: modbus
      {{- include "modbus" . | indent 4 }}
      register:
        address: 13019 # Battery voltage
        type: input
        decode: int16
      scale: 0.1
    - source: modbus
      {{- include "modbus" . | indent 4 }}
      register:
        address: 13020 # Battery current
        type: input
        decode: int16
      scale: 0.1

leonidrm, 2024-10-21:
Here is entire template in case someone will need it:
```yaml annotate
template: sungrow-sht-hybrid
covers: ["sungrow"]
products:
  - brand: Sungrow
    description:
      generic: SH Series Hybrid Inverter
capabilities: ["battery-control"]
requirements:
  description:
    de: Verbindungen über das WiNet-S-Dongle (WiFi oder LAN) funktionieren nur mit aktueller Firmware. Ältere Versionen liefern nicht alle benötigten Daten (Leistung, Ladestand).
    en: Connections via the WiNet-S dongle (WiFi or LAN) only work with the latest firmware. Older versions do not provide all required data (power, state of charge).
params:
  - name: usage
    choice: ["grid", "pv", "battery"]
    allinone: true
  - name: modbus
    choice: ["rs485", "tcpip"]
    baudrate: 9600
  - name: timeout
  - name: capacity
    advanced: true
render: |
  type: custom
  {{- if eq .usage "grid" }}
  power:
    source: modbus
    {{- include "modbus" . | indent 2 }}
    register:
      type: input
      address: 13009 # Export power
      decode: int32s
    scale: -1
  energy:
    source: modbus
    {{- include "modbus" . | indent 2 }}
    register:
      address: 13036 # Total Import Energy, 0.1kWh
      type: input
      decode: uint32s
    scale: 0.1
  currents:
  - source: calc
    div:
    - source: modbus
      {{- include "modbus" . | indent 4 }}
      register:
        type: input
        address: 5602 # Meter Phase A Active Power, 1W
        decode: int32s
    - source: modbus
      {{- include "modbus" . | indent 4 }}
      register:
        type: input
        address: 5018 # Phase A voltage, 0.1V
        decode: uint16
      scale: 0.1
  - source: calc
    div:
    - source: modbus
      {{- include "modbus" . | indent 4 }}
      register:
        type: input
        address: 5604 # Meter Phase B Active Power, 1W
        decode: int32s
    - source: modbus
      {{- include "modbus" . | indent 4 }}
      register:
        type: input
[... 148 lines cut ...]
          - name: power
            type: int
            config:
              source: modbus
              {{- include "modbus" . | indent 12 }}
              timeout: {{ .timeout }}
              register:
                address: 5627 # BDC rated power
                type: input
                decode: uint16
              scale: 100
          out:
          - name: power
            type: int
            config:
              source: modbus
              {{- include "modbus" . | indent 12 }}
              timeout: {{ .timeout }}
              register:
                address: 13051 # Charge/discharge power
                type: writesingle
                decode: uint16
  capacity: {{ .capacity }} # kWh
  {{- end }}
`

andig, 2024-10-21:
Sehr cool. Magst Du einen PR dafür machen? Vielen Dank!

andig, 2024-10-21:
You can open a PR without any permissions!

In terms of scope: is the battery power the only change? Then the implementation could use the current template and add a `switch` Plugin. One case for 13000 && (Discharge || Charge) being 0 (the change here) and a default case for the current solution.

I'd still appreciate a ticket with Sungrow. evcc has implemented the interface as per Modbus documentation. Inverter does not seem to follow that.

andig, 2024-10-21:
Our documentation is `TI_20211231_Communication Protocol of Residential Hybrid Inverter_V1.0.23_EN` from 2022. If anybody has more recent version that might help too.

wapman1979, 2024-10-24:
> Here is entire template in case someone will need it:

hi @leonidrm ich habe das gleiche Issue und such schon ein paar wochen nach einer lösung...
Daher will ich da unbeding testen :-)

Wo kopierts du deinen yaml code genau rein?
in das evcc.yaml?

leonidnf, 2024-10-24:
@wapman1979 Unfortunately right now is a little bit more complicated. You need either to edit the `templates/definition/meter/sungrow-hybrid.yaml` from evcc repository and rebuild the app, either to create a separate template in `templates/definition/meter/` directory and use it later in `evcc.yaml` configuration.

This week I'll try to find a little time and will create a PR to Evcc repository and then the fix will come to everyone just by updating the application to latest version.

andig, 2024-10-25:
Here's the diff of the two templates:

[image]

I would really like to avoid duplicating everything else. Could you please add the SHT modbus documentation here so we can take a look how to auto-switch if possible?

Afrouper, 2024-10-26:
Hi,

I have also the same problem and interested in the PR.
I am not familiar with the modbus protocol and a newbie with my 1 week old PV 😉

But I found a new "[version](https://community.openhab.org/uploads/short-url/8AQhNuy0LHYTsM8awDaWMbYiWeD.pdf)" of the specification; perhaps this will help: 
[Communication Protocol of Residential Hybrid Inverter-V1.1.4.pdf](https://github.com/user-attachments/files/17531639/Communication.Protocol.of.Residential.Hybrid.Inverter-V1.1.4.pdf)

leonidnf, 2024-10-26:
@Afrouper Thank you for the updated documentation. This explains everything!
@andig  Sungrow changed the running state to power flows (point 11). We should change the address from 13000 to 13001
[image]

I have tried to push the branch and make a PR but I'm getting a permission error:
[image]

@andig can you please change the address from `sungrow-hybrid.yaml:123 and 131` to 13001?

Afrouper, 2024-10-26:
Thanks.

@leonidnf You should (and cannot) not push direct to the evcc projects master. I think you have to create a fork of the evcc repo. In this fork you can make the changes and then create a PR to the evcc project.

@andig Should this Issue be still open and not closed?

Thanks Christian

Afrouper, 2024-10-27:
> @andig can you please change the address from `sungrow-hybrid.yaml:123 and 131` to 13001?

Hi @leonidnf,
thanks to time change this night I have some time to read the specifications. I think that sungrow did not do some breaking changes - they only do some renaming and did not change the meanings of the address. So the configured address 13000 seems to be correct because the addresses started at 1 and not 0; so you have to add 1 to the configured addresses in der YAML file.
(No garantee - AI helped me with this stuff 😇)

Please double check. Might there be a fault in the sungrow implementation?

leonidrm, 2024-10-27:
Hi @Afrouper,
Unfortunately, I'm not an expert in Modbus, but I have general understanding how this works. Currently, my setup works on an evcc-fork that uses `battery-current` and `battery-voltage` to calculate the `battery-power`. 
Now I'm testing `13001` and it gives same results.

In this screenshot we have results for `13000` address. The `battery-power` can not be calculated and its value is shown as `0`:
[image]

In following screenshot we have results for `13001` address. The `battery-power` was able to be calculated and we see a  value of `-231W`
[image]

andig, 2024-10-27:
Sungrow has documented a change to 13000/13001 but totally fails to mention which devices this applies to. Could someone please contact Sungrow and ask how to detect which register is correct? It *MUST* be clear from outside without guessing. At least that would be my hope…

Afrouper, 2024-10-27:
> Could someone please contact Sungrow and ask how to detect which register is correct?

I just created service request CL2410270267 in the [sungrow portal](https://gsp.sungrow.cn) - hope I get an answer.

But an additional question...
According to appendix "4.2 Power Flow Status" it is the same as in the older documentation ("Appendix 1.2 Running state"). For me it sounds like they just renamed "Battery maintenance" to "Power Flow Status" - the address 13001 is still the same.

@andig Just for my understanding. When writing 13000 in the `sungrow-hybrid.yaml` evcc will access modbus address "13001", correct? As I understand there is an offset of 1.
With this in mind all other addresses in `sungrow-hybrid.yaml` seems to be correct.

@leonidrm Thanks for testing - looks good. Are you sure the when reading from 13001 you do not get data from "Daily PV Generation"?

Thanks and sorry if I am totally wrong with the meanings from modbus and the addresses.

andig, 2024-10-27:
> @andig Just for my understanding. When writing 13000 in the sungrow-hybrid.yaml evcc will access modbus address "13001", correct? As I understand there is an offset of 1.

No, no and no. 13000 means 13000. The Sungrow docs have +1 offset. 13001 per doc is 13000 physically.

Could anyone please give access to info@evcc.io for the SH15T so we can verify what's going on (VPN, Tailscale etc...). Thanks!

andig, 2024-10-27:
@leonidnf I have compared the old and the new docs. While they have changed the naming, 13001 is still the same register (and physically 13000). It is not clean what might have changed here.

wapman1979, 2024-10-27:
@andig 
would it be possible to add a third sungrow inverter categorie which is using the configuration from @leonidrm?
So others could test/use it meanwhile the clarification with Sungrow is on going?

I also raised this issue a couple of weeks ago.
What I saw, was that users of the new SHxxT Series all were struggling with that Battery issue.
Also within the mkaiser integration. But there it was solved with the change of the calculation, simlar to that what @leonidrm now suggested here.

For me it would be a big comfort step in using evcc.
Currently, I have to care manually that the car is not discharging the battery.

Afrouper, 2024-10-27:
Thanks for your explanation.

> Could anyone please give access to [info@evcc.io](mailto:info@evcc.io) for the SH15T so we can verify what's going on (VPN, Tailscale etc...). Thanks!

I will try to give you access to the SH15T via Tailscale; but first I have to change and secure my network setup 😇
The SHxxT series do not have a separate ethernet port; modbus must be handled via WiNet-S Dongle.

Thanks Christian

Afrouper, 2024-10-29:
I got an answer from sungrow (translated from german):

> Dear customer,
> We are sending you the latest document containing the communication protocols in the attachment. The document applies to both the SHxxT and SHxxRT series inverters. You will also find the information from the training on the SHxxT inverter series attached.
> 
> Please feel free to attend our next training for installers!
> Training for installers: https://germany-sungrow.clickmeeting.com/

[TI_20231019_Communication Protocol of Residential Hybrid Inverter_V1.1.2_EN.pdf](https://github.com/user-attachments/files/17560350/TI_20231019_Communication.Protocol.of.Residential.Hybrid.Inverter_V1.1.2_EN.pdf)

Hope this helps to clarify the meanings of the modbus addresses.

@andig: I've forwarded the original email to you.

andig, 2024-10-29:
@leonidnf @Afrouper just did a test via tailscale:

```
[shBattery] TRACE 2024/10/29 20:38:45 modbus: send 00 01 00 00 00 06 01 04 32 dd 00 01 # 13021
[shBattery] TRACE 2024/10/29 20:38:46 modbus: recv 00 01 00 00 00 05 01 04 02 03 01    # <- 500W
[shBattery] TRACE 2024/10/29 20:38:46 modbus: send 00 02 00 00 00 06 01 04 32 c8 00 01 # 13000
[shBattery] TRACE 2024/10/29 20:38:46 modbus: recv 00 02 00 00 00 05 01 04 02 00 00    # 0!
```

Log clearly shows that power flow status (13000 aka 13001) is 0. This MUST NOT be.

The same happens when I put the battery into charging mode:

```
[shBattery] TRACE 2024/10/29 20:42:55 modbus: send 00 01 00 00 00 06 01 04 32 dd 00 01
[shBattery] TRACE 2024/10/29 20:42:55 modbus: recv 00 01 00 00 00 05 01 04 02 1f 92    # <- 8kW
[shBattery] TRACE 2024/10/29 20:42:55 modbus: send 00 02 00 00 00 06 01 04 32 c8 00 01
[shBattery] TRACE 2024/10/29 20:42:55 modbus: recv 00 02 00 00 00 05 01 04 02 00 00    # 0!
```

Power flow status is still 0.

I can only suggest to contact Sungrow _why_ that is and show them the modbus traces.

leonidnf, 2024-10-29:
@andig can you please try same thing with 13001 address?
I'm using this address for last few days and the data is accurate (same as I see in iSolar cloud)

Afrouper, 2024-10-29:
> I can only suggest to contact Sungrow _why_ that is and show them the modbus traces.

I will send the information to sungrow support.

premultiply, 2024-10-29:
One more thing to test:

`37. Battery Current 5631 S16 0.1A` => Reg 5630 (Input)

What does it contain?

It SHOULD contain the signed value of
`84. Battery current 13021 U16 0.1A` => Reg 13020 (Input)

Afrouper, 2024-10-31:
> > I can only suggest to contact Sungrow _why_ that is and show them the modbus traces.
> 
> I will send the information to sungrow support.

Next week there should come an update to the WiNet-S2 Communication module to address this issue 😀

> Sehr geehrter Kunde,
> vielen Dank für Ihre Antwort.
> 
> Das Problem wird mit dem nächsten Update des WiNet-S2 behoben. Es sollte nächste Woche in iSC verfügbar sein. Aktualisieren Sie bitte das Kommuniaktionsmodul und lassen Sie uns bitte bis 7.11. wissen, ob das Ticket geschlossen werden kann.
> 
> Mit freundlichen Grüßen
> Sungrow Service Partner

So I have to hope I will also get an update for my WiNet-S module (pending with support)
Please check your updates and report if there is a fix.

andig, 2024-10-31:
Wir haben eine ähnliche Antwort bekommen:

> Bitte das Firmwareupdate des WiNet-S auf die Firmwareversion 27 durchführen welches gestern erschienen ist.

Kann das jemand ausprobieren?

Afrouper, 2024-10-31:
Hi,

heute Nachmittag noch nicht verfügbar jetzt schon.
Gerade das Update durchgeführt und kurz die Batterie per Netz geladen; sieht gut aus:

`

shBattery
---------
[shBattery] TRACE 2024/10/31 20:56:05 modbus: send 00 0b 00 00 00 06 01 04 32 dd 00 01
[shBattery] TRACE 2024/10/31 20:56:05 modbus: recv 00 0b 00 00 00 05 01 04 02 1f cf
[shBattery] TRACE 2024/10/31 20:56:05 modbus: send 00 0c 00 00 00 06 01 04 32 c8 00 01
[shBattery] TRACE 2024/10/31 20:56:05 modbus: recv 00 0c 00 00 00 05 01 04 02 00 22
[shBattery] TRACE 2024/10/31 20:56:05 modbus: send 00 0d 00 00 00 06 01 04 32 c8 00 01
[shBattery] TRACE 2024/10/31 20:56:05 modbus: recv 00 0d 00 00 00 05 01 04 02 00 22
[shBattery] TRACE 2024/10/31 20:56:05 modbus: send 00 0e 00 00 00 06 01 04 32 e2 00 02
[shBattery] TRACE 2024/10/31 20:56:05 modbus: recv 00 0e 00 00 00 07 01 04 04 05 d0 00 00
[shBattery] TRACE 2024/10/31 20:56:05 modbus: send 00 0f 00 00 00 06 01 04 32 de 00 01
[shBattery] TRACE 2024/10/31 20:56:05 modbus: recv 00 0f 00 00 00 05 01 04 02 00 39
Power:        -8143W
`

Auch in der UI sieht es gut aus.

Wenn ich die Antworten von Sungrow dann korrekt verstehe war der WiNet-S heute dran und der WiNet-S2 kommt nächste Woche.

Aus meiner Sicher erst mal OK. Beobachte morgen mal mit PV.

Danke an alle

closed 2024-10-31 by andig