evcc-io/evcc #16709: myToyota Schnittstelle

JohnnyTribi, 2024-10-18:
Es gibt von Toyota eine Schnittstelle welche bereits funktioniert:
https://github.com/DurgNomis-drol/mytoyota

Kann man das in EVCC einbauen um wenigstens den aktuellen Akkustand auszulesen?

andig, 2024-10-18:
Das kannst Du über die evcc Plugins einbinden, in dem Fall z.B. das `script` Plugin.

closed 2024-10-18 by andig

JohnnyTribi, 2024-10-23:
Hallo Andig. Habe leider herausgefunden, dass es mit dem Toyota Plugin nicht möglich ist den SOC vom Fahrzeug abzurufen. Mit ABRP über Enode kann ich ihn allerdings abrufen, insofern ist es grundsätzlich möglich... Gibt es eine Möglichkeit, dass Toyota hier doch noch implementiert wird?

andig, 2024-10-23:
Du kannst das jederzeit als `custom` vehicle integrieren. Da Toyota bisher nur ein Nischenplayer ist lohnt es leider nicht, da mehr Aufwand rein zu stecken :(

andre-bu, 2024-11-04:
Hallo, auch wenn das Usue schon geschlossen ist hange ich mich mal ran. Auch wenn Toyota im E- Markt keine Rolle spielt wäre es super wenn die hier mit aufgenommen werden. Gibt es denn wirklich keine Möglichkeit den Akkustand aus der my toyota App in EVCC zu intgrieren?

JohnnyTribi, 2024-11-05:
Hallo Andig
Ich verstehe den Punkt mit der „Nische“, denke aber, dass das bei Toyota nicht ganz zutrifft: Toyota ist einer der grössten Automobilhersteller weltweit und Vorreiter bei Hybrid- und Plug-in-Hybridfahrzeugen. Die Nachfrage nach ihren E-Fahrzeugen wächst stetig, und Toyota hat grosse Pläne für die Zukunft mit 70 neuen elektrifizierten Modellen bis 2030.
Ein Toyota-Plugin könnte also viele Nutzer ansprechen und wäre eine wertvolle Ergänzung für EVCC. Würde mich freuen, wenn ihr das nochmal überdenkt! :)

andig, 2024-11-05:
Frag gerne Toyota ob sie ein Api anbieten das wir nutzen können.

andre-bu, 2024-11-05:
Jupp mache ich. Es muss ja eigentlich ne Möglichkeit geben. Für den
Iobroker gibt es ja einen Adapter der unter anderem den Ladestand anzeigt.

andre-bu, 2024-11-05:
@andig So Anfrage an Toyota zwecks API ist raus. Mal sehen ob eine Rückmeldung kommt. Vielleicht können ja andere Interessenten auch noch eine Anfrage bei Toyota stellen. ;-) Ist sicher hilfreich wenn das Thema API da häufiger aufkommt.

JohnnyTribi, 2024-11-05:
Wohin hast du die Anfrage geschickt?

andre-bu, 2024-11-05:
@JohnnyTribi kundenbetreuung@toyota.de

JohnnyTribi, 2024-11-06:
Habe auch mal angefragt...