evcc-io/evcc #16721: Error not with GO-E Charger

BeginnerEVCC, 2024-10-18:
### Describe the bug

Following error pops up in EVCC. The function itself seems to work. Is anyone able to assist? Do NOT Close this case unless you have some proper advise for someone who is not a pro, thanks!

@andig - do not close this case unless you are able to assist. Anything else is just senseless in regards of being helpfull. 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Garage: charger status: Get \"http://192.XXX.XXX.XX/api/status?filter=alw,car,eto,nrg,wh,trx,cards\": dial tcp 192.XXX.XXX.XX:80: connect: no route to host
```

### Log details

```text
Garage: charger status: Get \"http://192.XXX.XXX.XX/api/status?filter=alw,car,eto,nrg,wh,trx,cards\": dial tcp 192.XXX.XXX.XX:80: connect: no route to host
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

mfuchs1984, 2024-10-18:
You should open a discussion instead, this is obviously not an evcc issue.

mfuchs1984, 2024-10-18:
As mentioned, create at discussions instead and provide more context. https://github.com/evcc-io/evcc/discussions

closed 2024-10-18 by andig