evcc-io/evcc #16726: volvo-connect: unauthorized_client

triple-it, 2024-10-18:
### Describe the bug

Hi..
Is the (hard-coded) client suddenly become invalid?

```
[volvo-cc] TRACE 2024/10/18 21:28:49 POST https://volvoid.eu.volvocars.com/as/token.oauth2
[volvo-cc] TRACE 2024/10/18 21:28:49 access_token_manager_id=JWTh4Yf0b&grant_type=password&password=***&scope=openid+vehicle%3Aattributes+energy%3Arecharge_status+energy%3Abattery_charge_level+energy%3Aelectric_range+energy%3Aestimated_charging_time+energy%3Acharging_connection_status+energy%3Acharging_system_status+conve%3Afuel_status+conve%3Aodometer_status+conve%3Aenvironment&username=***
--
{"error":"unauthorized_client"}
[main  ] ERROR 2024/10/18 21:28:49 creating vehicle db:1 failed: cannot create vehicle type 'template': cannot create vehicle type 'volvo-connected': unexpected status: 400 (Bad Request)

```
This is in the code

```
const (
	managerId = "JWTh4Yf0b"
	basicAuth = "Basic aDRZZjBiOlU4WWtTYlZsNnh3c2c1WVFxWmZyZ1ZtSWFEcGhPc3kxUENhVXNpY1F0bzNUUjVrd2FKc2U0QVpkZ2ZJZmNMeXc="
)
```

### Steps to reproduce

curl -X POST https://volvoid.eu.volvocars.com/as/token.oauth2 \
  -H "Authorization: Basic aDRZZjBiOlU4WWtTYlZsNnh3c2c1WVFxWmZyZ1ZtSWFEcGhPc3kxUENhVXNpY1F0bzNUUjVrd2FKc2U0QVpkZ2ZJZmNMeXc=" \
  -H "Content-Type: application/x-www-form-urlencoded" \
  -d "access_token_manager_id=JWTh4Yf0b" \
  -d "grant_type=password" \
  -d "username=invalid_user_does_not_matter" \
  -d "password=invalid_pass_does_not_matter" \
  -d "scope=openid vehicle:attributes energy:recharge_status energy:battery_charge_level energy:electric_range energy:estimated_charging_time energy:charging_connection_status energy:charging_system_status conve:fuel_status conve:odometer_status conve:environment"
  
  {"error":"unauthorized_client"}
  
  

### Configuration details

```yaml
Not appliccable
```

### Log details

```text
{"error":"unauthorized_client"}
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.13 (not yet relevant to issue,  current master version has same credentials)

andig, 2024-10-19:
Seems Volvo has made changes. Do we know which?

andig, 2024-10-19:
See https://github.com/TA2k/ioBroker.volvo/issues/37#issuecomment-2423673823

mmierke, 2024-10-20:
I don't know what they have changed, but I can confirm that Volvo2mqtt is working again.

mmierke, 2024-10-20:
https://github.com/Dielee/volvo2mqtt

mmierke, 2024-10-20:
They have updated to V2 API. What ever that means.

mmierke, 2024-10-20:
There is a volvo.py in that repository src/volvo.py

andig, 2024-10-20:
Seems this requires 2FA now which we don't support. Long time ago, we've asked Volvo to unboard us on their official APIs. This hasn't happened. We're not going to add more ugly workaround.

mmierke, 2024-10-20:
So back to the generic car.

mmierke, 2024-10-20:
But is this
As of version v1.9.0, this addon uses the same OTP authentication as the Volvo app. The following steps are required for authentication in exactly this order:

    Setup volvo2Mqtt, either via Docker, or via HA addon (take a look at the "Setup" section below)

    Fill in your settings and start volvo2Mqtt

    Your log will show the following lines
    Waiting for otp code... Please check your mailbox and post your otp code to the following mqtt topic "volvoAAOS2mqtt/otp_code". Retry 0/15
    Waiting for otp code... Please check your mailbox and post your otp code to the following mqtt topic "volvoAAOS2mqtt/otp_code". Retry 1/15
    etc ...

    Now, open your mailbox and copy your OTP Code

    Open HomeAssistant and search for the entity ID text.volvo_otp

    Paste your OTP into the text entity and press Enter
    6a. (optional) If your OTP cannot be safed, try to paste your OTP code via MQTT to the following topic: volvoAAOS2mqtt/otp_code

    OTP Post Script

If everything has worked, your addon is now authenticated. In the future, OTP authentication only needs to be done when updating, not when restarting the container.

really an ugly workaround?

FritsStau, 2024-10-21:
I've got a bad request 400 error on my Volvo template running Evcc as an addon in HS. Does this mean that we cannot use the templates in evcc due to the 2FA on Volvo's side?

MichielVanwelsenaere, 2024-10-21:
> I've got a bad request 400 error on my Volvo template running Evcc as an addon in HS. Does this mean that we cannot use the templates in evcc due to the 2FA on Volvo's side?

yes, that's exactly what it means.
+1 on this issue.

mmierke, 2024-10-21:
What is the reason behind 2FA not being supported?

mmierke, 2024-10-21:
I understand, that in the yaml approach it is complicated to integrate 2FA, but with the new configuration screen it should be possible to redirect to Volvo and deal with 2FA.

There is even a kind of exmaple flow.

https://github.com/volvo-cars/developer-portal-api-samples/tree/main/oauth2-code-flow-sample

andig, 2024-10-21:
That's exactly the API that Volvo has not onboarded us on... Once they do that's the way to go.

triple-it, 2024-10-21:
> That's exactly the API that Volvo has not onboarded us on... Once they do that's the way to go.
Ah, so you didn't receive a client_id? And the app is not yet published by Volvo?
Who should we have to e-mail at Volvo to ask for more information? Do you have any contacts for us?

mmierke, 2024-10-21:
I think you can get a client id without even having a Volvo. But somehow the evcc user needs to login with his or her client id to get connected to their cars.

@andig Where are you stuck in the process of onboarding?

https://developer.volvocars.com/apis/docs/authorisation/

mmierke, 2024-10-21:
But you should be able to onboard even with your github account

[image]

triple-it, 2024-10-21:
Idea: Creating a volvoAAOS2mqtt vehicle type?
(Using: https://github.com/Dielee/volvo2mqtt)

Playing with the idea, i unfortunately see that the DemoCar VIN is not very supportive.. so some real data is needed.

```
app-1  | Oct 21 13:14:22 volvo2mqtt [1] - INFO: Vin: ['LPSEFAVS2NPOLDEMO'] found!
app-1  | Oct 21 13:14:22 volvo2mqtt [1] - ERROR: API Call failed. Status Code: 403. Error: {
app-1  |   "status" : 403,
app-1  |   "operationId" : "ff862ec3-f787-4a09-89db-f21d6a3d1472",
app-1  |   "error" : {
app-1  |     "message" : "FORBIDDEN",
app-1  |     "description" : "No relationship to UUID"
app-1  |   }
app-1  | }
app-1  | Oct 21 13:14:22 volvo2mqtt [1] - INFO: Failed, Battery Charge Level is unfortunately not supported by your vehicle.

```

But a real VIN is very responsive with the following data every 300 seconds on mqtt...

```
volvoAAOS2mqtt/availability online
homeassistant/sensor/ANONYMIZED_VIN_battery_charge_level/state 75.0
homeassistant/sensor/volvoAAOS2mqtt/ANONYMIZED_VIN_battery_charge_level/config {"name": "Battery Charge Level", "object_id": "volvo_ANONYMIZED_VIN_battery_charge_level", "schema": "state", "icon": "mdi:battery-70", "state_topic": "homeassistant/sensor/ANONYMIZED_VIN_battery_charge_level/state", "device": {"identifiers": ["volvoAAOS2mqtt_ANONYMIZED_VIN"], "manufacturer": "Volvo", "model": "EX30", "name": "EX30 (2025) - ANONYMIZED_VIN"}, "unique_id": "volvoAAOS2mqtt_ANONYMIZED_VIN_battery_charge_level", "availability_topic": "volvoAAOS2mqtt/availability", "device_class": "battery", "unit_of_measurement": "%", "state_class": "measurement"}
homeassistant/sensor/ANONYMIZED_VIN_battery_capacity/state 66.0
homeassistant/sensor/ANONYMIZED_VIN_electric_range/state 356
homeassistant/sensor/ANONYMIZED_VIN_estimated_charging_time/state 0
homeassistant/sensor/ANONYMIZED_VIN_charging_system_status/state Done
homeassistant/sensor/ANONYMIZED_VIN_charging_connection_status/state Connected AC
homeassistant/sensor/ANONYMIZED_VIN_odometer/state 1436
homeassistant/sensor/ANONYMIZED_VIN_last_data_update/state 21.10.2024, 12:55:40
homeassistant/sensor/ANONYMIZED_VIN_active_schedules/attributes {"timers": []}
homeassistant/binary_sensor/ANONYMIZED_VIN_window_front_left/state OFF
homeassistant/binary_sensor/ANONYMIZED_VIN_window_front_right/state OFF
homeassistant/binary_sensor/ANONYMIZED_VIN_window_rear_left/state OFF
homeassistant/binary_sensor/ANONYMIZED_VIN_window_rear_right/state OFF
homeassistant/binary_sensor/ANONYMIZED_VIN_door_front_left/state OFF
homeassistant/binary_sensor/volvoAAOS2mqtt/ANONYMIZED_VIN_door_front_left/config {"name": "Door Front Left", "object_id": "volvo_ANONYMIZED_VIN_door_front_left", "schema": "state", "icon": "mdi:car-door-lock", "state_topic": "homeassistant/binary_sensor/ANONYMIZED_VIN_door_front_left/state", "device": {"identifiers": ["volvoAAOS2mqtt_ANONYMIZED_VIN"], "manufacturer": "Volvo", "model": "EX30", "name": "EX30 (2025) - ANONYMIZED_VIN"}, "unique_id": "volvoAAOS2mqtt_ANONYMIZED_VIN_door_front_left", "availability_topic": "volvoAAOS2mqtt/availability", "device_class": "door"}
homeassistant/binary_sensor/ANONYMIZED_VIN_door_front_right/state OFF
homeassistant/binary_sensor/volvoAAOS2mqtt/ANONYMIZED_VIN_door_front_right/config {"name": "Door Front Right", "object_id": "volvo_ANONYMIZED_VIN_door_front_right", "schema": "state", "icon": "mdi:car-door-lock", "state_topic": "homeassistant/binary_sensor/ANONYMIZED_VIN_door_front_right/state", "device": {"identifiers": ["volvoAAOS2mqtt_ANONYMIZED_VIN"], "manufacturer": "Volvo", "model": "EX30", "name": "EX30 (2025) - ANONYMIZED_VIN"}, "unique_id": "volvoAAOS2mqtt_ANONYMIZED_VIN_door_front_right", "availability_topic": "volvoAAOS2mqtt/availability", "device_class": "door"}
homeassistant/binary_sensor/ANONYMIZED_VIN_door_rear_left/state OFF
homeassistant/binary_sensor/volvoAAOS2mqtt/ANONYMIZED_VIN_door_rear_left/config {"name": "Door Rear Left", "object_id": "volvo_ANONYMIZED_VIN_door_rear_left", "schema": "state", "icon": "mdi:car-door-lock", "state_topic": "homeassistant/binary_sensor/ANONYMIZED_VIN_door_rear_left/state", "device": {"identifiers": ["volvoAAOS2mqtt_ANONYMIZED_VIN"], "manufacturer": "Volvo", "model": "EX30", "name": "EX30 (2025) - ANONYMIZED_VIN"}, "unique_id": "volvoAAOS2mqtt_ANONYMIZED_VIN_door_rear_left", "availability_topic": "volvoAAOS2mqtt/availability", "device_class": "door"}
homeassistant/binary_sensor/ANONYMIZED_VIN_door_rear_right/state OFF
homeassistant/binary_sensor/volvoAAOS2mqtt/ANONYMIZED_VIN_door_rear_right/config {"name": "Door Rear Right", "object_id": "volvo_ANONYMIZED_VIN_door_rear_right", "schema": "state", "icon": "mdi:car-door-lock", "state_topic": "homeassistant/binary_sensor/ANONYMIZED_VIN_door_rear_right/state", "device": {"identifiers": ["volvoAAOS2mqtt_ANONYMIZED_VIN"], "manufacturer": "Volvo", "model": "EX30", "name": "EX30 (2025) - ANONYMIZED_VIN"}, "unique_id": "volvoAAOS2mqtt_ANONYMIZED_VIN_door_rear_right", "availability_topic": "volvoAAOS2mqtt/availability", "device_class": "door"}
homeassistant/binary_sensor/ANONYMIZED_VIN_tailgate/state OFF
homeassistant/binary_sensor/ANONYMIZED_VIN_engine_hood/state OFF
homeassistant/binary_sensor/ANONYMIZED_VIN_tank_lid/state ON
homeassistant/binary_sensor/ANONYMIZED_VIN_sunroof/state OFF
homeassistant/switch/ANONYMIZED_VIN_climate_status/state OFF
homeassistant/lock/ANONYMIZED_VIN_lock_status/state LOCKED
[... cut ...]
So this requires only to have a volvo2mqtt-app (https://github.com/Dielee/volvo2mqtt) installed, and a volvo-mqtt vehicle type to be created.

I would rather have a nice and clean programmed evcc go implementation of the volvo API in place of course.

But in the meantime, while we wait for volvo, it doesn't feel too kludgy for me to:
- Create a mqtt vehicle type  (Could be re-used for future mqtt vehicles ...)
- Pull in a well supported homeassistant vehicle data via MQTT: volvoAAOS2mqtt

Any ideas/thoughts for the volvoAAOS2mqtt vehicle type support?

andig, 2024-10-21:
We have the same for Mazda, happy tonsuppoert a PR.

mmierke, 2024-10-21:
ecvv yaml

vehicles:
  - name: myCar
    title: Volvo XC60 Recharge
    type: custom
    icon: car
    capacity: 19
    phases: 2
    onIdentify:
      mode: pv # enable PV-charging when vehicle is identified (minpv|now|off|pv)
    soc: # battery soc (%)
      source: http
      uri: http://192.168.178.202:8123/api/states/sensor.volvo_<<whatever>>_battery_charge_level
      method: GET # default HTTP method
      headers:
        - authorization: Bearer <<some bearer token>>
        - content-type: application/json
      jq: .state | tonumber # parse response json
    range: # optional electric range (km)
      source: http
      uri: http://192.168.178.202:8123/api/states/sensor.volvo_<<whatever>>_electric_range
      method: GET # default HTTP method
      headers:
        - authorization: Bearer <<some bearer token>>
        - content-type: application/json
      jq: .state | tonumber # parse response json
    odometer:
      source: http
      uri: http://192.168.178.202:8123/api/states/sensor.volvo_<<whatever>>_odometer
      method: GET # default HTTP method
      headers:
        - authorization: Bearer <<some bearer token>>
        - content-type: application/json
      jq: .state | tonumber # parse response json
    status:
      source: http
      uri: http://192.168.178.202:8123/api/states/sensor.xc60_charging_status_a_f
      method: GET # default HTTP method
      headers:
        - authorization: Bearer <<some bearer token>>
        - content-type: application/json
      jq: .state  # parse response json  

Template for helper sensor sensor.xc60_charging_status_a_f

{% if is_state("sensor.volvo_<<whatever>>_charging_connection_status","Disconnected")-%}
  A
{%- elif is_state("sensor.volvo_<<whatever>>_charging_connection_status","Connected AC")-%}
  {% if is_state("sensor.volvo_<<whatever>>_charging_system_status","Charging")-%}
    C
  {%- else -%}
    B
  {%- endif %}  
{%- else -%}
  F
{%- endif %}

mmierke, 2024-10-21:
evcc yaml:

vehicles:
  - name: myCar
    title: Volvo XC60 Recharge
    type: custom
    icon: car
    capacity: 19
    phases: 2
    onIdentify:
      mode: pv # enable PV-charging when vehicle is identified (minpv|now|off|pv)
    soc: # battery soc (%)
      source: http
      uri: http://xxx.xxx.xxx.xxx:8123/api/states/sensor.volvo_<whatever>_battery_charge_level
      method: GET # default HTTP method
      headers:
        - authorization: Bearer <some bearer token>
        - content-type: application/json
      jq: .state | tonumber # parse response json
    range: # optional electric range (km)
      source: http
      uri: http://xxx.xxx.xxx.xxx:8123/api/states/sensor.volvo_<whatever>_electric_range
      method: GET # default HTTP method
      headers:
        - authorization: Bearer <some bearer token>
        - content-type: application/json
      jq: .state | tonumber # parse response json
    odometer:
      source: http
      uri: http://xxx.xxx.xxx.xxx:8123/api/states/sensor.volvo_<whatever>_odometer
      method: GET # default HTTP method
      headers:
        - authorization: Bearer <some bearer token>
        - content-type: application/json
      jq: .state | tonumber # parse response json
    status:
      source: http
      uri: http://xxx.xxx.xxx.xxx:8123/api/states/sensor.xc60_charging_status_a_f
      method: GET # default HTTP method
      headers:
        - authorization: Bearer <some bearer token>
        - content-type: application/json
      jq: .state  # parse response json  

helper sensor (sensor.xc60_charging_status_a_f):

{% if is_state("sensor.volvo_<whatever>_charging_connection_status","Disconnected")-%}
  A
{%- elif is_state("sensor.volvo_<whatever>_charging_connection_status","Connected AC")-%}
  {% if is_state("sensor.volvo_<whatever>_charging_system_status","Charging")-%}
    C
  {%- else -%}
    B
  {%- endif %}  
{%- else -%}
  F
{%- endif %}

mmierke, 2024-10-21:
So basically, because I use evcc as an add on in ha, I have created a vehicle, that pulls the values from HA directly

mmierke, 2024-10-21:
You need to create a bearer token

Robbi12, 2024-10-22:
Hallo. mmierke ich hätte eine Frage. Funktioniert das auch wenn ich homeassistant nicht habe. Ich habe nur den iobroker. Was muss ich denn nun genau eingeben.

Vielleicht kannst du mir helfen

triple-it, 2024-10-22:
Gutentag Robbi12.

I think mmierke uses the REST api from homeassistant, to pull in the data for evcc (where evcc is again an addon for ha to make things complicated :-) ).

I am using evcc using #16771 with just volvo2mqtt without homeassistant.
So I am using an evcc "volvo2mqtt" vehicle template, instead of volvo or volvo-connected, which just uses [volvo2mqtt](https://github.com/Dielee/volvo2mqtt) which was written for homeassistant, but is standalone usable.

In the template there are homeassistant topic paths, but I am not using homeassistant....

```
  soc:
    source: mqtt
    topic: homeassistant/sensor/{{ .vin }}_battery_charge_level/state
    timeout: {{ .timeout }}
```

Maybe you could test that path too?

martijnvangeffen, 2024-10-22:
Quistion why dont we move to full oauth as per Powershell i can still pull the data:

[image]

martijnvangeffen, 2024-10-22:
> which just uses [volvo2mqtt](https://github.com/Dielee/volvo2mqtt) which was written for homeassistant, but is standalone usable.

Can you explain a little more on how to do this ?

triple-it, 2024-10-22:
> > which just uses [volvo2mqtt](https://github.com/Dielee/volvo2mqtt) which was written for homeassistant, but is standalone usable.
> 
> Can you explain a little more on how to do this ?

sure...

Install Dielee/volvo2mqtt, just follow their instructions. (This is with OTP btw)
Install evcc/vehicle/volvo2mqtt template which is lurking the data from Dielee/volvo2mqtt

Some more info and test config is placed in PR https://github.com/evcc-io/evcc/pull/16771

triple-it, 2024-10-22:
> Quistion why dont we move to full oauth as per Powershell i can still pull the data: (No 2FA done anywhere)
> 
> ![image](https://private-user-images.githubusercontent.com/37099973/378974205-7d861da4-ab5b-478a-b744-7b0272e3a47b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk2MzQzODcsIm5iZiI6MTcyOTYzNDA4NywicGF0aCI6Ii8zNzA5OTk3My8zNzg5NzQyMDUtN2Q4NjFkYTQtYWI1Yi00NzhhLWI3NDQtN2IwMjcyZTNhNDdiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDIyVDIxNTQ0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNlYzY3MGM1NWY1Y2NkNjM0MjBkZGZhM2ZiOWM1MTc3Mzg2N2U2ODQzNjViMmY4NDVjOWEwYjllNDc3Nzc2MGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.F9Xtl9fucVbRZM__Dn54M40WSzErXQLcuFPhml4loNU)

What is the state of your published API Application?
e.g.
https://developer.volvocars.com/account/#your-api-applications

For the one with vcc-api-key that starts with: 16..e8633

"Status: Publication under review" is mine.....

nikagersonlohman, 2024-10-23:
Anyone find a way to "revoke" the authorization? In my case the authorization flow returns:

1. USERNAME_PASSWORD_REQUIRED (after  /as/authorization.oauth2)
2. COMPLETED (after action=checkUsernamePassword)
3. An access_token (after token.oauth2)
4. But when using the access token it returns:

{
  "status" : 403,
  "operationId" : "1cbf623f-5516-4f96-acd2-a5f1d268d5ed",
  "error" : {
    "message" : "FORBIDDEN",
    "description" : "Access Denied"
  }
}

So the token returned somehow is incorrect. When I do the same from Postman, all is ok...

Robbi12, 2024-10-23:
So richtig steige ich da nicht durch. Ich habe evcc auf nem Proxmox Container. Wie installiere ich dort volvo2mqtt rein? Oder einen neuen Container anlegen? Ich habe auch bei Iobroker einen mqtt server, ebenfalls in meiner zigbee2mqtt installation einen. Diese läuft wieder in einem extra Container.

triple-it, 2024-10-23:
> So richtig steige ich da nicht durch. Ich habe evcc auf nem Proxmox Container. Wie installiere ich dort volvo2mqtt rein? Oder einen neuen Container anlegen? Ich habe auch bei Iobroker einen mqtt server, ebenfalls in meiner zigbee2mqtt installation einen. Diese läuft wieder in einem extra Container.

I just followed the docker compose, so I have an extra mqtt service where the port is exposed.
You could change that of course from expose -> ports :

```
  mqtt:
    container_name: mqtt
    image: eclipse-mosquitto
#    expose:
#      - 1883
    ports:
      - "1883:1883"  # Make port 1883 available on your host
    command: [ "/usr/sbin/mosquitto", "-c", "/mosquitto-no-auth.conf" ]

```

To open that port on the host machine

But they also mention a docker oneliner on https://github.com/Dielee/volvo2mqtt
And it looks like with the parameter broker: you can configure an ip/host if you want to connect and reuse some existing mqtt. (e.g. zigbee2mqtt )

`docker run -d --pull=always -e CONF_updateInterval=300 -e CONF_babelLocale='de' -e CONF_mqtt='@json {"broker": "", "username": "", "password": "", "port": 1883}' -e CONF_volvoData='@json {"username": "", "password": "", "vin": "", "vccapikey": ["key1", "key2"]}' -e TZ='Europe/Berlin' --name volvo2mqtt ghcr.io/dielee/volvo2mqtt:latest`

mmierke, 2024-10-23:
@Robbi12 Entweder du verwendest Home Assistant oder installierst Volvo2MQTT in einem eigenen Container. Ich verwende nur einen MQTT Server. Mosquitto in einem eigenen Proxmox Container. Ich verwende evcc und volvo2mqtt als add ons in Home Assistant, aber mqtt oder einen modbus proxy als standalone container. Das war bei mir eine bewußte Entscheidung. Wenn jemand einen einfacheren Wert kennt, wie ich in evcc an Daten aus HA komme (ich meine nicht mit mqtt) wäre ich interessiert.

PeterZinn, 2024-10-23:
Könnte mir kurz jemand erklären, wie ich den volvo2mqtt template in evcc installiere? Aktuell kommt beim Start noch 'template not found'. Evcc läuft bei mir auf einem Raspi. Oder muss ich auf ein neues Release warten? Vielen Dank!!

mmierke, 2024-10-23:
Es gibt, soweit ich weiß, kein volvo2mqtt template. Du musst dir entweder die Daten von HA holen, so wie oben beschrieben oder ein mqtt basiertes vehicle aufbauen. In beiden Fällen findet sich die Anleitung nicht unter Volvo, sondern unter Manuell.

triple-it, 2024-10-23:
Hi.. we have just checked in the volvo2mqtt into the master (1 hour ago)..
https://github.com/evcc-io/evcc/blob/master/templates/definition/vehicle/volvo2mqtt.yaml

So the nightly build of tonight 2 AM will hopefully make it easy testable... from tomorrow on.

```
docker run \
-d \
--name evcc \
-v $(pwd)/evcc.yaml:/etc/evcc.yaml \
-v $(pwd)/.evcc:/root/.evcc \
-p 7070:7070 \
-p 8887:8887 \
-p 7090:7090/udp \
-p 9522:9522/udp \
evcc/evcc:nightly
```

mmierke, 2024-10-23:
Okay, missed that one. Looking forward to it.

Robbi12, 2024-10-23:
Ich bekomme es irgendwie nicht hin. Habe einen neuen debian Container aufgesetzt. Dann dort Docker installiert.

Dann habe ich diese Befehlsreihe mit meinen Daten ergänzt:

docker run -d --pull=always -e CONF_updateInterval=300 -e CONF_babelLocale='de' -e CONF_mqtt='@json {"broker": "", "username": "", "password": "", "port": 1883}' -e CONF_volvoData='@json {"username": "", "password": "", "vin": "", "vccapikey": ["key1", "key2"]}' -e TZ='Europe/Berlin' --name volvo2mqtt ghcr.io/dielee/volvo2mqtt:latest

Ich denke meine Daten müssen zwischen die ""

Dann soll ich volvomqtt starten. Wie mache ich das?

mmierke, 2024-10-23:
Also ehrlich gesagt, finde ich die Lösung mit Docker unter Proxmox nicht wirklich gut. Mir fehlt der Sinn dahinter docker, also eine Virtualisierungsplattform, auf Proxmox, einer Virtualisierungsplattform, zu installieren. Docker läuft bei mir auf dem NAS.

mmierke, 2024-10-23:
> Hi.. we have just checked in the volvo2mqtt into the master (1 hour ago).. https://github.com/evcc-io/evcc/blob/master/templates/definition/vehicle/volvo2mqtt.yaml
>
> So the nightly build of tonight 2 [...]

Do you have an idea on how to retrieve sensors from HA in the evcc add-on without requiring rest or mqtt?

mmierke, 2024-10-23:
> Ich bekomme es irgendwie nicht hin. Habe einen neuen debian Container aufgesetzt. Dann dort Docker installiert.
>
> Dann habe ich diese Befehlsreihe mit meinen Daten ergänzt:
>
> docker run -d [...]

Schon mal darüber nachgedacht unter Proxmox eine Supervised HA Instanz zu installieren und in dieser volvo2mqtt als add-on laufen zu lassen? Das ist einfacher als dieser ganze Umweg über Docker unter Proxmox

martijnvangeffen, 2024-10-23:
Got every thing working now only need my vehicle to pull from MQTT but yaml keeps failing ,  any comments on what i am missing??    or how i can update evcc native install on linux to have nightly build?

vehicles:
  - name: EX30
    title: EX30
    type: custom
    timeout: 720h
    capacity: 51 # in kWh
    vin: "abc..."
    soc:
      source: mqtt
      topic: homeassistant/sensor/{{ .vin }}_battery_charge_level/state
      timeout: {{ .timeout }}
    status:
      source: combined
      plugged:
        source: mqtt
        topic: homeassistant/sensor/{{ .vin }}_charging_connection_status/state
        timeout: {{ .timeout }}
      charging:
        source: mqtt
        topic: homeassistant/sensor/{{ .vin }}_charging_system_status/state
        timeout: {{ .timeout }}
    range:
      source: mqtt
      topic: homeassistant/sensor/{{ .vin }}_electric_range/state
      timeout: {{ .timeout }}

martijnvangeffen, 2024-10-23:
got it working by hard coding VIN. 

Only thing open is missing SOC , but it reads range correctly
[image]

nikagersonlohman, 2024-10-24:
> Anyone find a way to "revoke" the authorization? In my case the authorization flow returns:
>
> 1. USERNAME_PASSWORD_REQUIRED (after  /as/authorization.oauth2)
> 2. COMPLETED (after [...]

A header in one of the other calls caused the access_token to be invalid (even though it was generated)

triple-it, 2024-10-24:
The evcc nightly has been build...

You can test-drive the volvo2mqtt-evcc-vehicle template easy via a docker-compose like:

```

services:
  evcc:
    image: evcc/evcc:nightly
    container_name: evcc
    env_file:
      - .env
    volumes:
      - ./evcc.yaml:/etc/evcc.yaml
      - ./.evcc:/root/.evcc
    ports:
      - "${EVCC_PORT_7070}:7070"
      - "${EVCC_PORT_8887}:8887"
      - "${EVCC_PORT_7090_UDP}:7090/udp"
      - "${EVCC_PORT_9522_UDP}:9522/udp"

  volvo2mqtt:
    image: ghcr.io/dielee/volvo2mqtt:latest
    container_name: volvo2mqtt
    env_file:
      - .env

  mqtt:
    container_name: mqtt
    image: eclipse-mosquitto
    expose:
      - 1883
#    ports:
#      - "1883:1883"  # Maak poort 1883 beschikbaar op je hostmachine
    command: [ "/usr/sbin/mosquitto", "-c", "/mosquitto-no-auth.conf" ]

# Push the OTP to mqtt with
#docker exec -it mqtt  mosquitto_pub -t "volvoAAOS2mqtt/otp_code" -m "<OTP_FROM_EMAIL>" 

```

Run:

```
docker compose build
docker compose up

```
With the following .env

```
# EVCC instellingen
EVCC_PORT_7070=7070
EVCC_PORT_8887=8887
EVCC_PORT_7090_UDP=7090
EVCC_PORT_9522_UDP=9522

# Volvo2MQTT instellingen
VOLVO_USERNAME="username"
VOLVO_PASSWORD="pass"
VIN="vin"
VCC_API_KEY="vcc_api_key"
VCC_API_KEY1="vcc_api_key1"
VCC_API_KEY2="vcc_api_key2"
#MQTT_BROKER="192.168.50.246"
#MQTT_BROKER="localhost"
MQTT_BROKER="mqtt"
MQTT_PORT="1883"
TZ=Europe/Berlin
CONF_babelLocale='de'
CONF_updateInterval=300

CONF_mqtt="@json {\"broker\": \"${MQTT_BROKER}\", \"username\": \"\", \"password\": \"\", \"port\": \"${MQTT_PORT}\"}"

CONF_volvoData="@json {\"username\": \"${VOLVO_USERNAME}\", \"password\": \"${VOLVO_PASSWORD}\", \"vin\": \"${VIN}\", \"vccapikey\": [\"${VCC_API_KEY1}\", \"${VCC_API_KEY2}\"]}"

#echo $CONF_mqtt
#echo $CONF_volvoData

```

Robbi12, 2024-10-25:
Nun nach unendlichen Versuchen habe ich es hinbekommen. 
Homeassistant als normale Version in einer virtuellen Maschine installiert mit proxmox helfer script.
Dann konnte ich volvo2mqtt installieren. Habe es nur mit einem mqtt server direkt in homeassistant hinbekommen

Hatte dann das Problem dass ich in der Yaml bei evcc keine zwei mqtt server hinbekommen. Denn ich werte daten mit mqtt von evcc nach iobroker aus.

Dort dann den evcc adapter installiert. Nun funktioniert es. Riesen aufwand nur weil volvo was dichtmacht.

Das Problem. Ich kenne mit mit Homeassistant null komma null aus

mmierke, 2024-10-25:
Und für HA unter proxmox gibt es doch hier alles was man braucht https://helper-scripts.com/

lahlor2, 2024-10-26:
Ich muss mich nun hier auch nochmal einmischen. Ich als auch Freund&Bekannte nutzen evcc als "Standalone" Lösung auf einem Linux Server (Pi, Custom Build Ubunutu etc.) und haben bisher das volvo-connect template verwendet.
Dies schlägt nun bei uns allen fehl (XC90, XC40..). 
Ich verstehe, dass es seitens Volvo ein Anpassung gab, aber ich verstehe nicht, wieso ich nun mit MQTT, HomeAssistant oder Proxmox oder sonstigen 3rd Party Lösungen einen Workaround finden muss. 1. ist es viel zu komplex meiner Meinung nach, 2. brauchte ich es vorher ja auch nicht.
Daher meine direkte (nicht böse gemeinte) Frage: 
Wird es in naher Zukunft seitens EVCC eine direkte implementierung der "neuen" API Anbindung von Volvo geben, sodass es ohne Zusatzsoftware wieder einsetzbar ist?

mmierke, 2024-10-26:
@lahlor2 Wenn ich andig's früheren Kommentar und das Schließen des Issues richtig interpretiere, hat sich der Fall für ihn erledigt.

mmierke, 2024-10-26:
@Robbi12 Wenn du magst, können wir uns dein Setup mal zusammen ansehen.

lahlor2, 2024-10-26:
> @lahlor2 Wenn ich andig's früheren Kommentar und das Schließen des Issues richtig interpretiere, hat sich der Fall für ihn erledigt.

Ich bin jetzt nicht genau durchgestiegen aber wenn ich es richtig lese, dann kann evcc kein Volvo mehr anbieten? Verstehe ich das richtig?

Habe mich dann mal an die ganzen MQTT HA etc Themen gesetzt und mal getestet. Als Laie, der so erstmal mit MQTT/HA und co nichts zu tun hat - alles sehr unverständlich und unlösbar.
Gibt es irgendjemanden, der das Thema mal inhaltlich so aufarbeiten kann, dass man es versteht und vorallem weioß was ich nun genau brauche?
Habe z.B. kein HA und nur eine kleine Ubuntu Linux Kiste.
Wenn ich z.B. wie in einem Post hier nur die volvo2mqtt Lösung via Docker installieren, dann kommt im Docker Log folgender Fehler.

<snip>
ct 26 23:02:20 volvo2mqtt [1] - INFO: Starting volvo2mqtt version v1.10.3
Traceback (most recent call last):
  File "/volvoAAOS2mqtt/./main.py", line 14, in <module>
    connect()
  File "/volvoAAOS2mqtt/mqtt.py", line 39, in connect
    client.connect(settings["mqtt"]["broker"], port)
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 914, in connect
    return self.reconnect()
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1044, in reconnect
    sock = self._create_socket_connection()
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 3685, in _create_socket_connection
    return socket.create_connection(addr, timeout=self._connect_timeout, source_address=source)
  File "/usr/local/lib/python3.9/socket.py", line 856, in create_connection
    raise err
  File "/usr/local/lib/python3.9/socket.py", line 844, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
</snip>

Ich habe "außerhalb" von Docker einen Mosquitto MQTT Broker laufen, ob der aber nun auch korrekt luft etc - keine Ahnung. Also mMn alles sehr undurchsichtig und eine super Bastellösung.
Versteht mich nicht falsch - aber das ist mehr als nicht zufriedenstellend.
Ich bin für jede Hilfe dankbar, da ich hinter mir noch 2 Installationen stehen habe, die dasselbe Problem haben und schnellstmöglich nach eine Lösung schreien.

Robbi12, 2024-10-26:
Hallo ich bräuchte auch noch Hilfe. Habe nun stundenlang probiert.
Ich würde gerne Volvo2mqtt mit Homeassistant laufen lassen. Dies klappt auch, aber nur mit einem mqtt broker auf homeassistant. Egal was ich eingebe (Internetadresse von meinem MQTT Broker in Iobroker oder den Port) nichts wird genommen. Ich kann lediglich mit auto_broker und auto_port arbeiten.

Nun würde ich aber gerne vom Iobroker evcc steuern. Dies klappt nicht so gut mit dem EVCC Adapter. Vorher hatte es wunderbar mit dem mqtt broker vom iobroker geklappt, dessen Daten in Evcc waren. In Evcc habe ich ja nun den Homeassistant Broker drin. Oder könnte ich in EVCC zwei MQTT Broker eintragen und wenn ja, wie?

lahlor2, 2024-10-26:
> > @lahlor2 Wenn ich andig's früheren Kommentar und das Schließen des Issues richtig interpretiere, hat sich der Fall für ihn erledigt.
> 
> Ich bin jetzt nicht genau durchgestiegen aber wenn ich es richtig lese, dann kann evcc kein Volvo mehr anbieten? Verstehe ich das richtig?
> 
> Habe mich dann mal an die ganzen MQTT HA etc Themen gesetzt und mal getestet. Als Laie, der so erstmal mit MQTT/HA und co nichts zu tun hat - alles sehr unverständlich und unlösbar. Gibt es irgendjemanden, der das Thema mal inhaltlich so aufarbeiten kann, dass man es versteht und vorallem weioß was ich nun genau brauche? Habe z.B. kein HA und nur eine kleine Ubuntu Linux Kiste. Wenn ich z.B. wie in einem Post hier nur die volvo2mqtt Lösung via Docker installieren, dann kommt im Docker Log folgender Fehler.
> 
> ```
> ct 26 23:02:20 volvo2mqtt [1] - INFO: Starting volvo2mqtt version v1.10.3
> Traceback (most recent call last):
>   File "/volvoAAOS2mqtt/./main.py", line 14, in <module>
>     connect()
>   File "/volvoAAOS2mqtt/mqtt.py", line 39, in connect
>     client.connect(settings["mqtt"]["broker"], port)
>   File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 914, in connect
>     return self.reconnect()
>   File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1044, in reconnect
>     sock = self._create_socket_connection()
>   File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 3685, in _create_socket_connection
>     return socket.create_connection(addr, timeout=self._connect_timeout, source_address=source)
>   File "/usr/local/lib/python3.9/socket.py", line 856, in create_connection
>     raise err
>   File "/usr/local/lib/python3.9/socket.py", line 844, in create_connection
>     sock.connect(sa)
> ConnectionRefusedError: [Errno 111] Connection refused
> ```
> 
> Ich habe "außerhalb" von Docker einen Mosquitto MQTT Broker laufen, ob der aber nun auch korrekt luft etc - keine Ahnung. Also mMn alles sehr undurchsichtig und eine super Bastellösung. Versteht mich nicht falsch - aber das ist mehr als nicht zufriedenstellend. Ich bin für jede Hilfe dankbar, da ich hinter mir noch 2 Installationen stehen habe, die dasselbe Problem haben und schnellstmöglich nach eine Lösung schreien.

Mein lieber Herr Gesangsverein war das ein Ritt 😆 

Ich möchte nun mal meine Erfahrungen und Ergebnisse zusammenfassen, damit vielleicht auch andere eine schnelle Volvo Lösung haben.

Grundlage bei mir:
- evcc Versionn v0.131.1 auf Ubuntu
- Mosquitto MQTT Broker
- Docker

Wie ihr evcc installiert wisst ihr 😆 
Die andere Software ist neben der evcc Installation auf dem Ubunutu installiert. Ich denke es lässt sich auch einfach auf andere Linux Derivate adaptieren. Wichtig ist sicher nachfolgend die Config etc.

Mosquitto

`sudo apt-add-repository ppa:mosquitto-dev/mosquitto-ppa
sudo apt-get update
sudo apt-get install mosquitto`

Dann noch eine passende Konfiguration erstellen unter /etc/mosquitto/mosquitto.conf
Achtung hier muss eure IP eingetragen werden.

`# Place your local configuration in /etc/mosquitto/conf.d/
#
# A full description of the configuration file is at
# /usr/share/doc/mosquitto/examples/mosquitto.conf.example

persistence true
persistence_location /var/lib/mosquitto/

log_dest file /var/log/mosquitto/mosquitto.log

include_dir /etc/mosquitto/conf.d
per_listener_settings false
allow_anonymous false
password_file /etc/mosquitto/password
bind_address <EureIPdesServers>
`
Die "password" Datei erstellen wir dann mit Hilfe des Befehls

`mosquitto_passwd -c /etc/mosquito/password user`

Hier kommt der User/Passwort rein, der dann den Broker mit Daten beladen kann/darf. Der Nutzer in dem Beispiel heißt "user".
[... 58 lines cut ...]

Abschließend muss dann evcc beigebracht werden die Daten via MQTT zu beziehen. Hierzu meine evcc.yaml Konfiguration mit den relevanten Daten. Neu ist der mqtt Part und hier muss dann entsprechend die IP des MQTT Brokers Mosquitto rein, sowie der Benutzername/Passwort.

`vehicles:
- name: XC90
  type: template
  template: volvo2mqtt
  title: XC90 # Wird in der BenutzeroberflÃ¤che angezeigt (optional)
  vin: XXXXXXXXX # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
  capacity: 18.8 # AkkukapazitÃ¤t in kWh (optional)

mqtt:  # Mqtt broker is required!
  broker: xxx.xxx.xxx.xxx:1883
  topic: #
  #clientid: evcc
  user: user
  password: xxxxxx
`
Nach einem Neustart des evcc Dienstes habe ich dann via 

`evcc vehicle`

eine Abfrage gestartet und siehe da, alles wieder an seinem Platz!

Ich hoffe ich kann mit dieser Zusammenfassung und Anleitung dem einem oder anderen helfen!

lahlor2, 2024-10-27:
> > > @lahlor2 Wenn ich andig's früheren Kommentar und das Schließen des Issues richtig interpretiere, hat sich der Fall für ihn erledigt.
> > 
> > 
> > Ich bin jetzt nicht genau durchgestiegen aber wenn ich es richtig lese, dann kann evcc kein Volvo mehr anbieten? Verstehe ich das richtig?
> > Habe mich dann mal an die ganzen MQTT HA etc Themen gesetzt und mal getestet. Als Laie, der so erstmal mit MQTT/HA und co nichts zu tun hat - alles sehr unverständlich und unlösbar. Gibt es irgendjemanden, der das Thema mal inhaltlich so aufarbeiten kann, dass man es versteht und vorallem weioß was ich nun genau brauche? Habe z.B. kein HA und nur eine kleine Ubuntu Linux Kiste. Wenn ich z.B. wie in einem Post hier nur die volvo2mqtt Lösung via Docker installieren, dann kommt im Docker Log folgender Fehler.
> > ```
> > ct 26 23:02:20 volvo2mqtt [1] - INFO: Starting volvo2mqtt version v1.10.3
> > Traceback (most recent call last):
> >   File "/volvoAAOS2mqtt/./main.py", line 14, in <module>
> >     connect()
> >   File "/volvoAAOS2mqtt/mqtt.py", line 39, in connect
> >     client.connect(settings["mqtt"]["broker"], port)
> >   File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 914, in connect
> >     return self.reconnect()
> >   File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1044, in reconnect
> >     sock = self._create_socket_connection()
> >   File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 3685, in _create_socket_connection
> >     return socket.create_connection(addr, timeout=self._connect_timeout, source_address=source)
> >   File "/usr/local/lib/python3.9/socket.py", line 856, in create_connection
> >     raise err
> >   File "/usr/local/lib/python3.9/socket.py", line 844, in create_connection
> >     sock.connect(sa)
> > ConnectionRefusedError: [Errno 111] Connection refused
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Ich habe "außerhalb" von Docker einen Mosquitto MQTT Broker laufen, ob der aber nun auch korrekt luft etc - keine Ahnung. Also mMn alles sehr undurchsichtig und eine super Bastellösung. Versteht mich nicht falsch - aber das ist mehr als nicht zufriedenstellend. Ich bin für jede Hilfe dankbar, da ich hinter mir noch 2 Installationen stehen habe, die dasselbe Problem haben und schnellstmöglich nach eine Lösung schreien.
> 
> Mein lieber Herr Gesangsverein war das ein Ritt 😆
> 
> Ich möchte nun mal meine Erfahrungen und Ergebnisse zusammenfassen, damit vielleicht auch andere eine schnelle Volvo Lösung haben.
> 
> Grundlage bei mir:
> 
> * evcc Versionn v0.131.1 auf Ubuntu
> * Mosquitto MQTT Broker
> * Docker
> 
> Wie ihr evcc installiert wisst ihr 😆 Die andere Software ist neben der evcc Installation auf dem Ubunutu installiert. Ich denke es lässt sich auch einfach auf andere Linux Derivate adaptieren. Wichtig ist sicher nachfolgend die Config etc.
> 
> Mosquitto
> 
> `sudo apt-add-repository ppa:mosquitto-dev/mosquitto-ppa sudo apt-get update sudo apt-get install mosquitto`
> 
> Dann noch eine passende Konfiguration erstellen unter /etc/mosquitto/mosquitto.conf Achtung hier muss eure IP eingetragen werden.
> 
> ` persistence true persistence_location /var/lib/mosquitto/
> 
> log_dest file /var/log/mosquitto/mosquitto.log
> 
> include_dir /etc/mosquitto/conf.d per_listener_settings false allow_anonymous false password_file /etc/mosquitto/password bind_address ` Die "password" Datei erstellen wir dann mit Hilfe des Befehls
> 
> `mosquitto_passwd -c /etc/mosquito/password user`
> 
> Hier kommt der User/Passwort rein, der dann den Broker mit Daten beladen kann/darf. Der Nutzer in dem Beispiel heißt "user".
> 
> Abschließend den Service einmal neu starten
> 
> `sudo systemctl restart mosquitto.service`
> 
> Dann noch einmal den Status prüfen, der "grün" sein sollte.
[... 32 lines cut ...]
> 
> Abschließend muss dann evcc beigebracht werden die Daten via MQTT zu beziehen. Hierzu meine evcc.yaml Konfiguration mit den relevanten Daten. Neu ist der mqtt Part und hier muss dann entsprechend die IP des MQTT Brokers Mosquitto rein, sowie der Benutzername/Passwort.
> 
> ` vehicles:
> 
> * name: XC90
>   type: template
>   template: volvo2mqtt
>   title: XC90 # Wird in der BenutzeroberflÃ¤che angezeigt (optional)
>   vin: XXXXXXXXX # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
>   capacity: 18.8 # AkkukapazitÃ¤t in kWh (optional)
> 
> mqtt: # Mqtt broker is required! broker: xxx.xxx.xxx.xxx:1883 topic: # #clientid: evcc user: user password: xxxxxx ` Nach einem Neustart des evcc Dienstes habe ich dann via
> 
> `evcc vehicle`
> 
> eine Abfrage gestartet und siehe da, alles wieder an seinem Platz!
> 
> Ich hoffe ich kann mit dieser Zusammenfassung und Anleitung dem einem oder anderen helfen!

Ich habe noch eine Sache zu ergänzen. Die Ausgabe via "evcc vehicle" klappt und zeit mir auch aktuelle Werte an (siehe Screenshot) allerdings auf der Webseite von evcc ist das Auto immer "nicht verbunden". Gibt es hierzu Ideen, denn dann wäre es mMn alles wieder im Lot und am Laufen 😄 

[image]

[image]

mmierke, 2024-10-27:
@lahlor2 Ich würde mir mal Proxmox ansehen. Ich habe so einen kleinen Zauberwürfel mit 16GB RAM und 512GB SSSD. Auf dem läuft Proxmox als Virtualisierungsplattform.
[image]
Interessant sind hier eigentlich nur der Container 100 mqtt und die virtuelle Maschine 500 haos. 
Bei mir laufen beide auf Debian. Beide habe ich mithilfe dieser Seite aufgebaut => https://helper-scripts.com/

Bei haos würde ich zumindest ssh aktivieren um mithilfe von WinScp und VsCode direkt auf die Konfigurationsdateien zugreifen zu können.
Mqtt habe ich im Container als Service eingerichtet, damit er bei einem Reboot direkt wieder gestartet wird.

Evcc und Volvo2Mqtt laufen bei mir als Addons im HA. Das ist aber nicht zwingend notwendig.

@Robbi12 Wenn ihr mehr Details haben möchtet, können wir gerne ein Slack aufmachen.

Robbi12, 2024-10-27:
Danke mmierke.

So habe ich es ähnlich. Du hast einen eigenen Container mit mqtt. Ich habe dort zigbee2mqtt laufen mit broker drin und auch iobroker mit einem mqtt broker drin.

Also ähnlich.

Was mich interessiert und überhaupt nicht schaffe, wie ich volvo2mqtt im homeassistant einstellen muss, dass er auf meinen Broker (bei dir MQTT bei 100) zugreifen kann. Funktioniert einfach nicht.

mmierke, 2024-10-27:
Also erstmal habe ich meinem mqtt eine feste IP Addresse gegeben. Das geht über die Netzwerkeinstellungen für den Container und proxmox. Gateway nicht vergessen, sonst funktioniert kein update mehr. Und dann hat mein Mosquitto Server noch ein Passwort. Ich weiß nicht, ob das dein Problem ist. Hier ein Beispiel für die Werte auf der Configuration Seite des Addons in HA.

broker: 192.168.123.456
port: "1883"
username: mosquitto
password: "sdgffgsfgdfg"

mmierke, 2024-10-27:
Ich hatte früher auch ioBroker und bin dann langsam auf HA umgezogen. Um mir das Leben zu erleichtern habe ich zuerst mqtt in einen eigenen Container verlagert und zigbee2mqtt und in deinem Fall ioBroker mit diesem Verbunden. Also aus 2 mach 1 mqtt broker. Das vereinfacht auch deine gesamte Architektur.

Robbi12, 2024-10-27:
Danke für deine Mühe. Genau so mache ich es. Es funktionier alles bei mir zigbee2mqtt usw.

Wenn ich jedoch die Daten im Homeassistant bei volvo2mqtt eingebe kommt immer folgendes.
Oct 27 12:23:57 volvo2mqtt [106] - INFO: Starting volvo2mqtt version v1.10.2
Traceback (most recent call last):
  File "/volvoAAOS2mqtt/main.py", line 14, in <module>
    connect()
  File "/volvoAAOS2mqtt/mqtt.py", line 39, in connect
    client.connect(settings["mqtt"]["broker"], port)
  File "/usr/lib/python3.12/site-packages/paho/mqtt/client.py", line 914, in connect
    return self.reconnect()
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/paho/mqtt/client.py", line 1044, in reconnect
    sock = self._create_socket_connection()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/paho/mqtt/client.py", line 3685, in _create_socket_connection
    return socket.create_connection(addr, timeout=self._connect_timeout, source_address=source)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/socket.py", line 865, in create_connection
    raise exceptions[0]
  File "/usr/lib/python3.12/socket.py", line 850, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Nehme ich den eigenen mosquito mqtt im homeassistant mit

auto_broker
auto_port
auto_user
auto_password

funktioniert es.

mmierke, 2024-10-27:
Setzt mal dend Port in ""

[15 more replies]