evcc-io/evcc #16740: CO2 optimised charging in combination with a plan should not go abvove charge limit early

MPW1412, 2024-10-20:
### Describe the bug

I've set a charge limit, e.g. 85%. And I set a plan to charge the car to 100% to 9 o'clock. And I've also enabled Grünstrom API to optimize for CO2.

What has happened:

The car was charged to 100% at 4 am already and sat with a full battery for hours, which is not good for the battery.

What I'd expect EVCC to do is:

Charge CO2 optimized until the standard chargelimit of 85 % (or whatever user has set) and charge the rest without regards to CO2 very close to departure time.

Best,
Matthias

### Steps to reproduce

1. Enable Grünstrom API
2. Set a charge limit below 100%
3. Deactivate charge limits in the car
4. Set a plan to charge to 100%

### Configuration details

```yaml
Nothing particular.
```

### Log details

```text
none
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.130.13

andig, 2024-10-20:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-10-20 by andig

MPW1412, 2024-10-20:
@andig, this is an algorithmic thing, not an error to debug. Please read again and reopen the issue.