evcc-io/evcc #16744: support phase currents, and three-phase with enphase envoy

pdeliot, 2024-10-20:
To use enphase template for circuit, phase current is required.

This information information is available for single and tree-phase.

Three-phase support is also required (for  me ;-) )...

Attached sample file with Tree-phase.

[envoy.json](https://github.com/user-attachments/files/17449652/envoy.json)
Below a little sample extract of available data:
`
{
    "consumption": [
        {
            "type": "eim",
            "activeCount": 1,
            "measurementType": "net-consumption",
            "readingTime": 1729414585,
            "wNow": 5364.339,
            "whLifetime": 48631628.574,
            "lines": [
                {
                    "wNow": 452.054,
                    "whLifetime": 20136453.749,
                    "varhLeadLifetime": 5727861.458,
                },
                {
                    "wNow": -98.144,
                    "whLifetime": 1634255.145,
                    "varhLeadLifetime": 4811478.811,
                },
                {
                    "wNow": 5010.429,
                    "whLifetime": 26860919.68,
                    "varhLeadLifetime": 1345115.396,
                }
            ]
        }
    ]
}
`

pdeliot, 2024-10-20:
I can try to work on it if you can point me a sample template for tree-phase and/or phases currents...

VolkerK62, 2024-10-20:
here you can see an example, how to add `currents` to a template https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/shelly-3em.yaml

it should be the same syntax as for `power`
https://github.com/evcc-io/evcc/blob/27e45a2b31861acb9be891b0fa3c29e325677f81/templates/definition/meter/enphase.yaml#L23

the `jq` should be
`jq: .consumption[] | select(.measurementType == "net-consumption").lines[0].rmsCurrent`

where
`lines[0] = phase 1`
`lines[1] = phase 2`
`lines[2] = phase 3`

pdeliot, 2024-10-20:
Thanks,

I've started a first implementation that looks to work.

`
  powers:
    - source: http
      uri: http://{{ .host }}/production.json?details=1
      {{- if .token }}
      auth:
        type: bearer
        password: {{ .token }}
      insecure: true
      {{- end }}
      jq: .consumption[] | select(.measurementType == "net-consumption").lines[0].rmsCurrent
    - source: http
      uri: http://{{ .host }}/production.json?details=1
      {{- if .token }}
      auth:
        type: bearer
        password: {{ .token }}
      insecure: true
      {{- end }}
      jq: .consumption[] | select(.measurementType == "net-consumption").lines[1].rmsCurrent
    - source: http
      uri: http://{{ .host }}/production.json?details=1
      {{- if .token }}
      auth:
        type: bearer
        password: {{ .token }}
      insecure: true
      {{- end }}
      jq: .consumption[] | select(.measurementType == "net-consumption").lines[2].rmsCurrent
  currents:
    - source: http
      uri: http://{{ .host }}/production.json?details=1
      {{- if .token }}
      auth:
        type: bearer
        password: {{ .token }}
      insecure: true
      {{- end }}
      jq: .consumption[] | select(.measurementType == "net-consumption").lines[0].rmsCurrent
    - source: http
      uri: http://{{ .host }}/production.json?details=1
      {{- if .token }}
      auth:
        type: bearer
        password: {{ .token }}
      insecure: true
      {{- end }}
      jq: .consumption[] | select(.measurementType == "net-consumption").lines[1].rmsCurrent
    - source: http
      uri: http://{{ .host }}/production.json?details=1
      {{- if .token }}
      auth:
        type: bearer
        password: {{ .token }}
      insecure: true
      {{- end }}
      jq: .consumption[] | select(.measurementType == "net-consumption").lines[2].rmsCurrent
  voltages:
    - source: http
      uri: http://{{ .host }}/production.json?details=1
      {{- if .token }}
      auth:
        type: bearer
        password: {{ .token }}
      insecure: true
      {{- end }}
      jq: .consumption[] | select(.measurementType == "net-consumption").lines[0].rmsVoltage
    - source: http
      uri: http://{{ .host }}/production.json?details=1
      {{- if .token }}
      auth:
        type: bearer
        password: {{ .token }}
      insecure: true
      {{- end }}
      jq: .consumption[] | select(.measurementType == "net-consumption").lines[1].rmsVoltage
    - source: http
      uri: http://{{ .host }}/production.json?details=1
      {{- if .token }}
      auth:
        type: bearer
        password: {{ .token }}
      insecure: true
      {{- end }}
      jq: .consumption[] | select(.measurementType == "net-consumption").lines[2].rmsVoltage
`
Now I need to check this is working on single-phase systems.

Is is expected to have 2 version of the template? A parameter ? Or it will deal with error on phase 1 and 2 ?

VolkerK62, 2024-10-21:
But `powers` is `wNow` and not `rmsCurrent`.

pdeliot, 2024-10-21:
> But `powers` is `wNow` and not `rmsCurrent`.

Yes! Fixed.

pdeliot, 2024-10-21:
Now I have severall questions

- Understand  how to deal with single and three phase definision
- Is it possible to use ONE query for several indicators (to avoid querying once per value. All the values are in the returned json)?
- How can I push the contribution?

VolkerK62, 2024-10-21:
>Understand how to deal with single and three phase definision

How does the JSON für single-phase looks like?

>Is it possible to use ONE query for several indicators (to avoid querying once per value. All the values are in the returned json)?

Not sure, but I don´t think so.

>How can I push the contribution?

create a PR.
Best way would be, to modify the actual template, if it is working under all circumstances

pdeliot, 2024-10-23:
> How does the JSON für single-phase looks like?
I'm working on a solution with optional 'phases" parameter default 1 for implemetation.

But maybe other ways exists...

On phase can be retrieved this fashion (simplified):

`    {
        "type": "eim",
        "activeCount": 1,
        "measurementType": "net-consumption",
        "readingTime": 1729537922,
        "wNow": 725.998,
        "whLifetime": 81596.841,
        "rmsCurrent": 3.771,
        "rmsVoltage": 241.879,
        "pwrFactor": 0.8,
        "lines": [
            {
                "wNow": 725.998,
                "whLifetime": 81596.841,
                "rmsCurrent": 3.771,
                "rmsVoltage": 241.879,
                "pwrFactor": 0.8,
            }
        ]
    }
`

And Three-pahse this way (simplified):

`
        {
            "type": "eim",
            "activeCount": 1,
            "measurementType": "net-consumption",
            "readingTime": 1729684620,
            "wNow": 4640.513,
            "whLifetime": 48824472.782,
            "pwrFactor": 0.83,
            "lines": [
                {
                    "wNow": 253.166,
                    "whLifetime": 20186173.813,
                    "rmsCurrent": 2.875,
                    "rmsVoltage": 245.513,
                    "pwrFactor": 0.36,
                },
                {
                    "wNow": -495.77,
                    "whLifetime": 1644809.589,
                    "rmsCurrent": -2.447,
                    "rmsVoltage": 242.702,
                    "pwrFactor": -1.0,
                },
                {
                    "wNow": 4883.117,
                    "whLifetime": 26993489.38,
                    "rmsCurrent": 22.817,
                    "rmsVoltage": 239.245,
                    "pwrFactor": 0.89,
                    "whToday": 0,
                }
            ]
        }

`

VolkerK62, 2024-10-23:
>I'm working on a solution with optional 'phases" parameter default 1 for implemetation.

in that case it is breaking change for all 3p configuration
Best way would be a kind of automatic detection.

Don´t know, what would be the best way for that.
One possibility ist the sum of voltages
<300 = 1p
>300 = 3p
But I am sure, this would be the solution of an amateur 😃

pdeliot, 2024-10-23:
It should not break existing 3p configuration, because the template today is treating 3p as 1p.
Then if no change is done on "evcc.yaml" to add the "phases' with default '1' parameter value, nothing will change (or just new feature becoming available).

It is possible to detect, but I must work to understand how to use calculations in yaml  ;-)
Maybe counting rows (then phases) in JSON. But it may be too late, because the template is already parssed and loaded, before any measure to comme back.

I'm fighting a little bit with 'currents'... 
Is is supported for 1p installations?

I do not see how to define 'currents' (or maybe 'current' ?) for only one phase.

VolkerK62, 2024-10-24:
>I do not see how to define 'currents' (or maybe 'current' ?) for only one phase.

it must be `currents` and it must be three 
In case of 1p, you must set set the Phases 2 and 3 to `0` 

I tried the following with a one phase shelly and it works:
```
  currents:
    - source: http
      uri: 192.168.178.88/status
      jq: .meters[0].power
    - source: http
      uri: 192.168.178.88/status
      jq: if .meters[1].power == null then 0 else .meters[1].power end
    - source: http
      uri: 192.168.178.88/status
      jq: if .meters[2].power == null then 0 else .meters[2].power end
```

pdeliot, 2024-10-24:
I've followed this implementation and it looks working.. test in progress.

Doe exists a documentation on "how to contribute" .. how to create/select a branch and create a PR ?

pdeliot, 2024-10-24:
Does 'powers' and 'currents' can be negative (in case of injection) ?
Or should be set to 0 when negative?

pdeliot, 2024-10-24:
I've written a new template with 'currents', but only tested on my installation with CTs.

ivoks, 2024-10-25:
Feel free to attach it and I can test it in my setup.

pdeliot, 2024-10-25:
[enphase.yaml.gz](https://github.com/user-attachments/files/17517909/enphase.yaml.gz)
The template for test

ivoks, 2024-10-25:
Thank you! I've ran the test and I'm quite sure that the PV part is not correct. By looking at the consumption, you are getting info about consumption and not production. PV should tell you what the production of the PV is.

If you turn this into '.production[] | select(.measurementType == "production")', then for all three phases on my system it would look like this:
$ curl http://172.16.12.43/production.json?details=1 2>/dev/null | jq 'if .production[] | select(.measurementType == "production").lines[1].wNow == null then 0 else .production[] | select(.measurementType == "production").lines[0].wNow end'
0.0
$ curl http://172.16.12.43/production.json?details=1 2>/dev/null | jq 'if .production[] | select(.measurementType == "production").lines[1].wNow == null then 0 else .production[] | select(.measurementType == "production").lines[1].wNow end'
-0.688
$ curl http://172.16.12.43/production.json?details=1 2>/dev/null | jq 'if .production[] | select(.measurementType == "production").lines[1].wNow == null then 0 else .production[] | select(.measurementType == "production").lines[2].wNow end'
-1.291

And these are all wrong. Maybe this would be a better approach:

$ curl http://172.16.12.43/production.json?details=1 2>/dev/null | jq 'if .production[] | select(.measurementType == "production").activeCount >= 1 then .production[] | select(.measurementType == "production").lines[0].wNow else empty end'
$ curl http://172.16.12.43/production.json?details=1 2>/dev/null | jq 'if .production[] | select(.measurementType == "production").activeCount > 1 then .production[] | select(.measurementType == "production").lines[1].wNow else empty end'
$ curl http://172.16.12.43/production.json?details=1 2>/dev/null | jq 'if .production[] | select(.measurementType == "production").activeCount > 1 then .production[] | select(.measurementType == "production").lines[2].wNow else empty end'
$ 

First if statement checks number of active CTs/phases. Those without CTs should have 0 activeCount (as I do). And then, instead of returning 0 (which is not really correct), it would probably be better to return empty value, so that evcc knows that the data is not valid. And then you can apply the same approach for currents:

$ curl http://172.16.12.43/production.json?details=1 2>/dev/null | jq 'if .production[] | select(.measurementType == "production").activeCount >= 1 then .production[] | select(.measurementType == "production").lines[0].rmsCurrent else empty end'
$ curl http://172.16.12.43/production.json?details=1 2>/dev/null | jq 'if .production[] | select(.measurementType == "production").activeCount > 1 then .production[] | select(.measurementType == "production").lines[1].rmsCurrent else empty end'
$ curl http://172.16.12.43/production.json?details=1 2>/dev/null | jq 'if .production[] | select(.measurementType == "production").activeCount > 1 then .production[] | select(.measurementType == "production").lines[2].rmsCurrent else empty end'
$ 

and voltages:

$ curl http://172.16.12.43/production.json?details=1 2>/dev/null | jq 'if .production[] | select(.measurementType == "production").activeCount >= 1 then .production[] | select(.measurementType == "production").lines[0].rmsVoltage else empty end'
$ curl http://172.16.12.43/production.json?details=1 2>/dev/null | jq 'if .production[] | select(.measurementType == "production").activeCount > 1 then .production[] | select(.measurementType == "production").lines[1].rmsVoltage else empty end'
$ curl http://172.16.12.43/production.json?details=1 2>/dev/null | jq 'if .production[] | select(.measurementType == "production").activeCount > 1 then .production[] | select(.measurementType == "production").lines[2].rmsVoltage else empty end'
$ 

Can you check if these give you correct values in your environment?

pdeliot, 2024-10-26:
Hi Ante,

I have few time to work on this during this week-end .

But your suggestion looks better than my implementation.
I will test it and get back to you.

pdeliot, 2024-11-01:
I'm currently testing a version based on your suggestions Ante.
[enphase.yaml.gz](https://github.com/user-attachments/files/17599350/enphase.yaml.gz)
Maybe you can try it at your side.

BeneWilh, 2024-11-01:
could you give a hint where to put the enphase.yaml on a linux distribution? thx

ivoks, 2024-11-01:
@pdeliot building right now and I'll give it a try.
@BeneWilh evcc is a golang app, so it's statically compiled and any changes need to be made in source tree. On top of that templates are built into the resulting binary.

ivoks, 2024-11-01:
It's looking good. However, it's nighttime right now, so I'll be able to confirm tomorrow when some sunlight hits the panels.

pdeliot, 2024-11-06:
Ok.. looks fine at my side...
Now I need to understand how to make a PR...

pdeliot, 2024-11-10:
How to contribute?
How to create a branch ?
How to get access?

ivoks, 2024-11-11:
@pdeliot as the link bellow the comment text area suggest, you should read [contributing guidlines](https://github.com/evcc-io/evcc/blob/6c9ae85d85d95d44b85e5e9beefb61e93ec87c5f/CONTRIBUTING.md). As for forking, branching, and merging, it really depends on your level of expertise with git. If you have never done such a thing, it would be wise to read github docs and use github for this effort.

Good reads would be:
[Forking](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/working-with-forks/fork-a-repo)
[Editing](https://docs.github.com/en/repositories/working-with-files/managing-files/editing-files)
[Creating pull request](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request)

Evcc project is at [https://github.com/evcc-io/evcc](https://github.com/evcc-io/evcc).