evcc-io/evcc #16746: Elli Pro - Permanente Fehler im Log trotz LAN Anbindung

trichte, 2024-10-20:
### Describe the bug

Im Einsatz ist die Wallbox Elli Pro.
Die Ladepunkt Informationen sind ausgegarut und im LOG sind Fehler zur Verbidung der Elli Wallbox enthalten.
Die Wallbox ist per LAN über einen Switch angebunden.

### Steps to reproduce

EVCC starten

### Configuration details

```yaml
eters:
  - name: my_grid
    type: template
    template: goodwe-hybrid  
    usage: grid
    
#   template: goodwe-wifi
#   uri: 192.168.178.162

    # Modbus TCP
    modbus: tcpip
    id: 247
    host: 192.168.178.40 # Hostname
    port: 502 # Port 

  - name: my_pv
    type: template
    template: goodwe-hybrid
    usage: pv

#   template: goodwe-wifi 
#   uri: 192.168.178.162

    # Modbus TCP
    modbus: tcpip
    id: 247
    host: 192.168.178.40 # Hostname
    port: 502 # Port 

  - name: my_battery
    type: template

    template: goodwe-hybrid
    usage: battery

#   template: goodwe-wifi
#   uri: 192.168.178.162
 
    # Modbus TCP
    modbus: tcpip
    id: 247
    host: 192.168.178.40 # Hostname
    port: 502 # Port 

chargers:
- type: template
  template: ellipro 
  ski: '...'  
  ip: 192.168.178.36  
  name: my_wallbox

vehicles:
- name: my_car
  type: template
  template: vw 
  title: Passat Hybrid V2 
  user: *
  password: *
  vin: *
  capacity: 20
  timeout: 10s  
   

loadpoints:
- title: Carport
  charger: my_wallbox
#  vehicles: my_car
#  mode: pv
#  phases: 3
#  mincurrent: 6
#  maxcurrent: 16

site:
  title: Home PleRi # display name for UI
  meters:
    grid: my_grid # grid meter
    pv: my_pv # list of pv inverters/ meters
    battery: my_battery # list of battery meters
  residualPower: 100

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.27 # 27ct/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          MHcCAQEEIDWvRHuIipj4C0xqJZi5w/Slj+k6IXJREC8Wo5g1tsMaoAoGCCqGSM49
          AwEHoUQDQgAE7NFLdnf9t7j1xHeUL2afuNC7cA5PvPzeQeIMRl6M8uR6vofvzmHi
          LGRaNQKEwJCSEdDzvZcMSJ/Y3d8cJLJMDQ==
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          MIIBvTCCAWOgAwIBAgIRAjfQTlmaCLeDxIW7JI6TS7EwCgYIKoZIzj0EAwIwPjEL
          MAkGA1UEBhMCREUxDTALBgNVBAoTBEVWQ0MxCTAHBgNVBAsTADEVMBMGA1UEAwwM
          RVZDQ19IRU1TXzAxMB4XDTI0MDUwNTEwMjEzNVoXDTM0MDUwMzEwMjEzNVowPjEL
          MAkGA1UEBhMCREUxDTALBgNVBAoTBEVWQ0MxCTAHBgNVBAsTADEVMBMGA1UEAwwM
          RVZDQ19IRU1TXzAxMFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAE7NFLdnf9t7j1
          xHeUL2afuNC7cA5PvPzeQeIMRl6M8uR6vofvzmHiLGRaNQKEwJCSEdDzvZcMSJ/Y
          3d8cJLJMDaNCMEAwDgYDVR0PAQH/BAQDAgeAMA8GA1UdEwEB/wQFMAMBAf8wHQYD
          VR0OBBYEFF9GqcWZCEo+3bR2vCGy7nb5XZlKMAoGCCqGSM49BAMCA0gAMEUCIGD+
          SMyOkRkVSTcgg4vEJ6yBXmatdWPxnS6ic3Ll6OjpAiEAmLrR+Mxeytzdosriurr4
          2cNHenzHMcYbmCV8/nykxLY=
          -----END CERTIFICATE-----
```

### Log details

```text
[eebus ] DEBUG 2024/10/20 12:25:02 delaying connection to 064e14546d136b43ac0c00e6b924528d24729ee4 by 16.834s to minimize double connection probability
[site ] DEBUG 2024/10/20 12:25:03 ----
[lp-1 ] DEBUG 2024/10/20 12:25:03 charge power: 0W
[lp-1 ] DEBUG 2024/10/20 12:25:03 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/20 12:25:05 pv power: 7900W
[site ] DEBUG 2024/10/20 12:25:08 battery soc: 99%
[site ] DEBUG 2024/10/20 12:25:08 battery power: 0W
[site ] DEBUG 2024/10/20 12:25:09 grid meter: -2950W
[site ] DEBUG 2024/10/20 12:25:10 site power: -2850W
[lp-1 ] DEBUG 2024/10/20 12:25:10 charger status: A
[eebus ] DEBUG 2024/10/20 12:25:19 trying to connect to 064e14546d136b43ac0c00e6b924528d24729ee4 at wallbox-2113A1A304.local.
[eebus ] DEBUG 2024/10/20 12:25:19 initiating connection to 064e14546d136b43ac0c00e6b924528d24729ee4 at wallbox-2113A1A304.local.:4712/ship/
[eebus ] DEBUG 2024/10/20 12:25:21 connection to 064e14546d136b43ac0c00e6b924528d24729ee4 failed: dial tcp: lookup wallbox-2113A1A304.local. on 192.168.178.1:53: no such host
[eebus ] DEBUG 2024/10/20 12:25:21 trying to connect to 064e14546d136b43ac0c00e6b924528d24729ee4 at 192.168.178.36
[eebus ] DEBUG 2024/10/20 12:25:21 initiating connection to 064e14546d136b43ac0c00e6b924528d24729ee4 at 192.168.178.36:4712/ship/
[eebus ] DEBUG 2024/10/20 12:25:28 ski 064e14546d136b43ac0c00e6b924528d24729ee4 connected
[eebus ] DEBUG 2024/10/20 12:25:28 Send: read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/10/20 12:25:29 064e14546d136b43ac0c00e6b924528d24729ee4 websocket read error: websocket: close 1006 (abnormal closure): unexpected EOF
[eebus ] DEBUG 2024/10/20 12:25:29 ski 064e14546d136b43ac0c00e6b924528d24729ee4 disconnected
[eebus ] DEBUG 2024/10/20 12:25:29 mdns: announce
[eebus ] DEBUG 2024/10/20 12:25:29 mdns: using zeroconf
[eebus ] DEBUG 2024/10/20 12:25:29 delaying connection to 064e14546d136b43ac0c00e6b924528d24729ee4 by 13.807s to minimize double connection probability
[site ] DEBUG 2024/10/20 12:25:33 ----
[lp-1 ] DEBUG 2024/10/20 12:25:33 charge power: 0W
[lp-1 ] DEBUG 2024/10/20 12:25:33 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/20 12:25:38 pv power: 8306W
[eebus ] DEBUG 2024/10/20 12:25:37 064e14546d136b43ac0c00e6b924528d24729ee4 connection is in error state
[site ] DEBUG 2024/10/20 12:25:39 battery soc: 99%
[site ] DEBUG 2024/10/20 12:25:39 battery power: 0W
[site ] DEBUG 2024/10/20 12:25:40 grid meter: -3280W
[site ] DEBUG 2024/10/20 12:25:40 site power: -3180W
[lp-1 ] DEBUG 2024/10/20 12:25:40 charger status: A
[eebus ] DEBUG 2024/10/20 12:25:42 trying to connect to 064e14546d136b43ac0c00e6b924528d24729ee4 at wallbox-2113A1A304.local.
[eebus ] DEBUG 2024/10/20 12:25:42 initiating connection to 064e14546d136b43ac0c00e6b924528d24729ee4 at wallbox-2113A1A304.local.:4712/ship/
[eebus ] DEBUG 2024/10/20 12:25:43 connection to 064e14546d136b43ac0c00e6b924528d24729ee4 failed: dial tcp: lookup wallbox-2113A1A304.local. on 192.168.178.1:53: no such host
[eebus ] DEBUG 2024/10/20 12:25:43 trying to connect to 064e14546d136b43ac0c00e6b924528d24729ee4 at 192.168.178.36
[eebus ] DEBUG 2024/10/20 12:25:43 initiating connection to 064e14546d136b43ac0c00e6b924528d24729ee4 at 192.168.178.36:4712/ship/
[eebus ] DEBUG 2024/10/20 12:25:44 connection to 064e14546d136b43ac0c00e6b924528d24729ee4 failed: dial tcp 192.168.178.36:4712: connect: connection refused
[eebus ] DEBUG 2024/10/20 12:25:44 mdns: announce
[eebus ] DEBUG 2024/10/20 12:25:44 mdns: using zeroconf
[eebus ] DEBUG 2024/10/20 12:25:44 delaying connection to 064e14546d136b43ac0c00e6b924528d24729ee4 by 14.047s to minimize double connection probability
[eebus ] DEBUG 2024/10/20 12:25:48 mdns: remove - ski: 064e14546d136b43ac0c00e6b924528d24729ee4 name: Elli-Wallbox-2113A1A304 brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-2113A1A304 register: false host: wallbox-2113A1A304.local. port: 4712 addresses: []
[eebus ] DEBUG 2024/10/20 12:25:58 trying to connect to 064e14546d136b43ac0c00e6b924528d24729ee4 at wallbox-2113A1A304.local.
[eebus ] DEBUG 2024/10/20 12:25:58 initiating connection to 064e14546d136b43ac0c00e6b924528d24729ee4 at wallbox-2113A1A304.local.:4712/ship/
[eebus ] DEBUG 2024/10/20 12:26:00 connection to 064e14546d136b43ac0c00e6b924528d24729ee4 failed: dial tcp: lookup wallbox-2113A1A304.local. on 192.168.178.1:53: no such host
[eebus ] DEBUG 2024/10/20 12:26:00 trying to connect to 064e14546d136b43ac0c00e6b924528d24729ee4 at 192.168.178.36
[eebus ] DEBUG 2024/10/20 12:26:00 initiating connection to 064e14546d136b43ac0c00e6b924528d24729ee4 at 192.168.178.36:4712/ship/
[eebus ] DEBUG 2024/10/20 12:26:00 connection to 064e14546d136b43ac0c00e6b924528d24729ee4 failed: dial tcp 192.168.178.36:4712: connect: connection refused
[eebus ] DEBUG 2024/10/20 12:26:00 mdns: announce
[eebus ] DEBUG 2024/10/20 12:26:00 mdns: using zeroconf
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

andig, 2024-10-20:
Keine Fehler zu sehen.

closed 2024-10-20 by andig

trichte, 2024-10-20:
Es sind doch Fehler drin:
... hier mit failed
[eebus ] DEBUG 2024/10/20 12:26:00 connection to 064e14546d136b43ac0c00e6b924528d24729ee4 failed: dial tcp 192.168.178.36:4712: connect: connection refused

.. hier mit no such host ... warum auch immer auf eine irgendwie andere IP 192.168.178.1:53: no such host
[eebus ] DEBUG 2024/10/20 12:26:00 connection to 064e14546d136b43ac0c00e6b924528d24729ee4 failed: dial tcp: lookup wallbox-2113A1A304.local. on 192.168.178.1:53: no such host

trichte, 2024-10-20:
Quintessenz der Ladepunkt im EVCC ist am Ende ausgegraut

andig, 2024-10-20:
> 192.168.178.1:53: no such host

Da wird nichtmal der DNS Server gefunden. Du hast ein Netzwerkproblem, warum auch immer.

trichte, 2024-10-20:
Was soll denn das für ein Netzwerkproblem sein? Die 192.168.178.1 ist die Fritzbox! Brauch EVCC hier was besonderes an Einstellung?

andig, 2024-10-20:
> Was soll denn das für ein Netzwerkproblem sein? 

Kann ich Dir nicht sagen. 

> Brauch EVCC hier was besonderes an Einstellung?

Nein

> 192.168.178.1:53: no such host

Die Fehlermeldung ist eindeutig. Die Lösung ist ebenso einfach: Fehler ohne Docker nachstellen. Nicht umsonst steht im Issuetemplate, dass wir für Docker keinen Support geben!