evcc-io/evcc #16747: ID.4 Fahrzeugladelimit wird falsch angezeigt

longtom99, 2024-10-20:
### Describe the bug

Ich habe im Auto Sofortladen auf 100% eingestellt, es wird aber in der evcc Oberfläsche mit 80% angezeigt.
[image]

Wie kann ich die letzte Nightly Build in mein Portainer Docker installieren?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
meters:
- type: template
  template: solax
  id: 1
  host: 192.168.90.99
  port: 502
  usage: pv
  modbus: tcpip
  name: pv
- name: gridshelly
  type: template
  template: shelly-pro-3em
  usage: grid
  host: 192.168.90.98
- name: netz
  type: custom
  power:
    source: modbus
    id: 1
    uri: 192.168.90.90:502
    timeout: 5s
    register:
      address: 5
      type: holding
      decode: int32s

chargers:
- name: wallbox
  type: template
  template: eebus
  ski: d3842e77fb5461a2e407c12684e22e34452beb65 # SKI (Subject Key Identifier)
  ip: 192.168.1.99 # IP-Adresse (optional)

vehicles:
- type: template
  template: vw
  title: ID.4
  user: harald.jung@al-enterprise.com
  password: t7aUouW,r
  vin: WVGZZZE2ZPE016782
  capacity: 77
  timeout: 10s
  name: ev

loadpoints:
- title: Parkplatz
  charger: wallbox
  mode: pv
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 3m # threshold must be exceeded for this long
    threshold: 0 # maximum import power (W)

site:
  title: Zaus
  meters:
    grid: netz
    pv:
    - pv

tariffs:
  grid:
    type: awattar
    region: at # optional, choose at for Austria
    charges: 0.012 # optional, additional charges per kWh
    tax: 0.2 # optional, additional tax (0.1 for 10%)

eebus:
  certificate:
    public: |
      -----BEGIN CERTIFICATE-----
```

### Log details

```text
Log zu groß für hier?
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.130.13

longtom99, 2024-10-20:
[_evcc_logs_id4.txt](https://github.com/user-attachments/files/17450011/_evcc_logs_id4.txt)

wittmannchris, 2024-10-20:
Na einfach bei „Plan“ die 100% einstellen

wittmannchris, 2024-10-20:
Zufällig das battery active care in der App eingeschalten ?

longtom99, 2024-10-20:
Ja

andig, 2024-10-20:
Bitte trace log fürs Auto.

longtom99, 2024-10-20:
Wie mache ich die?

TomF79, 2024-10-20:
> Wie mache ich die?

Steck das Auto nochmal ab und vergewissere dich das im Auto auf 100% eingestellt ist und versuche es nochmal, auch ein Neustart von evcc kann nicht schaden.

Bei mir funktioniert es wie es sein soll.

longtom99, 2024-10-20:
OK, morgen dann.
Jetzt ist es mal voll geladen.

TomF79, 2024-10-20:
> OK, morgen dann. Jetzt ist es mal voll geladen.

Du musst dabei ja nichts laden, du siehst es auch nach dem anstecken.

Schimi1983, 2024-10-20:
ich empfehle dir, aus der config oben... deine VW zugangsdaten zu entfernen...
sonst kann jeder auf deine "Apollonia" zugreifen....

Ich würde danach auch das passwort ändern....

andig, 2024-10-20:
Das Passwort ist verbrannt. Unbedingt ändern!!! Um genau sowas zu verhindern gibts

    evcc discuss

wittmannchris, 2024-10-21:
[image]

ist in deiner APP hier etwas aktiviert ?

wittmannchris, 2024-10-21:
Ne VW bzw das Auto hat die Oberhand. EVCC steuert da nichts . 

Ausschalten in der App und über EVCC begrenzen macht da Sinn.

wittmannchris, 2024-10-21:
Probiers doch mal ob es das überhaupt ist. Also alles ausschalten in der APP

andig, 2024-10-21:
Wenn das Auto 80% sagt dann glauben wir ihm. Kein Fehler erkennbar.

closed 2024-10-21 by andig

longtom99, 2024-10-21:
> Wenn das Auto 80% sagt dann glauben wir ihm. Kein Fehler erkennbar.

Anscheinend ist das von VW so gewollt, weil in der App sehe ich auch 80%.
Hier sollte ja auch 100% erscheinen wenn ich das Sofortladen Limit erhöhe.

Passt, Danke!