evcc-io/evcc #16757: Slight logic error somewhere in syncCharger

Maddimax, 2024-10-21:
### Describe the bug

When:
* My Vehicle (Tesla + TWC3) is at 100% and finished charging
* Mode is set to "fast"

I will get errors in the log:
```
...
[lp-1 ] WARN 2024/10/21 07:51:00 charger logic error: current mismatch (got 0A, expected 16A)
[lp-1 ] DEBUG 2024/10/21 07:51:00 max charge current: 16A
[site ] DEBUG 2024/10/21 07:51:10 ----
[lp-1 ] DEBUG 2024/10/21 07:51:10 charge power: 0W
...
```
This will try to set a new charge current, which will keep the car online instead of falling asleep.

### Steps to reproduce

1. Tesla + TWC3
2. Teslalogger as data source, esphome tesla ble via home assistant
3. My config
4. Set charging to "Fast"
5. Let car charge to 100%

=> Car will never enter sleep mode.

or:

1. Have fully charged car, mode set to "Solar"
2. Set mode to "Fast"
=> Car will wake up

### Configuration details

```yaml
mqtt:
  broker: 10.1.42.4:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  user: <secret>
  password: <secret>
chargers:
  - name: CarportWallbox
    type: template
    template: twc3
    host: 10.1.42.194
tariffs:
  grid:
    type: tibber
    token: "<secret>" # access token
    homeid: "<secret>" # optional if multiple homes associated with account
  feedin:
    type: fixed
    price: 0.08 # 0.12 [currency]/kWh
  co2:
    type: grünstromindex
    zip: <secret>
vehicles:
  - name: madmobil
    type: custom
    title: MadMobil
    capacity: 55
    icon: car
    soc: # battery soc (%)
      source: mqtt
      topic: teslalogger/json/LRW3E7FR8NC508779/currentjson
      jq: .battery_level
    status:
      source: combined
      plugged:
        source: mqtt
        topic: teslalogger/json/LRW3E7FR8NC508779/currentjson
        jq: .plugged_in
      charging:
        source: mqtt
        topic: teslalogger/json/LRW3E7FR8NC508779/currentjson
        jq: .charging
    range:
      source: mqtt
      topic: teslalogger/json/LRW3E7FR8NC508779/currentjson
      jq: .battery_range_km
    odometer:
      source: mqtt
      topic: teslalogger/json/LRW3E7FR8NC508779/currentjson
      jq: .odometer
    climater:
      source: mqtt
      topic: teslalogger/json/LRW3E7FR8NC508779/currentjson
      jq: .is_preconditioning
    getMaxCurrent:
      source: mqtt
      topic: teslalogger/json/LRW3E7FR8NC508779/currentjson
      jq: .charge_current_request
    limitsoc:
      source: mqtt
      topic: teslalogger/json/LRW3E7FR8NC508779/currentjson
      jq: .charge_limit_soc
    wakeup:
      source: http
      uri: http://10.1.42.4:8123/api/services/button/press
      method: POST
      headers:
        - content-type: application/json
        - Authorization: Bearer <secret>
      insecure: false
      timeout: 10s
      body: '{"entity_id": "button.tesla_ble_5af060_wake_up"}'
    chargeenable:
      source: http
      uri: http://10.1.42.4:8123/api/services/switch/turn_{{if .chargeenable}}on{{else}}off{{end}}
      method: POST
      headers:
        - content-type: application/json
        - Authorization: Bearer <secret>
      insecure: false
      timeout: 10s
      body: '{"entity_id": "switch.tesla_ble_5af060_charger_switch"}'
    maxcurrent:
      source: http
      uri: http://10.1.42.4:8123/api/services/number/set_value
      method: POST
      headers:
        - content-type: application/json
        - Authorization: Bearer <secret>
      insecure: false
      timeout: 10s
      body: '{"entity_id": "number.tesla_ble_5af060_charging_amps", "value": {{.maxcurrent}} }'
loadpoints:
  - title: Carport # display name for UI
    charger: CarportWallbox # Wallbox Referenz
    vehicle: madmobil # Referenz auf Standardfahrzeug
    mode: pv # charge mode (off, now, minpv, pv)
meters:
  - name: tibber_grid
    type: template
    template: tibber-pulse
    usage: grid
    token: <secret>
    homeid: <secret> # optional
  - name: my_pv # Leistung PV
    type: custom
    power:
      source: mqtt
      topic: huawei/powerin
  - name: carport_pv
    type: custom
    power:
      source: mqtt
      topic: hoymiles/powerin
site:
  - title: Zuhause # display name for UI
    meters:
      grid: tibber_grid # tibber_grid # grid meter reference
      pv:
        - my_pv
        - carport_pv
interval: 15s
```
```

### Log details

```text
[lp-1 ] WARN 2024/10/21 07:51:00 charger logic error: current mismatch (got 0A, expected 16A)
[lp-1 ] DEBUG 2024/10/21 07:51:00 max charge current: 16A
[site ] DEBUG 2024/10/21 07:51:10 ----
[lp-1 ] DEBUG 2024/10/21 07:51:10 charge power: 0W
[lp-1 ] DEBUG 2024/10/21 07:51:10 charge currents: [0 0 0.3]A
[site ] DEBUG 2024/10/21 07:51:10 pv power: 0W
[site ] DEBUG 2024/10/21 07:51:10 grid meter: 326W
[site ] DEBUG 2024/10/21 07:51:10 grid currents: [0 0 0]A
[site ] DEBUG 2024/10/21 07:51:10 site power: 326W
[lp-1 ] DEBUG 2024/10/21 07:51:10 charge voltages: [3.4 0 0]V
[lp-1 ] DEBUG 2024/10/21 07:51:10 charger status: B
[lp-1 ] WARN 2024/10/21 07:51:10 charger logic error: current mismatch (got 0A, expected 16A)
[lp-1 ] DEBUG 2024/10/21 07:51:10 max charge current: 16A
[site ] DEBUG 2024/10/21 07:51:20 ----
[lp-1 ] DEBUG 2024/10/21 07:51:20 charge power: 0W
[lp-1 ] DEBUG 2024/10/21 07:51:20 charge currents: [0 0 0.3]A
[site ] DEBUG 2024/10/21 07:51:20 pv power: 0W
[site ] DEBUG 2024/10/21 07:51:20 grid meter: 324W
[site ] DEBUG 2024/10/21 07:51:20 grid currents: [0 0 0]A
[site ] DEBUG 2024/10/21 07:51:20 site power: 324W
[lp-1 ] DEBUG 2024/10/21 07:51:20 charge voltages: [3.4 0 0]V
[lp-1 ] DEBUG 2024/10/21 07:51:20 charger status: B
[lp-1 ] WARN 2024/10/21 07:51:20 charger logic error: current mismatch (got 0A, expected 16A)
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.130.13

andig, 2024-10-21:
evcc erwartet, den freigegebenen Ladestrom zu sehen, nicht den tatsächlichen.

Maddimax, 2024-10-21:
Könnte er das ignorieren wenn er nicht am laden ist?

andig, 2024-10-21:
Dass er bei Vorgabestrom 0 nicht lädt ist ja kein Wunder- das kann also kein Kriterium für "kein Fehler" sein.