evcc-io/evcc #16763: EVCC Startfehler bei Solaredge SE10K RWB WR

CSH12, 2024-10-21:
### Describe the bug

- EVCC läuft unter HomeAssistant
- In EVCC ist der Modbus-Proxy konfiguriert (damit HA und EVCC auf die Daten des WR Zugriff haben aber nur ein Zugang zum WR selbst existiert) 
- Ich nutze aktuell die v0.129 da die 0.130.xx mit meinen WBs (Wallbox PulsarPlus) ein Problem hat und nicht läuft.

[main ] INFO 2024/10/21 12:48:55 evcc 0.129.0
[main ] INFO 2024/10/21 12:48:55 using config file: /config/evcc.yaml
[db ] INFO 2024/10/21 12:48:55 using sqlite database: /config/evcc.db
[mqtt ] INFO 2024/10/21 12:48:56 connecting evcc-2019568367 at tcp://192.168.2.68:1883
[mqtt ] DEBUG 2024/10/21 12:48:56 tcp://192.168.2.68:1883 connected
[main ] INFO 2024/10/21 12:48:56 listening at :7070
[proxy-5200] DEBUG 2024/10/21 12:48:56 modbus proxy for 192.168.2.66:1502 listening at :5200
**_[main ] FATAL 2024/10/21 12:48:56 cannot create meter 'my_pv': cannot create meter type 'template': cannot create meter type 'custom': power: add[0]: not a SunSpec device_**
[main ] FATAL 2024/10/21 12:48:56 will attempt restart in: 15m0s
[main ] INFO 2024/10/21 12:48:56 new version available: 0.130.13

### Steps to reproduce

Restart EVCC
Nach einer gewissen Zeit läuft irgendwann EVCC, wenn ich es dann mal neu starte, umkonfiguriere, HA neu starte, dauert es dann wieder EWIG bis EVCC sich dann wieder einkriegt, und doch noch startet 

### Configuration details

```yaml
##########################################################################################################
#--------Verbindung und LOG -------------------------------------------------------------------------------
##########################################################################################################
sponsortoken: ************

modbusproxy:
  - port: 5200
    uri: 192.168.2.66:1502
     # readonly: true

network:
  schema: http
  host: 192.168.2.68
  port: 7070

mqtt:
  broker: 192.168.2.68:1883
  topic: evcc
  user: ************
  password: ************
  
log: debug
levels:
  cache: error
  # site: debug
  # lp-1: debug
  # lp-2: debug
  # tesla: trace
  
interval: 30s # control cycle interval

##########################################################################################################
#--------Einzelne Geräte -------------------------------------------------------------------------------
##########################################################################################################

chargers:
  - name: PulsarPlus_rechts    #191221
    type: template
    template: pulsarplus
    #timeout: 5m # optional
    stationid: 191221
    #connector: 1
    #remotestart: true

  # - name: PulsarPlus_mitte    #191252
    # type: template
    # template: pulsarplus
    # #timeout: 5m # optional
    # stationid: 191252
    # #connector: 1
    # #remotestart: true

meters:
  - name: my_pv
    type: template
    template: solaredge-hybrid
    usage: pv
    modbus: tcpip
    id: 1
    host: 192.168.2.68 # Hostname
    port: 5200 # Port
    timeout: 10s # optional
    watchdog: 60s # optional
      
  - name: my_grid
    type: template
    template: solaredge-hybrid
    usage: grid
    modbus: tcpip
    id: 1
    host: 192.168.2.68 # Hostname
    port: 5200 # Port
    timeout: 10s # optional
    watchdog: 60s # optional

  - name: my_battery
    type: template
    template: solaredge-hybrid
    usage: battery
    modbus: tcpip
    id: 1
    host: 192.168.2.68 # Hostname
    port: 5200 # Port
    timeout: 10s # optional
    watchdog: 60s # optional

# # loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Garage_rechts
    charger: PulsarPlus_rechts
    # vehicle: Test
    # meter: charge # charge meter
    mode: pv
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
      
  # - title: Garage_mitte
    # charger: PulsarPlus_mitte
    # # vehicle: Test
    # # meter: charge # charge meter
    # mode: pv
    # enable: # pv mode enable behavior
      # delay: 1m # threshold must be exceeded for this long
      # threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    # disable: # pv mode disable behavior
      # delay: 3m # threshold must be exceeded for this long
      # threshold: 0 # maximum import power (W)

# # vehicle definitions
# # name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# # for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - name: EQA
    type: template
    template: mercedes
    title: EQA 300 # Wird in der Benutzeroberfläche angezeigt (optional)
    icon: car # Icon in der Benutzeroberfläche (optional)
    user: "************" # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    region: EMEA
    accesstoken: ************
    refreshtoken: ************
    vin: ************ # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
    capacity: 66.5 # Akkukapazität in kWh (optional)
    phases: 3 # Die maximale Anzahl der Phasen welche genutzt werden können (optional)
    mode: PV # Möglich sind Off, Now, MinPV und PV, oder leer wenn keiner definiert werden soll (optional)
    minCurrent: 6 # Definiert die minimale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll (optional)
    maxCurrent: 16 # Definiert die maximale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll (optional)
    # #identifiers: # Kann meist erst später eingetragen werden, siehe: https://docs.evcc.io/docs/features/vehicle (optional)
    # #priority: # Priorität des Ladepunktes oder Fahrzeugs in Relation zu anderen Ladepunkten oder Fahrzeugen für die Zuweisung von PV-Energie (optional)

site:
  - title: Zuhause
    meters:
      grid: my_grid
      pv: my_pv
      battery: my_battery
    residualPower: 500 # additional household usage margin
    # prioritySoc: 75 # give home battery priority up to this soc (empty to disable)
    # bufferSoc: 0 # continue charging on battery above soc (0 to disable)
    # bufferStartSoc: 50 # start charging on battery above soc (0 to disable)
    maxGridSupplyWhileBatteryCharging: 50 # ignore battery charging if AC consumption is above this value
    # smartCostLimit: 0 # set cost limit for automatic charging in PV mode

# # tariffs are the fixed or variable tariffs
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price (or price zones)
    type: fixed
    price: 0.3564 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.082 # [currency]/kWh

  co2:
    type: grünstromindex # Beispiel GrünstromIndex (nur Deutschland)
    zip: ************ # Deine Postleitzahl
```

### Log details

```text
[main ] INFO 2024/10/21 12:48:55 evcc 0.129.0
[main ] INFO 2024/10/21 12:48:55 using config file: /config/evcc.yaml
[db ] INFO 2024/10/21 12:48:55 using sqlite database: /config/evcc.db
[mqtt ] INFO 2024/10/21 12:48:56 connecting evcc-2019568367 at tcp://192.168.2.68:1883
[mqtt ] DEBUG 2024/10/21 12:48:56 tcp://192.168.2.68:1883 connected
[main ] INFO 2024/10/21 12:48:56 listening at :7070
[proxy-5200] DEBUG 2024/10/21 12:48:56 modbus proxy for 192.168.2.66:1502 listening at :5200
**_[main ] FATAL 2024/10/21 12:48:56 cannot create meter 'my_pv': cannot create meter type 'template': cannot create meter type 'custom': power: add[0]: not a SunSpec device_**
[main ] FATAL 2024/10/21 12:48:56 will attempt restart in: 15m0s
[main ] INFO 2024/10/21 12:48:56 new version available: 0.130.13

Zusätzlich kommt hier meistens dann noch das Problem auf, dass es irgendwelche Rate Limits von GIT gibt, die dann 45min benötigen, bis wieder ein Zugriff möglich ist. Gibt es dafür auch eine Lösung?
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.129

andig, 2024-10-21:
> I have verified that the issue is reproducible with the latest nightly build

Wirklich?

CSH12, 2024-10-21:
Nein, da wie gesagt EVCC mit den 0.130.xx ja bei mir wegen den Wallbox-Problemen leider gar nicht läuft... Ohne das Häkchen Konnte ich aber den Issue nicht anlegen

andig, 2024-10-21:
> Nein, da wie gesagt EVCC mit den 0.130.xx ja bei mir wegen den Wallbox-Problemen leider gar nicht läuft...

Was meinst Du genau? Pulsar sollte zwischenzeitlich funktionieren.

> Ohne das Häkchen konnte ich aber den Issue nicht anlegen

Das kann auch auf Windows getestet werden. In jedem Fall brauchts ein trace log um zu sehen was da los ist, idealerweise für 

    evcc meter

CSH12, 2024-10-21:
jetzt hat sich gerade EVCC wieder eingekriegt, läuft stabil und mein EV kommt gleich zum Laden nach Hause. Außerdem scheint gerade bei uns die Sonne, da kann ich jetzt nicht wieder ~2h warten bis der Reboot geklappt hat ;-)

Aber wenn die Sonne dann weg ist/das Auto geladen ist, kann ich gerne ein Trace machen. kein Thema. 
Reicht dir da der Trace aus dem Web-Frontend? Oder habt ihr irgendwo eine Beschreibung wie das gemacht werden soll?

CSH12, 2024-10-21:
Hier das Log nach dem Neustart (über HA-Addon)
[evcc-20241021-135202-debug.log](https://github.com/user-attachments/files/17464244/evcc-20241021-135202-debug.log)

andig, 2024-10-21:
> In jedem Fall brauchts ein trace log

CSH12, 2024-10-21:
heißt?

andig, 2024-10-21:
Prima. Dann wäre es gut das auch zu speichern ;)

CSH12, 2024-10-21:
Oh nein, Sorry... Mein Fehler... Falscher Anhang :-O
[evcc-20241021-183120-trace.log](https://github.com/user-attachments/files/17464428/evcc-20241021-183120-trace.log)
Heißt dann sogar "trace".log wenn man das richtige File nimmt.

andig, 2024-10-21:
> ISt halt so von euch programmiert

Isso 🤓

CSH12, 2024-10-21:
Ja, das habt ihr gut gemacht - extra für DAU's wie mich 👍 

Und hier noch als Ergänzung dann mit dem GIT Error oben drauf
[evcc-20241021-185855-trace.log](https://github.com/user-attachments/files/17464547/evcc-20241021-185855-trace.log)

Jetzt ist gerade wieder schön die Endlosschleife aktiv bis sich EVCC dann irgendwann wieder erholt hat...

CSH12, 2024-10-21:
gleiche Meldung auch mit der 0.130.13
[evcc-20241021-191114-trace.log](https://github.com/user-attachments/files/17464646/evcc-20241021-191114-trace.log)

andig, 2024-10-21:
Hier bricht mehrere male die Verbindung an:

```
[proxy-5200] DEBUG 2024/10/21 19:09:53 modbus proxy for 192.168.2.66:1502 listening at :5200
[db    ] TRACE 2024/10/21 19:09:53 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
[sunspec] TRACE 2024/10/21 19:09:53 modbus: send 00 01 00 00 00 06 01 03 9c 40 00 02
[proxy-5200] TRACE 2024/10/21 19:09:53 read holdings: id 1 addr 40000 qty 2
[proxy-5200] TRACE 2024/10/21 19:09:54 read holding: read tcp 192.168.2.68:53588->192.168.2.66:1502: read: connection reset by peer
[sunspec] TRACE 2024/10/21 19:09:54 modbus: recv 00 01 00 00 00 03 01 83 04
[sunspec] TRACE 2024/10/21 19:09:54 modbus: send 00 02 00 00 00 06 01 03 c3 50 00 02
[proxy-5200] TRACE 2024/10/21 19:09:54 read holdings: id 1 addr 50000 qty 2
[proxy-5200] TRACE 2024/10/21 19:09:54 read holding: dial tcp 192.168.2.66:1502: connect: connection refused
[sunspec] TRACE 2024/10/21 19:09:54 modbus: recv 00 02 00 00 00 03 01 83 04
[sunspec] TRACE 2024/10/21 19:09:54 modbus: send 00 03 00 00 00 06 01 03 00 00 00 02
[proxy-5200] TRACE 2024/10/21 19:09:54 read holdings: id 1 addr 0 qty 2
[proxy-5200] TRACE 2024/10/21 19:09:54 read holding: dial tcp 192.168.2.66:1502: connect: connection refused
[sunspec] TRACE 2024/10/21 19:09:54 modbus: recv 00 03 00 00 00 03 01 83 04
[sunspec] TRACE 2024/10/21 19:09:54 modbus: send 00 04 00 00 00 06 01 03 9c 40 00 02
[proxy-5200] TRACE 2024/10/21 19:09:54 read holdings: id 1 addr 40000 qty 2
[proxy-5200] TRACE 2024/10/21 19:09:54 read holding: dial tcp 192.168.2.66:1502: connect: connection refused
[sunspec] TRACE 2024/10/21 19:09:54 modbus: recv 00 04 00 00 00 03 01 83 04
[sunspec] TRACE 2024/10/21 19:09:54 modbus: send 00 05 00 00 00 06 01 03 c3 50 00 02
[proxy-5200] TRACE 2024/10/21 19:09:54 read holdings: id 1 addr 50000 qty 2
[proxy-5200] TRACE 2024/10/21 19:09:54 read holding: dial tcp 192.168.2.66:1502: connect: connection refused
[sunspec] TRACE 2024/10/21 19:09:54 modbus: recv 00 05 00 00 00 03 01 83 04
[sunspec] TRACE 2024/10/21 19:09:54 modbus: send 00 06 00 00 00 06 01 03 00 00 00 02
[proxy-5200] TRACE 2024/10/21 19:09:54 read holdings: id 1 addr 0 qty 2
[proxy-5200] TRACE 2024/10/21 19:09:54 read holding: dial tcp 192.168.2.66:1502: connect: connection refused
[sunspec] TRACE 2024/10/21 19:09:54 modbus: recv 00 06 00 00 00 03 01 83 04
[main  ] FATAL 2024/10/21 19:09:54 cannot create meter 'my_pv': cannot create meter type 'template': cannot create meter type 'custom': power: add[0]: not a SunSpec device
[main  ] FATAL 2024/10/21 19:09:54 will attempt restart in: 15m0s
```

Warum auch immer. Du könntest testen, obs am Proxy liegt oder am Gerät.

closed 2024-10-21 by andig

CSH12, 2024-10-21:
Ich hatte ja bevor ich es auf Proxy umgestellt habe das ganze ohne Proxy am laufen (mit 2 Modbus-Geräten parallel die mit dem WR kommuniziert haben :scream:) 

Das Problem bestand da genauso schon, also würde ich mal behaupten, dass es nicht am Proxy liegt. 
Solaredge - als börsennotierte, international-tätige Firma ja keine Fehler (bzw. würde nie einen zugeben) also kann es daran auch nicht liegen.... 😄 

weitere Vorschläge?

CSH12, 2024-10-21:
[main ] INFO 2024/10/21 20:54:38 listening at :7070
[proxy-5200] DEBUG 2024/10/21 20:54:38 modbus proxy for 192.168.2.66:1502 listening at :5200
[db ] TRACE 2024/10/21 20:54:38 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
[sunspec] TRACE 2024/10/21 20:54:38 modbus: send 00 01 00 00 00 06 01 03 9c 40 00 02
[proxy-5200] TRACE 2024/10/21 20:54:38 read holdings: id 1 addr 40000 qty 2
[proxy-5200] TRACE 2024/10/21 20:54:38 read holding: read tcp 192.168.2.68:55876->192.168.2.66:1502: read: connection reset by peer
[sunspec] TRACE 2024/10/21 20:54:38 modbus: recv 00 01 00 00 00 03 01 83 04
[sunspec] TRACE 2024/10/21 20:54:38 modbus: send 00 02 00 00 00 06 01 03 c3 50 00 02
[proxy-5200] TRACE 2024/10/21 20:54:38 read holdings: id 1 addr 50000 qty 2
[proxy-5200] TRACE 2024/10/21 20:54:38 read holding: dial tcp 192.168.2.66:1502: connect: connection refused
[sunspec] TRACE 2024/10/21 20:54:38 modbus: recv 00 02 00 00 00 03 01 83 04
[sunspec] TRACE 2024/10/21 20:54:38 modbus: send 00 03 00 00 00 06 01 03 00 00 00 02
[proxy-5200] TRACE 2024/10/21 20:54:38 read holdings: id 1 addr 0 qty 2
[proxy-5200] TRACE 2024/10/21 20:54:38 read holding: dial tcp 192.168.2.66:1502: connect: connection refused
[sunspec] TRACE 2024/10/21 20:54:38 modbus: recv 00 03 00 00 00 03 01 83 04
[sunspec] TRACE 2024/10/21 20:54:38 modbus: send 00 04 00 00 00 06 01 03 9c 40 00 02
[proxy-5200] TRACE 2024/10/21 20:54:38 read holdings: id 1 addr 40000 qty 2
[proxy-5200] TRACE 2024/10/21 20:54:38 read holding: dial tcp 192.168.2.66:1502: connect: connection refused
[sunspec] TRACE 2024/10/21 20:54:38 modbus: recv 00 04 00 00 00 03 01 83 04
[sunspec] TRACE 2024/10/21 20:54:38 modbus: send 00 05 00 00 00 06 01 03 c3 50 00 02
[proxy-5200] TRACE 2024/10/21 20:54:38 read holdings: id 1 addr 50000 qty 2
[proxy-5200] TRACE 2024/10/21 20:54:38 read holding: dial tcp 192.168.2.66:1502: connect: connection refused
[sunspec] TRACE 2024/10/21 20:54:38 modbus: recv 00 05 00 00 00 03 01 83 04
[sunspec] TRACE 2024/10/21 20:54:38 modbus: send 00 06 00 00 00 06 01 03 00 00 00 02
[proxy-5200] TRACE 2024/10/21 20:54:38 read holdings: id 1 addr 0 qty 2
[proxy-5200] TRACE 2024/10/21 20:54:38 read holding: dial tcp 192.168.2.66:1502: connect: connection refused
[sunspec] TRACE 2024/10/21 20:54:38 modbus: recv 00 06 00 00 00 03 01 83 04
[main ] FATAL 2024/10/21 20:54:38 cannot create meter 'my_pv': cannot create meter type 'template': cannot create meter type 'custom': power: add[0]: not a SunSpec device
[main ] FATAL 2024/10/21 20:54:38 will attempt restart in: 15m0s

Das sieht aber immer gleich aus. auch jetzt nachdem ich wieder erst auf 0.129 zurück bin und mittlerweile wieder zum testen auf 0.130.13 hoch ist das auch wieder genau das gleiche Fehlerbild wie vorher in dem anderen log...
Scheint also was systematisches zu sein. Was und Wo kann ich aber leider nicht ausmachen.

CSH12, 2024-10-22:
Kleines Statusupdate: 
Leider funktioniert die 0.130.13 immer noch nicht. Ich habe jetzt die ganze Nacht EVCC versuchen lassen sich einkriegen und doch noch zu starten, aber es hat leider überhaupt nicht funktioniert. Immer noch der gleiche Fehler der sich nicht beheben lässt. 
Einzige alternative wäre das ganze über MQTT zu lösen - hab ich aber gerade keine Lust zu. 

--> Ich habe jetzt wieder die 0.129 zurückgeholt und nach kurzem beschweren, dass kein SunSpec Modul vorliegt hat es dann doch funktioniert (Für mich ist das leider immer noch die einzige einigermaßen funktionierende Version im Moment)

andig, 2024-10-22:
Dein WR sagt

> --> ergo: es muss schon was mit dem EVCC Plugin/Template zu tun haben

Das ist ein Geräteproblem. Einziger Lösungsansatz den ich sehe wäre ein Delay einzubauen um ihn langsamer abzufragen (so hat 129 das eigentlich unnötigerweise auch getan).

CSH12, 2024-10-22:
der Proxy auch und die Integration die auf die Proxy Daten referenziert läuft auch... Ich sehe daher das Geräteproblem nicht.

Langsamer würde doch aber auch gar nix bringen, da ich doch die Daten per Proxy ausgelesen habe und dann direkt auf die Proxydaten zugreife?!?

CSH12, 2024-10-22:
Und jetzt? 
was tun wir um das Problemchen zu beheben? 
wie gesagt denke ich nicht, dass es ein gerätespezifisches Thema ist. Vielleicht SolarEdge spezifisch…

andig, 2024-10-23:
Bitte Zugang zum WR per VPN o.ä. An info@evcc.io

CSH12, 2024-10-23:
Ich hab gerade mal die 0.131 auprobiert...

Einmal hat zwar das Wechselrichter-Thema mit dem PulsarPlus gezuckt - hier hab ich dann auf das pulsarplus-fw5 Plugin umgestellt. Hier bekomme ich noch eine Warning([lp-1 ] WARN 2024/10/23 19:23:31 charger out of sync: expected disabled, got enabled), hier muss ich die Flags die @mfuchs genannt hatte mal noch zusätzlich ausprobieren.

**Aber ansonsten gar es in Bezug auf das SolarEdge/SunSpec Thema überhaupt keine Beschwerden!!!** 
Habt ihr da jetzt auch noch was angepasst?

andig, 2024-10-23:
Nein...