evcc-io/evcc #16765: tq em420 - Aufgenommene Leistung stimmt nicht

crazycusti, 2024-10-21:
### Describe the bug

Moin!

Ich habe hier eine Heidelberg Ampfilid Energy Control Wallbox mit seperaten Zähler TQ-Group 420 , in evcc wird mir die aufgenommene Leistung aber mit Faktor *1000? anzeigt. Ich vermute mal das Register wird nicht korrekt im Scale ausgelesen.. Die Box ist erst 1 Woche alt, Zähler ebenso mit aktuellster 3.40 Firmware.

### Steps to reproduce

1. Logs auslesen.
2. Konfiguration überprüfen, EM420 Webinterface prüfen
3. Neustart evcc
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: x

sponsortoken: x

interval: 25s # control cycle interval

meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.178.110
  name: grid1
- type: template
  template: sma-hybrid
  id: 3
  host: 192.168.178.148
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: sma-hybrid
  id: 3
  host: 192.168.178.148
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3
- type: template
  template: sma-inverter-modbus
  id: 3
  host: 192.168.178.131
  port: 502
  usage: pv
  modbus: tcpip
  name: pv4
- type: template
  template: sma-inverter-modbus
  id: 3
  host: 192.168.178.126
  port: 502
  usage: pv
  modbus: tcpip
  name: pv5
- type: template
  template: sma-inverter-modbus
  id: 3
  host: 192.168.178.187
  port: 502
  usage: pv
  modbus: tcpip
  name: pv6
- name: em420
  type: template
  template: tq-em420
  usage: charger
  host: 192.168.178.94 # IP-Adresse oder Hostname
  token: x
- type: template
  template: sma-inverter-modbus
  id: 3
  host: 192.168.178.112
  port: 502
  usage: pv
  modbus: tcpip
  name: pv7
- type: template
  template: shelly-1pm
  usage: pv
  host: 192.168.178.191
  name: pv8

chargers:
- type: template
  template: heidelberg
  name: wallbox7
  modbus: rs485tcpip
  id: 2
  host: 192.168.178.123 # Hostname
  port: 8899 # Port

vehicles:
 - name: my_car
   type: template
   template: audi
   title: Audi e-tron
   user: x
   password: x
   vin: x
#   capacity: 95

loadpoints:
- title: Hof
  charger: wallbox7
  vehicle: my_car
  mode: pv
  meter: em420
  phases: 3

site:
  title: Klosterhörn
  meters:
    grid: grid1
    pv:
    - pv2
    - pv4
    - pv5
    - pv6
    - pv7
    - pv8
    battery:
    - battery3

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.26 # EUR/kWh
  feedin:
    type: fixed
    price: 0.06 # EUR/kWh
```

### Log details

```text
Okt 21 15:26:08 Smarthome evcc[3823]: [lp-1  ] DEBUG 2024/10/21 15:26:08 vehicle odometer: 113662km
Okt 21 15:26:08 Smarthome evcc[3823]: [tq-em420] TRACE 2024/10/21 15:26:08 GET http://192.168.178.94:80/api/json/local/values/smart-meter
Okt 21 15:26:08 Smarthome evcc[3823]: [sma   ] TRACE 2024/10/21 15:26:08 recv 192.168.178.110: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
Okt 21 15:26:08 Smarthome evcc[3823]: [tq-em420] TRACE 2024/10/21 15:26:08 {"smart-meter":{"configuration_id":"84b6174e6d51f10809793fc105951e0851def70b","status":"STATUS_OK","timestamp":{"seconds":1729517168,"nanos":592178297},"values":{"active_energy_+":69564000,"active_energy_+_L1":24308300,"active_energy_+_L2":22834400,"active_energy_+_L3":22421200,"active_energy_-":0,"active_energy_-_L1":0,"active_energy_-_L2":0,"active_energy_-_L3":0,"active_power_+":4070700,"active_power_+_L1":1349300,"active_power_+_L2":1373700,"active_power_+_L3":1347700,"active_power_-":0,"active_power_-_L1":0,"active_power_-_L2":0,"active_power_-_L3":0,"apparent_energy_+":70277900,"apparent_energy_+_L1":24835600,"apparent_energy_+_L2":22930600,"apparent_energy_+_L3":22511900,"apparent_energy_-":0,"apparent_energy_-_L1":0,"apparent_energy_-_L2":0,"apparent_energy_-_L3":0,"apparent_power_+":4099800,"apparent_power_+_L1":1359600,"apparent_power_+_L2":1383600,"apparent_power_+_L3":1356700,"apparent_power_-":0,"apparent_power_-_L1":0,"apparent_power_-_L2":0,"apparent_power_-_L3":0,"current_L1":5903,"current_L2":5959,"current_L3":5886,"power_factor":993,"power_factor_L1":992,"power_factor_L2":993,"power_factor_L3":993,"reactive_energy_+":0,"reactive_energy_+_L1":0,"reactive_energy_+_L2":0,"reactive_energy_+_L3":0,"reactive_energy_-":7156000,"reactive_energy_-_L1":3215100,"reactive_energy_-_L2":2011600,"reactive_energy_-_L3":1929200,"reactive_power_+":0,"reactive_power_+_L1":0,"reactive_power_+_L2":0,"reactive_power_+_L3":0,"reactive_power_-":487900,"reactive_power_-_L1":166600,"reactive_power_-_L2":165100,"reactive_power_-_L3":156100,"supply_frequency":49998,"voltage_L1":230540,"voltage_L2":232296,"voltage_L3":230599}}}
Okt 21 15:26:08 Smarthome evcc[3823]: [lp-1  ] DEBUG 2024/10/21 15:26:08 charge voltages: [231 232 231]V
Okt 21 15:26:08 Smarthome evcc[3823]: [lp-1  ] DEBUG 2024/10/21 15:26:08 detected connected phases: 3p
Okt 21 15:26:08 Smarthome evcc[3823]: [lp-1  ] DEBUG 2024/10/21 15:26:08 charge total import: 69564.000kWh
Okt 21 15:26:09 Smarthome evcc[3823]: [heidel] TRACE 2024/10/21 15:26:09 modbus: send 02 04 00 05 00 01 21 f8
Okt 21 15:26:09 Smarthome evcc[3823]: [sma   ] TRACE 2024/10/21 15:26:09 recv 192.168.178.110: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
Okt 21 15:26:09 Smarthome evcc[3823]: [heidel] TRACE 2024/10/21 15:26:09 modbus: recv 02 04 02 00 07 bc f2
Okt 21 15:26:09 Smarthome evcc[3823]: [lp-1  ] DEBUG 2024/10/21 15:26:09 charger status: C
Okt 21 15:26:09 Smarthome evcc[3823]: [lp-1  ] DEBUG 2024/10/21 15:26:09 charge start energy: 69564.000kWh
Okt 21 15:26:09 Smarthome evcc[3823]: [lp-1  ] INFO 2024/10/21 15:26:09 car connected
Okt 21 15:26:09 Smarthome evcc[3823]: [lp-1  ] DEBUG 2024/10/21 15:26:09 pv timer elapse
Okt 21 15:26:09 Smarthome evcc[3823]: [lp-1  ] DEBUG 2024/10/21 15:26:09 pv timer inactive
Okt 21 15:26:09 Smarthome evcc[3823]: [lp-1  ] DEBUG 2024/10/21 15:26:09 charge total import: 69564.000kWh
Okt 21 15:26:09 Smarthome evcc[3823]: [lp-1  ] DEBUG 2024/10/21 15:26:09 charge start energy: 69564.000kWh
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.130.13

andig, 2024-10-21:
Was sagt denn

    evcc meter --log trace,db:error

n0cturne, 2024-10-22:
In deiner Konfig hast du als usage "charger" beim Zähler:
  - name: em420
    type: template
    template: tq-em420
    usage: charger
    host: 192.168.178.94 # IP-Adresse oder Hostname
    token: x

In der Dokue für Zähler an einer Wallbox wird hier jedoch immer "charge" verwendet. Vielleicht ist das auch das Problem?

andig, 2024-10-22:
Das wird komplett ignoriert. Der externe Zähler hat ja nur einen Verwendungszweck.

An-Wi, 2024-10-22:
> Das wird komplett ignoriert. Der externe Zähler hat ja nur einen Verwendungszweck.

Mh, korrgiere mich, wenn ich das falsch verstehe, aber der "externe Zähler" wird ihr doch anders als üblich verwendet.
Ich selbst habe auch den TQ EM-3xx im Einsatz als "Netzverknüpfungszähler" (usage: grid).
Der Grid-Zähler von @crazycusti ist ein ganz anderen: "sma-home-manager" laut yaml.
Der TQ EM-420 wird hier doch als "Wallbox-Zähler" (usage: charge[r]) verwendet. 

Wenn das flag "usage" ignoriert wird, woher weißt du dann, dass es der Wallbox-Zähler und kein Grid-Zähler ist?

n0cturne, 2024-10-22:
und der Zähler wird auch beim Loadpoint verwendet:
loadpoints:
  - title: Hof
    charger: wallbox7
    vehicle: my_car
    mode: pv
    meter: **em420**
    phases: 3

Oder genügt das Bereits?

andig, 2024-10-22:
Gegenfrage: was soll er denn anderes messen als die angeschlossenen Ströme und Leistungen?

VolkerK62, 2024-10-22:
https://github.com/evcc-io/evcc/blob/76650945c1540539fbd204fddb74f8fe9218caae/templates/definition/meter/tq-em420.yaml#L29

`usage` spielt hier überhaupt keine Rolle. Deshalb kommt auch kein Fehler, trotz des falschen wordings.

>Wenn das flag "usage" ignoriert wird, woher weißt du dann, dass es der Wallbox-Zähler und kein Grid-Zähler ist?

Entscheidend ist die Zuordnung in `site` oder `loadpoint`.

`usage` ist wichtig bzgl. der richtigen Register Abfragen (z.B. Hybrid-Wechselricher) und/oder beim richtigen Skalieren.

n0cturne, 2024-10-22:
aber geht es hier nicht eventuell um die Skalierung (_Leistung aber mit Faktor *1000_)?

VolkerK62, 2024-10-22:
laut dem Log oben, kommen folgende Leistungswerte:
```
 "active_power_+": 4070700,
 "active_power_-": 0,
```
Die Differenz wird in der Implementierung durch 1000 geteilt
https://github.com/evcc-io/evcc/blob/76650945c1540539fbd204fddb74f8fe9218caae/meter/tq-em420.go#L147

Somit sollte eine Leistung von 4070,7W angezeigt werden.

Aber es fehlt die Antwort auf https://github.com/evcc-io/evcc/issues/16765#issuecomment-2427099116

crazycusti, 2024-10-23:
@andig 

```
[main  ] INFO 2024/10/23 18:05:42 evcc 0.130.13
[main  ] INFO 2024/10/23 18:05:42 using config file: /root/evcc.yaml
[tq-em420] TRACE 2024/10/23 18:05:42 GET http://192.168.178.94:80/api/json/local/values/smart-meter
[sma   ] TRACE 2024/10/23 18:05:42 send 192.168.178.110: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/23 18:05:42 recv 192.168.178.110: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/23 18:05:42 new energy meter at 192.168.178.110 - Serial=3013357031
[sma   ] TRACE 2024/10/23 18:05:42 send 192.168.178.110: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[tq-em420] TRACE 2024/10/23 18:05:42 {"smart-meter":{"configuration_id":"84b6174e6d51f10809793fc105951e0851def70b","status":"STATUS_OK","timestamp":{"seconds":1729699542,"nanos":583875718},"values":{"active_energy_+":124596300,"active_energy_+_L1":42895100,"active_energy_+_L2":41204500,"active_energy_+_L3":40496500,"active_energy_-":0,"active_energy_-_L1":0,"active_energy_-_L2":0,"active_energy_-_L3":0,"active_power_+":6730100,"active_power_+_L1":2248800,"active_power_+_L2":2264500,"active_power_+_L3":2216800,"active_power_-":0,"active_power_-_L1":0,"active_power_-_L2":0,"active_power_-_L3":0,"apparent_energy_+":125622500,"apparent_energy_+_L1":43635100,"apparent_energy_+_L2":41352900,"apparent_energy_+_L3":40634700,"apparent_energy_-":0,"apparent_energy_-_L1":0,"apparent_energy_-_L2":0,"apparent_energy_-_L3":0,"apparent_power_+":6745200,"apparent_power_+_L1":2254100,"apparent_power_+_L2":2269400,"apparent_power_+_L3":2221700,"apparent_power_-":0,"apparent_power_-_L1":0,"apparent_power_-_L2":0,"apparent_power_-_L3":0,"current_L1":9825,"current_L2":9875,"current_L3":9626,"power_factor":998,"power_factor_L1":998,"power_factor_L2":998,"power_factor_L3":998,"reactive_energy_+":0,"reactive_energy_+_L1":0,"reactive_energy_+_L2":0,"reactive_energy_+_L3":0,"reactive_energy_-":11574900,"reactive_energy_-_L1":4987100,"reactive_energy_-_L2":3372600,"reactive_energy_-_L3":3215100,"reactive_power_+":0,"reactive_power_+_L1":0,"reactive_power_+_L2":0,"reactive_power_+_L3":0,"reactive_power_-":450200,"reactive_power_-_L1":153500,"reactive_power_-_L2":149300,"reactive_power_-_L3":147300,"supply_frequency":50001,"voltage_L1":229511,"voltage_L2":229879,"voltage_L3":230858}}}
[sma   ] TRACE 2024/10/23 18:05:42 recv 192.168.178.110: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/23 18:05:42 new energy meter at 192.168.178.110 - Serial=3013357031
[sma   ] TRACE 2024/10/23 18:05:42 found device 3013357031 at 192.168.178.110
pv5
---
[modbus] TRACE 2024/10/23 18:05:42 modbus: send 00 01 00 00 00 06 03 04 78 37 00 02
[modbus] TRACE 2024/10/23 18:05:42 modbus: recv 00 01 00 00 00 07 03 04 04 00 00 00 00
[modbus] TRACE 2024/10/23 18:05:42 modbus: send 00 02 00 00 00 06 03 03 77 31 00 04
[modbus] TRACE 2024/10/23 18:05:42 modbus: recv 00 02 00 00 00 0b 03 03 08 00 00 00 00 00 63 72 e2
Power:  0W
Energy: 6517.5kWh

pv4
---
[modbus] TRACE 2024/10/23 18:05:42 modbus: send 00 01 00 00 00 06 03 04 78 37 00 02
[modbus] TRACE 2024/10/23 18:05:43 modbus: recv 00 01 00 00 00 07 03 04 04 00 00 00 00
[modbus] TRACE 2024/10/23 18:05:43 modbus: send 00 02 00 00 00 06 03 03 77 31 00 04
[modbus] TRACE 2024/10/23 18:05:43 modbus: recv 00 02 00 00 00 0b 03 03 08 00 00 00 00 00 4d b1 c3
Power:  0W
Energy: 5091.8kWh

battery3
--------
[modbus] TRACE 2024/10/23 18:05:43 modbus: send 00 01 00 00 00 06 03 04 7a a3 00 02
[modbus] TRACE 2024/10/23 18:05:43 modbus: recv 00 01 00 00 00 07 03 04 04 00 00 00 00
[modbus] TRACE 2024/10/23 18:05:43 modbus: send 00 02 00 00 00 06 03 04 7a a1 00 02
[sma   ] TRACE 2024/10/23 18:05:43 recv 192.168.178.110: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[modbus] TRACE 2024/10/23 18:05:43 modbus: recv 00 02 00 00 00 07 03 04 04 00 00 00 00
[modbus] TRACE 2024/10/23 18:05:43 modbus: send 00 03 00 00 00 06 03 03 7a a9 00 04
[sma   ] TRACE 2024/10/23 18:05:43 send discover package
[modbus] TRACE 2024/10/23 18:05:43 modbus: recv 00 03 00 00 00 0b 03 03 08 00 00 00 00 00 4b 28 5d
[modbus] TRACE 2024/10/23 18:05:43 modbus: send 00 04 00 00 00 06 03 03 78 7d 00 02
[sma   ] TRACE 2024/10/23 18:05:43 recv 192.168.178.126: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2024/10/23 18:05:43 send 192.168.178.126: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/23 18:05:43 recv 192.168.178.110: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2024/10/23 18:05:43 recv 192.168.178.131: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2024/10/23 18:05:43 recv 192.168.178.148: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry]
[modbus] TRACE 2024/10/23 18:05:43 modbus: recv 00 04 00 00 00 07 03 03 04 00 00 00 05
[... cut ...]
``` 

@VolkerK62 

Hab mich wahrscheinlich etwas umständlich ausgedrückt, es geht nicht um die aktuelle Leistung, sondern um die aufgenomme, siehe:

em420
-----
Power:          6730W
Energy:         124596.3kWh
Current L1..L3: 9.82A 9.88A 9.63A
Voltage L1..L3: 230V 230V 231V

In Wirklichkeit sollten es 124kWh sein, nicht 124596. Power stimmt, Energy nicht.

Und ja, der Zähler wird als Wallbox-Zähler missbraucht, mir bewusst das dass Teil teurer ist als die Wallbox selbst, aber der war halt über..

VolkerK62, 2024-10-23:
>Energy: 124596.3kWh

stimmt, die Einheit ist falsch. Das sind Wh.

andig, 2024-10-23:
PR welcome

VolkerK62, 2024-10-23:
ich kümmer mich drum

PR #16810 (2024-10-23): tq em420 correction energy
fix #16765 
Wert wurde in Wh ausgegeben, aber mit Einheit kWh
closed 2024-10-23 by andig