evcc-io/evcc #16768: [lp-1 ] ERROR: charger disable: set charging profile: Rejected

CSH12, 2024-10-21:
### Describe the bug

Sobald ich auf 0.130.13 upgedated habe, bekomme ich vom LP1 den Fehler, dass das Ladeprofil nicht akzeptiert wurde. 
[lp-1 ] ERROR 2024/10/21 20:09:23 charger disable: set charging profile: Rejected

Ich hatte auch mal nachgelesen, dass man wohl noch ein Remotestart-Flag setzen kann, bringt aber weder bei dieser noch bei vorhergehenden Versionen Besserung.

### Steps to reproduce

gibt nix zu reproduzieren, ist einfach dauerhaft da.

### Configuration details

```yaml
##########################################################################################################
#--------Verbindung und LOG -------------------------------------------------------------------------------
##########################################################################################################
sponsortoken: ************

modbusproxy:
  - port: 5200
    uri: 192.168.2.66:1502
     # readonly: true

network:
  schema: http
  host: 192.168.2.68
  port: 7070

mqtt:
  broker: 192.168.2.68:1883
  topic: evcc
  user: ************
  password: ************
  
log: debug
levels:
  cache: error
  # site: debug
  # lp-1: debug
  # lp-2: debug
  # tesla: trace
  
interval: 30s # control cycle interval

##########################################################################################################
#--------Einzelne Geräte -------------------------------------------------------------------------------
##########################################################################################################

chargers:
  - name: PulsarPlus_rechts    #191221
    type: template
    template: pulsarplus
    #timeout: 5m # optional
    stationid: 191221
    #connector: 1
    #remotestart: true

  # - name: PulsarPlus_mitte    #191252
    # type: template
    # template: pulsarplus
    # #timeout: 5m # optional
    # stationid: 191252
    # #connector: 1
    # #remotestart: true

meters:
  - name: my_pv
    type: template
    template: solaredge-hybrid
    usage: pv
    modbus: tcpip
    id: 1
    host: 192.168.2.68 # Hostname
    port: 5200 # Port
    timeout: 10s # optional
    watchdog: 60s # optional
      
  - name: my_grid
    type: template
    template: solaredge-hybrid
    usage: grid
    modbus: tcpip
    id: 1
    host: 192.168.2.68 # Hostname
    port: 5200 # Port
    timeout: 10s # optional
    watchdog: 60s # optional

  - name: my_battery
    type: template
    template: solaredge-hybrid
    usage: battery
    modbus: tcpip
    id: 1
    host: 192.168.2.68 # Hostname
    port: 5200 # Port
    timeout: 10s # optional
    watchdog: 60s # optional

# # loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Garage_rechts
    charger: PulsarPlus_rechts
    # vehicle: Test
    # meter: charge # charge meter
    mode: pv
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
      
  # - title: Garage_mitte
    # charger: PulsarPlus_mitte
    # # vehicle: Test
    # # meter: charge # charge meter
    # mode: pv
    # enable: # pv mode enable behavior
      # delay: 1m # threshold must be exceeded for this long
      # threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    # disable: # pv mode disable behavior
      # delay: 3m # threshold must be exceeded for this long
      # threshold: 0 # maximum import power (W)

# # vehicle definitions
# # name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# # for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - name: EQA
    type: template
    template: mercedes
    title: EQA 300 # Wird in der Benutzeroberfläche angezeigt (optional)
    icon: car # Icon in der Benutzeroberfläche (optional)
    user: "************" # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    region: EMEA
    accesstoken: ************
    refreshtoken: ************
    vin: ************ # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
    capacity: 66.5 # Akkukapazität in kWh (optional)
    phases: 3 # Die maximale Anzahl der Phasen welche genutzt werden können (optional)
    mode: PV # Möglich sind Off, Now, MinPV und PV, oder leer wenn keiner definiert werden soll (optional)
    minCurrent: 6 # Definiert die minimale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll (optional)
    maxCurrent: 16 # Definiert die maximale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll (optional)
    # #identifiers: # Kann meist erst später eingetragen werden, siehe: https://docs.evcc.io/docs/features/vehicle (optional)
    # #priority: # Priorität des Ladepunktes oder Fahrzeugs in Relation zu anderen Ladepunkten oder Fahrzeugen für die Zuweisung von PV-Energie (optional)

site:
  - title: Zuhause
    meters:
      grid: my_grid
      pv: my_pv
      battery: my_battery
    residualPower: 500 # additional household usage margin
    # prioritySoc: 75 # give home battery priority up to this soc (empty to disable)
    # bufferSoc: 0 # continue charging on battery above soc (0 to disable)
    # bufferStartSoc: 50 # start charging on battery above soc (0 to disable)
    maxGridSupplyWhileBatteryCharging: 50 # ignore battery charging if AC consumption is above this value
    # smartCostLimit: 0 # set cost limit for automatic charging in PV mode

# # tariffs are the fixed or variable tariffs
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price (or price zones)
    type: fixed
    price: 0.3564 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.082 # [currency]/kWh

  co2:
    type: grünstromindex # Beispiel GrünstromIndex (nur Deutschland)
    zip: ************ # Deine Postleitzahl
```

### Log details

```text
[lp-1 ] DEBUG 2024/10/21 20:14:53 charge voltages: [236 253 253]V
[lp-1 ] DEBUG 2024/10/21 20:14:53 detected connected phases: 3p
[lp-1 ] DEBUG 2024/10/21 20:14:53 charge total import: 715.118kWh
[lp-1 ] DEBUG 2024/10/21 20:14:53 charger status: A
[ocpp ] TRACE 2024/10/21 20:14:53 enqueued CALL [1555718446, SetChargingProfile] for 191221
[ocpp ] TRACE 2024/10/21 20:14:53 dispatched request 1555718446 for 191221
[ocpp ] TRACE 2024/10/21 20:14:53 sent JSON message to 191221: [2,"1555718446","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"stackLevel":1,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-10-21T18:13:53Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0}]}}}]
[ocpp ] TRACE 2024/10/21 20:14:53 started timeout timer for 191221
[ocpp ] TRACE 2024/10/21 20:14:53 received JSON message from 191221: [3,"1555718446",{"status": "Rejected"}]
[ocpp ] TRACE 2024/10/21 20:14:53 handling incoming CALL RESULT [1555718446] from 191221
[ocpp ] TRACE 2024/10/21 20:14:53 completed request 1555718446 for 191221
[lp-1 ] ERROR 2024/10/21 20:14:53 charger disable: set charging profile: Rejected
[ocpp ] TRACE 2024/10/21 20:14:53 191221 ready to transmit again
[ocpp ] TRACE 2024/10/21 20:14:53 timeout canceled for 191221
[ocpp ] TRACE 2024/10/21 20:14:55 enqueued CALL [2111269898, TriggerMessage] for 191221
[ocpp ] TRACE 2024/10/21 20:14:55 dispatched request 2111269898 for 191221
[ocpp ] TRACE 2024/10/21 20:14:55 sent JSON message to 191221: [2,"2111269898","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2024/10/21 20:14:55 started timeout timer for 191221
[ocpp ] TRACE 2024/10/21 20:14:55 received JSON message from 191221: [3,"2111269898",{"status": "Accepted"}]
[ocpp ] TRACE 2024/10/21 20:14:55 handling incoming CALL RESULT [2111269898] from 191221
[ocpp ] TRACE 2024/10/21 20:14:55 completed request 2111269898 for 191221
[ocpp ] TRACE 2024/10/21 20:14:55 191221 ready to transmit again
[ocpp ] TRACE 2024/10/21 20:14:55 timeout canceled for 191221
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.13

CSH12, 2024-10-21:
hier noch das logfile in Summe falls ihr mehr braucht als den Ausschnitt
[evcc-20241021-201434-trace.log](https://github.com/user-attachments/files/17465372/evcc-20241021-201434-trace.log)

andig, 2024-10-21:
Probier mal `pulsarplus-fw5`

CSH12, 2024-10-21:
du meinst als template?

mfuchs1984, 2024-10-21:
Es gab bisher zur Pulsar Plus mehrere Fehlerbilder
- mit der Version 5.X gab es den Fehler, den du hier siehst "set charging profile: Rejected" https://github.com/evcc-io/evcc/discussions/15986
- mit der 6.4.19 gab es den Fehler nicht,aber es funktioniert wohl die Ladestromregelung nicht https://github.com/evcc-io/evcc/issues/15724

Das Problem mit der 5.X könnte gelöst werden. Mit dem aktuellen Stand von evcc hilft das template ocpp-abb mit remotestart: true und stacklevelzero" true. In der nightly gibt es nun das template pulsarplus-fw5, das diese Dinge automatisch setzt.

Der Ball für das Problem mit der 6.4.19 liegt beim Hersteller. 

Du hast 6.1.19 und wie es aussieht dasselbe Problem wie die wallboxen mit der 5.X, könntest also entweder das ocpp-abb template nehmen und die settings oben eintragen oder die nightly (oder das nächste Release) mit dem pulsarplus-fw5 ausprobieren.

CSH12, 2024-10-21:
@mfuchs1984: 
Danke für den Hinweis. Ich werde es ausprobieren, sobald mein EVCC wieder mit meinem WR spricht. 

Aktuell habe ich leider immer noch das Problem, dass der Kommunikationsaufbau nicht funktioniert (siehe #16763). Hierfür habe ich leider noch keine Lösung und im Moment ist es egal, ob ich die 0.129 oder die 0.130.13 verwende... will einfach nicht.

CSH12, 2024-10-23:
Ich habe jetzt mit der neuen 0.131 mal ausprobiert wie es läuft und bei dem Standard pulsarplus Template bekomme ich einen timeout. Verbindet sich damit nicht...

Daher habe ich auf das "pulsarplus-fw5" Template umgestellt und damit bekomme ich dann die folgende Warnmeldung 
[lp-1 ] WARN 2024/10/23 19:57:31 charger out of sync: expected disabled, got enabled

Dann hatte ich versucht die beiden Flags noch zu setzen `remotestart: true` bzw. `stacklevelzero: true` (mit beiden und nur remotestart) dann hat es aber direkt fehler gegeben, dass stacklevelzero ein invalid key ist und bei remotestart, dass der key schon verwendet wurde. 

ohne die beiden funktioniert ist, eben nur mit der Warnmeldung. Ob damit schließlich dann das laden funktioniert muss ich noch ausprobieren.

andig, 2024-10-23:
Wie immer (bitte tut uns und Euch den Gefallen) braucht es ein Logfile. Mit "geht" nicht lässt sich exakt _nichts_ anfangen.

mfuchs1984, 2024-10-23:
Am besten gleich ein logfile mit ocpp auf trace.

CSH12, 2024-10-23:
> Am besten gleich ein logfile mit ocpp auf trace.
klar, kein Thema :smile:

[evcc-20241023-202422-trace.log](https://github.com/user-attachments/files/17496686/evcc-20241023-202422-trace.log)
Ich hab es im Browser auf einen Full-Trace eingestellt - hoffe das hilft weiter

mfuchs1984, 2024-10-23:
Trotz metervalues: -Current.Offered,-Power.Offered im template sendet der charger weiter Power.Offered mit 11 kW, auch wenn er aus ist.

{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Offered","unit": "W","value": "11085"} ]}

Kannst du noch ein trace machen, in dem der Verbindungsaufbau mit der wallbox drin ist? Also einmal evcc neu starten und sobald es gestartet ist, den trace ziehen.

CSH12, 2024-10-23:
[evcc-20241023-211053-trace.log](https://github.com/user-attachments/files/17497197/evcc-20241023-211053-trace.log)
ich denke hier sollte die Startphase mit drinsein (sollte wieder ein Fulltrace sein)

CSH12, 2024-10-23:
Vielleicht ist das hier noch besser bzw. mit mehr daten... 
[evcc-20241023-212358-trace.log](https://github.com/user-attachments/files/17497303/evcc-20241023-212358-trace.log)

mfuchs1984, 2024-10-23:
@andig im neuen template ist ja `metervalues: -Current.Offered,-Power.Offered` drin, was dazu führen sollte, dass diese measurands nicht konfiguriert werden.

Dennoch sieht man in den trace logs, dass evcc Power.Offered konfiguriert 
`[ocpp  ] TRACE 2024/10/23 21:10:08 sent JSON message to 191221: [2,"2536277856","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Offered"}]`

`[ocpp  ] TRACE 2024/10/23 21:10:09 sent JSON message to 191221: [2,"188895320","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import,Energy.Active.Import.Register,Current.Import,Voltage,Power.Offered,SoC"}]`

Current.Offered ist dagegen nicht dabei. 

Ist da möglicherweise noch ein Fehler drin, der dazu führt, dass das Entfernen nicht vollständig klappt, wenn mehrere metervalues mit der "-" Notation angegeben sind? https://github.com/evcc-io/evcc/blob/0d18d6703f6fd00766131c05d34950f928c3c833/charger/ocpp/cp_setup.go#L21
Bin leider nicht fit genug in go, um diese Zeile ohne größere Recherche vollständig zu durchdringen.

andig, 2024-10-23:
Es darf nur ein Minus am Anfang geben, nicht je Wert.

mfuchs1984, 2024-10-23:
Bingo, dieses Minimalbeispiel zeigt, dass Power.Offered nicht entfernt wird

https://go.dev/play/p/co9Kf-VwXeK

mfuchs1984, 2024-10-23:
> Es darf nur ein Minus am Anfang geben, nicht je Wert.

Dann ist das template fehlerhaft 
https://github.com/evcc-io/evcc/blob/0d18d6703f6fd00766131c05d34950f928c3c833/templates/definition/charger/ocpp-pulsarplus-fw5.yaml#L26

CSH12, 2024-10-23:
heißt für mich: Fehler gefunden, und mit dem nächsten Bugfix-Release ist dann die Warning behoben, richtig? 
Laden funktioniert aber trotzdem, nehme ich mal an... und die beiden Schalter (remotestart/stacklevelzero) sind erstmal irrelevant.

mfuchs1984, 2024-10-23:
@CSH12 du kannst auch folgendes verwenden 

```
template: ocpp-abb
remotestart: true
metervalues: -Current.Offered,Power.Offered
````

CSH12, 2024-10-23:
Hab’s gerade der Einfachheit halber noch ausprobiert.
Es läd, auch wenn er mal kurz meckert und ne Warning ausspuckt (minpv und now hab ich probiert) 

Wenn es noch irgendwo anders Haken sollte probiere ich die Alternative von oben. 

Vielen Dank euch beiden!

mfuchs1984, 2024-10-24:
Sollte mit der heute erschienenen 0.131.1 gelöst sein.

cdyna43, 2024-10-25:
Habe auf die 0.131.1 aktualisiert.

chargers:
- type: template
  template: pulsarplus
  stationid: xxxxxx
  connector: 1  
  connecttimeout: 5m  
  timeout: 2m  
  name: wallbox4

Führt zu folgendem Fehler:
[main  ] FATAL 2024/10/25 06:55:04 cannot create charger 'wallbox4': cannot create charger type 'template': cannot create charger type 'ocpp': sponsorship required, see https://github.com/evcc-io/evcc#sponsorship 

Mit meiner "alten" funktionierenden Konfiguration unter der 0.130.13:
chargers:
- type: ocpp
  stationid: xxxxxx  
  connector: 1  
  connecttimeout: 5m  
  timeout: 2m  
  name: wallbox4
  getconfiguration: false
  #autostart: true
  stacklevelzero: true
  remotestart: true

kommt unter der 0.131.1 derselbe Fehler.

Brauche ich unter der 0.131.1 zwingend eine Sponsorship?

mfuchs1984, 2024-10-25:
1. Du musst pulsarplus-fw5 nehmen.
2. Ja, die Pulsar braucht sponsorship https://docs.evcc.io/docs/devices/chargers#pulsar-plus-pulsar-max-commander-2-copper-sb-fw-5x

andig, 2024-10-25:
Siehe Blog- das gilt für alle Ocpp Charger nachdem wir die Integration von grundauf neu entwickelt haben.

cdyna43, 2024-10-25:
Ah, vielen Dank. Dann schaue ich mir mal das Sponsoring an.

Frintrop, 2024-10-28:
Ich habe zwei Commander 2 Wallboxen und habe bislang das pulsarplus template verwendet.
Firmwareversion bei beiden ist 6.1.15.Sponsoring vorhanden.

Trotzdem erhalte ich

[lp-1  ] WARN 2024/10/28 16:35:36 charger logic error: current mismatch (got 32A, expected 0A)
[lp-1  ] ERROR 2024/10/28 16:35:37 charger disable: set charging profile: Rejected

Wodran kann das liegen ?

chargers:
 - name: WallboxLinks
   type: template
   template: pulsarplus
   #type: ocpp
   #timeout: 1m 
   stationid: 102249
   remotestart: true
   #stacklevelzero: true
   
 - name: WallboxRechts
   type: template
   template: pulsarplus
   #type: ocpp
   #timeout: 1m 
   stationid: 65065
   remotestart: true
   #stacklevelzero: true

mfuchs1984, 2024-10-28:
Probier's Testweise mal mit pulsarplus-fw5 (remotestart: true muss dann raus, weil schon im template).

Frintrop, 2024-10-29:
Ok. Das funktioniert ... vielleicht in der Doku anpassen für andere Commander 2 Benutzer ?!

mfuchs1984, 2024-10-29:
@andig template für commander 2 identisch zu pulsarplus-fw5 erstellen?

andig, 2024-10-29:
Ahh, das Ding ist gar nicht von Pulsar? Dann sollte es reichen, das Produkt im Pulsar Template zu ergänzen. Das wäre super nett, danke!

mfuchs1984, 2024-10-29:
Es ist von pulsar, aber nicht die pulsar plus, sondern die pulsar commander 2.

andig, 2024-10-29:
Für mich ist leider nicht nachvollziehbar, welche Probleme Pulsar da in welchen Produkten und FW-Versionen hat. Whatever helps...

premultiply, 2024-10-29:
Die Firma heisst "Wallbox" 🙂

mfuchs1984, 2024-10-29:
Ja, Wallbox ist richtig, Wallbox from hell wäre aber ein besserer Name, wenn man das ocpp Chaos sieht.

typxxi, 2024-10-30:
> Ahh, das Ding ist gar nicht von Pulsar? Dann sollte es reichen, das Produkt im Pulsar Template zu ergänzen. Das wäre super nett, danke!

Keine Ahnung, aber meiner Meinung liegt da ein Mißverständnis vor.

"Das Ding ist gar nicht von Pulsar" kann nicht stimmen, denn die Pulsar und Pulsar Plus sind Produkte der Herstellerfirma WALLBOX aus Barcelona, der wir alle die Nutzung des Wallbox Begriffes wohl zu verdanken haben. 
Die Copper ist ein Schwester Produkt und so weiter, alles aus dem Hause Wallbox, aber wie gleich die Templates sind, das weiß ich nicht, nur ist die Copper sicher nicht vom Hersteller Pulsar. 

Von daher andere fragen, wie gleich die sind oder nicht, aber sicher ist , dass der Hersteller Wallbox heißt.

mfuchs1984, 2024-11-04:
@Frintrop die Commander 2 hat einen RFID Reader, richtig?

PR #16817 (2024-10-24): Pulsar Plus (v5.x): fix template
Syntax for measurands removal was wrong, so Power.Offered was not removed as intended. Fixes #16768