evcc-io/evcc #16777: Qcells: cannot disable

andig, 2024-10-22:
Du meinst:

```
(nested2) hendrikn@MacBook-Pro-von-Hendrik log % evcc charger --log trace,db:error --disable
[main ] INFO 2024/10/20 16:58:32 evcc 0.130.13
[main ] INFO 2024/10/20 16:58:32 using config file: /etc/evcc.yaml
[solax ] TRACE 2024/10/20 16:58:33 modbus: send 00 01 00 00 00 06 46 06 06 27 00 03
[solax ] TRACE 2024/10/20 16:58:33 modbus: recv 00 01 00 00 00 06 46 06 06 27 00 03
(nested2) hendrikn@MacBook-Pro-von-Hendrik log % evcc charger --log trace

[main ] INFO 2024/10/20 16:58:35 evcc 0.130.13
[main ] INFO 2024/10/20 16:58:35 using config file: /etc/evcc.yaml
[solax ] TRACE 2024/10/20 16:58:35 modbus: send 00 01 00 00 00 06 46 04 00 0b 00 01
[solax ] TRACE 2024/10/20 16:58:35 modbus: recv 00 01 00 00 00 05 46 04 02 00 0e
[solax ] TRACE 2024/10/20 16:58:35 modbus: send 00 02 00 00 00 06 46 04 00 10 00 02
[solax ] TRACE 2024/10/20 16:58:35 modbus: recv 00 02 00 00 00 07 46 04 04 00 00 09 a5
[solax ] TRACE 2024/10/20 16:58:35 modbus: send 00 03 00 00 00 06 46 04 00 04 00 03
[solax ] TRACE 2024/10/20 16:58:35 modbus: recv 00 03 00 00 00 09 46 04 06 00 00 00 00 00 26
[solax ] TRACE 2024/10/20 16:58:35 modbus: send 00 04 00 00 00 06 46 04 00 00 00 03
[solax ] TRACE 2024/10/20 16:58:35 modbus: recv 00 04 00 00 00 09 46 04 06 5c 8a 5b fd 5c a2
[solax ] TRACE 2024/10/20 16:58:35 modbus: send 00 05 00 00 00 06 46 04 00 1d 00 01
[solax ] TRACE 2024/10/20 16:58:35 modbus: recv 00 05 00 00 00 05 46 04 02 00 02
[solax ] TRACE 2024/10/20 16:58:35 modbus: send 00 06 00 00 00 06 46 03 06 0d 00 01
[solax ] TRACE 2024/10/20 16:58:35 modbus: recv 00 06 00 00 00 05 46 03 02 00 01
Power: 14W
Energy: 246.9kWh
Current L1..L3: 0A 0A 0.38A
Voltage L1..L3: 237V 235V 237V
Charge status: C
Enabled: true
```

@premultiply das Schreiben von 

	solaxRegCommandControl  = 0x0627 // uint16

Scheint nicht zu funktionieren.

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/16660#discussioncomment-11013879_

andig, 2024-10-29:
Ping @premultiply

premultiply, 2024-11-01:
https://github.com/evcc-io/evcc/pull/13202#issuecomment-2045621340

Welche Firmware wird hier verwendet?
Offensichtlich funktioniert es erst ab Firmware 1.13 mit dem Start des Ladevorgangs zuverlässig.

hnehnes, 2024-11-01:
1.13 wurde durch 1.12 ersetzt

andig, 2024-11-01:
Closing in favor of https://github.com/evcc-io/evcc/discussions/16660

closed 2024-11-01 by andig