evcc-io/evcc #16780: Audi integration not working anymore after MyAudi API changes

tw-mloschwitz, 2024-10-22:
### Describe the bug

Integrating EVCC with the myAudi API was working fine until a few weeks ago, when Audi introduced large-scale API changes. Since then, connecting to myAudi does not work anymore, i.e. EVCC cannot properly read out car parameters or any car details. This has haunted other pieces of software such as Tibber, too.

Please let me know if there is anything I can do to help debug this or to provide additional information. Thank you.

### Steps to reproduce

1. Fire up EVCC
2. See a "Garage: charger status: InternalError:" error message and no details on the myAudi cars

### Configuration details

```yaml
vehicles:
- type: template
  template: audi
  title: e-tron S Sportback
  user: secret
  password: topsecret
  vin: alsotopsecret
  capacity: 95
  name: ev1
```

### Log details

```text
No Audi specific error messages in the log, just the desired detail is not shown anymore.
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.130.13

andig, 2024-10-22:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-10-22 by andig

tw-mloschwitz, 2024-10-22:
What is the correct way to enable any debugging for the Audi template? Tried 

log: debug
levels:
  site: debug
  lp-1: debug
  cache: error
  db: error
  ocpp: trace
  audi: trace
  
but that is apparently not it. I am not seeing *any* debugging output from any API calls to myAudi, neither in the CLI nor in the web UI.

andig, 2024-10-22:
Does https://github.com/TA2k/ioBroker.vw-connect work for you?

tw-mloschwitz, 2024-10-22:
Yes: 
[image]

tw-mloschwitz, 2024-10-22:
I guess I'm a bit of an Idiot -- after changing the template from audi to etron, evcc started working again and connecting properly to myAudi. I wonder why it was working before, but i'm quite happy with it now, so thanks again and sorry for the noise.