evcc-io/evcc #16783: eyond charging tariff

waterfall0815, 2024-10-22:
Eyond offers a tariff for charging of electric cars. Normally there is a communication between the eyond backend and your car required. But eyond also offers interfaces to certain wallboxes/chargers. 
When charging your car according to the recommendations of eyon ("netzdienlich") you get so called FlexPoints that effectively reduce the price you pay per kWh. Unfortunatelly neither my car nor my charger is supported.

Is it possible to implement an interface from evcc to the eyond backend to use the time recommendations for grid charging? Could maybe handled like API for dynamic tariffs.

andig, 2024-10-22:
Not sure I follow. What‘s the benefit?

waterfall0815, 2024-10-22:
ok, it was just a thought. If the car is not compatible, it should in this case also work, if you have a certain wallbox.

When letting the app directly control the charging plans of your car there is no additional control possible. The app can also control in PV-mode but does not use actual measurement data but only weather data (inaccurate).

For my opinion this tariff has some advantages as it is a fixed tariff at all (so there are no price surprises when you need current for your heatpump or other home appliances), but when you charge your car from grid in times it is recommended by eyond you get a discount compared to the fix price you pay per kWh for the current you use to charge your car.

Hopefully my explanation is understandable

andig, 2024-10-22:
I still fail to see why this should be added.