evcc-io/evcc #16786: Increase Charging Plan Slots to more than 42 hours if more data is available

moritzmair, 2024-10-23:
**Is your feature request related to a problem? Please describe.**
I'm developing a Energy Forecast service where there is price data available for 48 hours or even 96 hours (paid). For now it is not possible to see all those forecasted hours. Therefore I'm also not sure if this information is used for planned charging.

**Describe the solution you'd like**
If there is more data available then 42 hours in the future I would like to see and use of all it.

**Additional context**
for more context see my little introduction to the tool: https://github.com/evcc-io/evcc/discussions/14211

andig, 2024-10-23:
Fixed in master

closed 2024-10-23 by andig

andig, 2024-10-23:
Sorry, I had to revert https://github.com/evcc-io/evcc/commit/fc377138d1debfea0d4bfc749911f1be580bf73b, it was a wrong fix. That said: I do not see any limitation to 48 slots in the code.

moritzmair, 2024-10-23:
In the frontend only 42 hours are shown. I guess that 48 hours are used. It would be awesome if 96 of forecast or more could be used if available (and bonus: shown in the frontend).