evcc-io/evcc #16794: Failed to start with 0.131

softservicegit, 2024-10-23:
### Describe the bug

starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/10/23 14:15:54 evcc 0.131.0
[main  ] INFO 2024/10/23 14:15:54 using config file: /config/evcc.yaml
[main  ] FATAL 2024/10/23 14:16:11 cannot create meter 'carport_pv_slave': cannot create meter type 'template': cannot create meter type 'custom': currents: [0] sunspec model not found: [{103 0 AphA}]
[main  ] FATAL 2024/10/23 14:16:11 will attempt restart in: 15m0s

### Steps to reproduce

1. INstall update. 
2.
3.
...

### Configuration details

```yaml
meters:
  - name: solaredge_grid
    type: template
    template: solaredge-inverter
    usage: grid
    modbus: tcpip
    id: 1
    host: 192.168.10.6 # Hostname
    port: 526
    # host: 192.168.10.66 # Hostname
    # port: 5206
    timeout: 3s

  - name: house_battery_master
    type: template
    template: solaredge-hybrid
    usage: battery
    modbus: tcpip
    id: 1
    host: 192.168.10.6 # Hostname
    port: 526
    # host: 192.168.10.66 # Hostname
    # port: 5206
    timeout: 3s

  - name: house_pv_master
    type: template
    template: solaredge-hybrid
    usage: pv
    modbus: tcpip
    id: 1
    host: 192.168.10.6 # Hostname
    port: 526
    # host: 192.168.10.66 # Hostname
    # port: 5206
    timeout: 3s

  - name: house_pv_slave
    type: template
    template: solaredge-hybrid
    usage: pv
    modbus: tcpip
    id: 3
    host: 192.168.10.6 # Hostname
    port: 527
    # host: 192.168.10.66 # Hostname
    # port: 5207
    timeout: 3s

  - name: pool_pv_slave
    type: template
    template: solaredge-inverter
[... 204 lines cut ...]
        mode: always
        interval: 5m
```

### Log details

```text
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/10/23 14:15:54 evcc 0.131.0
[main  ] INFO 2024/10/23 14:15:54 using config file: /config/evcc.yaml
[main  ] FATAL 2024/10/23 14:16:11 cannot create meter 'carport_pv_slave': cannot create meter type 'template': cannot create meter type 'custom': currents: [0] sunspec model not found: [{103 0 AphA}]
[main  ] FATAL 2024/10/23 14:16:11 will attempt restart in: 15m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131

andig, 2024-10-23:
Could you please show

    evcc sunspec

For your SolarEdge?

typxxi, 2024-10-23:
I had also issues that it evcc was not booting but no logs due to the HA addon which only offers a short log and said it was waiting which then was repeated every 2 minutes or so.

I decided to stop and went back to 1.30.13 due to the issue.

softservicegit, 2024-10-23:
Sorry, took me a while to learn how to come into the docker container :)

/app # evcc sunspec 192.168.10.6:528 -i 5
[main  ] INFO 2024/10/23 14:53:52 evcc 0.130.13
--------- Model 1 common ---------
Mn         SolarEdge     string
Md   SE4000H-RW000BEN    string
Opt                      string
Vr           0004.002    string
SN           75032C20    string
DA               5.00    uint16
Pad            0x0065    pad
--------- Model 101 inverter ---------
A                    0.47    uint16
AphA                 0.47    uint16
AphB                  n/a    uint16
AphC                  n/a    uint16
A_SF                   -2    sunssf
PPVphAB            236.70    uint16
PPVphBC               n/a    uint16
PPVphCA               n/a    uint16
PhVphA             236.70    uint16
PhVphB                n/a    uint16
PhVphC                n/a    uint16
V_SF                   -1    sunssf
W                   68.15    int16
W_SF                   -2    sunssf
Hz                  49.98    uint16
Hz_SF                  -3    sunssf
VA                 112.45    int16
VA_SF                  -2    sunssf
VAr                 89.45    int16
VAr_SF                 -2    sunssf
PF                  60.60    int16
PF_SF                  -2    sunssf
WH             3039209.00    acc32
WH_SF                   0    sunssf
DCA                  0.19    uint16
DCA_SF                 -5    sunssf
DCV                369.70    uint16
DCV_SF                 -1    sunssf
DCW                 69.19    int16
DCW_SF                 -2    sunssf
TmpCab                n/a    int16
TmpSnk              31.97    int16
TmpTrns               n/a    int16
TmpOt                 n/a    int16
Tmp_SF                 -2    sunssf
St                      4    enum16
StVnd                   0    enum16
Evt1                  n/a    bitfield32
Evt2                  n/a    bitfield32
EvtVnd1        0x00000000    bitfield32
EvtVnd2               n/a    bitfield32
EvtVnd3               n/a    bitfield32
EvtVnd4        0x00000000    bitfield32

typxxi, 2024-10-23:
I have tried it again under HA Addon 

and I got this log

"Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/10/23 15:07:38 evcc 0.131.0
[main  ] INFO 2024/10/23 15:07:38 using config file: /config/evcc.yaml
[mqtt  ] INFO 2024/10/23 15:07:39 connecting evcc-2126174612 at tcp://192.168.178.2:1883
[mqtt  ] DEBUG 2024/10/23 15:07:39 tcp://192.168.178.2:1883 connected
[main  ] INFO 2024/10/23 15:07:39 listening at :7070
[Garage-1] DEBUG 2024/10/23 15:07:39 waiting for chargepoint: 5m0s
[ocpp  ] DEBUG 2024/10/23 15:08:37 charge point connected: Garage
[ocpp  ] TRACE 2024/10/23 15:08:37 enqueued CALL [1093396971, ChangeAvailability] for Garage
[ocpp  ] TRACE 2024/10/23 15:08:37 dispatched request 1093396971 for Garage
[ocpp  ] TRACE 2024/10/23 15:08:37 sent JSON message to Garage: [2,"1093396971","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
[ocpp  ] TRACE 2024/10/23 15:08:37 started timeout timer for Garage
[ocpp  ] TRACE 2024/10/23 15:09:07 timeout for client Garage, canceling message
[ocpp  ] TRACE 2024/10/23 15:09:07 completed request 1093396971 for Garage
[ocpp  ] TRACE 2024/10/23 15:09:07 request 1093396971 for Garage timed out
[ocpp  ] TRACE 2024/10/23 15:09:07 Garage ready to transmit again
[Garage-1] DEBUG 2024/10/23 15:09:07 failed configuring availability: timeout
[ocpp  ] TRACE 2024/10/23 15:09:07 enqueued CALL [1613254401, GetConfiguration] for Garage
[ocpp  ] TRACE 2024/10/23 15:09:07 dispatched request 1613254401 for Garage
[ocpp  ] TRACE 2024/10/23 15:09:07 sent JSON message to Garage: [2,"1613254401","GetConfiguration",{}]
[ocpp  ] TRACE 2024/10/23 15:09:07 started timeout timer for Garage
[ocpp  ] TRACE 2024/10/23 15:09:37 timeout for client Garage, canceling message
[ocpp  ] TRACE 2024/10/23 15:09:37 completed request 1613254401 for Garage
[ocpp  ] TRACE 2024/10/23 15:09:37 request 1613254401 for Garage timed out
[ocpp  ] TRACE 2024/10/23 15:09:37 Garage ready to transmit again
[main  ] FATAL 2024/10/23 15:09:37 cannot create charger 'wallbox4': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
[main  ] FATAL 2024/10/23 15:09:37 will attempt restart in: 15m0s

...

[ocpp  ] TRACE 2024/10/23 15:12:00 received JSON message from Garage: [2,"1623527199","MeterValues",{"meterValue": [{"timestamp": "2024-10-23T13:11:59Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "702201.2"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1-N","measurand": "Voltage","unit": "V","value": "226.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2-N","measurand": "Voltage","unit": "V","value": "234.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3-N","measurand": "Voltage","unit": "V","value": "233.0"} ]} ],"connectorId": 1}]
[ocpp  ] TRACE 2024/10/23 15:12:00 handling incoming CALL [1623527199, MeterValues] from Garage
[ocpp  ] TRACE 2024/10/23 15:12:00 sent CALL RESULT [1623527199] for Garage
[ocpp  ] TRACE 2024/10/23 15:12:00 sent JSON message to Garage: [3,"1623527199",{}]
[ocpp  ] TRACE 2024/10/23 15:12:10 enqueued CALL [2379615324, TriggerMessage] for Garage
[ocpp  ] TRACE 2024/10/23 15:12:10 dispatched request 2379615324 for Garage
[ocpp  ] TRACE 2024/10/23 15:12:10 sent JSON message to Garage: [2,"2379615324","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/10/23 15:12:10 started timeout timer for Garage
[ocpp  ] TRACE 2024/10/23 15:12:10 received JSON message from Garage: [3,"2379615324",{"status": "Accepted"}]
[ocpp  ] TRACE 2024/10/23 15:12:10 handling incoming CALL RESULT [2379615324] from Garage
[ocpp  ] TRACE 2024/10/23 15:12:10 completed request 2379615324 for Garage
[ocpp  ] TRACE 2024/10/23 15:12:10 Garage ready to transmit again
[ocpp  ] TRACE 2024/10/23 15:12:10 timeout canceled for Garage
[ocpp  ] TRACE 2024/10/23 15:12:10 received JSON message from Garage: [2,"115155578","MeterValues",{"meterValue": [{"timestamp": "2024-10-23T13:12:09Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "702201.2"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1-N","measurand": "Voltage","unit": "V","value": "226.0"},{"format": "Raw","location":  [... cut ...]
But it looks like that it successfully reached my inverter and stopped that from charging the battery cause the AC charging is no longer running. I had tried that home storage charging based on price < 29 cent but it did not work. So i went to my inverter and set up a schedule to charge now from AC but this has been stopped now. 

Here you can see the red area which is energy from the grid used to charge the battery with 3 kW max and AC power from the grid if there is no surplus.

Then there was break cause by the installation and start of evcc 1.31.0 and then i was reable to restart the AC charging via Growatt Cloud. 
[image]

typxxi, 2024-10-23:
and now it has turned red again within 5 minutes
[ocpp  ] TRACE 2024/10/23 15:23:51 handling incoming CALL [6283719, StatusNotification] from Garage
[ocpp  ] TRACE 2024/10/23 15:23:51 sent CALL RESULT [6283719] for Garage
[ocpp  ] TRACE 2024/10/23 15:23:51 sent JSON message to Garage: [3,"6283719",{}]
[main  ] FATAL 2024/10/23 15:23:51 cannot create charger 'wallbox4': cannot create charger type 'template': cannot create charger type 'ocpp': sponsorship required, see https://github.com/evcc-io/evcc#sponsorship
[main  ] FATAL 2024/10/23 15:23:51 will attempt restart in: 15m0s
[ocpp  ] TRACE 2024/10/23 15:23:56 enqueued CALL [1913146757, TriggerMessage] for Garage
[ocpp  ] TRACE 2024/10/23 15:23:56 dispatched request 1913146757 for Garage
[ocpp  ] TRACE 2024/10/23 15:23:56 sent JSON message to Garage: [2,"1913146757","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/10/23 15:23:56 started timeout timer for Garage
[ocpp  ] TRACE 2024/10/23 15:23:56 received JSON message from Garage: [3,"1913146757",{"status": "Accepted"}]
[ocpp  ] TRACE 2024/10/23 15:23:56 handling incoming CALL RESULT [1913146757] from Garage
[ocpp  ] TRACE 2024/10/23 15:23:56 completed request 1913146757 for Garage
[ocpp  ] TRACE 2024/10/23 15:23:56 Garage ready to transmit again
[ocpp  ] TRACE 2024/10/23 15:23:56 timeout canceled for Garage
[ocpp  ] TRACE 2024/10/23 15:23:56 received JSON message from Garage: [2,"750216102","MeterValues",{"meterValue": [{"timestamp": "2024-10-23T13:23:56Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "702201.2"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1-N","measurand": "Voltage","unit": "V","value": "227.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2-N","measurand": "Voltage","unit": "V","value": "234.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3-N","measurand": "Voltage","unit": "V","value": "232.0"} ]} ],"connectorId": 1}]
[ocpp  ] TRACE 2024/10/23 15:23:56 handling incoming CALL [750216102, MeterValues] from Garage
[ocpp  ] TRACE 2024/10/23 15:23:56 sent CALL RESULT [750216102] for Garage
[ocpp  ] TRACE 2024/10/23 15:23:56 sent JSON message to Garage: [3,"750216102",{}]
[ocpp  ] TRACE 2024/10/23 15:24:06 enqueued CALL [3703372465, TriggerMessage] for Garage
[ocpp  ] TRACE 2024/10/23 15:24:06 dispatched request 3703372465 for Garage
[ocpp  ] TRACE 2024/10/23 15:24:06 sent JSON message to Garage: [2,"3703372465","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/10/23 15:24:06 started timeout timer for Garage
[ocpp  ] TRACE 2024/10/23 15:24:06 received JSON message from Garage: [3,"3703372465",{"status": "Accepted"}]
[ocpp  ] TRACE 2024/10/23 15:24:06 handling incoming CALL RESULT [3703372465] from Garage
[ocpp  ] TRACE 2024/10/23 15:24:06 completed request 3703372465 for Garage
[ocpp  ] TRACE 2024/10/23 15:24:06 Garage ready to transmit again
[ocpp  ] TRACE 2024/10/23 15:24:06 timeout canceled for Garage
[ocpp  ] TRACE 2024/10/23 15:24:06 received JSON message from Garage: [2,"1472806988","MeterValues",{"meterValue": [{"timestamp": "2024-10-23T13:24:06Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "702201.2"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1-N","measurand": "Voltage","unit": "V","value": "227.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2-N","measurand": "Voltage","unit": "V","value": "233.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3-N","measurand": "Voltage","unit": "V","value": "232.0"} ]} ],"connectorId": 1}]
[ocpp  ] TRACE 2024/10/23 15:24:06 handling incoming CALL [1472806988, MeterValues] from Garage
[ocpp  ] TRACE 2024/10/23 15:24:06 sent CALL RESULT [1472806988] for Garage
[ocpp  ] TRACE 2024/10/23 15:24:06 sent JSON message to Garage: [3,"1472806988",{}]
[ocpp  ] TRACE 2024/10/23 15:24:16 enqueued CALL [521495271, TriggerMessage] for Garage
[ocpp  ] TRACE 2024/10/23 15:24:16 dispatched request 521495271 for Garage
[ocpp  ] TRACE 2024/10/23 15:24:16 sent JSON message to Garage: [2,"521495271","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[... cut ...] tlet","context": "Trigger","phase": "L2-N","measurand": "Voltage","unit": "V","value": "234.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3-N","measurand": "Voltage","unit": "V","value": "232.0"} ]} ],"connectorId": 1}]
[ocpp  ] TRACE 2024/10/23 15:24:56 handling incoming CALL [965163551, MeterValues] from Garage
[ocpp  ] TRACE 2024/10/23 15:24:56 sent CALL RESULT [965163551] for Garage
[ocpp  ] TRACE 2024/10/23 15:24:56 sent JSON message to Garage: [3,"965163551",{}]

andig, 2024-10-23:
@typxxi can you please immediately stop spamming this issue?

andig, 2024-10-23:
/cc @ivoks this is caused by https://github.com/evcc-io/evcc/pull/16675 not taking 1p inverters into account.

closed 2024-10-23 by andig

Dacia-Spring, 2024-10-23:
Hallo, die geposteten Zugangsdaten für die VW-Autos.
Sollten die nicht mal wegzensiert werden?
Ich konnte damit mich problemlos einloggen….

andig, 2024-10-23:
@softservicegit unbedingt Passwort ändern!!! In Zukunft am Besten

    evcc discuss

benutzen.

ivoks, 2024-10-24:
> /cc @ivoks this is caused by #16675 not taking 1p inverters into account.

Indeed. Hm, sorry for the trouble @softservicegit. I'll see if I can come up with a better approach.