evcc-io/evcc #16795: home storage charging below 29 cent is not working

typxxi, 2024-10-23:
### Describe the bug

my current tarif is below 29 cent and evcc should now charge the home storage battery but it is not charging.
screenshot shows the green = valid charging time. 
Growatt SPH 4600 inverter with JK BMS and 14 kWh battery

[image]

latest screenshot 
[image]

### Steps to reproduce

1. during a "cheap" hour I opened the home battery configuration
2.then I chose a price of 29 cent
3.and it did not start charging even though no car was connected or charging
...
LOG

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug      
  # alternativ
  # log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error
  ocpp: trace

# unique installation id
plant: b4e6c38871bb59bd6f892b4fa72af41423514af3f6177bba1d789184ab602bf3

##########################################
# control cycle interval                 #
# kann kritisch sein, wenn < 30 Sekunden #
##########################################

interval: 12s   # meterinterval: 12s unter chargers sollte mit synchron gehalten werden
                # wenn das zu kurz ist, dann gibt es den folgenden Fehler, von daher das Intervall verlängern
                # Garage: charger logic error: current mismatch (got 11.2A measured, expected 8.73A) 
                # https://github.com/evcc-io/evcc/discussions/15761#discussioncomment-10893056

meters:
  - type: template
    template: growatt-hybrid
    id: 1
    device: /dev/ttyUSB0
    baudrate: 9600
    comset: 8N1
    usage: grid
    modbus: rs485serial
    name: grid1

  - type: template
    template: growatt-hybrid
    id: 1
    device: /dev/ttyUSB0
    baudrate: 9600
    comset: 8N1
    usage: pv
    modbus: rs485serial
    name: pv2

  - type: template
    template: growatt-hybrid
    id: 1
    device: /dev/ttyUSB0
    baudrate: 9600
    comset: 8N1
    usage: battery
    modbus: rs485serial
    name: battery3
    capacity: 14.0

# chargers:
#   - type: template
#     template: pulsarplus
#     timeout: 2m
#     name: wallbox4

###   LÖSUNG   von   hier   ###
###   https://github.com/evcc-io/evcc/issues/15581#issuecomment-2337136867

chargers:
#   5 ZEILEN LÖSUNG für 1.30.13 von https://github.com/evcc-io/evcc/discussions/15986#discussioncomment-10674568
  - type: template
    template: ocpp-abb
    remotestart: true     ###   >1.30.x   ###   NEU FOLGE von https://github.com/evcc-io/evcc/issues/15581 
    meterinterval: 12s    ###   from 10s to 12s 09.10.2024 >1.30.1x = optional
                          ###   ganz oben noch als generelles interval: 12s 
    metervalues: Energy.Active.Import.Register,Power.Active.Import,SoC,Current.Import,Voltage  # >1.30.1x

    # metervalues: Energy.Active.Import.Register,Power.Active.Import,SoC,Power.Offered,Current.Import,Voltage  # >1.30.1x

    stationid: Garage ### NEU = OCCP ID
    connector: 1 ### soll alt sein
    timeout: 2m
    name: wallbox4

vehicles:
  - name: eGOLF
    title: GOLF
    type: custom
    ###       type: template
    ###       template: offline
    icon: car
    capacity: 31.6 # net capacity in kWh

    phases: 2

    soc: # optional SoC (%)
      source: mqtt
      topic: export/golf/soc_permanent
    range: # optional electric range (km)
      source: mqtt
      topic: export/golf/reichweite_permanent
    odometer: # optional odometer (km)
      source: mqtt
      topic: export/golf/km_stand_permanent

loadpoints:
  - title: Garage
    charger: wallbox4
    mode: pv # charge mode (off, now, minpv, pv)
    vehicle: eGOLF # set default vehicle (disables vehicle detection)
    ###   NEU: SCHWELLEN   ###
    enable:
      delay: 1m
      threshold: -1500 # laden, sobald 1 Minute lang min 1.500 W eingespeist worden sind
    disable:
      delay: 2m
      threshold: 1000 # laden beenden, wenn 2 Minuten lang mehr als 1.000 W Netzstrom bezogen wurden

site:
  title: Zuhaus
  meters:
    grid: grid1
    #
    # für saubere Überschussdarstellung bei mehreren Wechselrichtern
    # pv Zähler deaktivieren
    # https://github.com/evcc-io/evcc/discussions/16574#discussioncomment-10892601
    # 
    pv: pv2  
    battery:
      - battery3
  residualPower: 111

tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: tibber
    token: RcnWXBLfW5ZLf2yuHShZ9X3hubZgvIR7azaD7SUt084
    # type: fixed
    # price: 0.272 # 0.272 [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.0774 #  seit 19.8.2024

# mqtt message broker
mqtt:
  broker: 192.168.178.2:1883 # MQTT-Broker & Port
  topic: evcc # alles wird im MQTT Explorer unter TOPIC EVCC erscheinen
  user: mymqtt
  password: mymqttpi

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: Soc updated
      msg: Battery charged to ${vehicleSoc:%.0f}%
    guest: # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
```

### Log details

```text
full log attached above
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

Current version: 0.130.13

andig, 2024-10-23:
Please reproduce using

    evcc meter --battery-mode charge

Closing until then.

closed 2024-10-23 by andig

typxxi, 2024-10-23:
I did that in the HA terminal and I have gotten this

[image]

What else can I do ?