evcc-io/evcc #16798: FR: aktuelle Co2 Werte beim Laden inkludieren

SolarPower2024, 2024-10-23:
**Is your feature request related to a problem? Please describe.**
Ich möchte gerne die Co2 Einsparung tracken und dafür müssen die Co2 Werte in den Ladevorgängen verfügbar sein. Es geht rein um das Tracking und Darstellung in der Übersicht, nicht um das Laden per Planner über Co2. 

Für Personen in Deutschland ist das simple, weil man einfach den Grünstromindex verwenden kann. Für Leute aus anderen Ländern ist es nicht möglich, außer mit electricitymap, wobei die Integration hier nur in der teuren Bezahlvariante möglich ist. 

**Describe the solution you'd like**
Ich würde mich freuen wenn man weitere Möglichkeiten einbinden würde, die aktuellen Co2 Werte zu tracken. 
Folgende Möglichkeiten als Beispiel:
1. Den FREE-Tier von electricitymap möglich zu machen (live Daten sind hier ja zugänglich) 
2. Die API von Nowtricity wäre eine Möglichkeit mit sehr vielen Ländern zur Auswahl https://www.nowtricity.com/emissions-api/ 
3. Eine Möglichkeit zu schaffen den Wert per MQTT an evcc bereitzustellen. 

Option 1 oder 2 wären für die Anzahl an Usern wohl die bessere Wahl, persönlich wäre für mich Option 3 auch kein Problem. 

**Describe alternatives you've considered**
In der Diskussion https://github.com/evcc-io/evcc/discussions/11949#discussioncomment-10918732 habe ich dank Volker bereits per MQTT eine Notlösung für mich selber gefunden. Allerdings hatte dies den Nachteil, dass die Preise im Planer falsch angezeigt wurden. (in den Screenshots der Diskussion ersichtlich) Und weiters funktioniert diese Notlösung in der aktuellen 0.131 Version nicht mehr. Hier bekomme ich eine Warnmeldung, dass der Planer Tarif nicht erstellt werden kann. 

Ich wäre sehr froh, wenn es hier eine weitere Möglichkeit des Co2 Tracking geben könnte, speziell jetzt mit den neuen Grafiken in der Ladeübersicht möchte ich die Co2 Statistik auch gerne sehen. 

Vielen Dank vorab für eine Prüfung meines Anliegen.

andig, 2024-10-23:
> Den FREE-Tier von electricitymap möglich zu machen (live Daten sind hier ja zugänglich)

Was fehlt dafür?

> Eine Möglichkeit zu schaffen den Wert per MQTT an evcc bereitzustellen.

Das kannst Du jetzt schon.

closed 2024-10-23 by andig

SolarPower2024, 2024-10-23:
> > Den FREE-Tier von electricitymap möglich zu machen (live Daten sind hier ja zugänglich)
> 
> Was fehlt dafür?
> 
Es kommt die Fehlermeldung, dass der Tarif nicht erstellt werden kann. Tarif wurde so eingetragen wie in der Doku beschrieben, nur mit Free-Tier Token.
```
co2:
  type: template
  template: electricitymaps
  uri: https://api-access.electricitymaps.com/2w...1g/ # HTTP(S) Adresse
  token: BcVeUbyoypOyX
  zone: AT # siehe https://api.electricitymap.org/v3/zones
```
Fehlermeldung:
[site ] WARN 2024/10/23 16:05:17 planner: tariff not available: cannot create tariff type 'template': cannot create tariff type 'electricitymaps': unexpected status: 500 (Internal Server Error)

> > Eine Möglichkeit zu schaffen den Wert per MQTT an evcc bereitzustellen.
> 
> Das kannst Du jetzt schon.
> 
Leider nein, hatte ich ja die letzten 2 Wochen wie verlinkt. Aber das funktioniert in der aktuellen Version nicht mehr und es ist auch mehr eine Notlösung, weil dadurch die Preise falsch angezeigt werden.

```
co2:
  type: custom
  price:
    source: mqtt
    topic: HA/EVCC/co2
```
MQTT Wert ist übertragen:
[mqtt ] TRACE 2024/10/23 15:53:29 recv HA/EVCC/co2: '119.0'

[image]

Außer es gibt hier eine andere Möglichkeit den aktuellen Co2 Wert zu senden und ich mache hier einen Fehler?

> > Die API von Nowtricity wäre eine Möglichkeit mit sehr vielen Ländern zur Auswahl
> 
> Gerne PR.

Bin nur ein Anwender, ich kann leider keinen PR erstellen bzw. nicht Programmieren.

andig, 2024-10-23:
> Leider nein, hatte ich ja die letzten 2 Wochen wie verlinkt. Aber das funktioniert in der aktuellen Version nicht mehr und es ist auch mehr eine Notlösung, weil dadurch die Preise falsch angezeigt werden.

Das sollte gehen. Heute nachmittag war da ein Fehler von heute drin der auch grad behoben wurde.

SolarPower2024, 2024-10-23:
Alles klar, vielen Dank für die schnelle Rückmeldung.