evcc-io/evcc #16800: 3. Register beim Hausspeicher für "Entladepausen", um preisbasiert den AC Strom zu schalten und so den Akku zu schonen

typxxi, 2024-10-23:
Im Moment gibt es 
1. Nutzung der Batterie seitens evcc 
2. Netzladen der Batterie (AC laden)

[image]

Wenn nun mehr und mehr Funktionen hier rein wandern, dann fehlt mir so etwas wie eine AC Preis basierte "Entlade Pause", wo ich den Strom nicht aus dem Akku erzeuge, sondern einfach den Netzstrom nutze, um am Morgen in den teuren Tibber Stunden das Haus noch mit der so (auf)gesparten Energie aus dem Akku versorgen zu können. 

Würde ich den Strom hingegen selbst in den günstigen Stunden laden, um noch durch die Morgenstunden abzudecken, würden sofort aus 24 Cent aufgrund der 20% Verluste eben 29 Cent. Wenn ich aber den 24 Cent Netzstrom direkt beziehe und dem Akku so 4h Pause gönne, schaffe ich am Morgen es noch gut die 2 teuren und Lastintensiven Spitzenstunden aus dem Akku zu versorgen.

So behalte ich dann alles an einem Ort, was die Einstellungen anbelangt.

Die Funktion "Netzstrom als Entladepause" wäre wie beim Netzladen von einem Preis  X Cent abhängig. 
Ggf. macht es noch Sinn, diese Pause mit vom SOC zu einer Uhrzeit wie Mitternacht mit abhängig zu machen, also bei SOC um 00:00 < 55% Pause und ggf. auch Höchstzahl an Pause. 

00:00 Uhr SOC < 55%:   max. 3h (günstigste) wenn Preis <26 Cent

Aktuell mach ich das in Homeassistant per Automationen, aber 2 oder 3 Orte (Wechselrichter / Home Assistant / evcc), das einzupflegen macht weniger Sinn

closed 2024-10-23 by andig

typxxi, 2024-10-23:
Dann hat HAUSAKKU LADEN auch nicht mit EV Laden zu tun, oder ?

Ich hab ja geschrieben, dass dann, wenn das AC Akku Laden hier mit reinrutscht, dann genau so gut der Punkt Entladepause mit dazu gehört, denn der basiert schließlich auch auf den dynamischen Preisen. Und die anderen Rahmenbedingungen werden beim Akku Laden auch schon in einem Topic diskutiert, dass dort Randbedingungen fehlen würden u.a. wie lange das AC Laden zum Preis unter X Cent denn nun gelten solle, diese Nacht oder so weit die Tibber Preise reichen oder immer.

Das gleiche in grün, nur dass ich das zusammengefasst habe, was dort auch erst letzte Woche aufgetaucht ist in einem Punkt oder Feature Request. 

Dem hier , wo nataltis die Thematik erkannte
https://github.com/evcc-io/evcc/issues/16673#issuecomment-2418647485

mit 

_"Ja, sehe ich. Spannende Frage ist nur ob wir das zeitbasiert (bspw. solange wie ich in die Zukunft schauen kann, Tarif, fix 24h, ...) machen sollten oder ob wir das "Ereignisbasiert" (bis zum Abstecken) machen. Am klassischen Ladepunkt finde ich das "bis zum Abstecken" gut und sinnvoll. Wir haben aber beim "Speicher aus dem Netz laden" genau das gleiche Thema. Da würde nur eine zeitbasierte Logik funktionieren."_

Von daher passt Entladepause am Ende genauso zu EVCC wie preisbasiertes Akku AC Laden. Alles nur eine Frage der Zeit, denn mit jedem Tibber Kunden mehr, der mal im Winter den Akku leer gefahren hat und die teure Preiszone vor sich hat, wird das Bewusstsein steigen, denn

den Akku nicht laden und den günstigen AC Strom nutzen spart ja am Ende Energie und entlastet die Umwelt, wenn ich den AC Strom zu günstigen Zeiten stattdessen nutzen kann.

So bleibt am Ende der schwarze Peter wo, wenn es um das Thema CO2 Bilanz geht ?

Ich denke weiter, dass das Feature genau hierhin passt, wenn man preisbasiertes AC Laden implementiert, denn daneben ist die günstigere und saubere Alternative noch die Entladepause, weil ich damit 20% Verluste einsparen kann. Sonst wird ja nur bis 30 Cent geladen, um die 45 Cent Spitze abzufangen, hab dabei aber 5 bis 6 Cent Rundreise Verluste.

andig, 2024-10-23:
Dann müssen wir es ja nicht doppelt posten 👍🏻