evcc-io/evcc #16816: Spotty Energie

Ferenz12, 2024-10-23:
Hello! Please add the dynamic electricity price spottyenergie (api) from Austria.
 Thanks

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

andig, 2024-10-23:
Gibts ein Api?

Ferenz12, 2024-10-24:
JaVon meinem iPhone gesendet

TomF79, 2024-10-24:
> Gibts ein Api?

### Technische Details zur Schnittstelle:

Server: https://i.spottyenergie.at/
Ressource: /api/prices/[priceType]/[contractId]
Aufruf: HTTP GET
Verschlüsselung: SSL (HTTPS)
Datenformat: JSON

### Erforderliche Pfadparameter:

priceType: MARKET, CONSUMPTION, GENERATION

contractId: Deine Vertrags-ID im UUID-Format

### Optionale Abfrageparameter:

timezone: Definiert die Zeitzone für Zeitstempel. Standardmäßig werden alle Zeitstempel in UTC zurückgegeben. Du kannst dies überschreiben. Unterstützter Wert: at

### Beispiel-URL mit allen Parametern: 
https://i.spottyenergie.at/api/prices/MARKET/ffffffff-4444-6666-2222-aaaaaabbbbbb?timezone=at

### Beispielantwort:

[{"from":"2024-08-19T00:00:00+02:00","price":8.79},{"from":"2024-08-19T01:00:00+02:00","price":8.21},{"from":"2024-08-19T02:00:00+02:00","price":8.50},{"from":"2024-08-19T03:00:00+02:00","price":8.38},{"from":"2024-08-19T04:00:00+02:00","price":8.21},{"from":"2024-08-19T05:00:00+02:00","price":8.71},{"from":"2024-08-19T06:00:00+02:00","price":11.97},{"from":"2024-08-19T07:00:00+02:00","price":14.38},{"from":"2024-08-19T08:00:00+02:00","price":12.75},{"from":"2024-08-19T09:00:00+02:00","price":11.26},{"from":"2024-08-19T10:00:00+02:00","price":9.49},{"from":"2024-08-19T11:00:00+02:00","price":8.85},{"from":"2024-08-19T12:00:00+02:00","price":9.43},{"from":"2024-08-19T13:00:00+02:00","price":8.21},{"from":"2024-08-19T14:00:00+02:00","price":8.21},{"from":"2024-08-19T15:00:00+02:00","price":7.60},{"from":"2024-08-19T16:00:00+02:00","price":8.24},{"from":"2024-08-19T17:00:00+02:00","price":7.80},{"from":"2024-08-19T18:00:00+02:00","price":10.14},{"from":"2024-08-19T19:00:00+02:00","price":12.19},{"from":"2024-08-19T20:00:00+02:00","price":12.09},{"from":"2024-08-19T21:00:00+02:00","price":10.52},{"from":"2024-08-19T22:00:00+02:00","price":8.50},{"from":"2024-08-19T23:00:00+02:00","price":7.72}]

Ferenz12, 2024-10-24:
Vielen Dank 😊 Von meinem iPhone gesendet

https://i.spottyenergie.at/api/prices/MARKET/ffffffff-4444-6666-2222-aaaaaabbbbbb?timezone=at

andig, 2024-10-24:
Das lässt sich als custom Tarif abbilden. Gerne PR.

Ferenz12, 2024-10-24:
Servus, das Funktioniert leider nicht. Von Spotty wird mir das so ausgegeben:

 

  {

        "from": "2024-10-24T00:00:00+02:00",

        "price": 9.4

 

Es wird mir kein end ausgegeben von Spotty

Ich habe in die evcc.yaml jetzt eingetragen:

 

tariffs:

  grid:

    type: custom

    price:

      source: http

      uri: https://example.com/api/price (nur hier das von Spotty übernommen)

 

Vielen Dank

andig, 2024-10-24:
Schau Dir einfach die Beispiele an, insbesondere allinpower.

andig, 2024-10-24:
Reopening in case someone wants to help.

Ferenz12, 2024-10-25:
Sorry aber ich kann dir leider nicht ganz folgen. Allinpower ist ein template also was soll ich mir da abschauen und ich weiß weder wo dieses template hinterlegt bzw konfiguriert ist?! Ich habe jetzt den SmartEngergie genommen (der ruft die Daten von der gleichen Börse ab) und habe bei charges 0.0258 und bei tax -0.2 (ohne Mwst) eingegeben. Jetzt passt es einigermaßen.

closed 2024-10-29 by andig