evcc-io/evcc #16822: Custom tariffs don't work any more

arboeh, 2024-10-24:
### Describe the bug

The custom tariffs no longer seem to work since update to 0.131.0.
I also get a warning and can't choose charging plans:
`WARN 2024/10/24 07:14:09 planner: no matching rate`

The mqtt and http endpoints of my example are working!

[image]

### Steps to reproduce

1. Change Configuration to custom tariff

### Configuration details

```yaml
tariffs:
  currency: EUR
  grid:
    type: custom
    price:
      source: mqtt
      topic: < MQTT TOPIC TO GET PRICE >
      timeout: 25h

tariffs:
  currency: EUR
  grid:
    type: custom
    price:
      source: http
      uri: https://api.ostrom.de/v1/tariffs/city-id?cityId=2147&usage=3000
      jq: .ostrom[0].workingPriceBrutto
      scale: 0.01
```

### Log details

```text
WARN 2024/10/24 07:14:09 planner: no matching rate
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.0

VolkerK62, 2024-10-24:
same here
```
tariffs:
  currency: EUR
  grid:
    type: custom
    price:
      source: mqtt
      topic: Haus/Preis_Bezug
      timeout: 25h
```
```
pi@raspberrypi:~ $ evcc tariff --log trace
[main  ] INFO 2024/10/24 08:37:32 evcc 0.131.0
[main  ] INFO 2024/10/24 08:37:33 using config file: /etc/evcc.yaml
[db    ] INFO 2024/10/24 08:37:33 using sqlite database: /var/lib/evcc/evcc.db
[db    ] TRACE 2024/10/24 08:37:33 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/10/24 08:37:33 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/10/24 08:37:33 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/10/24 08:37:33 SELECT * FROM `settings` 116 <nil>
[mqtt  ] INFO 2024/10/24 08:37:33 connecting evcc-633872853 at tcp://localhost:1883
[mqtt  ] DEBUG 2024/10/24 08:37:33 tcp://localhost:1883 connected
[db    ] TRACE 2024/10/24 08:37:33 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/10/24 08:37:33 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/10/24 08:37:33 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/10/24 08:37:33 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/10/24 08:37:33 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/10/24 08:37:33 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/10/24 08:37:33 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
grid:
[mqtt  ] TRACE 2024/10/24 08:37:33 recv Haus/Preis_Bezug: '0.378'
From    To    Price/Cost

pi@raspberrypi:~ $ 
```

arboeh, 2024-10-24:
Fyi: I just downgraded to V0.130.13 (using partitial backup) and it works again...

andig, 2024-10-24:
> have verified that the issue is reproducible with the latest nightly build

Really?

MobilettoSoft, 2024-10-24:
Exactly the same here as the OP. It works if I revert to 0.130.13.

VolkerK62, 2024-10-24:
is fixed in the latest Nightly

arboeh, 2024-10-24:
Okay, thanks. As I know it's not possible to install nightly addons in Home Assistant. So, Im guilty and I apoligize very much.

closed 2024-10-24 by andig

VolkerK62, 2024-10-24:
try 0.131.1

MobilettoSoft, 2024-10-24:
working again with 0.131.1