evcc-io/evcc #16823: PV Überschuss Laden in der Nacht aktiv - ohne PV Überschuss und ohne Hausakku ?

typxxi, 2024-10-24:
### Describe the bug

Gestern Abend kam das Auto mit SOC 37% an und wurde um 23:03 angeschlosssen.

evcc war im 
* PV Modus 
* Netzladen unter 20 Cent eingestellt
* und laden bis 95% 

Heut morgen war das Auto voll geladen und ich kann den Auslöser nicht erkennen.

1. Die Tibber Preise lagen weit jenseits 25 Cent, um 23 Uhr bei 0,30€ und dann fallend 
[image]

2. AC Laden stand und steht auf 20 Cent
[image]

3. evcc meldet 23.03 Uhr als Verbindungsbeginn
[image]

4. nach 6:36 h waren 18,7 kWh geladen - vermutlich durch das 6A Limit gedrosselt gewesen
[image]

Ich meine, der Mindestladestand ist auf 25% eingestellt, bis wohin dann schnell geladen würde, aber nicht 95%.

Die einzige Besonderheit oder auch Zufall, der mir jetzt ins Auge stach, war 
**Hausakku Netzladen unter 30 Cent**
[image]

Aber der Hausakku war um 23 Uhr in die Entladepause bis 7 Uhr gegangen, damit er noch die teuren Zeiten am Morgen überbrücken konnte
[image]

### Steps to reproduce

1.evcc im PV Modus mit 95% Ladeziel
2.Netzladen Auto unter 20 Cent
3.Hausakku Netz laden unter 30 Cent
[evcc-20241024-083803-trace.log](https://github.com/user-attachments/files/17503047/evcc-20241024-083803-trace.log)
...

### Configuration details

```yaml
###################################################
# 2024-10-09 - 13:45 = vehicle golf: 2 not phases #
###################################################

network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug      
  # alternativ
  # log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error
  ocpp: trace

# unique installation id
plant: 1234

##########################################
# control cycle interval                 #
# kann kritisch sein, wenn < 30 Sekunden #
##########################################

interval: 12s   # meterinterval: 12s unter chargers sollte mit synchron gehalten werden
                # wenn das zu kurz ist, dann gibt es den folgenden Fehler, von daher das Intervall verlängern
                # Garage: charger logic error: current mismatch (got 11.2A measured, expected 8.73A) 
                # https://github.com/evcc-io/evcc/discussions/15761#discussioncomment-10893056

meters:
  - type: template
    template: growatt-hybrid
    id: 1
    device: /dev/ttyUSB0
    baudrate: 9600
    comset: 8N1
    usage: grid
    modbus: rs485serial
    name: grid1

  - type: template
    template: growatt-hybrid
    id: 1
    device: /dev/ttyUSB0
    baudrate: 9600
    comset: 8N1
    usage: pv
    modbus: rs485serial
    name: pv2

  - type: template
    template: growatt-hybrid
    id: 1
    device: /dev/ttyUSB0
    baudrate: 9600
    comset: 8N1
    usage: battery
    modbus: rs485serial
    name: battery3
    capacity: 14.0

# chargers:
#   - type: template
#     template: pulsarplus
#     timeout: 2m
#     name: wallbox4

###   LÖSUNG   von   hier   ###
###   https://github.com/evcc-io/evcc/issues/15581#issuecomment-2337136867

chargers:
#   5 ZEILEN LÖSUNG für 1.30.13 von https://github.com/evcc-io/evcc/discussions/15986#discussioncomment-10674568
  - type: template
    template: ocpp-abb
    remotestart: true     ###   >1.30.x   ###   NEU FOLGE von https://github.com/evcc-io/evcc/issues/15581 
    meterinterval: 12s    ###   from 10s to 12s 09.10.2024 >1.30.1x = optional
                          ###   ganz oben noch als generelles interval: 12s 
    metervalues: Energy.Active.Import.Register,Power.Active.Import,SoC,Current.Import,Voltage  # >1.30.1x

    # metervalues: Energy.Active.Import.Register,Power.Active.Import,SoC,Power.Offered,Current.Import,Voltage  # >1.30.1x

    stationid: Garage ### NEU = OCCP ID
    connector: 1 ### soll alt sein
    timeout: 2m
    name: wallbox4

vehicles:
  - name: eGOLF
    title: GOLF
    type: custom
    ###       type: template
    ###       template: offline
    icon: car
    capacity: 31.6 # net capacity in kWh

    phases: 2

    soc: # optional SoC (%)
      source: mqtt
      topic: export/golf/soc_permanent
    range: # optional electric range (km)
      source: mqtt
      topic: export/golf/reichweite_permanent
    odometer: # optional odometer (km)
      source: mqtt
      topic: export/golf/km_stand_permanent

loadpoints:
  - title: Garage
    charger: wallbox4
    mode: pv # charge mode (off, now, minpv, pv)
    vehicle: eGOLF # set default vehicle (disables vehicle detection)
    ###   NEU: SCHWELLEN   ###
    enable:
      delay: 1m
      threshold: -1500 # laden, sobald 1 Minute lang min 1.500 W eingespeist worden sind
    disable:
      delay: 2m
      threshold: 1000 # laden beenden, wenn 2 Minuten lang mehr als 1.000 W Netzstrom bezogen wurden

site:
  title: Zuhaus
  meters:
    grid: grid1
    #
    # für saubere Überschussdarstellung bei mehreren Wechselrichtern
    # pv Zähler deaktivieren
    # https://github.com/evcc-io/evcc/discussions/16574#discussioncomment-10892601
    # 
    pv: pv2  
    battery:
      - battery3
  residualPower: 111

tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: tibber
    token: 123
    # type: fixed
    # price: 0.272 # 0.272 [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.0774 #  seit 19.8.2024

# mqtt message broker
mqtt:
  broker: 192.168.178.2:1883 # MQTT-Broker & Port
  topic: evcc # alles wird im MQTT Explorer unter TOPIC EVCC erscheinen
  user: 1234
  password: 1234

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: Soc updated
      msg: Battery charged to ${vehicleSoc:%.0f}%
    guest: # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
```

### Log details

```text
Log im Anhang oben unter Steps to reproduce
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

Current version: 0.130.13

andig, 2024-10-24:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-10-24 by andig

typxxi, 2024-10-24:
I have added the log file here 
[image]

typxxi, 2024-10-24:
> > Heut morgen war das Auto voll geladen und ich kann den Auslöser nicht erkennen.
> 
> Mach gerne eine Diskussion auf. Wenn kein Fehler erkennbar ist gehört es nicht in die Issues.

Fehler ist doch erkennbar, dass geladen wurde, obschon abends um 23 Uhr nicht PV Überschuss geladen werden konnte und sollte. 

Ich wollte sicher nicht den Akku heute Nacht voll haben, wenn im Osnabrücker Land mal so ein sonniger Herbsttag vorhergesagt war wie heute. Auto ist heute nicht bewegt worden, daher steht es noch auf 95% wie heute früh, von daher kann ich heute Abend auch nix testen, ob das wieder passiert oder wann und warum.

Vielleicht morgen Abend und dann muss ich schauen, was die HomeAssitant Protokolle bei trace dann hergeben.

andig, 2024-10-24:
> Fehler ist doch erkennbar

Dann mach bitte eine konkrete Fehlerbeschreibung mit Bezug aufs Logfile. Die ganzen Screenshots brauchts nicht.