evcc-io/evcc #16826: REST API: lodpoint ID mismatch

Coernel82, 2024-10-24:
### Describe the bug

 IDs of loadpoints are different when accessing different API-urls:
when accessing the state-url the loadpoints start with ``0``, when accessing the POST Urls the loadpoints beginnt with ``1``

### Steps to reproduce

### GET
''/api/state'' → loadpoints are 0, 1, 2, 3 ....
``/api/loadpoints/0/plan``  → ``404`` while 1, 2, 3, 4 is ok - so it is shifted +1

### POST
/api/loadpoints/0/mode/<mode> → ``404`` while 1, 2, 3, ,4 is ok - shifted as well!
probably all the other POST loadpoints as well

### Configuration details

```yaml
unrelated
```

### Log details

```text
unrelated
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.1

VolkerK62, 2024-10-24:
>Alle API IDs (z.B. die Loadpoint ID) beginnen bei 1.

https://docs.evcc.io/docs/integrations/rest-api

andig, 2024-10-24:
> /api/state → loadpoints are 0, 1, 2, 3 ....

That's your interpretation!

closed 2024-10-24 by andig