evcc-io/evcc #16828: Failed to start with 0.131.1

denniswham, 2024-10-24:
### Describe the bug

starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/10/24 12:56:43 evcc 0.131.1
[main  ] INFO 2024/10/24 12:56:43 using config file: /config/evcc.yaml
[main  ] INFO 2024/10/24 12:56:44 listening at :7070
[main  ] FATAL 2024/10/24 12:56:44 cannot create meter 'solaredge': cannot create meter type 'template': cannot create meter type 'custom': power: read tcp 10.0.20.10:43826->10.0.20.160:1502: read: connection reset by peer
[main  ] FATAL 2024/10/24 12:56:44 will attempt restart in: 15m0s

### Steps to reproduce

1. Install 0.131.1
2.
3.
...

### Configuration details

```yaml
meters:
  - name: solaredge
    type: template
    template: solaredge-inverter 
    usage: pv
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: x.x.x.x  
    port: 1502
    timeout: 10s
```

### Log details

```text
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/10/24 12:56:43 evcc 0.131.1
[main  ] INFO 2024/10/24 12:56:43 using config file: /config/evcc.yaml
[main  ] INFO 2024/10/24 12:56:44 listening at :7070
[main  ] FATAL 2024/10/24 12:56:44 cannot create meter 'solaredge': cannot create meter type 'template': cannot create meter type 'custom': power: read tcp 10.0.20.10:43826->10.0.20.160:1502: read: connection reset by peer
[main  ] FATAL 2024/10/24 12:56:44 will attempt restart in: 15m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.1

denniswham, 2024-10-24:
Same issue as #16794

ivoks, 2024-10-24:
@denniswham issue #16794 has been resolved with 0.131.1, as the patch that introduced the problem in 0.131 has been reverted for 0.131.1. Problem that you are seeing is different; connection reset by peer means that solaredge terminated the connection. It didn't complain about invalid request.

andig, 2024-10-24:
> Same issue as https://github.com/evcc-io/evcc/issues/16794

Its not even the same error message, so no.

Still, this is a network error, nothing evcc can likely do. If the error is reproducible allow info@evcc.io connection via VPN, SSH or similar.

closed 2024-10-24 by andig