evcc-io/evcc #16829: Adding tariffs to dynamic electricity price

timpetersson81, 2024-10-24:
Hi, here from Denmark.

I have Dynamic electricity price, but we allso here in Denmark have supplying tarifs for supply.

and Addon to the dynamic part looking like this could be awesome, here I can add costs

tariffs:
  currency: DKK
  grid:
    type: template
    template: energinet
    region: dk2 # optional
    zones:
      - days: Mo-So
        hours: 00-06
        charges: 0.1431
      - days: Mo-So
        hours: 06-17
        charges: 0.4293 # DKK/kWh
      - days: Mo-So
        hours: 17-21
        charges: 1.28 # DKK/kWh
      - days: Mo-So
        hours: 21-00
        charges: 0.4293 # DKK/kWh

As you can se the cost during 17-21 is high, so only relying in the time price can be very expensive.

andig, 2024-10-24:
Sorry, it's unclear what you're asking.

VolkerK62, 2024-10-24:
dynamic tariff with time-based charges?

timpetersson81, 2024-10-25:
Here in Denmark, transport costs for electricity are called tariffs, but we also have variable electricity prices by the hour. Where I live, we have 4 prices for transport over the day. I would like the spot prices for electricity + transport costs.

from 00-06 they are currently 0.1431 DKK
from 06-17 they are currently 0.4293 DKK

What I asked for where a solution where I can add the ekstra cost by the hour to the dynamic electricity prices.
`
tariffs:
  currency: DKK
  grid:
  type: template
  template: energinet
  region: dk2 # optional
  zones:
     - days: Mo-So
        hours: 00-06
        charges: 0.1431
     - days: Mo-So
        hours: 06-17
        charges: 0.4293 # DKK/kWh
     - days: Mo-So
        hours: 17-21
        charges: 1.28 # DKK/kWh
     - days: Mo-So
        hours: 21-00
        charges: 0.4293 # DKK/kWh
`

My thoughts were that it could do something like the above. Where on the price between 00-06, "charges" were added on top of the imported price, and so on.

ReinholdEckert, 2024-10-26:
Hello Tim!
I guess it would be very complicated for evcc to cover all combinations that might be needed. I am thinking to build a solution for that, maybe this solution could provide the correct tariff for evcc via API. Are you interested?

andig, 2024-10-26:
This sounds as we want to have the capability to have zones for charges, taxes etc. too. Similar to us having price zones for the fixed tariff. Is that right?

ReinholdEckert, 2024-10-26:
I think the thing is: most people are inserting the energy costs only, which is only half the truth as you have grid costs (fixed, per kWh and maybe even dynamic for specific time frames) too. So when comparing costs you have to compare the energy + grid costs vs. money you get for selling power. In Austria with cheap energy costs the grid costs actually are quite equal atm.
So what would be needed is a cost calculator that involves all those costs (which are different in every country or even states). As I said, I could imagine building a service for that if it is not planned as a feature in evcc.

andig, 2024-10-26:
If you have a different request (?) please open a new issue.

timpetersson81, 2024-10-26:
> This sounds as we want to have the capability to have zones for charges, taxes etc. too. Similar to us having price zones for the fixed tariff. Is that right?

Yes that sounds about it

timpetersson81, 2024-10-26:
> If you have a different request (?) please open a new issue.

Idont :)

typxxi, 2024-10-29:
> Here in Denmark, transport costs for electricity are called tariffs, but we also have variable electricity prices by the hour. Where I live, we have 4 prices for transport over the day. I would like [...]

at the end nordpool and entso-e have allowed to use formulas inside their integrations by adding a line in the yaml file.

Might be worth to have a look how that might fit too cause both are scandinavian platforms if I remember correctly.
Nordpool had an issue till 10 days ago when I found an error for german users when they switched to a new model and used VAT from the backend instead of those formulas. Unfortunatelly someone was unsure and wrote 23% VAT for germany and the prices from the nordpool source for my tibber prices never matched the tibber app / cloiud prices again.

Till I found out that the upgrade used the backend and that someone had written 23% VAT with a remark that he is unsure if that is true. Which is a mess cause which user will look deep into the code to find such line there  ? 

Anyhow: here is a code example, how the consumer price for germans has to be calculated

where 0.19161 cents stand for the fees you had mentioned which are already incl. german vat
then there is the vat on top of the kWh generation price per each hour (a day ahead price)
which both will be added on top of the  kWh generation price. 

I had been using this line to achieve it in the previous months till they changed things and I had to find out about the issue with that false 19% vat for germany mentioned above when they changed the architecture.

Might offer more flexibility with such kind of user formulas  instead of administrating all EU power generation and distribution prices here inside evcc with so many models and dealers and exchanges.  
{{ ( 0.19161 + current_price * 0.19 ) | float }}

Of cause this is from my Home Assistant experience with Entso-E and nordpool but it worked and I liked it cause there were many references and examples available which all had worked till nordpool did the upgrade from the decentralized formula to the central approach but with the wrong VAT.

timpetersson81, 2024-10-29:
> at the end nordpool and entso-e have allowed to use formulas inside their integrations by adding a line in the yaml file.
>
> Might be worth to have a look how that might fit too cause both are [...]

I have no clue what you are talking about. The best Way for an integration with EVCC is not to add info at a yaml file in Home assistant

andig, 2024-10-30:
Closing in favor of https://github.com/evcc-io/evcc/issues/16977

closed 2024-10-30 by andig

andig, 2024-10-30:
@timpetersson81 how many additional parameters per zone would you need for this case? Afaikt no additional parameters since `charges` already exists, right? It would just need be enabled to be per-zone instead of globally?

timpetersson81, 2024-10-30:
Just 1 charges pr zone, but i need to add 4 zones to the api price. And zones are different time intervals