evcc-io/evcc #16830: Inverter Work Mode after fast or planed charge default mode to last mode

farcorben, 2024-10-24:
Hallo Team evcc ^^

mir ist aufgefallen, das immer wenn man eine Funktion "geplantes Laden" oder "Netzladen" nutzt, der Wechselrichter im abschluss in den "normal" modus zurück gesetzt wird. In meinem definitions Fall wäre das der "self use" Modus. Definiert hier im Template 

https://github.com/evcc-io/evcc/blob/a804a775235ab0c350b41a6f59c642dbeca262f5/templates/definition/meter/solax.yaml#L81-L138

Leider ist es so das der "self Use" Modus nicht immer der favorisierte Modus ist.
In meinem Fall wäre "default" der Modus "Enhanced TOU". Es könnte aber natürlich auch ein ganz anderer sein z.b. SmartSchedule, Peak Shaving oder oder etc..

[image]
[image]

Meine Erwartung wäre jedoch das nach Abschluss des Ladezyklus der zuvor konfiguriere Modus wiederhergestellt wird. 
Bedeutet, evcc sollte zuerst feststellen welcher Betriebsmodus vorher definiert war, denn Modus entsprechend ändern, und im Abschluss den zuvor ermittelten Work-mode zurücksetzen.

andig, 2024-10-24:
> Bedeutet, evcc sollte zuerst feststellen welcher Betriebsmodus vorher definiert war, denn Modus entsprechend ändern, und im Abschluss den zuvor ermittelten Work-mode zurücksetzen.

Das geht leider nicht ganz so einfach. Alternative: Template anpassen und Wunschmodus dort einstellen.

farcorben, 2024-10-24:
> > Bedeutet, evcc sollte zuerst feststellen welcher Betriebsmodus vorher definiert war, denn Modus entsprechend ändern, und im Abschluss den zuvor ermittelten Work-mode zurücksetzen.
> 
> Das geht leider nicht ganz so einfach. Alternative: Template anpassen und Wunschmodus dort einstellen.

Das würd bedeuten das sich jeder sein eigenes Template modifiziert der nicht entsprechend dem default workmode "self use" nutzt... das erscheint mir sehr Unpraktikabel

Wäre es eine idee einen optionalen Schalter zu schaffen der es dem user möglich macht den "default" zu definieren?
ähnlich diesem:

```evcc.yaml
meters:
  - name: my_battery
    type: template
    template: solax
    usage: battery
    default_workmode: 1 # 1 = self use, 21 = Enhanced TOU
```

andig, 2024-10-24:
Davon sprach ich, ja.

closed 2024-11-01 by andig

andihh71, 2024-12-14:
Moin,
ich verstehe es hier noch nicht richtig. Was muss ich denn nun machen, dass der letzteingestellte Modus am WR bleibt?

andig, 2024-12-14:
Das Template muss angepasst werden damit der Modus überhaupt einstellbar wird.

andihh71, 2024-12-14:
Danke für den Hinweis. Gibt es dazu eine Anleitung? Ich weiß leider nicht wie und was genau gemeint ist. Sorry für meine Unwissenheit 🫣

farcorben, 2024-12-14:
Hey @andihh71 ,

das is ganz einfach. Schau mal in die Solax Doku:
https://docs.evcc.io/docs/devices/meters#hybrid-x1x3-g3g4

Abschnitt unter "Batterie" und "advanced options"

Du musst in der config folgendes einfügen.

```
meters:
  - name: my_battery
    type: template
    template: solax
    usage: battery
  ...
    defaultmode: 0 # optional
```

wobei es folgende mode`s gibt:

```
0: self Use
1: Feedin Priority
2: Backup mode
3: Manual mode
4: Peak shaving
5: Enhanced TOU
6: SmartSchedule
```

andihh71, 2024-12-16:
Hallo @farcorben, 
das habe ich bei mir instinktiv auf defaultmode: 2 eingestellt, leider ohne Erfolg. Es stellt sich nach wie vor von Schnell automatisch auf Eigenverbrauch, wenn der Ladestecker gezogen wird.

andihh71, 2024-12-17:
Moin,
ich habe derzeit die aktuelle Version von EVCC laufen. Anbei hänge ich mal Bilder von der Yaml und der Firmware an. Anders kann ich es gerade nicht machen. 
Der Wechselrichter ist vom Typ X3-G4. 
[image]
[image]
[image]

farcorben, 2024-12-17:
hey @andihh71,

der WR Code hab ich keine Ahnung was das genau sein soll ^^

Bei mir läuft die 
ARM Version 1.44-1.09
DSP Version 1.46

ich kommuniziere mit meiner Solax X3-Gen4 direkt über `tcip` wie folgt. Damit funktioniert es, bei mir, Problemlos..

```yaml
meters:
...
- type: template
  template: solax
  id: 1
  host: 192.168.0.64
  port: 502
  usage: battery
  modbus: tcpip
  name: battery
  capacity: "12"
  defaultmode: 5 # TOU
```

@andig , hast du vielleicht noch eine idee? Kann es an der connection "tcpip" vs "rs485serial" liegen?

andig, 2024-12-17:
Ich hab immer eine Idee wenn ich das Problem verstehe. Da würde sicher ein Logfile helfen ;)

andihh71, 2024-12-18:
Moin,

ich möchte mich noch mal bedanken. Jetzt funktioniert es so, wie es sein sollte. Ich verstehe zwar nicht, wieso plötzlich, aber vielleicht hatte ein Neustart alleine nicht gereicht, sondern erst beim zweiten erneuten Neustart. 
Viele Grüße 
Andi