evcc-io/evcc #16837: SofarSolar von PV auf Schnell Laden ->Akku Bezug

Maddin1977, 2024-10-24:
### Describe the bug

Hallo

Ich nutze die 0.131.0
Sofarsolar Hyd15  FW00061
Sofarsolar AmassGTX3000 H4

Wenn ich von PV auf Schnell Laden stelle
Werden kurz 5.xkW vom Akku und 3.xkW vom Netzt gezogen.
[image]

Bei der nächsten Aktualisierung schreibt er Batterie gesperrt und wechselt komplett auf Netz Bezug.

Wenn ich nun wieder zurück stelle auf PV
reduziert sich die Lade-Leistung auf 2.7kW aber bezieht diese aus dem Akku
[image]

Erst wenn ich auf Aus stelle. 
Schaltet das Laden ab.

Er sollte denke ich aber gar nicht die Batterie belasten wenn so eingestellt?
[image]

Nun wieder auf PV und es wird nicht geladen.
[image]

Denke das sollte so nicht sein.
Fehler wird keiner ausgeworfen.
Reproduzierbar ist das ganze.

Danke und Grüße 

### Steps to reproduce

1.von PV auf Schnell Laden
2.zurück auf PV
3.Aus
4.PV

### Configuration details

```yaml
Reiche ich nach wenn nötig
```

### Log details

```text
Kein Fehler im Log
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.0

VolkerK62, 2024-10-24:
auch ohne Fehler ist ein Log zur Fehleranalyse notwendig

Maddin1977, 2024-10-24:
> ohne auch Fehler einlog zur Fehleranalyse

hab ich oben noch dazu gepackt von diesem Zeitraum

Chris45671, 2024-10-24:
Also er hört erst im PV Modus auf, wenn der eingestellte Timer abgelaufen ist
```

[lp-1 ] DEBUG 2024/10/24 18:31:06 pv disable in 2m29s
[lp-1 ] DEBUG 2024/10/24 18:31:06 pv disable timer remaining: 2m29s
[site ] DEBUG 2024/10/24 18:31:33 ----
```

Maddin1977, 2024-10-24:
> ```
> disable:
>       # ausschalten, wenn 30 Minuten lang mehr als 2000 W aus dem Netz bezogen werden
>       delay: 30m
>       threshold: 2000
> ```

da heißt es aber Netzbezug? er zieht aber doch von der Batterie die Leistung?
oder versteh ich was falsch?

PS:
das ganze muss ich dann in die yaml einfügen?

Maddin1977, 2024-10-24:
> Der Strombezug nach dem umschalten auf den schnellen Modus ist bei meinem Kostal identisch. Auch wenn er den Akku sperrt, dauert es einen Moment bis dies umgesetzt ist. In der Zeit werden dann vielleicht 50-100 Watt bezogen. Danach wird die Batterie dann gesperrt. Der andere Punkt ist, die Batterie bzw. Bezug vom Strom wird in der Ansicht nicht im Sekundentakt aktualisiert. Da ist auch etwas Verzug.

das würde mich ja nicht stören wenn es nur kurz ist.
und nur paar Watt
aber 5.1kW auf dauer macht den Akku schnell leer
das die Anzeige nicht Aktuell ist und etwas verzögerung ist mir aufgefallen.

Chris45671, 2024-10-24:
Dies ist laut Doku aber der entscheidende Punkt. Einen anderen kenne ich nicht. Während dem delay zieht er bei mir auch Strom aus der Batterie, da der Akku ja nicht gesperrt ist. 
Mit dem delay geht es aus meiner Sicht darum wenn zum Beispiel einer Wolke kommt, der Ladevorgang nicht gleich gestoppt wird. 
Dies kann ja nach Auto auch zu Problemen führen das sie bei zu vielen Pausen den Ladevorgang abbrechen.

Chris45671, 2024-10-24:
Die 5,1 kW beziehen sich auf eine Stunde bei dieser Leistung. 5,1 (5100 Watt) : 60 und dann ca. 1 Minute oder zwei, dann weißt Du was er wirklich gezogen hat.

Chris45671, 2024-10-24:
Ich hoffe Du schaust nie wenn Du eine Kaffeemaschine oder eine Waschmaschine anmachst. Diese ziehen auch über 2kW ;-)

Maddin1977, 2024-10-24:
> Ich hoffe Du schaust nie wenn Du eine Kaffeemaschine oder eine Waschmaschine anmachst. Diese ziehen auch über 2kW ;-)

das ist jetzt am Anfang so, da evcc komplett Neu ist für mich und auch das erste mal das E-Auto angesteckt wurde.
Da ist man(n) etwas nervöser

Frage:
war das so gemeint mit dem einpflegen
[image]

Chris45671, 2024-10-24:
Ja so ist es gemeint. 
Siehe Doku: https://docs.evcc.io/docs/features/solar-charging

Die Werte müssen natürlich dem Wunsch entsprechend angepasst werden. 

Ich lade zum Beispiel 5 min weiter.

Maddin1977, 2024-10-24:
> Ja so ist es gemeint. Siehe Doku: https://docs.evcc.io/docs/features/solar-charging
> 
> Die Werte müssen natürlich dem Wunsch entsprechend angepasst werden.
> 
> Ich lade zum Beispiel 5 min weiter.

Okay....
Fazit es funktioniert also alles wie es soll/programmiert ist.
Ich war zu ungeduldig und unerfahren das es solche Delay's gibt ?

Chris45671, 2024-10-24:
Ein bischen vielleicht.
Man kann noch das eine oder andere Optimieren.

Bitte dran denken, wenn Du im Beispiel oben die 30 übernommen hast, dann zieht er 30min den Strom aus der Batterie bis er stoppt.