evcc-io/evcc #16843: OCPP Current Mismatch - sits at 8A when solar excess greater

t3h-n00b, 2024-10-25:
### Describe the bug

Charger sits at 8A in Solar mode (when solar excess is >8A)

### Steps to reproduce

1. Whenever the solar excess is >8A, the EV charges at 8A
2. Errors are reported - confirmed that the charge rate is actually 8A however
`[lp-1 ] WARN 2024/10/25 11:50:06 charger logic error: current mismatch (got 8A, expected 19.6A)`
3. Fast charging (32A) works fine, Min charging (6A) also works fine

[evcc-20241025-113333-trace.log](https://github.com/user-attachments/files/17514349/evcc-20241025-113333-trace.log)

### Configuration details

```yaml
# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - name: grid_usage
    type: custom
    usage: grid
    power:
      source: http
      uri: http://192.168.2.20/emeter/0
      jq: .power
  - name: solar_generation
    type: template
    template: shelly-1pm
    usage: pv
    host: 192.168.2.20 # IP address or hostname
    user: # User account (e.g. email address, user id, etc.) (optional)
    password: # Password of the user account (use single quotes in case of leading zeros) (optional)
    channel: 1 # optional

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for documentation see https://docs.evcc.io/docs/devices/chargers
chargers:
- name: EVSwitch
  type: template
  template: ocpp
  stationid: SN10052312237863 # Station ID of the charging point. Only required if multiple OCPP charging stations are set up to assign them correctly.

# vehicle definitions
vehicles:
  - name: mustang_mache
    type: offline
    title: Mustang Mach-E # display name for UI
    capacity: 72 # battery capacity (kWh)

# site describes the EVU connection, PV and home battery
site:
  title: Home # display name for UI
  meters:
    grid: grid_usage # grid meter not yet available
    pv:
      - solar_generation # list of pv inverters/ meters
  maxGridSupplyWhileBatteryCharging: 0 # ignore battery charging if AC consumption is above this value

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Home # display name for UI
    charger: EVSwitch # charger
    mode: "off" # default charge mode to apply when vehicle is disconnected; use "off" to disable by default if charger is publicly available

    # remaining settings are experts-only and best left at default values
    priority: 0 # relative priority for concurrent charging in PV mode with multiple loadpoints (higher values have higher priority)
    enable: # pv mode enable behavior
      delay: 30s # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
```

### Log details

```text
Attached
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.0 (0d18d670)

t3h-n00b, 2024-10-25:
Alright, something must have changed somewhere in my settings, as my fast charging at 32A yesterday is now occuring at 8A. Given there is nothing in my config for 8A, could it be somewhere else?

t3h-n00b, 2024-10-25:
After a bit more digging, this appears to be version related. Rolling back to 0.130.7 seems to have fixed my issues (I didn't roll back to any versions between .7 and .131 yet to test where the bug was introduced).

closed 2024-10-25 by andig

t3h-n00b, 2024-10-29:
Apologies for the slow response, however I've managed to do a bit of testing.

- The 8A issue persists on 0.131.4 (as well as 131.0)
- I've tested on 0.130.7 and 0.130.8, they don't have this issue (although they aren't perfect and have a different 8A issue - whereby in Solar mode it will occasionally set the limit to 8A when there is ample excess solar available, and sometimes will sit there for 5+ minutes before it 'corrects' to the correct setting)
- I haven't had a chance to test on 0.130.9+ yet to see where the issue was introduced.

Trace logs from 0.130.4 set to Fast Charge (Current.Offered is always set to 8A, even though the limit is set to 32A):
`[site  ] DEBUG 2024/10/29 17:24:06 ----
[site  ] DEBUG 2024/10/29 17:24:06 grid power: -707W
[site  ] DEBUG 2024/10/29 17:24:06 pv power: 1115W
[site  ] DEBUG 2024/10/29 17:24:06 site power: -707W
[ocpp  ] TRACE 2024/10/29 17:24:06 enqueued CALL [477232814, SetChargingProfile] for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:06 dispatched request 477232814 for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:06 sent JSON message to **SERIAL**: [2,"477232814","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":3,"stackLevel":5,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-10-29T06:23:06Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":32}]}}}]
[ocpp  ] TRACE 2024/10/29 17:24:06 started timeout timer for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:09 enqueued CALL [2998029851, TriggerMessage] for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:11 received JSON message from **SERIAL**: [3,"477232814",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/10/29 17:24:11 handling incoming CALL RESULT [477232814] from **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:11 completed request 477232814 for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:11 **SERIAL** ready to transmit again
[ocpp  ] TRACE 2024/10/29 17:24:11 dispatched request 2998029851 for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:11 sent JSON message to **SERIAL**: [2,"2998029851","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/10/29 17:24:11 started timeout timer for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:11 timeout canceled for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:11 enqueued CALL [4232443617, SetChargingProfile] for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:11 received JSON message from **SERIAL**: [3,"2998029851",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/10/29 17:24:11 handling incoming CALL RESULT [2998029851] from **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:11 completed request 2998029851 for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:11 enqueued CALL [2294781436, TriggerMessage] for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:11 timeout canceled for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:11 **SERIAL** ready to transmit again
[ocpp  ] TRACE 2024/10/29 17:24:11 dispatched request 4232443617 for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:11 sent JSON message to **SERIAL**: [2,"4232443617","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":3,"stackLevel":5,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-10-29T06:23:11Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":32}]}}}]
[ocpp  ] TRACE 2024/10/29 17:24:11 started timeout timer for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:12 received JSON message from **SERIAL**: [2,"241029062406-9903","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-10-29T06:24:06Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"0.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"441967.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}],"transactionId":1730181981}]
[ocpp  ] TRACE 2024/10/29 17:24:12 handling incoming CALL [241029062406-9903, MeterValues] from **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:12 sent CALL RESULT [241029062406-9903] for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:12 sent JSON message to **SERIAL**: [3,"241029062406-9903",{}]
[ocpp  ] TRACE 2024/10/29 17:24:12 received JSON message from **SERIAL**: [3,"4232443617",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/10/29 17:24:12 handling incoming CALL RESULT [4232443617] from **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:12 completed request 4232443617 for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:12 **SERIAL** ready to transmit again
[ocpp  ] TRACE 2024/10/29 17:24:12 dispatched request 2294781436 for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:12 sent JSON message to **SERIAL**: [2,"2294781436","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/10/29 17:24:12 started timeout timer for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:12 timeout canceled for **SERIAL**
[ocpp  ] TRACE 2024/10/29 17:24:13 received JSON message from **SERIAL**: [2,"241029062407-9904","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-10-29T06:24:07Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"0.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"441967.0","context":"Sample. [... cut ...]
[ocpp ] TRACE 2024/10/29 15:00:28 send **SERIAL**: [3,"241029040028-8121",{}]
[ocpp ] TRACE 2024/10/29 15:00:29 recv **SERIAL**: [3,"3811392802",{"status":"Accepted"}]
[lp-1 ] DEBUG 2024/10/29 15:00:29 max charge current: 18.5A
[ocpp ] TRACE 2024/10/29 15:00:29 send **SERIAL**: [2,"558090396","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[lp-1 ] DEBUG 2024/10/29 15:00:29 set charge mode: pv
[site ] DEBUG 2024/10/29 15:00:29 ----
[lp-1 ] DEBUG 2024/10/29 15:00:29 charge power: 1844W
[lp-1 ] DEBUG 2024/10/29 15:00:29 charge currents: [7.6 0 0]A`

andig, 2024-10-29:
> I have verified that the issue is reproducible with the latest nightly build

Could you please share that log?

t3h-n00b, 2024-10-29:
> > I have verified that the issue is reproducible with the latest nightly build
> 
> Could you please share that log?

No worries, this is the latest nightly

```text
[lp-1 ] INFO 2024/10/30 08:32:17 car connected
[lp-1 ] DEBUG 2024/10/30 08:32:17 vehicle api refresh
[lp-1 ] DEBUG 2024/10/30 08:32:17 pv timer elapse
[lp-1 ] DEBUG 2024/10/30 08:32:17 pv timer inactive
[lp-1 ] DEBUG 2024/10/30 08:32:17 charge total import: 442.113kWh
[lp-1 ] DEBUG 2024/10/30 08:32:17 charger vehicle id: evcc
[lp-1 ] DEBUG 2024/10/30 08:32:20 set charge mode: now
[lp-1 ] DEBUG 2024/10/30 08:32:20 pv timer reset
[lp-1 ] DEBUG 2024/10/30 08:32:20 pv timer inactive
[site ] DEBUG 2024/10/30 08:32:20 ----
[lp-1 ] DEBUG 2024/10/30 08:32:20 charge power: 0W
[lp-1 ] DEBUG 2024/10/30 08:32:20 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/30 08:32:20 grid power: 80W
[site ] DEBUG 2024/10/30 08:32:20 pv power: 869W
[site ] DEBUG 2024/10/30 08:32:20 site power: 80W
[lp-1 ] DEBUG 2024/10/30 08:32:20 charge total import: 442.113kWh
[lp-1 ] DEBUG 2024/10/30 08:32:20 charger status: B
[ocpp ] TRACE 2024/10/30 08:32:20 enqueued CALL [2774195174, SetChargingProfile] for **SERIAL**
[ocpp ] TRACE 2024/10/30 08:32:20 dispatched request 2774195174 for **SERIAL**
[ocpp ] TRACE 2024/10/30 08:32:20 sent JSON message to **SERIAL**: [2,"2774195174","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":3,"stackLevel":5,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-10-29T21:31:20Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":32}]}}}]
[ocpp ] TRACE 2024/10/30 08:32:20 started timeout timer for **SERIAL**
[ocpp ] TRACE 2024/10/30 08:32:21 received JSON message from **SERIAL**: [3,"2774195174",{"status":"Accepted"}]
[ocpp ] TRACE 2024/10/30 08:32:21 handling incoming CALL RESULT [2774195174] from **SERIAL**
[ocpp ] TRACE 2024/10/30 08:32:21 completed request 2774195174 for **SERIAL**
[lp-1 ] DEBUG 2024/10/30 08:32:21 max charge current: 32A
[ocpp ] TRACE 2024/10/30 08:32:21 **SERIAL** ready to transmit again
[ocpp ] TRACE 2024/10/30 08:32:21 dispatched request 626817248 for **SERIAL**
[ocpp ] TRACE 2024/10/30 08:32:21 timeout canceled for **SERIAL**
[ocpp ] TRACE 2024/10/30 08:32:21 enqueued CALL [626817248, SetChargingProfile] for **SERIAL**
[ocpp ] TRACE 2024/10/30 08:32:21 sent JSON message to **SERIAL**: [2,"626817248","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":3,"stackLevel":5,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-10-29T21:31:21Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":32}]}}}]
[ocpp ] TRACE 2024/10/30 08:32:21 started timeout timer for **SERIAL**
[ocpp ] TRACE 2024/10/30 08:32:22 received JSON message from **SERIAL**: [3,"626817248",{"status":"Accepted"}]
[ocpp ] TRACE 2024/10/30 08:32:22 handling incoming CALL RESULT [626817248] from **SERIAL**
[ocpp ] TRACE 2024/10/30 08:32:22 completed request 626817248 for **SERIAL**
[ocpp ] TRACE 2024/10/30 08:32:22 **SERIAL** ready to transmit again
[ocpp ] TRACE 2024/10/30 08:32:22 timeout canceled for **SERIAL**
[lp-1 ] DEBUG 2024/10/30 08:32:22 charger enable
[lp-1 ] DEBUG 2024/10/30 08:32:22 wake-up timer: start
[ocpp ] TRACE 2024/10/30 08:32:22 received JSON message from **SERIAL**: [2,"241029213222-5524","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-10-29T21:32:22Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"0.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"442113.0","context":"Sample.Periodic"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Sample.Periodic"}]}],"transactionId":1730234735}]
[ocpp ] TRACE 2024/10/30 08:32:22 handling incoming CALL [241029213222-5524, MeterValues] from **SERIAL**
[ocpp ] TRACE 2024/10/30 08:32:22 sent CALL RESULT [241029213222-5524] for **SERIAL**
[ocpp ] TRACE 2024/10/30 08:32:22 sent JSON message to **SERIAL**: [3,"241029213222-5524",{}]
[ocpp ] TRACE 2024/10/30 08:32:23 received JSON message from **SERIAL**: [2,"241029213222-5525","StatusNotification",{"connectorId":1,"status":"Charging","errorCode":"NoError","timestamp":"2024-10-29T21:32:20Z","vendorId":"EN+","info":"{\"reason\":\"plugOutGun\",\"cpv\":0,\"rv\":0}"}]
[ocpp ] TRACE 2024/10/30 08:32:23 handling incoming CALL [241029213222-5525, StatusNotification] from **SERIAL**
[ocpp ] TRACE 2024/10/30 08:32:23 sent CALL RESULT [241029213222-5525] for **SERIAL**
[ocpp ] TRACE 2024/10/30 08:32:23 sent JSON message to **SERIAL**: [3,"241029213222-5525",{}]
[ocpp ] TRACE 2024/10/30 08:32:24 received JSON message from **SERIAL**: [2,"241029213223-5526","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-10-29T21:32:23Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"0.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"1.2"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"442113.0","context":"Sample.Periodic"},{"measurand":"Power.Active.Import","unit":"W","value":"292.0","context":"Sample.Periodic"}]}],"transactionId":1730234735}]
[ocpp ] TRACE 2024/10/30 08:32:24 handling incoming CALL [241029213223-5526, MeterValues] from **SERIAL**
[... cut ...]
[ocpp ] TRACE 2024/10/30 08:34:15 sent JSON message to **SERIAL**: [3,"241029213414-5552",{"status":"Accepted"}]
[site ] DEBUG 2024/10/30 08:34:17 ----
[lp-1 ] DEBUG 2024/10/30 08:34:17 charge power: 1848W
[lp-1 ] DEBUG 2024/10/30 08:34:17 charge currents: [7.7 0 0]A
[site ] DEBUG 2024/10/30 08:34:17 grid power: 1421W
[site ] DEBUG 2024/10/30 08:34:17 pv power: 974W
[site ] DEBUG 2024/10/30 08:34:17 site power: 1421W
[lp-1 ] DEBUG 2024/10/30 08:34:17 detected active phases: 1p
[lp-1 ] DEBUG 2024/10/30 08:34:17 charge total import: 442.162kWh
[lp-1 ] DEBUG 2024/10/30 08:34:17 charger status: C
```

t3h-n00b, 2024-10-29:
Alright, looks like the issue is introduced in 0.130.9 - logs from this version:

```text
[lp-1 ] DEBUG 2024/10/30 08:46:48 charge power: 0W
[lp-1 ] DEBUG 2024/10/30 08:46:48 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/30 08:46:48 pv power: 950W
[site ] DEBUG 2024/10/30 08:46:48 grid meter: -389W
[site ] DEBUG 2024/10/30 08:46:48 site power: -389W
[lp-1 ] DEBUG 2024/10/30 08:46:48 charge total import: 442.182kWh
[lp-1 ] DEBUG 2024/10/30 08:46:48 charger status: B
[lp-1 ] DEBUG 2024/10/30 08:46:48 charge start energy: 442.182kWh
[lp-1 ] INFO 2024/10/30 08:46:48 car connected
[lp-1 ] DEBUG 2024/10/30 08:46:48 vehicle api refresh
[lp-1 ] DEBUG 2024/10/30 08:46:48 pv timer elapse
[lp-1 ] DEBUG 2024/10/30 08:46:48 pv timer inactive
[lp-1 ] DEBUG 2024/10/30 08:46:48 charge total import: 442.182kWh
[lp-1 ] DEBUG 2024/10/30 08:46:48 charger vehicle id: evcc
[ocpp ] TRACE 2024/10/30 08:46:48 send **SERIAL**: [2,"2969005108","GetCompositeSchedule",{"connectorId":1,"duration":60}]
[ocpp ] TRACE 2024/10/30 08:46:49 recv **SERIAL**: [2,"241029214647-5630","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-10-29T21:46:47Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"0.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"442182.0","context":"Sample.Periodic"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Sample.Periodic"}]}],"transactionId":1}]
[ocpp ] TRACE 2024/10/30 08:46:49 send **SERIAL**: [3,"241029214647-5630",{}]
[ocpp ] TRACE 2024/10/30 08:46:49 recv **SERIAL**: [3,"2969005108",{"status":"Accepted","connectorId":1,"chargingSchedule":{"chargingRateUnit":"A","duration":60,"chargingSchedulePeriod":[{"startPeriod":0,"limit":0}]}}]
[ocpp ] TRACE 2024/10/30 08:46:49 recv **SERIAL**: [2,"241029214647-5631","StatusNotification",{"connectorId":1,"status":"SuspendedEVSE","errorCode":"NoError","timestamp":"2024-10-29T21:46:45Z","vendorId":"EN+","info":"{\"reason\":\"plugInGun\",\"cpv\":0,\"rv\":0}"}]
[ocpp ] TRACE 2024/10/30 08:46:49 send **SERIAL**: [3,"241029214647-5631",{}]
[ocpp ] TRACE 2024/10/30 08:46:50 recv **SERIAL**: [2,"241029214648-5632","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":0,\"cpVolt\":950,\"transactionId\":1,\"totalPowerFactor\":0,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":0,\"ccVolt\":0,\"offset0\":2013,\"offset1\":2015,\"leakcurr\":0,\"AMBTemp\":227,\"lockStatus\":1,\"ctCurrent\":0}"}]
[ocpp ] TRACE 2024/10/30 08:46:50 send **SERIAL**: [3,"241029214648-5632",{"status":"Accepted"}]
[lp-1 ] INFO 2024/10/30 08:46:52 vehicle updated: unknown -> Mustang Mach-E
[lp-1 ] DEBUG 2024/10/30 08:46:53 set charge mode: now
[lp-1 ] DEBUG 2024/10/30 08:46:53 pv timer reset
[lp-1 ] DEBUG 2024/10/30 08:46:53 pv timer inactive
[site ] DEBUG 2024/10/30 08:46:53 ----
[lp-1 ] DEBUG 2024/10/30 08:46:53 charge power: 0W
[lp-1 ] DEBUG 2024/10/30 08:46:53 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/30 08:46:53 pv power: 945W
[site ] DEBUG 2024/10/30 08:46:53 grid meter: -380W
[site ] DEBUG 2024/10/30 08:46:53 site power: -380W
[lp-1 ] DEBUG 2024/10/30 08:46:53 charge total import: 442.182kWh
[lp-1 ] DEBUG 2024/10/30 08:46:53 charger status: B
[ocpp ] TRACE 2024/10/30 08:46:53 send **SERIAL**: [2,"917725333","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":3,"stackLevel":5,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-10-29T21:46:53Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":32}]}}}]
[ocpp ] TRACE 2024/10/30 08:46:54 recv **SERIAL**: [3,"917725333",{"status":"Accepted"}]
[lp-1 ] DEBUG 2024/10/30 08:46:54 max charge current: 32A
[ocpp ] TRACE 2024/10/30 08:46:54 send **SERIAL**: [2,"1317749953","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":3,"stackLevel":5,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-10-29T21:46:54Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":32}]}}}]
[ocpp ] TRACE 2024/10/30 08:46:55 recv **SERIAL**: [3,"1317749953",{"status":"Accepted"}]
[lp-1 ] DEBUG 2024/10/30 08:46:55 charger enable
[lp-1 ] DEBUG 2024/10/30 08:46:55 wake-up timer: start
[ocpp ] TRACE 2024/10/30 08:46:56 recv **SERIAL**: [2,"241029214654-5633","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-10-29T21:46:54Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"0.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"442182.0","context":"Sample.Periodic"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Sample.Periodic"}]}],"transactionId":1}]
[ocpp ] TRACE 2024/10/30 08:46:56 send **SERIAL**: [3,"241029214654-5633",{}]
[ocpp ] TRACE 2024/10/30 08:46:57 recv **SERIAL**: [2,"241029214655-5634","StatusNotification",{"connectorId":1,"status":"Charging","errorCode":"NoError","timestamp":"2024-10-29T21:46:52Z","vendorId":"EN+","info":"{\"reason\":\"plugOutGun\",\"cpv\":0,\"rv\":0}"}]
[ocpp ] TRACE 2024/10/30 08:46:57 send **SERIAL**: [3,"241029214655-5634",{}]
[... cut ...] 1}]
[ocpp ] TRACE 2024/10/30 08:47:59 send **SERIAL**: [3,"241029214757-5660",{}]
[ocpp ] TRACE 2024/10/30 08:48:00 recv **SERIAL**: [2,"241029214758-5661","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":133,\"cpVolt\":625,\"transactionId\":1,\"totalPowerFactor\":0,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":0,\"ccVolt\":0,\"offset0\":2013,\"offset1\":2016,\"leakcurr\":20,\"AMBTemp\":230,\"lockStatus\":1,\"ctCurrent\":0}"}]
```

t3h-n00b, 2024-10-29:
Alright, just to throw more into it - I downgraded back to 0.130.8 and had the same issue where it was sitting on 8A. Change the max limit to 31A (then back to 32A) must have kicked something back into gear. Will go back to the latest version (probably later in the week, as need to actually charge the car now!) but there may be something in that.

```text
[site ] DEBUG 2024/10/30 09:02:41 ----
[lp-1 ] DEBUG 2024/10/30 09:02:41 charge power: 1825W
[lp-1 ] DEBUG 2024/10/30 09:02:41 charge currents: [7.6 0 0]A
[site ] DEBUG 2024/10/30 09:02:42 pv power: 1031W
[site ] DEBUG 2024/10/30 09:02:42 grid meter: 1310W
[site ] DEBUG 2024/10/30 09:02:42 site power: 1310W
[lp-1 ] DEBUG 2024/10/30 09:02:42 detected active phases: 1p
[lp-1 ] DEBUG 2024/10/30 09:02:42 !! session: chargeRater.chargedEnergy=0.0 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/10/30 09:02:42 charge total import: 442.282kWh
[lp-1 ] DEBUG 2024/10/30 09:02:42 charger status: C
[ocpp ] TRACE 2024/10/30 09:02:42 recv SN10052312237863: [2,"241029220241-5811","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":133,\"cpVolt\":628,\"transactionId\":1,\"totalPowerFactor\":0,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":0,\"ccVolt\":0,\"offset0\":2013,\"offset1\":2018,\"leakcurr\":0,\"AMBTemp\":235,\"lockStatus\":1,\"ctCurrent\":0}"}]
[ocpp ] TRACE 2024/10/30 09:02:42 send SN10052312237863: [3,"241029220241-5811",{"status":"Accepted"}]
[ocpp ] TRACE 2024/10/30 09:02:51 recv SN10052312237863: [2,"241029220250-5812","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-10-29T22:02:50Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"8.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"7.6"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"442287.0","context":"Sample.Periodic"},{"measurand":"Power.Active.Import","unit":"W","value":"1824.0","context":"Sample.Periodic"}]}],"transactionId":1}]
[ocpp ] TRACE 2024/10/30 09:02:51 send SN10052312237863: [3,"241029220250-5812",{}]
[ocpp ] TRACE 2024/10/30 09:02:52 recv SN10052312237863: [2,"241029220251-5813","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":133,\"cpVolt\":631,\"transactionId\":1,\"totalPowerFactor\":0,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":0,\"ccVolt\":0,\"offset0\":2013,\"offset1\":2018,\"leakcurr\":0,\"AMBTemp\":235,\"lockStatus\":1,\"ctCurrent\":0}"}]
[ocpp ] TRACE 2024/10/30 09:02:52 send SN10052312237863: [3,"241029220251-5813",{"status":"Accepted"}]
[ocpp ] TRACE 2024/10/30 09:03:01 recv SN10052312237863: [2,"241029220300-5814","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-10-29T22:03:00Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"8.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"7.6"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"442293.0","context":"Sample.Periodic"},{"measurand":"Power.Active.Import","unit":"W","value":"1825.0","context":"Sample.Periodic"}]}],"transactionId":1}]
[ocpp ] TRACE 2024/10/30 09:03:01 send SN10052312237863: [3,"241029220300-5814",{}]
[lp-1 ] DEBUG 2024/10/30 09:03:02 set charge mode: off
[site ] DEBUG 2024/10/30 09:03:02 ----
[lp-1 ] DEBUG 2024/10/30 09:03:02 charge power: 1825W
[lp-1 ] DEBUG 2024/10/30 09:03:02 charge currents: [7.6 0 0]A
[site ] DEBUG 2024/10/30 09:03:02 pv power: 1023W
[site ] DEBUG 2024/10/30 09:03:02 grid meter: 1317W
[site ] DEBUG 2024/10/30 09:03:02 site power: 1317W
[lp-1 ] DEBUG 2024/10/30 09:03:02 detected active phases: 1p
[lp-1 ] DEBUG 2024/10/30 09:03:02 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/10/30 09:03:02 charge total import: 442.293kWh
[lp-1 ] DEBUG 2024/10/30 09:03:02 charger status: C
[ocpp ] TRACE 2024/10/30 09:03:02 send SN10052312237863: [2,"1159351071","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":3,"stackLevel":5,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-10-29T22:03:02Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0}]}}}]
[ocpp ] TRACE 2024/10/30 09:03:02 recv SN10052312237863: [2,"241029220301-5815","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":133,\"cpVolt\":621,\"transactionId\":1,\"totalPowerFactor\":0,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":0,\"ccVolt\":0,\"offset0\":2013,\"offset1\":2018,\"leakcurr\":0,\"AMBTemp\":236,\"lockStatus\":1,\"ctCurrent\":0}"}]
[ocpp ] TRACE 2024/10/30 09:03:02 send SN10052312237863: [3,"241029220301-5815",{"status":"Accepted"}]
[ocpp ] TRACE 2024/10/30 09:03:03 recv SN10052312237863: [3,"1159351071",{"status":"Accepted"}]
[lp-1 ] DEBUG 2024/10/30 09:03:03 charger disable
[lp-1 ] DEBUG 2024/10/30 09:03:03 wake-up timer: stop
[ocpp ] TRACE 2024/10/30 09:03:05 recv SN10052312237863: [2,"241029220304-5816","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-10-29T22:03:04Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"8.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"7.6"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"442298.0","context":"Sample.Periodic"},{"measurand":"Power.Active.Import","unit":"W","val [... cut ...]
[ocpp ] TRACE 2024/10/30 09:04:22 recv SN10052312237863: [2,"241029220421-5838","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":133,\"cpVolt\":621,\"transactionId\":1,\"totalPowerFactor\":0,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":0,\"ccVolt\":0,\"offset0\":2013,\"offset1\":2018,\"leakcurr\":0,\"AMBTemp\":240,\"lockStatus\":1,\"ctCurrent\":0}"}]
[ocpp ] TRACE 2024/10/30 09:04:22 send SN10052312237863: [3,"241029220421-5838",{"status":"Accepted"}]
[lp-1 ] DEBUG 2024/10/30 09:04:29 set charge mode: off
```

t3h-n00b, 2024-11-07:
Iterated versions a few more times, 130.11 currently fine (issue introduced in 130.12,13 or 131.0 then)

Still have the other issue of the charge dropping to 8A for 1-2 minutes throughout the charge cycle randomly.

t3h-n00b, 2024-11-07:
Alright, 0.130.12 seems to be the 'problematic' version. I can _occasionally_ get the full charge power (either in Fast mode, or the full Solar value), but 90%+ of the time, it's limited to 8A.

0.130.11 and older aren't perfect, but they are pretty consistent in that they will give 32A in Fast mode all the time, and the 'failback' to 8A in Solar mode is much less frequent.

```text
site ] DEBUG 2024/11/08 08:51:48 ----
[lp-1 ] DEBUG 2024/11/08 08:51:49 charge power: 1848W
[lp-1 ] DEBUG 2024/11/08 08:51:49 charge currents: [7.6 0 0]A
[site ] DEBUG 2024/11/08 08:51:49 pv power: 4052W
[site ] DEBUG 2024/11/08 08:51:49 grid meter: -1468W
[site ] DEBUG 2024/11/08 08:51:49 site power: -1468W
[lp-1 ] DEBUG 2024/11/08 08:51:49 charge voltages: [241 0 0]V
[lp-1 ] DEBUG 2024/11/08 08:51:49 detected connected phases: 1p
[lp-1 ] DEBUG 2024/11/08 08:51:49 detected active phases: 1p
[lp-1 ] DEBUG 2024/11/08 08:51:49 charge total import: 534.742kWh
[lp-1 ] DEBUG 2024/11/08 08:51:49 charger status: C
[lp-1 ] WARN 2024/11/08 08:51:49 charger logic error: current mismatch (got 8A, expected 32A)
[ocpp ] TRACE 2024/11/08 08:51:49 enqueued CALL [730716023, SetChargingProfile] for **SERIAL**
[ocpp ] TRACE 2024/11/08 08:51:49 dispatched request 730716023 for **SERIAL**
[ocpp ] TRACE 2024/11/08 08:51:49 sent JSON message to **SERIAL**: [2,"730716023","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":3,"stackLevel":5,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-11-07T21:51:49Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":32}]}}}]
[ocpp ] TRACE 2024/11/08 08:51:49 started timeout timer for **SERIAL**
[ocpp ] TRACE 2024/11/08 08:51:50 received JSON message from **SERIAL**: [3,"730716023",{"status":"Accepted"}]
[ocpp ] TRACE 2024/11/08 08:51:50 handling incoming CALL RESULT [730716023] from **SERIAL**
[ocpp ] TRACE 2024/11/08 08:51:50 completed request 730716023 for **SERIAL**
[lp-1 ] DEBUG 2024/11/08 08:51:50 max charge current: 32A
[ocpp ] TRACE 2024/11/08 08:51:50 **SERIAL** ready to transmit again
[ocpp ] TRACE 2024/11/08 08:51:50 timeout canceled for **SERIAL**
[ocpp ] TRACE 2024/11/08 08:51:53 received JSON message from **SERIAL**: [2,"241107215153-17127","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-11-07T21:51:53Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"8.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"7.7"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"241.2","context":"Sample.Periodic"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"534747.0","context":"Sample.Periodic"},{"measurand":"Power.Active.Import","unit":"W","value":"1861.0","context":"Sample.Periodic"}]}],"transactionId":1730955963}]
[ocpp ] TRACE 2024/11/08 08:51:53 handling incoming CALL [241107215153-17127, MeterValues] from **SERIAL**
[ocpp ] TRACE 2024/11/08 08:51:53 sent CALL RESULT [241107215153-17127] for **SERIAL**
[ocpp ] TRACE 2024/11/08 08:51:53 sent JSON message to **SERIAL**: [3,"241107215153-17127",{}]
[ocpp ] TRACE 2024/11/08 08:51:54 received JSON message from **SERIAL**: [2,"241107215153-17128","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":133,\"cpVolt\":621,\"transactionId\":1730955963,\"totalPowerFactor\":0,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":0,\"ccVolt\":0,\"offset0\":2013,\"offset1\":2019,\"leakcurr\":0,\"AMBTemp\":287,\"lockStatus\":1,\"ctCurrent\":0}"}]
[ocpp ] TRACE 2024/11/08 08:51:54 handling incoming CALL [241107215153-17128, DataTransfer] from **SERIAL**
[ocpp ] TRACE 2024/11/08 08:51:54 sent CALL RESULT [241107215153-17128] for **SERIAL**
[ocpp ] TRACE 2024/11/08 08:51:54 sent JSON message to **SERIAL**: [3,"241107215153-17128",{"status":"Accepted"}]
[ocpp ] TRACE 2024/11/08 08:52:03 received JSON message from **SERIAL**: [2,"241107215202-17129","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-11-07T21:52:02Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"8.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"7.7"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"241.4","context":"Sample.Periodic"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"534752.0","context":"Sample.Periodic"},{"measurand":"Power.Active.Import","unit":"W","value":"1861.0","context":"Sample.Periodic"}]}],"transactionId":1730955963}]
[ocpp ] TRACE 2024/11/08 08:52:03 handling incoming CALL [241107215202-17129, MeterValues] from **SERIAL**
[ocpp ] TRACE 2024/11/08 08:52:03 sent CALL RESULT [241107215202-17129] for **SERIAL**
[ocpp ] TRACE 2024/11/08 08:52:03 sent JSON message to **SERIAL**: [3,"241107215202-17129",{}]
[ocpp ] TRACE 2024/11/08 08:52:04 received JSON message from **SERIAL**: [2,"241107215203-17130","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":133,\"cpVolt\":621,\"transactionId\":1730955963,\"totalPowerFactor\":0,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":0,\"ccVolt\":0,\"offset0\":2013,\"offset1\":2019,\"leakcurr\":0,\"AMBTemp\":287,\"lockStatus\":1,\"ctCurrent\":0}"}]
[ocpp ] TRACE 2024/11/08 08:52:04 handling incoming CALL [241107215203-17130, DataTransfer] from **SERIAL**
[... cut ...] aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":0,\"ccVolt\":0,\"offset0\":2013,\"offset1\":2019,\"leakcurr\":0,\"AMBTemp\":287,\"lockStatus\":1,\"ctCurrent\":0}"}]
[ocpp ] TRACE 2024/11/08 08:52:24 handling incoming CALL [241107215223-17137, DataTransfer] from **SERIAL**
[ocpp ] TRACE 2024/11/08 08:52:24 sent CALL RESULT [241107215223-17137] for **SERIAL**
[ocpp ] TRACE 2024/11/08 08:52:24 sent JSON message to **SERIAL**: [3,"241107215223-17137",{"status":"Accepted"}]
```

andig, 2024-11-08:
Your charge is faulty:

```
[ocpp ] TRACE 2024/11/08 08:51:49 sent JSON message to **SERIAL**: [2,"730716023","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":3,"stackLevel":5,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-11-07T21:51:49Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":32}]}}}]
[ocpp ] TRACE 2024/11/08 08:51:50 received JSON message from **SERIAL**: [3,"730716023",{"status":"Accepted"}]

[ocpp ] TRACE 2024/11/08 08:51:53 received JSON message from **SERIAL**: [2,"241107215153-17127","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-11-07T21:51:53Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"8.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"7.7"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"241.2","context":"Sample.Periodic"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"534747.0","context":"Sample.Periodic"},{"measurand":"Power.Active.Import","unit":"W","value":"1861.0","context":"Sample.Periodic"}]}],"transactionId":1730955963}]
```

It accepts the 32A limit and later claims the limit is 8A. Either that's a bug in the charger or you have something else setting those 8A. Either ask vendor what's wrong or just remove the invalid 8A measurement to silence the evcc error:

    metervalues: -Current.Offered

That still won't make it charge with 32A though.