evcc-io/evcc #16851: Margins, Fees, Taxes and More Fees on dynamic electricity price

RenatusRo, 2024-10-25:
As tariffs are trending today yet another perspective:

None of the Belgian energy companies provides an API.
Prices are given as formulas based on the belpex market index X

```
grid:
  marginA * X + marginB
feedin:
  marginC * X - marginD
```

The final grid: price/kWh then becomes
`(marginA * X + marginB + various_fees) * VATaxThing + MoreFees`

MoreFees being a composition of capacity tariff + fixed provider fee
broken down to an estimation per kWh based on year demand an average capacity. 

**Describe the solution you'd like**

A possibility to describe a tariff formula based on entsoe.

**Describe alternatives you've considered**

The current entsoe/charges/tax gives a good start but stays a bit off the reality.
And using @VolkerK62 nice life hack https://github.com/evcc-io/evcc/discussions/16465#discussioncomment-10839844 looks cumbersome.
Of course I could build a day ahead price array myself outside evcc and do a `type: custom` to myself.

andig, 2024-10-25:
We already have charges and taxes. Aren't they sufficient? What tariff are we even talking about here?

RenatusRo, 2024-10-25:
At least Margin is missing:

AFAIK now it is

```
func (t *embed) totalPrice(price float64) float64 {
	return (price + t.Charges) * (1 + t.Tax)
}
```
for grid we would need
```
func (t *embed) totalPrice(price float64) float64 {
	return ((t.Margin* price + t.Charges) * (1 + t.Tax)) + t.MoreFees
}
```
t.Margin, t.Charges being specific for grid. o.k, t.MoreFees
could be calculated into t.Charges

and for feedin
```
func (t *embed) totalPrice(price float64) float64 {
	return ((t.Margin* price + t.Charges)
}
```
t.Margin, t.Charges being specific for feedin

andig, 2024-10-25:
I don't understand:

- are you saying you need to be able to calculate charges as percentage instead of fixed value (and call that margin)
- and that there are additional non-taxable uplifts

?

RenatusRo, 2024-10-25:
exactly, charges as percentage and non-taxable uplifts. ( https://trevion.be/tariefkaarten/ )

closed 2024-10-27 by andig

andig, 2024-10-30:
I tend to revert this and replace this PR with the ability to define an arbitrary formula (using Go syntax):

    math.Abs(price) * 1.19 + 0.32

If we do that we need to understand which and how many variables need to go into that formula and if the variables are fixed or zone-based (see https://github.com/evcc-io/evcc/issues/16829).