evcc-io/evcc #16852: Günstiges Netzladen funktioniert nach Update von 0.131.0 auf 0.131.1 nicht mehr

martinscheller, 2024-10-25:
### Describe the bug

Ich verwendete die Funktion unter Home Assistant in Kombination mit einem Kostal Plenticore Wechselrichter, einer BYD Batterie, einer OpenWB 2 und dem Tibber Stundentarif. Bisher lief es trotz "experimenteller UI-Funktion" absolut problemlos. Schwellwert für den Preis über das UI konfiguriert und zum besagten Zeitpunkt wurde die Batterie vom Netz geladen. 

### Steps to reproduce

1. Preis so eingestellt, daß heute um 12:00 Uhr die Batterie geladen werden soll.
2. UI bestätigt günstiges Netzladen
3. Trace gezogen und unten zur Verfügung gestellt 

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 10.11.12.21
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
# modbusproxy:
#    - port: 5200
#      uri: 10.11.12.160:1502
#       rtu: true
#       readonly: true # use `deny` to raise modbus errors

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
[... 340 lines cut ...]
[sunspec] TRACE 2024/10/25 12:01:09 modbus: send 64 1d 00 00 00 06 47 03 9d 76 00 3e
[sunspec] TRACE 2024/10/25 12:01:09 modbus: recv 64 1d 00 00 00 7f 47 03 7c 00 19 28 00 23 40 58 20 00 00 00 00 00 00 00 00 00 00 00 13 00 00 00 00 00 00 03 80 00 00 00 00 00 00 00 00 00 01 00 04 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 10 63 11 a0 00 00 00 20 00 00 00 00 00 20 00 00 00 00 00 00 fd d9 07 d0 13 88 f7 0c 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff ff ff fe ff fe 00 00
[sunspec] TRACE 2024/10/25 12:01:09 model 802 block 0 point SoC: 19
[site  ] DEBUG 2024/10/25 12:01:09 battery soc: 19%
[sunspec] TRACE 2024/10/25 12:01:09 modbus: send 64 1e 00 00 00 06 47 03 9d 4c 00 14
[site  ] DEBUG 2024/10/25 12:01:09 battery power: -2292W
[sunspec] TRACE 2024/10/25 12:01:09 modbus: recv 64 1e 00 00 00 2b 47 03 28 00 02 44 43 5f 53 54 52 49 4e 47 5f 32 20 20 20 20 20 00 c8 0c 8b 02 84 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[sunspec] TRACE 2024/10/25 12:01:09 model 160 block 2 point DCW: 644
[sunspec] TRACE 2024/10/25 12:01:09 modbus: send 64 1f 00 00 00 06 47 03 9d 30 00 08
[sunspec] TRACE 2024/10/25 12:01:09 modbus: recv 64 1f 00 00 00 13 47 03 10 ff fe ff ff 00 00 00 00 00 00 00 00 00 03 00 00
[sunspec] TRACE 2024/10/25 12:01:09 modbus: send 64 20 00 00 00 06 47 03 9d 60 00 14
[sunspec] TRACE 2024/10/25 12:01:09 modbus: recv 64 20 00 00 00 2b 47 03 28 00 03 44 43 5f 53 54 52 49 4e 47
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.1

andig, 2024-10-25:
Bitte vollständiges Log, Debug reicht. Bitte auch /api/state zum Zeitpunkt des günstigen ladens.

andig, 2024-10-25:
@naltatis smartCost zieht nur wenn es _keinen_ aktiven Ladeplan gibt. Evtl. führt das hier zu Verwirrung? Zumindest passen UI Indikation und Code nicht so richtig zueinander:

```go
case mode == api.ModeMinPV || mode == api.ModePV:
	// cheap tariff
	if smartCostActive && lp.EffectivePlanTime().IsZero() {
		err = lp.fastCharging()
		lp.resetPhaseTimer()
		lp.elapsePVTimer() // let PV mode disable immediately afterwards
		break
	}
```

martinscheller, 2024-10-25:
> Bitte vollständiges Log, Debug reicht. Bitte auch /api/state zum Zeitpunkt des günstigen ladens.

Wo finde ich das vollständige Logfile zu EVCC unter Home Assistant? Wie komme ich an die Daten von /api/state? :-)

andig, 2024-10-25:
@iseeberg79 die haben beide Batteriesteuerung. Wenn es vorher ging sollte es jetzt auch gehen.

martinscheller, 2024-10-25:
Ich habe die Funktion gestern am frühen Morgen noch genutzt und die Batterie darüber geladen. Ich bin mir ziemlich sicher, daß ich da die Version 0.131.0 genutzt habe. Leider weiß ich nicht, ob bzw. wo man die Home Assistant sehen kann, wann welche Zusatzmodule mit welchen Versionsständen upgedated wurden.

martinscheller, 2024-10-25:
> Versuche bitte das andere Template, wenn die Bedingungen bei dir gegeben sind: wenn es dann geht, passt es eigentlich

Ich werde berichten. Vielen Dank an alle für´s schnelle Feedback!

martinscheller, 2024-10-26:
Das andere Template hat leider keine Änderung gebracht. Gibt es eine Möglichkeit auf die 0.130.13 Version downzugraden?

martinscheller, 2024-10-26:
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 10.11.12.21
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
# modbusproxy:
#    - port: 5200
#      uri: 10.11.12.160:1502
#       rtu: true
#       readonly: true # use `deny` to raise modbus errors

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - name: PV_Garage
    type: template
    template: kostal-plenticore-gen2
    usage: pv

    # Modbus TCP
    modbus: tcpip
    id: 71
    host: 10.11.12.160 # Hostname
    port: 1502 # Port
    endianness: little # optional
    # maxacpower: 8 # Akkukapazität in kWh (optional)
    watchdog: 60s # optional
[... 226 lines cut ...]
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: Soc updated
      msg: Battery charged to ${vehicleSoc:%.0f}%
    guest: # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
  services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>

martinscheller, 2024-10-26:
[evcc.yaml.pdf](https://github.com/user-attachments/files/17531215/evcc.yaml.pdf)

Vielen Dank vorab. Konfig anbei. Liegt es evtl. daran, daß ich nicht den "eingebauten" Modbus Proxy verwende, sondern das Add On in HA?

martinscheller, 2024-10-26:
Danke, ich versuche es gleich nochmal. Wo ist der Unterschied Gen 1 und 2 beim Plenticore? Ich habe Gen 1 (2019). Inwiefern spielt das für die evcc.yaml eine Rolle?

martinscheller, 2024-10-26:
[image]

Das ist seltsam. Ich habe eben den Preis so konfiguriert, daß die Batterie geladen werden müßte. Wird sie aber nicht. Es wird lediglich das Entladen unterbunden. Ändere ich den Preis wieder, wird sie wieder entladen.

martinscheller, 2024-10-26:
[image]

martinscheller, 2024-10-26:
Ja, die BYD Hausbatterie soll geladen werden.

martinscheller, 2024-10-26:
Sorry, da haben sich unsere Posts überschnitten. 

**_**Wenn die Funktion vorher bei dir lief, geht sie bei deinem System scheinbar und du kannst umstellen.**_**

Umstellen worauf?

martinscheller, 2024-10-26:
[image]

Ja.

MiGro1194, 2024-10-27:
Kann das vielleicht auch an Tibber liegen? Bei mir geht es nämlich auch nicht mehr und in der neuen Home Assistant Version 2024.10.4 wurde die py für Tibber angepasst. 

_[Bump pyTibber to 0.30.3 (](https://github.com/Danielhiversen)[@Danielhiversen](https://github.com/Danielhiversen) - [#128860](https://github.com/home-assistant/core/pull/128860))_

MiGro1194, 2024-10-27:
Bei mir war es ein geänderter Token @martinscheller bei Tibber: [Tibber Pulse](https://docs.evcc.io/docs/devices/meters#tibber-pulse).

martinscheller, 2024-10-27:
> Bei mir war es ein geänderter Token @martinscheller bei Tibber: [Tibber Pulse](https://docs.evcc.io/docs/devices/meters#tibber-pulse).

Danke für den Hinweis. Gerade überprüft, der Token hat sich bei mir nicht geändert. Das Laden des Autos funktioniert jedenfalls noch preisbasiert und folgende korrekte Anzeige sehe ich auf dem Plenticore:

[image]

martinscheller, 2024-10-27:
[image]

[image]

Laut Dokumentation funktioniert Netzladen mit Gen 1 Plenticore nicht (mehr). Ich habe HW Version 0100. Das ist ärgerlich, zumal es ja vor 0.131.x wunderbar funktioniert hat.

martinscheller, 2024-10-27:
> Schick doch mal bitte die Übersicht mit deiner Firmware und Hardware-Release Version.

Plenticore 8.5 

[image]

Ich versuche über Home Assistant die Plenticore Register über Modbus auszulesen. Auf welche Register wären das?

martinscheller, 2024-10-27:
Ohne "Auf" ;-)

martinscheller, 2024-10-27:
[image]

Scheint wohl doch nicht so einfach zu sein. MIt o.g. Konfig kommt als Wert "0" zurück.

martinscheller, 2024-10-28:
> Im Screenshot oben ist mir aufgefallen, dass die externe Batteriesteuerung bei dir jetzt auch angezeigt wird (bei Batteriesperre/Netzladung?). Es braucht eine Weile (bis zu mehreren Minuten) bis die Netzladung aktiv wird.

Das war der Screenshot, während das Auto geladen wurde.

martinscheller, 2024-10-28:
[image]

Soll ich die die aktuellste Version updaten?

martinscheller, 2024-10-28:
> > Kleine Ursache, große Wirkung ;-) Der SoC wird nun korrekt angezeigt.
> 
> Du kannst das Register 1028 noch hinzufügen, dann sehen wir genau, was denn eigentlich bei Batteriesperre und Netzladen in den Registern los ist.

[image]

Erledigt, auch das 1030 hinzugefügt und neu gestartet. Leider zeigen die beiden 1028 und 1030 nur "0" an. 
Verstehe ich nicht. Laut Kostal Doku sind doch alle 3 vom gleichen Typ:

[image]

martinscheller, 2024-10-28:
Ich werde berichten, allerdings wohl erst Ende der Woche. Aktuell ist der Hausakku SoC bei 100% und ich bin nun einige Tage beruflich unterwegs.

martinscheller, 2024-10-28:
@iseeberg79: Es funktioniert wieder! Vielen Dank für deine Hilfe! Konnte es eben noch testen. Davor mußte ich einen Hard Reset meines Raspberry 4 machen, weil HA nicht mehr reagiert hat. Das viele Rebooten hat ihm offenbar nicht gefallen...

So sah es während des Netzladens des Hausakkus aus. Die Verzögerung waren nur wenige Sekunden, nachdem ich den Preis eingestellt hatte. 

[image]

[image]

[image]

So sah es aus, als ich den Preis wieder "gesenkt" habe. Auch das dauerte nur wenige Sekunden. Erst wurde die Akkuladung gestoppt, dann war der Akku kurz zum Entladen blockiert und kurz danach wurde wieder der Akku ganz normal entladen.   

[image]

[image]

[image]

closed 2024-10-28 by martinscheller

mbuhansen, 2024-11-02:
> Alles OK soweit. Folgendes ist jetzt interessant: wie sehen die Register bei "Batteriesperre (Schnellladen)" und "Netzladen (Hausbatterie)" aus.
> 
> Erwartung ist:
> 
> 1. bei Batteriesperre in 1028: 0, 1042 unverändert; externe Batteriesteuerung in der Webansicht des WR aktiv; praktisch keine Ladung/Entladung
> 2. bei normalem Betrieb verschwindet die externe Steuerung aus der Webansicht des WR und es wird nach etwas mehr als einer Minute wieder passend zum Verbrauch entladen oder geladen (PV Überschuss)
> 3. bei Netzladung in 1028: -100, 1042 unverändert (Achtung, dauert etwas, aber <5 Minuten) - die Ladeleistung steigt träge an, bis zum Maximalen Wert (abhängig von der Leistung der Batterie)
> 4. bei normalem Betrieb aus Netzladen in 1028: 0, 1042 unverändert; die Netzladung nimmt träge ab; nach etwas mehr als zwei Minuten ist die externe Steuerung wieder inaktiv, es wird wieder passend zum Verbrauch entladen oder geladen (PV Überschuss) (identisch zu 2)
> 
> Bei Änderung im Register 1042 während der Batteriesperre: altes Template, keine Netzladungsfunktion

@iseeberg79 Can you please share your yaml from HA or script to get Home assistant to write -100 to adresse 1028.
thanks