evcc-io/evcc #16860: 1.131.1 won't start because of sunspec

buliwyf42, 2024-10-25:
### Describe the bug

After updating to evcc_0.131.1+1729823465 
Evcc won't start with the following error.

In my evcc.yaml there is no mention of a sunpsec device.
Also the devices in den UI Config are no Sunspec devices.

 
[image]

I also had a look into the evcc.db but even there i cannot find any sunspec reference anymore. 

### Steps to reproduce

After updating to evcc_0.131.1+1729823465 
Evcc won't start with the following error.

In my evcc.yaml there is no mention of a sunpsec device.
Also the devices in den UI Config are no Sunspec devices.

 
[image]

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.home.lan # .local suffix announces the hostname on MDNS
  port: 7070

log:  info
levels:
  cache: error

sponsortoken: xxxx

interval: 15s # control cycle interval

tariffs:
    currency: EUR
    grid:
        type: fixed
        price: 0.2819
        zones:
          - days: Mo-Fr
            hours: 16-22
            price: 0.2769
          - days: Mo-Fr
            hours: 22-0
            price: 0.2318
          - days: Mo-Fr
            hours: 0-6
            price: 0.2318
          - days: Sa,So
            price: 0.2318
    feedin:
        type: fixed
        price: 0.075
    co2:
        type: grünstromindex
        zip: xxxxx

meters:
  - name: my_fronius_smartmeter
    type: template
    template: fronius-gen24
    usage: grid
    host: 192.168.215.156

chargers:
  - name: wallbox2
    type: template
    template: go-e-v3
    host: 192.168.211.33

loadpoints:
  - title: go-e rechts
    charger: wallbox2
    enable:
      threshold: 0
      delay: 1m
    disable:
      threshold: 0
      delay: 1m

site:
  title: xxxxxxxxx
  meters:
    grid: my_fronius_smartmeter
  residualPower: 100
```

### Log details

```text
2024-10-25T15:45:35.895268+02:00 evcc evcc[16259]: [main  ] FATAL 2024/10/25 15:45:35 cannot create meter 'my_fronius_smartmeter': cannot create meter type 'template': cannot create meter type 'mbmd': unknown meter type: SUNSPEC
2024-10-25T15:45:35.895332+02:00 evcc evcc[16259]: [main  ] FATAL 2024/10/25 15:45:35 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.1+1729823465

andig, 2024-10-25:
Thank you for the report. That's a blunder in the `fronius-gen24` template.

andig, 2024-10-25:
Nightly in 30min

closed 2024-10-25 by andig

buliwyf42, 2024-10-25:
Now it won't start with another error. 
[image]

2024-10-25T19:37:51.541828+02:00 evcc evcc[17387]: [main  ] FATAL 2024/10/25 19:37:51 cannot create meter 'my_fronius_smartmeter': cannot create meter type 'template': cannot create meter type 'custom': power: read tcp 192.168.203.61:49018->192.168.215.156:502: i/o timeout
2024-10-25T19:37:51.551076+02:00 evcc evcc[17387]: [main  ] FATAL 2024/10/25 19:37:51 will attempt restart in: 15m0s

0.131.0 starts after downgrade without issue

andig, 2024-10-25:
Mhhm. Not sure how I can help with an i/o timeout. Could you give remote access to the inverter via SSH, VPN, Tailscale or anything to info@evcc.io so I can see what's wrong?

buliwyf42, 2024-10-25:
Unfortunately, not at the moment. The timeout only occurs with the latest nightly. Does a pcap file help ?

buliwyf42, 2024-10-25:
> [info@evcc.io](mailto:info@evcc.io)

You should have received a invitation to my evcc lxc container.

andig, 2024-10-25:
See https://github.com/evcc-io/evcc/issues/16860#issuecomment-2438456711. Only the inverter is of interest.

buliwyf42, 2024-10-25:
> See [#16860 (comment)](https://github.com/evcc-io/evcc/issues/16860#issuecomment-2438456711). Only the inverter is of interest.

You can access the Inverter from my evcc host via ssh?

buliwyf42, 2024-10-27:
I did some testing, the issue is only with the configured Grid Meter from the evcc.yaml.
The same config works for a solar meter or battery configured via the UI as Solar&Battery
If i remove the config for the grid meter from evcc.yaml and configure it via the ui , it won't work either. 
Solar&battery
[image]
Grid Meter
[image]

andig, 2024-10-27:
See https://github.com/evcc-io/evcc/issues/16860#issuecomment-2438655967