evcc-io/evcc #16865: Cupra Hybrid-Fahrzeuge Daten des elektrischen Antriebs verwenden

pvdoda, 2024-10-25:
### Describe the bug

Cupra Hybrid Fahrzeuge können über die CupraConnect Gen4 Integration eingebunden werden. Es ist sinnvoll hier die Angaben (Ladestand, Reichweite)  des elektrischen Antriebs (`"fuelType":"electric"`) zu priorisieren. Derzeit wird immer der primäre Antrieb (`"fuelType":"gasoline"`) genutzt. 

### Steps to reproduce

1. Cupra Hybrid-Fahrzeug konfigurieren
2. konfiguriertes Fahrzeug in EVCC auswählen

### Configuration details

```yaml
Konfigurationsauszug:
vehicles:
- name: cupraleonvz
  type: template
  template: cupra
  title: Cupra Leon ST
  user: XXXXX
  password: *****
  capacity: 10.4
  phases: 1
  minCurrent: 6
  maxCurrent: 16
  mode: pv
```

### Log details

```text
Auszug mit relevantem API:

[cupra ] TRACE 2024/10/25 20:01:52 GET https://ola.prod.code.seat.cloud.vwgroup.com/v5/users/XXXX/vehicles/XXXX/mycar
[cupra ] TRACE 2024/10/25 20:01:53 {"engines":{"primary":{"fuelType":"gasoline","rangeKm":500.0,"levelPct":98.0},"secondary":{"fuelType":"electric","rangeKm":0.0,"levelPct":1.0}},"services":{"charging":{"status":"NotReadyForCharging","targetPct":100,"currentPct":1,"chargeMode":"invalid","active":false,"remainingTime":0,"progressBarPct":1.0},"climatisation":{"status":"Off","targetTemperatureCelsius":19.0,"targetTemperatureFahrenheit":66.0,"climatisationTrigger":"unsupported","active":false,"remainingTime":0,"progressBarPct":0.0},"windowHeating":{"status":"Off","active":false}}}
...
[cupra ] TRACE 2024/10/25 20:39:53 {"engines":{"primary":{"fuelType":"gasoline","rangeKm":500.0,"levelPct":98.0},"secondary":{"fuelType":"electric","rangeKm":0.0,"levelPct":1.0}},"services":{"charging":{"status":"Charging","targetPct":100,"currentPct":1,"chargeMode":"timer","active":true,"remainingTime":220,"progressBarPct":1.0},"climatisation":{"status":"Off","targetTemperatureCelsius":19.0,"targetTemperatureFahrenheit":66.0,"climatisationTrigger":"unsupported","active":false,"remainingTime":0,"progressBarPct":0.0},"windowHeating":{"status":"Off","active":false}}}
...
[cupra ] TRACE 2024/10/25 20:55:23 {"engines":{"primary":{"fuelType":"gasoline","rangeKm":500.0,"levelPct":98.0},"secondary":{"fuelType":"electric","rangeKm":2.0,"levelPct":5.0}},"services":{"charging":{"status":"Charging","targetPct":100,"currentPct":5,"chargeMode":"timer","active":true,"remainingTime":210,"progressBarPct":5.0},"climatisation":{"status":"Off","targetTemperatureCelsius":19.0,"targetTemperatureFahrenheit":66.0,"climatisationTrigger":"unsupported","active":false,"remainingTime":0,"progressBarPct":0.0},"windowHeating":{"status":"Off","active":false}}}
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.1 (7191ce9f)

andig, 2024-10-25:
Bitte Testzugang an info@evcc.io

andig, 2024-10-25:
The code is in https://github.com/evcc-io/evcc/blob/master/vehicle/seat/cupra/provider.go#L92, wanna try a PR?

pvdoda, 2024-10-25:
Ok, I’m trying to find time over the weekend to set up a development environment.

pvdoda, 2024-10-26:
I have implemented and tested a solution for this issue. What are the next steps to move forward?

closed 2024-10-29 by andig