evcc-io/evcc #16868: Add Device/Category: Solvis heating and Heatpump

lebochequirit, 2024-10-25:
I have no Wallbox nor electric car, but PV and Battery and a SolvisMax heating with a heatpump (Solvis Mia).
The heatpump is smartgrid ready and speaks modbus. It would communicate with my Kostal inverter if that was a younger generation. 

My idea would be to add integrations for Heatpumps and force them to run (and store heat) while either PV output exceeds the actual consumption over day or a dynamic electricity tariff is cheap, and to charge the battery during low price periods.
Home Assistant integration or any other wheather data could allow fine tuning to only do so if it's cold and there's a certain heat consumption over the day or during the night before.

Thanks for considering!

L.B.Q.R.

andig, 2024-10-25:
You can already do so, the the various discussions around heat pumps. If you feel a Solvis integration would help you can propose a custom Solvis charger formbeing converted into a template.

andig, 2024-10-31:
ping @lebochequirit anything to do here to move forward?

BeneWilh, 2024-11-01:
I do have a Solvis Max with Solvis LEA too, Solvis generally speaks Modbus TCP, but Solvis havent released the new Modbus register table so far... so you can read some values but at least i do not have the register numbers to set values relevant for the heatpump.
I am now trying it via D1 mini + 3 relais + resistors to set 12V / 8V / 4 V / 0 V on I1+ (Smartgrid contact)... see [https://forum.openwb.de/viewtopic.php?p=110933#p110933](url)

larsneo, 2024-11-06:
I do have a Solvis Max with Solvis Mia 14 too. As @BeneWilh already mentioned Solvis' modbus documentation is outdated and partly wrong. But there's already an Integration for Solvis within HomeAssistant with Heatpumps integrated: https://github.com/LarsK1/hass_solvis_control/ 
I guess it might be easier to control the heatpump via a couple of simple automations then to integrate into evcc.

Example for setting water and room temperature based on tibber price
[image]

closed 2024-11-18 by github-actions[bot]

andig, 2024-11-18:
Would have appreciated feedback/ pr.