evcc-io/evcc #16874: Neueinrichtung Wallbox easee Home -> API Fehler, obwohl Status-Page grün

CITAB, 2024-10-26:
### Describe the bug

Hallo zusammen,

wir haben gerade ein neues EVCC aufgesetzt und bekommen es nicht ans Laufen.
Nach Hinterlegung Sponsortoken und Eingrenzungen haben wir nun folgenden Fehler mit der Wallbox easee Home

> cannot create charger type 'easee': Get "https://api.easee.com/api/chargers/EH***/site": dial tcp: lookup api.easee.com: i/o timeout

Nach Sichtung diverser Beiträge hier (auch für eine lokale API gevotet), würde ich aber gerne das System mit der Wallbox easee Home unter den gegebenen Umständen gerne zum Laufen bekommen.
Die Statuspage von https://status.easee.com/ zeigt alles grün

Weiterhin die Frage, wie denn denn die Eingabewerte für die lokale Authentifizierung sind?
Aus dem Templatetext werde ich nicht schlau, d.h. welche Eingabe ist hier vorzunehmen: 0/1, true/false, etc.

>  authorize: # Steuert ob evcc die Authentifizierung am Charger vornimmt. Vorteil ist ein kontrollierter Ladestart. Nicht kompatibel mit RFID Identifikation von Fahrzeugen. (optional)

 Vielen Dank für die Unterstützung.

### Steps to reproduce

Passwort geändert, da gab es auch schon Probleme mit Sonderzeichen

### Configuration details

```yaml
chargers:
  - name: Wallbox
    type: template
    template: easee
    user: **** # Emailadresse
    password: **** # wie Login für Easee App oder Web Portal (https://easee.cloud)
    charger: EH****
    timeout: 50s # optional
    authorize: # Steuert ob evcc die Authentifizierung am Charger vornimmt. Vorteil ist ein kontrollierter Ladestart. Nicht kompatibel mit RFID Identifikation von Fahrzeugen. (optional)
```

### Log details

```text
[main ] INFO 2024/10/26 11:31:04 evcc 0.131.1
[main ] INFO 2024/10/26 11:31:04 using config file: /etc/evcc.yaml
[db ] INFO 2024/10/26 11:31:04 using sqlite database: /root/.evcc/evcc.db
[main ] INFO 2024/10/26 11:31:14 listening at :7070
[main ] FATAL 2024/10/26 11:31:15 failed configuring site: not found: grid
[main ] FATAL 2024/10/26 11:31:15 will attempt restart in: 15m0s
[main ] ERROR 2024/10/26 11:31:20 version check failed: Get "https://api.github.com/repos/evcc-io/evcc/releases/latest": dial tcp: lookup api.github.com: i/o timeout (installed: 0.131.1)
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.1

0x3d13f, 2024-10-26:
Ich sehe hier zwei Probleme:
`dial tcp: lookup api.easee.com: i/o timeout` Problem mit der Namensauflösung im Docker Container
`failed configuring site: not found: grid` Konfiguration ist nicht vollständig, ohne Grid Meter in der Site geht es nicht

CITAB, 2024-10-26:
Namensauflösung im Docker Container getestet, api.easee.com wird aufgelöst
 
Als Grid Meter ist hinterlegt

> meters:
  - name: VARTA
    type: template
    template: varta
    usage: battery
    host: 192.168.125.183 # IP-Adresse oder Hostname
    port: 502 # Port (optional)
    capacity: 50 # Akkukapazität in kWh (optional)

0x3d13f, 2024-10-26:
`usage: battery` ist kein Grid-Meter

mfuchs1984, 2024-10-26:
Dann musst du bei der Site bei grid VARTA eintragen, nicht grid.

naltatis, 2024-10-26:
> dial tcp: lookup api.easee.com: i/o timeout

Ja, sieht in der Tat als ob der Dockercontainer hier ein generelles Netzwerkproblem hat.

closed 2024-10-26 by andig