evcc-io/evcc #16890: Fehlerhafte Anzeige bzw. Berechnung der geladenen Energie

mareyem, 2024-10-26:
### Describe the bug

Hallo,

in der Ladehistorie werden immer wieder fehlerhaft berechnete Energiemengen angezeigt. 2 aufeinanderfolgende Ladevorgänge zeigen die gleiche geladene Energie, obwohl sie deutlich unterschiedlich lang waren. Eine Überprüfung der Zahlen anhand des CSV-Downloads zeigt, dass die Zählerstände korrekt sind, die Differenzberechnung aber nicht stimmt. Alle anderen Spalten scheinen korrekte Werte zu enthalten. siehe Screenshots. Ich habe die Spalte Excel-Rechnung eingefügt. Die rote Spalte EVCC-Energie ist die von EVCC berechnete Spalte. 

Ich habe eine Vermutung, dass das Ende von Ladevorgängen nicht korrekt erkannt wird, wenn das Auto abgesteckt wird. Scheinbar hängt das Verhalten von aktuellen Zustand des Autos ab - Laden bereits beendet oder noch nicht. Eine Systematik habe ich noch nicht entdeckt. 
Typischerweise stecke ich nur ab, indem ich den Ladeport im Auto entriegele und stecke beim nächsten Mal wieder an. EVCC lasse ich unberührt.

[image]
[image]

Bitte lasst mich wissen, ob ich bestimmt Logs liefern soll.

### Steps to reproduce

Na

### Configuration details

```yaml
Na
```

### Log details

```text
Na
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.1

andig, 2024-10-27:
> Ich habe eine Vermutung, dass das Ende von Ladevorgängen nicht korrekt erkannt wird, wenn das Auto abgesteckt wird

Es braucht ein Log dafür. Ich mache zu bis es eins gibt, Danke!

closed 2024-10-27 by andig

mareyem, 2024-10-27:
[evcc.log](https://github.com/user-attachments/files/17533307/evcc.log)

mareyem, 2024-10-27:
Hier ist das Log.

andig, 2024-10-27:
Was ist für dieses Log konkret falsch?

mareyem, 2024-10-27:
Naja, es deckt den Zeitraum ab, in dem die falschen Energiemengen berechnet wurden. Ich dachte, Du wolltest das haben? Da es nicht so ganz in der Materie stecke, wie Du, wäre es schön, wenn Du etwas genauer erläutern könntest, was Du benötigst, um dem Problem nachgehen zu können. 

Danke für Deine Unterstützung.

andig, 2024-10-27:
Da `db` im Logfile fehlt ist nicht zu erkennen, was da gespeichert wurde und ob es daher richtig ist.